arkq/bluez-alsa #482: LDAC as sink possible?

RaspiLearner, 2021-08-02:
Thanks to the great guide in wiki, I was able to compile bluez-alsa from source on my Raspberry Pi Zero.
It should act as a bluetooth receiver so I need bluez-alsa to be sink for the several codecs.

For configuration I used:
../configure --enable-aac --enable-ldac

and got this as replay:
checking for AAC... yes
checking for LDAC_ABR... yes
checking for LDAC_DEC... no
checking for LDAC_ENC... yes

Looking at configure.ac, if I didn't misunderstood, the script is looking for ldacBT_dec.
As far as I could see, [ldacBT](https://github.com/EHfive/ldacBT) only creates ldacBT_abr and ldacBT_enc but never (also in earlier versions) ldacBT_dec.

After compiling I could connect with my Android Smartphone and select AAC in developer options. When I tried LDAC it always switches back to AAC so to me it looks like LDAC as sink is not working.

Since I'm very new to this, I hope I did not misinterpret things.
Is it correct the LDAC is not possible as a sink at the moment or am I just missing something? If so, could you please give me a hint which way to go?

Thank for support!

Side question:
In contrast to the guide I couldn't find any packages for libopenaptx/libopenaptx-dev in the Raspbian Repository. Can someone confirm that this also has to be installed form source?

Arkq, 2021-08-03:
> As far as I could see, ldacBT only creates ldacBT_abr and ldacBT_enc but never (also in earlier versions) ldacBT_dec.

See this (last example): https://github.com/Arkq/bluez-alsa/wiki/Installation-from-source#additional-dependencies For decoding support you need full version of Sony's LDAC library (not the one available in AOSP). For that you will need to contact with Sony. Also please note that the library which Sony provides does not have pkg-config support, you will have to provide correct cflags and libs as in the wiki example.

borine, 2021-08-03:
> I couldn't find any packages for libopenaptx/libopenaptx-dev in the Raspbian Repository.

Debian introduces libopenaptx in release 11 ("bullseye") due to be released this month. It usually takes a while for Raspios to be updated to a new upstream Debian though. Ubuntu introduced it in release 20.10 "Groovy Gorilla".

anonymix007, 2021-08-05:
@Arkq does [this](https://github.com/hegdi/libldacdec) LDAC decoder library work out of the box? Or some modifications are required?

Arkq, 2021-08-05:
It won't work out of the box, it uses different API than Sony's LDAC. You
will have to modify BlueALSA decoder function.

anonymix007, 2021-08-05:
There are two possible solutions:
1. Add support to BlueALSA for that library
2. Edit library's API
Which one should be preferred?

Arkq, 2021-08-06:
It depends. If this library is already widely used and packaged with few
distributions, we can add support for it in bluez-alsa. However, if this
library is not widely known, I would opt for API change, or adding new API
as an adaptation layer for official Sony's LDAC library. The benefit of
having the same API is a seamless change between different implementation
of the same algorithm.

The Sony's LDAC API is not complicated. It uses the same handle as encoder
- getHandle function is the same. Then there are two functions, one for
codec initialization and one for actual decoding. API alone is not a
subject for patents, so using the same API will not break any law.

LDACBT_API int
ldacBT_init_handle_decode(
  HANDLE_LDAC_BT hLdacBT,
  int cm,
  int sf,
  int nshift,
  int var1,
  int var2 );

LDACBT_API int
ldacBT_decode(
  HANDLE_LDAC_BT hLdacBT,
  unsigned char *p_stream,
  void *p_pcm,
  LDACBT_SMPL_FMT_T fmt,
  int stream_sz,
  int *used_bytes,
  int *pcm_sz );

anonymix007, 2021-08-06:
Okay, I will give that a try. Given that probably no one uses this library, API change or adaptation layer seems to be better way.

anonymix007, 2021-08-11:
I added adaptation layer, it probably works, but I'm not able to test it and don't want to release without any tests.
I'm on LMDE 4, EHfive' spulseaudio bluetooth module was deleted (maybe incorrectly? At least, I can't see them in pacmd list-modules). I can remove device, pair and after few seconds it disconnects:
```
[CHG] Device XX:XX:XX:XX:XX:XX ServicesResolved: no
[CHG] Device XX:XX:XX:XX:XX:XX Connected: no
```
And then `connect XX:XX:XX:XX:XX:XX` gives
```
Failed to connect: org.bluez.Error.Failed
```
Any ideas?

Arkq, 2021-08-14:
Try connecting instead of pairing. That's first idea. Also, please check whether your host running bluealsa and remote device have matching Bluetooth profiles. Maybe there is nothing to connect. Also, sometimes it is required to have HFP enabled with wide band speech capability (some phones require that, e.g. iPhone).

anonymix007, 2021-08-14:
@Arkq 
It now connects (but only if connection is initiated on both devices and not every time) and plays audio with SBC, AAC, aptX (not HD though). But with LDAC it gives no sound probably because of this error: `Bluetooth: Dropping L2CAP data: receive buffer overflow`

Arkq, 2021-08-14:
Are you using Android with your tests? I've seen such errors when streaming
AAC from Android phone which was not conformant with the Bluetooth spec.
But it was long time ago (4 year maybe) when AAC was not in mosts Android
phones.

anonymix007, 2021-08-14:
@Arkq 
Yes, Android. Will look with Raspberry Pi later, but I think that problem is with library. What is receive buffer? Is it pcm.data or something else? What is it's size?

Arkq, 2021-08-14:
Buffer for received data is "bt.data" and it's size should be equal to
transport MTU. But the error you've got is from BlueZ, I think.

anonymix007, 2021-08-14:
@Arkq 
What actually happens? I can see debug messages from decoder library. Does this mean it actually decodes something? Or it receives no data to decode?

Arkq, 2021-08-14:
If you can see logs from decoder library, then you have to make sure that
the encoder (e.g. Android) has encoded the stream properly. But, "L2CAP"
log message does not seem like a log from decoder. I might be wrong though,
I've never been using the decoding library which you are trying to adopt to
bluealsa.

anonymix007, 2021-08-14:
I used to have WH-1000XM4 with LDAC sink support and it worked, so I assume it encodes stream properly.
L2CAP is from gnome-logs (it's also visible in dmesg), decoder messages - from ldacBT_decode which is called by bluealsa (and therefore printf prints messages to the same console bluealsa prints its debug messages).

Arkq, 2021-08-14:
There are few approaches how to debug it. The most "low level" is to
register Bluetooth traffic on host with bluealsa, e.g. with tcpdump,
wireshark, etc. Then extract raw assembled A2DP frames (the easiest way is
to use wireshark), then from these frames extract LDAC data (strip RTP
header) and concatenate all these frames and use decoder library to decode
the stream. If it will work then you will have to debug bluealsa in a
similar way. Save to a file raw LDAC data (without RTP header) and compare
it with data recorded with wireshark. I follow these steps when I encounter
serious troubles with encoder/decoder.

Arkq, 2021-08-14:
There is also other option, but I'm not sure whether it will work (I was not using it since long time ago). In the `src/a2dp-sbc.c` there is a function `a2dp_sink_dump()`. Remove `static` from it and add it's declaration to e.g. `a2dp.h`. Then in the file `src/a2dp-ldac.c` in line 345 replace `a2dp_ldac_dec_thread` with `a2dp_sink_dump`. This should create simple stream dump in  `/tmp` directory. Play few seconds of LDAC audio and attach the dump file here with log lines "BT read: xxx" which will say how long the frames are in the dump file. Then I will try to feed these data directly into the bluealsa and check whether Sony's decoder will decode your stream.

anonymix007, 2021-08-14:
@Arkq 
I added `a2dp_sink_dump` declaration to `ba-transport.h` because `a2dp.h` gave compilation error:In file included from 
```
../../src/ba-transport.h:24,
                 from ../../src/a2dp-sbc.h:19,
                 from ../../src/a2dp-sbc.c:11:
../../src/a2dp.h:131:29: warning: ‘struct ba_transport_thread’ declared inside parameter list will not be visible outside of this definition or declaration
 void *a2dp_sink_dump(struct ba_transport_thread *th);
                             ^~~~~~~~~~~~~~~~~~~
../../src/a2dp-sbc.c:317:7: error: conflicting types for ‘a2dp_sink_dump’
 void *a2dp_sink_dump(struct ba_transport_thread *th) {
       ^~~~~~~~~~~~~~
In file included from ../../src/ba-transport.h:24,
                 from ../../src/a2dp-sbc.h:19,
                 from ../../src/a2dp-sbc.c:11:
../../src/a2dp.h:131:7: note: previous declaration of ‘a2dp_sink_dump’ was here
 void *a2dp_sink_dump(struct ba_transport_thread *th);
       ^~~~~~~~~~~~~~
```
BT read: 669
[ba-a2dp-sink--ldac-.dump.tar.gz](https://github.com/Arkq/bluez-alsa/files/6986804/ba-a2dp-sink--ldac-.dump.tar.gz)

Arkq, 2021-08-14:
Ok, I will check that but in ~2 hours. Now I'm checking tomatoes and
cucumbers in grocery store :)

Arkq, 2021-08-14:
I assumed that the mode was stereo, 96000 Hz. There were no errors during decoding.
Decoded wave file: [decoded-LDAC.wav.zip](https://github.com/Arkq/bluez-alsa/files/6987146/decoded-LDAC.wav.zip)

anonymix007, 2021-08-14:
@Arkq
This is in logs then phone does not connect to PC: `SET_CONFIGURATION request rejected: Stream End Point in Use (19)`.

I also made my own sink [dump.zip](https://github.com/Arkq/bluez-alsa/files/6987261/tmp.zip). Here is the code for dump (also available in my libldacdec [fork](https://github.com/anonymix007/libldacdec))
```
int ldacBT_decode(HANDLE_LDAC_BT hLdacBT, uchar *p_stream, void *p_pcm, LDACBT_SMPL_FMT_T fmt, int stream_sz, int *used_bytes, int *pcm_sz){
    (void)stream_sz;
    ldacdec_t *dec = (ldacdec_t *)hLdacBT->hLDAC;
    
    FILE *fp = fopen("/tmp/ldac.dat","ab");	
    FILE *raw = fopen("/tmp/ldac.pcm","ab");	
    if(fp==NULL){
        fp=fopen("/tmp/ldac.dat","wb");	
    }
    if(raw==NULL){
        fp=fopen("/tmp/ldac.pcm","wb");	
    }    
    fwrite(p_stream, 1, stream_sz, fp);
    
    printf("LIBLDACDEC: Format: %d, ", fmt);
    
    if(fmt != LDACBT_SMPL_FMT_S16 && fmt != LDACBT_SMPL_FMT_S32 && fmt != LDACBT_SMPL_FMT_F32) return hLdacBT->error_code_api=517; //FIXME: Add other PCM types;
    
    if(hLdacBT->proc_mode != LDACBT_PROCMODE_DECODE) return hLdacBT->error_code_api=1000;

    int result = ldacDecode_type(dec, p_stream, p_pcm, used_bytes, fmt); 
    *pcm_sz = dec->frame.frameSamples;
    printf("decode result: %d, frameSamples: %d\n", result, dec->frame.frameSamples);
    if(result == -1) return hLdacBT->error_code_api=516;
   
    fclose(fp);
    fclose(raw);
    return  hLdacBT->error_code_api=result;
}
```
This should be just raw LDAC data and it is decoded by library but sound is not what I expect: 
[output.wav.zip](https://github.com/Arkq/bluez-alsa/files/6987262/output.wav.zip), it looks like channels were not deinterleaved.
Raw file is also not good.

Maybe I am using some params wrong? Can you please give short brief about all ldacBT_decode parameters?

Arkq, 2021-08-16:
```c
/* LDAC decode processing.
 *
 *  Arguments
 *      hLdacBt    HANDLE_LDAC_BT    LDAC handle.
 *      p_stream   unsigned char *   Input "ldac_transport_frame" sequence.
 *      p_pcm      void *            Output PCM signal sequence.
 *      fmt        LDACBT_SMPL_FMT_T Audio format type of output pcm.
 *      stream_sz  int               Size of input data. Unit:Byte.
 *      used_bytes int *             Used bytes of input data. Unit:Byte.
 *      pcm_sz     int *             Size of output PCM signal sequence. Unit:Byte.
 *  Return value
 *      int : 0 for success, -1 for failure.
 */
LDACBT_API int ldacBT_decode( HANDLE_LDAC_BT hLdacBT, unsigned char *p_stream, void *p_pcm,
                              LDACBT_SMPL_FMT_T fmt, int stream_sz, int *used_bytes, int *pcm_sz );
```

Definition of `LDACBT_SMPL_FMT_T` you've got in `inc/ldacBT.h` in AOSP libldac.

anonymix007, 2021-08-16:
@Arkq 
Yes, looks like I used them correct. I have this
```
LIBLDACDEC: Format: 4, LDACDEC: syncword: aa
result: 0, pcm_sz: 256, stream_sz: 656, used_bytes: 164
LIBLDACDEC: Format: 4, LDACDEC: syncword: aa
result: 0, pcm_sz: 256, stream_sz: 492, used_bytes: 164
LIBLDACDEC: Format: 4, LDACDEC: syncword: aa
result: 0, pcm_sz: 256, stream_sz: 328, used_bytes: 164
LIBLDACDEC: Format: 4, LDACDEC: syncword: aa
result: 0, pcm_sz: 256, stream_sz: 164, used_bytes: 164
```
in my logs continuously, it looks fine to me. I also tried 2 different phones: Galaxy A51 and Redmi 8A. Both of them didn't work but in different ways: A51 segfaulted my library (probably because I called fwrite with NULL argument, but it was caused by another error):
```
bluealsa: D: ../../src/a2dp-sbc.c:336: Opening BT dump file: /tmp/ba-a2dp-sink--ldac-.dump
bluealsa: D: ../../src/a2dp-sbc.c:350: IO loop: START: a2dp_sink_dump: A2DP Sink (LDAC)
bluealsa: D: ../../src/a2dp-sbc.c:358: BT read: 668
bluealsa: D: ../../src/a2dp-sbc.c:358: BT read: 17
...
bluealsa: D: ../../src/a2dp-sbc.c:358: BT read: 668
bluealsa: D: ../../src/a2dp-sbc.c:358: BT read: 13
bluealsa: D: ../../src/a2dp-sbc.c:358: BT read: 668
bluealsa: D: ../../src/a2dp-sbc.c:358: BT read: 17
...
bluealsa: D: ../../src/a2dp-sbc.c:358: BT read: 668
bluealsa: D: ../../src/a2dp-sbc.c:358: BT read: 13
...
bluealsa: D: ../../src/a2dp-sbc.c:358: BT read: 668
bluealsa: D: ../../src/a2dp-sbc.c:358: BT read: 17
```
[A51 dump](https://github.com/Arkq/bluez-alsa/files/6992972/ba-a2dp-sink--ldac-a51.dump.zip), [full logs](https://github.com/Arkq/bluez-alsa/files/6993038/logs.txt) and [decoder logs](https://github.com/Arkq/bluez-alsa/files/6993039/logs2.txt) attached. Interesting parts are (negative stream size?):
```
LIBLDACDEC: Format: 4, LDACDEC: syncword: aa
result: 0, pcm_sz: 256, stream_sz: 655, used_bytes: 164
LIBLDACDEC: Format: 4, LDACDEC: syncword: aa
result: 0, pcm_sz: 256, stream_sz: 491, used_bytes: 164
LIBLDACDEC: Format: 4, LDACDEC: syncword: aa
result: 0, pcm_sz: 256, stream_sz: 327, used_bytes: 164
LIBLDACDEC: Format: 4, LDACDEC: syncword: aa
result: 0, pcm_sz: 256, stream_sz: 163, used_bytes: 164
LIBLDACDEC: Format: 4, LDACDEC: syncword: aa
result: 0, pcm_sz: 256, stream_sz: 0, used_bytes: 164
LIBLDACDEC: Format: 4, LDACDEC: syncword: aa
result: 0, pcm_sz: 256, stream_sz: -164, used_bytes: 164
LIBLDACDEC: Format: 4, LDACDEC: syncword: aa
result: 0, pcm_sz: 256, stream_sz: -328, used_bytes: 164
LIBLDACDEC: Format: 4, LDACDEC: syncword: aa
result: 0, pcm_sz: 256, stream_sz: -492, used_bytes: 164
LIBLDACDEC: Format: 4, LDACDEC: syncword: 0
result: -1, pcm_sz: 256, stream_sz: -656, used_bytes: 164
bluealsa: D: ../../src/utils.c:647: Unknown error code: 0xffffffee (API: 4095, handle: 1023, block: 1006)
bluealsa: E: ../../src/a2dp-ldac.c:307: LDAC decoding error: Unknown error
*** Segmentation fault
```
Illegal sync word (probably related to BT read: 13/17, it happened before encoding parameters were changed):
```
LIBLDACDEC: Format: 4, LDACDEC: syncword: aa
result: 0, pcm_sz: 256, stream_sz: 655, used_bytes: 220
LIBLDACDEC: Format: 4, LDACDEC: syncword: aa
result: 0, pcm_sz: 256, stream_sz: 435, used_bytes: 220
LIBLDACDEC: Format: 4, LDACDEC: syncword: aa
result: 0, pcm_sz: 256, stream_sz: 215, used_bytes: 220
LIBLDACDEC: Format: 4, LDACDEC: syncword: 1
result: -1, pcm_sz: 256, stream_sz: 4, used_bytes: 220
bluealsa: E: ../../src/a2dp-ldac.c:307: LDAC decoding error: Invalid sync-word
```
And this is without parameter change (other try):
```
result: 0, pcm_sz: 256, stream_sz: 655, used_bytes: 110
LIBLDACDEC: Format: 4, LDACDEC: syncword: aa
result: 0, pcm_sz: 256, stream_sz: 545, used_bytes: 110
LIBLDACDEC: Format: 4, LDACDEC: syncword: aa
result: 0, pcm_sz: 256, stream_sz: 435, used_bytes: 110
LIBLDACDEC: Format: 4, LDACDEC: syncword: aa
result: 0, pcm_sz: 256, stream_sz: 325, used_bytes: 110
LIBLDACDEC: Format: 4, LDACDEC: syncword: aa
result: 0, pcm_sz: 256, stream_sz: 215, used_bytes: 110
LIBLDACDEC: Format: 4, LDACDEC: syncword: aa
result: 0, pcm_sz: 256, stream_sz: 105, used_bytes: 110
LIBLDACDEC: Format: 4, LDACDEC: syncword: 1
result: -1, pcm_sz: 256, stream_sz: 4, used_bytes: 110
bluealsa: E: ../../src/a2dp-ldac.c:307: LDAC decoding error: Invalid sync-word
*** Segmentation fault
```
Redmi 8A gave no sound and empty dump file (0 bytes), with SBC it played music just fine.

Arkq, 2021-08-16:
If you've got negative stream, something is not right with the ldacBT_decode API call. The decoding loop expects that in the `used_bytes` there will be the number of bytes used in a single decoding step - number of bytes for one LDAC frame. Then, this loop iterates over all frames in the RTP packet. So the total bytes of A2DP transport frames should equal to `rtp.frames * LDAC_i_frame_len`. So, either the RTP header is incorrect, or returned `used_bytes` is too big.

anonymix007, 2021-08-16:
@Arkq 
Can you check A51 dump? It printed 3 different BT read values (previous dump had only 1), I think this is may be somehow related.

anonymix007, 2021-08-19:
@Arkq 
I found the error: sound was present (but it wasn't inaudible because internally library has PCM float with range not [-1.0;1.0] but [SHRT_MIN;SHRT_MAX]). With this fix I was able to decode my first and second dumps. For first it was necessary to only add offset of 13 to every frame, for second - `while(ptr[0]!=0xAA){ptr++;filePosition++;}` too. Some frames are just to small, I have no ideas why.

However, it still won't work, I can see 2 problems:
1. Sound is played only half of the time at double speed (and there is nothing in other half ). This is probably because I misused some parameters (BlueALSA expects something another in them), but what is used incorrect? Can you check it?
2. Playback (even bad) stops after few seconds. This is probably has something to do with this: `bluealsa-aplay: D: ../../../utils/aplay/aplay.c:480: Device marked as inactive: XX:XX:XX:XX:XX:XX` and `Bluetooth: Dropping L2CAP data: receive buffer overflow`. It may not be caused by library but it still ruins everything. And I don't know how to fix this. Any ideas?

It would be great if someone checked [library](https://github.com/anonymix007/libldacdec) and confirmed that this issue is present on other devices.

Arkq, 2021-11-19:
According to https://github.com/Arkq/bluez-alsa/issues/482#issuecomment-906060241 the work is done here. Since it's not a strict bluez-alsa project issue I'm closing it then.

closed 2021-11-19 by Arkq

artem, 2022-05-19:
> not sure about original library from issue #104: it won't even install

I originally extracted this library from some FiiO player running android FW, so it is obviously linked against Bionic libc. You can try using the lib with something like https://github.com/libhybris/libhybris

anonymix007, 2022-08-18:
@Arkq 
Is it possible that debugging info about MTU is incorrect?
I changed default MTU in kernel to 688 (and this is value which is shown), but I can't find it in Wireshark logs. It only shows 672 (which was default) and few other values (IIRC 512, 1024, 1008).

So it seems that MTU is set to 672, but everything still works. There's no buffer overflow probably because of that 688 in kernel.

Looks like that BlueALSA should set incoming MTU, but I can't find any code related to that. Is there any easy way to set it to at least 673 for better LDAC sink compatibility?

Arkq, 2022-08-18:
The MTU for read/write comes from BlueZ. See this: https://github.com/Arkq/bluez-alsa/blob/master/src/ba-transport.c#L556 BlueALSA only respects this value, that's all.

anonymix007, 2022-08-18:
Oh, I think somebody [tried](https://gitlab.freedesktop.org/pipewire/pipewire/-/issues/1852#note_1499560) to use my library in Pipewire and got the exact same issue. So this probably should be reported to BlueZ as bug?

andreadaoud, 2022-09-28:
Hi, Could anyone please tell me whether LDAC sink is currently stable? Are there any known problems (except that we have to change MTU)? Thank you very much!

anonymix007, 2022-09-28:
@andreadaoud 
Yes, it's pretty stable. You may always check by yourself.
One of the problems is with Samsung phones. For some reason they also transfer small packets of unknown data which are not properly decoded. Simple patch for BlueALSA fixes that, see issue in my libldacdec fork

andreadaoud, 2022-09-29:
@anonymix007 Thank you very much for your great work on libldacdec! One more question: is the fixed-point branch ready to use?

anonymix007, 2022-09-29:
@andreadaoud 
Yes, it works fine. But you don't need it unless you're going to use device without FPU.