arkq/bluez-alsa #484: bluetoothctl not found, is there any way to pair an audio bluetooth device

honoju, 2021-08-11:
Hello,

I used buildroot to build an embedded target and I included bluez-alsa, bluez-tools and bluez-utils 5.x as I intend to develop a bluetooth audio application.

When I try to use the bluetoothctl tool, it is not found. I searched in /bin and /usr/bin and it is not there (bluealsa is in /usr/bin).

Does bluealsa provide a tool to pair bluetooth audio devices?

Regards,

Arkq, 2021-08-11:
> Does bluealsa provide a tool to pair bluetooth audio devices?

No, pairing/connecting Bluetooth devices is out of the scope of bluez-alsa project. The `bluetoothctl` tool is part of bluez project.

borine, 2021-08-11:
I don't use buildroot, but I believe that bluetoothctl is an optional part of the build. Make sure you select `[*] build CLI client` under `[*] bluez-utils 5.x`

honoju, 2021-08-13:
Thanks for the reply. I had already selected the CLI client under bluez-utils 5;x but I still can't have the bluetoothctl tool. I am getting this error :

/bin/sh: bluetoothctl: not found  

That's may be due to my development board (Ingenic Halley5) or the buildroot version used (2020.02.1). I am going to try to change those 2 parameters and see if it makes any better.

borine, 2021-09-11:
As this is not an issue with BlueALSA, but rather with buildroot and/or Bluez, I am closing it now. Please look to those two projects for help with getting bluetoothctl to install.

closed 2021-09-11 by borine