arkq/bluez-alsa #488: Versioning and last-connected function

graham8, 2021-08-24:
We have been building bluez-alsa from de58004c.  `bluealsa --version` says v3.1.0 but it seems the 3.1.0 release was way later at b09f373e. Is this expected?

I stumbled on this trying to get the last-connected function to work.  My BT speaker connects fine with an explicit conf node in .asoundrc but doesn't work if I rely on 20-bluealsa.conf (even building b09f373e).  There's no bluealsa in `aplay -L`.  `speaker-test -D bluealsa` doesn't complain but I get no sound.  How to troubleshoot?

borine, 2021-08-24:
> Is this expected?

I've just built from https://github.com/Arkq/bluez-alsa/commit/de58004c440f34078b6b138f8c7155ae36b26a10 using the wiki guide and I get
```shell
$ bluealsa --version
v3.0.0-16-gde58004
$
```
So I'm guessing there's somethng amiss with your build/install environment/process

> trying to get the last-connected function to work

I've now built from https://github.com/Arkq/bluez-alsa/commit/b09f373ea7dbc6e3ecbcb74d7299f5230cdc6e59
```shell
$ bluealsa --version
v3.1.0
$
```

To get the default bluealsa PCM to show with `aplay -L` it is necessary to enable extended namehints in the alsa config:
```
$ cat ~/.asoundrc
defaults.namehint.extended on
$
$ aplay -L
...
bluealsa
    Bluetooth Audio Hub
...
$
```

> speaker-test -D bluealsa doesn't complain but I get no sound

Me too. Looking at the debug output, the stream never actually starts: I see
```
D: ../../../src/asound/bluealsa-pcm.c:528: /org/bluealsa/hci0/dev_C4_67_B5_37_26_FC/a2dpsrc/sink: Prepared
 1 - Front Right
D: ../../../src/asound/bluealsa-pcm.c:381: /org/bluealsa/hci0/dev_C4_67_B5_37_26_FC/a2dpsrc/sink: Stopping
D: ../../../src/asound/bluealsa-pcm.c:528: /org/bluealsa/hci0/dev_C4_67_B5_37_26_FC/a2dpsrc/sink: Prepared
Time per period = 0.623431
 0 - Front Left
D: ../../../src/asound/bluealsa-pcm.c:381: /org/bluealsa/hci0/dev_C4_67_B5_37_26_FC/a2dpsrc/sink: Stopping
D: ../../../src/asound/bluealsa-pcm.c:528: /org/bluealsa/hci0/dev_C4_67_B5_37_26_FC/a2dpsrc/sink: Prepared
 ```

I've never tried `speaker-test` with bluealsa before, so I don't know if this has always been the result. However, aplay, arecord and my other applications all work OK with this build of bluealsa (except for portaudio applications such as `audacity` - see #487.

borine, 2021-08-24:
So strange things lurk within `speaker-test`. It requests a buffer size of the maximum available, then sets playback to start when the buffer is full. For the system default device I get:
```shell
$ speaker-test -D default

speaker-test 1.2.2

Playback device is default
Stream parameters are 48000Hz, S16_LE, 1 channels
Using 16 octaves of pink noise
Rate set to 48000Hz (requested 48000Hz)
Buffer size range from 2048 to 16384
Period size range from 1024 to 1024
Using max buffer size 16384
Periods = 4
was set period_size = 1024
was set buffer_size = 16384
```

So it has max buffer size of 16384 frames (~ 341 ms)

but for plughw:0,0 I get
```shell
 speaker-test -D plughw:0,0

speaker-test 1.2.2

Playback device is plughw:0,0
Stream parameters are 48000Hz, S16_LE, 1 channels
Using 16 octaves of pink noise
Rate set to 48000Hz (requested 48000Hz)
Buffer size range from 64 to 8544000
Period size range from 32 to 4272000
Using max buffer size 8544000
Periods = 4
Unable to set hw params for playback: Cannot allocate memory
Setting of hwparams failed: Cannot allocate memory
```

So it now has max buffer of 8544000 frames (~ 3 minutes !!), and the system refuses to allocate that amount memory. I cannot explain why the max buffer sizes differ.

With bluealsa I get:
```shell
$ speaker-test -D bluealsa

speaker-test 1.2.2

Playback device is bluealsa
Stream parameters are 48000Hz, S16_LE, 1 channels
Using 16 octaves of pink noise
D: ../../../src/asound/bluealsa-pcm.c:1057: Getting BlueALSA PCM: PLAYBACK 00:00:00:00:00:00 a2dp
D: ../../../src/asound/bluealsa-pcm.c:921: /org/bluealsa/hci0/dev_C4_67_B5_37_26_FC/a2dpsrc/sink: Setting constraints
Rate set to 48000Hz (requested 48000Hz)
Buffer size range from 960 to 524288
Period size range from 480 to 262144
Using max buffer size 524288
Periods = 4
```

So now the buffer is 524288 frames (~ 11 seconds). The system is able to allocate that memory, but each test sample is less than 11 seconds, so the start threshold is never reached and playback never starts - hence no sound from the speaker.

Basically, I think speaker-test is broken, there is no way I can see to enable it to work with bluealsa. It doesn't even work with plughw.

graham8, 2021-08-24:
Hmmm. No luck so far.

```
osmc@vero4kp2:~$ cat .asoundrc
defaults.namehint.extended on

osmc@vero4kp2:~$ aplay -L | grep blue
osmc@vero4kp2:~$
```

borine, 2021-08-24:
is that with bluez-alsa release 3.1.0 (commit https://github.com/Arkq/bluez-alsa/commit/b09f373ea7dbc6e3ecbcb74d7299f5230cdc6e59) ?
what version of ALSA are you running, and where is the file 20-bluealsa.conf installed ?
is the installed version of 20-bluealsa.conf the same as the one in the source (src/asound/20-bluealsa.conf) ?

graham8, 2021-08-24:
Hold on.  Forgot to connect first.

borine, 2021-08-24:
> Hold on. Forgot to connect first.

No need to connect. `aplay -L` reports the namehint from 20-bluealsa.conf. It does to try to open the device

graham8, 2021-08-24:
So I pulled b09f373 and built it and the .deb it makes includes 20-bluealsa.cof as pulled an writes it to alsa.conf.d.  Then I have a link in /etc/alsa/conf.d
```
osmc@vero4kp2:/etc/alsa/conf.d$ ls -l 20*
lrwxrwxrwx 1 root root 44 Aug 24 09:19 20-bluealsa.conf -> /usr/share/alsa/alsa.conf.d/20-bluealsa.conf
```
Not sure how to get the alsa version but I think alsa-utils uses the same numbers
```
osmc@vero4kp2:/etc/alsa/conf.d$ dpkg -l | grep alsa
ii  alsa-utils                            1.1.8-2                             armhf        Utilities for configuring and using ALSA
```

borine, 2021-08-24:
OK, I'm using alsa 1.2.2, but would be very surprised if that made any difference to the namehint processing. The path that libasound 1.1.8 looks in is `/etc/alsa/conf.d/20-bluealsa.conf`. Within the file/link at that path there should be:
```
defaults.bluealsa.device "00:00:00:00:00:00"
```

and

```
pcm.bluealsa {
	@args [ DEV PROFILE DELAY SRV ]
	@args.DEV {
		type string
		default {
			@func refer
			name defaults.bluealsa.device
		}
	}
	@args.PROFILE {
		type string
		default {
			@func refer
			name defaults.bluealsa.profile
		}
	}
	@args.DELAY {
		type integer
		default {
			@func refer
			name defaults.bluealsa.delay
		}
	}
	@args.SRV {
		type string
		default {
			@func refer
			name defaults.bluealsa.service
		}
	}
	type plug
	slave.pcm {
		type bluealsa
		service $SRV
		device $DEV
		profile $PROFILE
		delay $DELAY
	}
	hint {
		show {
			@func refer
			name defaults.namehint.extended
		}
		description "Bluetooth Audio Hub"
	}
}
```

That should be all that is needed for `aplay -L` to include bluealsa - unless some other file is overriding some part of those definitions.

graham8, 2021-08-24:
I can't see any difference, and ALSA is picking up other files/links in /etc/alsa/conf.d.
```
osmc@vero4kp2:/etc/alsa/conf.d$ cat 20-bluealsa.conf
# BlueALSA integration setup

# By default the predefined "bluealsa" PCM will use
# the most recently connected Bluetooth device.
defaults.bluealsa.device "00:00:00:00:00:00"
# Default to A2DP connection because that is what
# most people want to use - high quality audio.
defaults.bluealsa.profile "a2dp"
defaults.bluealsa.delay 0
defaults.bluealsa.battery "yes"
defaults.bluealsa.service "org.bluealsa"

ctl.bluealsa {
        @args [ BAT SRV ]
        @args.BAT {
                type string
                default {
                        @func refer
                        name defaults.bluealsa.battery
                }
        }
        @args.SRV {
                type string
                default {
                        @func refer
                        name defaults.bluealsa.service
                }
        }
        type bluealsa
        service $SRV
        battery $BAT
}

pcm.bluealsa {
        @args [ DEV PROFILE DELAY SRV ]
        @args.DEV {
                type string
                default {
                        @func refer
                        name defaults.bluealsa.device
                }
        }
        @args.PROFILE {
                type string
                default {
                        @func refer
                        name defaults.bluealsa.profile
                }
        }
        @args.DELAY {
                type integer
                default {
                        @func refer
                        name defaults.bluealsa.delay
                }
        }
        @args.SRV {
                type string
                default {
                        @func refer
                        name defaults.bluealsa.service
                }
        }
        type plug
        slave.pcm {
                type bluealsa
                service $SRV
                device $DEV
                profile $PROFILE
                delay $DELAY
        }
        hint {
                show {
                        @func refer
                        name defaults.namehint.extended
                }
                description "Bluetooth Audio Hub"
        }
}
```

borine, 2021-08-24:
ahh, my bad

I've just installed alsa-lib 1.1.8:

```shell
$ aplay -L | grep bluealsa
$
```

So it seems namehint processing has changed between alsa-lib 1.1.8 and 1.2.2. That does not affect bluealsa per se, but it does 
explain the `aplay -L` result

graham8, 2021-08-24:
Still, if 20-bluealsa.conf is where alsa can see it, why wouldn't we see bluealsa as a candidate pcm for aplay?

borine, 2021-08-24:
Because aplay -L invokes the namehint API `snd_device_name_hint()`. When an aplay opens a PCM it to play a file invokes the pcm API `snd_pcm_open()`. these are completely different code paths within libasound, they even operate on different internal copies of the alsa configuration. So aplay *can* play to a device (so long as it is connected) whether `aplay -L` lists it or not.

graham8, 2021-08-24:
Sort of got it.  aplay -D bluealsa is indeed working. But Kodi can't see the bluealsa device.  It uses snd_device_get_name_hint(). We'll figure something out.  Goal eventually is to have one node per device but I was looking for something that would work OOTB while we develop the auto configuration you and I have discussed elsewhere.

closed 2021-08-24 by graham8

borine, 2021-08-25:
> something that would work OOTB

either of the following will work, the only drawback is that both show the bluetooth output even when none is connected:

1. add an explicit node in your local config, with a hint, but do not call it "bluealsa": e.g

```
pcm.bluetooth {
    type empty
    slave.pcm "bluealsa"
    hint.show on
    hint.description "Bluetooth Audio"
}
```

or

2. add a namehint node to your local configuration: e.g.
```
namehint.pcm.bluetooth "bluealsa|DESCBluetooth Audio|IOIDOutput"
```

graham8, 2021-08-25:
Thanks. I had in mind something like (1).