arkq/bluez-alsa #490: Building gives an error

tommyzat, 2021-09-07:
Hello!

I'm trying to build alsa using ../configure --enable-aptx --with-libopenaptx --enable-msbc on Raspbian Buster after having installed openaptx following these instructions (https://github.com/Arkq/openaptx)

At first I am getting "No package 'libopenaptx' found", after which I set the APTX_CFLAGS and APTX_LIBS:
export APTX_LIBS="-I/usr/local/bin"
export APTX_CFLAGS="-I/usr/local/include"

The configure command then goes well, and when executing make I get this:

```
/usr/bin/ld: codec-aptx.o: in function `aptxenc_init':
/home/pi/blue-alsa/bluez-alsa/build/src/../../src/codec-aptx.c:32: undefined reference to `aptx_init'
/usr/bin/ld: codec-aptx.o: in function `aptxenc_encode':
/home/pi/blue-alsa/bluez-alsa/build/src/../../src/codec-aptx.c:130: undefined reference to `aptx_encode'
/usr/bin/ld: codec-aptx.o: in function `aptxdec_decode':
/home/pi/blue-alsa/bluez-alsa/build/src/../../src/codec-aptx.c:168: undefined reference to `aptx_decode_sync'
/usr/bin/ld: codec-aptx.o: in function `aptxenc_destroy':
/home/pi/blue-alsa/bluez-alsa/build/src/../../src/codec-aptx.c:320: undefined reference to `aptx_finish'
/usr/bin/ld: codec-aptx.o: in function `aptxdec_destroy':
/home/pi/blue-alsa/bluez-alsa/build/src/../../src/codec-aptx.c:320: undefined reference to `aptx_finish'
```

tommyzat, 2021-09-08:
I tried both L and I (capital I) for both export values. I'm not even sure if the folder content I'm pointing to is correct (see below):

APTX_LIBS (usr/local/bin) contains: aptxdec, aptxenc, aptxhddec, aptxhdenc
APTX_CFLAGs (usr/local/include) contains: openaptx.h

For reference, this is what openaptx outputs after "make install"

-- Up-to-date: /usr/local/lib/pkgconfig/openaptx.pc
-- Up-to-date: /usr/local/lib/pkgconfig/openaptxhd.pc
-- Up-to-date: /usr/local/share/man/man3/openaptx.h.3
-- Up-to-date: /usr/local/lib/libaptx.so
-- Up-to-date: /usr/local/include/openaptx.h
-- Up-to-date: /usr/local/lib/libaptxHD.so
-- Up-to-date: /usr/local/include/openaptx.h
-- Up-to-date: /usr/local/bin/aptxdec
-- Up-to-date: /usr/local/bin/aptxhddec
-- Up-to-date: /usr/local/bin/aptxenc
-- Up-to-date: /usr/local/bin/aptxhdenc

From here I chose what I thought was correct for the LIBS and CFLAGS

borine, 2021-09-08:
After installing openaptx, did you run
```shell
sudo ldconfig /usr/local/lib
```

tommyzat, 2021-09-08:
I have now typed the ldconfig command.

Now with:

export APTX_LIBS="-L/usr/local/lib"
export APTX_CFLAGS="-I/usr/local/include"

, I am still getting the same error as before.

/usr/local/lib contains libaptx.so, libaptxHD.so and a pkgconfig folder with openaptx.pc, openaptxhp.pc
/usr/local/include contains a file - openaptx.h

borine, 2021-09-09:
> I'm trying to build alsa using ../configure --enable-aptx --with-libopenaptx --enable-msbc on Raspbian Buster after having installed openaptx following these instructions (https://github.com/Arkq/openaptx)

Sorry, I've been a bit slow to spot the problem here. There are two alternative aptX inplementations that can be used by BlueALSA, and you need to select just one of them. The configure line above selects `libopenaptx` by @pali, whereas your aptX installation is for `openaptx` by @Arkq. The close similarity in names does create confusion for may users!

I've edited the BlueALSA "Installation from source" wiki page to try to clarify things a little. The @pali `libopenaptx` library is recommended, that is the one I use.

tommyzat, 2021-09-14:
Thank you! It's much clearer now.

closed 2021-09-14 by tommyzat

tommyzat, 2021-09-16:
I tried a fresh raspbian installation. Compiled fdk-aac and libopenaptx, and then bluealsa with `../configure --enable-aac --enable-aptx --enable-aptx-hd --with-libopenaptx --enable-ofono --enable-debug --disable-payloadcheck` (because raspbian runs on pulseaudio 12.2 < 13).

I didn't disable any pulse bluetooth audio modules since I'm following the **systemd integration** from https://github.com/Arkq/bluez-alsa/wiki/Systemd-integration.
Bluealsa is started using a service unit file `sudo systemctl enable bluealsa.service`
Bluealsa-aplay.service was not implemented.

systemctl bluetooth status
```
Sep 16 00:46:40 raspberrypi bluetoothd[663]: Endpoint registered: sender=:1.54 path=/org/bluez/hci0/A2DP/SBC/Sink/1
Sep 16 00:46:40 raspberrypi bluetoothd[663]: Endpoint registered: sender=:1.54 path=/org/bluez/hci0/A2DP/SBC/Sink/2
Sep 16 00:46:40 raspberrypi bluetoothd[663]: Endpoint unregistered: sender=:1.54 path=/org/bluez/hci0/A2DP/aptXHD/Sink/1
Sep 16 00:46:40 raspberrypi bluetoothd[663]: Endpoint unregistered: sender=:1.54 path=/org/bluez/hci0/A2DP/aptXHD/Sink/2
Sep 16 00:46:40 raspberrypi bluetoothd[663]: Endpoint unregistered: sender=:1.54 path=/org/bluez/hci0/A2DP/aptX/Sink/1
Sep 16 00:46:40 raspberrypi bluetoothd[663]: Endpoint unregistered: sender=:1.54 path=/org/bluez/hci0/A2DP/aptX/Sink/2
Sep 16 00:46:40 raspberrypi bluetoothd[663]: Endpoint unregistered: sender=:1.54 path=/org/bluez/hci0/A2DP/AAC/Sink/1
Sep 16 00:46:40 raspberrypi bluetoothd[663]: Endpoint unregistered: sender=:1.54 path=/org/bluez/hci0/A2DP/AAC/Sink/2
Sep 16 00:46:40 raspberrypi bluetoothd[663]: Endpoint unregistered: sender=:1.54 path=/org/bluez/hci0/A2DP/SBC/Sink/1
Sep 16 00:46:40 raspberrypi bluetoothd[663]: Endpoint unregistered: sender=:1.54 path=/org/bluez/hci0/A2DP/SBC/Sink/2
```

bluealsa --help
```
Available BT profiles:
  - a2dp-source	Advanced Audio Source (aptX-HD, aptX, AAC, SBC)
```

sudo bluealsa:
```
...
bluealsa: D: ../../src/main.c:419: Acquiring D-Bus service name: org.bluealsa
bluealsa: D: ../../src/main.c:424: Starting main dispatching loop
bluealsa: E: ../../src/main.c:85: Couldn't acquire D-Bus name: org.bluealsa
bluealsa: D: ../../src/main.c:427: Exiting main loop
```

/etc/dbus-1/system.d/bluealsa.conf:
```
<!-- This configuration file specifies the required security policies
     for BlueALSA core daemon to work. -->

<!DOCTYPE busconfig PUBLIC "-//freedesktop//DTD D-BUS Bus Configuration 1.0//EN"
 "http://www.freedesktop.org/standards/dbus/1.0/busconfig.dtd">
<busconfig>

  <!-- ../system.conf have denied everything, so we just punch some holes -->

  <policy user="root">
    <allow own_prefix="org.bluealsa"/>
    <allow send_destination="org.bluealsa"/>
  </policy>

  <policy group="audio">
    <allow send_destination="org.bluealsa"/>
  </policy>

    <policy user="bluealsa">
        <allow own_prefix="org.bluealsa"/>
        <allow send_destination="org.bluealsa"/>
        <allow send_destination="org.bluez"/>
    </policy>

</busconfig>
```

I also added the policy in system-local.conf: `/etc/dbus-1/system-local.conf`
```
<!DOCTYPE busconfig PUBLIC
 "-//freedesktop//DTD D-BUS Bus Configuration 1.0//EN"
 "http://www.freedesktop.org/standards/dbus/1.0/busconfig.dtd">

<busconfig>

    <policy user="bluealsa">
        <allow own_prefix="org.bluealsa"/>
        <allow send_destination="org.bluealsa"/>
        <allow send_destination="org.bluez"/>
    </policy>

</busconfig>
```

/etc/default/bluealsa:
`OPTIONS="-p a2dp-sink"`

With this new raspberry pi installation, this time SBC doesn't work, so sink isn't working at all, BUT connection is succesful, I do have the AptX, AptX HD and AAC options. There's just no sound being output. I have the old installation on another SD card but I believe I may have followed other installation practices, so I chose to try this on a new installation.

I'd appreciate it if you could direct me to where bluealsa log is stored.