arkq/bluez-alsa #491: portaudio / Couldn't get BlueALSA PCM: PCM not found

telemetrieTP23, 2021-09-07:
I have (at least i thought that) a well configured Alsa/bluezalsa Bluetooth Environment on my Raspberry Pi4.
	Because of Performance/Latency Issues I don't want to use Pulseaudio thats Standard on rp4 since December 2020.
	
	playing a Soundfile with aplay on Bluetooth Speakers works perfect, same when using vlc:
	pi@raspberrypi:~ aplay arrow_x.wav 
	D: ../../../src/asound/bluealsa-pcm.c:1057: Getting BlueALSA PCM: PLAYBACK AC:20:19:08:92:D9 a2dp
        D: ../../../src/asound/bluealsa-pcm.c:921: /org/bluealsa/hci0/dev_AC_20_19_08_92_D9/a2dpsrc/sink: Setting constraints
        Wiedergabe: WAVE 'arrow_x.wav' : Unsigned 8 bit, Rate: 11025 Hz, mono

	But when starting Pyo (uses portaudio) this happens:

	D: ../../../src/asound/bluealsa-pcm.c:1057: Getting BlueALSA PCM: CAPTURE AC:20:19:08:92:D9 a2dp
	ALSA lib ../../../src/asound/bluealsa-pcm.c:1061:(_snd_pcm_bluealsa_open) Couldn't get BlueALSA PCM: PCM not found
	D: ../../../src/asound/bluealsa-pcm.c:1057: Getting BlueALSA PCM: PLAYBACK AC:20:19:08:92:D9 a2dp
	D: ../../../src/asound/bluealsa-pcm.c:921: /org/bluealsa/hci0/dev_AC_20_19_08_92_D9/a2dpsrc/sink: Setting constraints
	D: ../../../src/asound/bluealsa-pcm.c:431: /org/bluealsa/hci0/dev_AC_20_19_08_92_D9/a2dpsrc/sink: Closing
	D: ../../../src/asound/bluealsa-pcm.c:1057: Getting BlueALSA PCM: CAPTURE AC:20:19:08:92:D9 a2dp
	ALSA lib ../../../src/asound/bluealsa-pcm.c:1061:(_snd_pcm_bluealsa_open) Couldn't get BlueALSA PCM: PCM not found
	D: ../../../src/asound/bluealsa-pcm.c:1057: Getting BlueALSA PCM: PLAYBACK AC:20:19:08:92:D9 a2dp
	D: ../../../src/asound/bluealsa-pcm.c:921: /org/bluealsa/hci0/dev_AC_20_19_08_92_D9/a2dpsrc/sink: Setting constraints
	D: ../../../src/asound/bluealsa-pcm.c:431: /org/bluealsa/hci0/dev_AC_20_19_08_92_D9/a2dpsrc/sink: Closing
	D: ../../../src/asound/bluealsa-pcm.c:1057: Getting BlueALSA PCM: CAPTURE AC:20:19:08:92:D9 sco
	D: ../../../src/asound/bluealsa-pcm.c:921: /org/bluealsa/hci0/dev_AC_20_19_08_92_D9/hfpag/source: Setting constraints
	D: ../../../src/asound/bluealsa-pcm.c:431: /org/bluealsa/hci0/dev_AC_20_19_08_92_D9/hfpag/source: Closing
	D: ../../../src/asound/bluealsa-pcm.c:1057: Getting BlueALSA PCM: PLAYBACK AC:20:19:08:92:D9 sco
	D: ../../../src/asound/bluealsa-pcm.c:921: /org/bluealsa/hci0/dev_AC_20_19_08_92_D9/hfpag/sink: Setting constraints
	D: ../../../src/asound/bluealsa-pcm.c:431: /org/bluealsa/hci0/dev_AC_20_19_08_92_D9/hfpag/sink: Closing
	D: ../../../src/asound/bluealsa-pcm.c:1057: Getting BlueALSA PCM: CAPTURE AC:20:19:08:92:D9 sco
	D: ../../../src/asound/bluealsa-pcm.c:921: /org/bluealsa/hci0/dev_AC_20_19_08_92_D9/hfpag/source: Setting constraints
	D: ../../../src/asound/bluealsa-pcm.c:431: /org/bluealsa/hci0/dev_AC_20_19_08_92_D9/hfpag/source: Closing
	D: ../../../src/asound/bluealsa-pcm.c:1057: Getting BlueALSA PCM: PLAYBACK AC:20:19:08:92:D9 a2dp
	D: ../../../src/asound/bluealsa-pcm.c:921: /org/bluealsa/hci0/dev_AC_20_19_08_92_D9/a2dpsrc/sink: Setting constraints
	D: ../../../src/asound/bluealsa-pcm.c:431: /org/bluealsa/hci0/dev_AC_20_19_08_92_D9/a2dpsrc/sink: Closing
 
	It seems like that portaudio take the A2DP sink as CAPTURE Device even in my .asoundrc only sco is configured with Capture possibilites
	Even when I remove SCO from .asoundrc, its still the same:

	D: ../../../src/asound/bluealsa-pcm.c:1057: Getting BlueALSA PCM: CAPTURE AC:20:19:08:92:D9 a2dp
        ALSA lib ../../../src/asound/bluealsa-pcm.c:1061:(_snd_pcm_bluealsa_open) Couldn't get BlueALSA PCM: PCM not found
        D: ../../../src/asound/bluealsa-pcm.c:1057: Getting BlueALSA PCM: PLAYBACK AC:20:19:08:92:D9 a2dp
        D: ../../../src/asound/bluealsa-pcm.c:921: /org/bluealsa/hci0/dev_AC_20_19_08_92_D9/a2dpsrc/sink: Setting constraints
        D: ../../../src/asound/bluealsa-pcm.c:431: /org/bluealsa/hci0/dev_AC_20_19_08_92_D9/a2dpsrc/sink: Closing
        D: ../../../src/asound/bluealsa-pcm.c:1057: Getting BlueALSA PCM: CAPTURE AC:20:19:08:92:D9 a2dp
        ALSA lib ../../../src/asound/bluealsa-pcm.c:1061:(_snd_pcm_bluealsa_open) Couldn't get BlueALSA PCM: PCM not found
        D: ../../../src/asound/bluealsa-pcm.c:1057: Getting BlueALSA PCM: PLAYBACK AC:20:19:08:92:D9 a2dp
        D: ../../../src/asound/bluealsa-pcm.c:921: /org/bluealsa/hci0/dev_AC_20_19_08_92_D9/a2dpsrc/sink: Setting constraints
        D: ../../../src/asound/bluealsa-pcm.c:431: /org/bluealsa/hci0/dev_AC_20_19_08_92_D9/a2dpsrc/sink: Closing  
        D: ../../../src/asound/bluealsa-pcm.c:1057: Getting BlueALSA PCM: PLAYBACK AC:20:19:08:92:D9 a2dp
        D: ../../../src/asound/bluealsa-pcm.c:921: /org/bluealsa/hci0/dev_AC_20_19_08_92_D9/a2dpsrc/sink: Setting constraints
        D: ../../../src/asound/bluealsa-pcm.c:431: /org/bluealsa/hci0/dev_AC_20_19_08_92_D9/a2dpsrc/sink: Closing

	Output from  bluealsa-aplay --list-pcm:

	bluealsa:SRV=org.bluealsa,DEV=AC:20:19:08:92:D9,PROFILE=a2dp
		JLab Rewind, trusted audio-card, playback
    		A2DP (SBC): S16_LE 2 channels 48000 Hz
	bluealsa:SRV=org.bluealsa,DEV=AC:20:19:08:92:D9,PROFILE=sco
    		JLab Rewind, trusted audio-card, playback
    		SCO (CVSD): S16_LE 1 channel 8000 Hz
	bluealsa:SRV=org.bluealsa,DEV=AC:20:19:08:92:D9,PROFILE=sco
    		JLab Rewind, trusted audio-card, capture
    		SCO (CVSD): S16_LE 1 channel 8000 Hz

Maybe somebody has further infos about this.... Thx a lot.

telemetrieTP23, 2021-09-09:
thanks borine !

i tryed both, compiled the newest version of alsa-lib and setting --keep-alive=1.

Some new Messages when launching pyo now appearing, 
on the starting its:

ALSA lib dlmisc.c:170:(snd_dlsym_verify) unable to verify version for symbol snd_func_private_integer
ALSA lib conf.c:4556:(_snd_config_evaluate) symbol snd_func_private_integer is not defined inside (null)
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such device or address
ALSA lib conf.c:3694:(snd_config_hook_load) Unable to expand filenames in the pre-load section
ALSA lib conf.c:3572:(snd_config_hooks_call) function snd_config_hook_load_for_all_cards returned error: No such device or address

and after this lines:

D: ../../../src/asound/bluealsa-pcm.c:1057: Getting BlueALSA PCM: PLAYBACK AC:20:19:08:92:D9 sco
D: ../../../src/asound/bluealsa-pcm.c:921: /org/bluealsa/hci0/dev_AC_20_19_08_92_D9/hfpag/sink: Setting constraints
D: ../../../src/asound/bluealsa-pcm.c:431: /org/bluealsa/hci0/dev_AC_20_19_08_92_D9/hfpag/sink: Closing

there comes:

ALSA lib pcm_direct.c:2023:(snd1_pcm_direct_parse_open_conf) Unknown field tstamp_type

Maybe i post it in the github issues of pyo,portaudio and alsa-lib too...

borine, 2021-09-09:
Those ALSA lib messages suggest that either your portaudio library or your bluealsa build (or both) are not compatible with your build of alsa-lib. Does `aplay -L` still work? if not I suspect the problem may be in your alsa-lib build, or possibly your alsa-lib installation is now part new-build and part original? I'm not an expert on diagnosing alsa-lib errors, but it does appear to me in the above log messages that the clients have different symbol definitions from the library - perhaps a mix of headers from one alsa-lib version and libraries from another?

When I'm building alsa-lib on Ubuntu, I configure it with:
```shell
configure --libdir=/usr/lib/x86_64-linux-gnu/
```

because that is how the Ubintu package does it. You may need to modify your own alsa-lib configure options to make it compatible with your system.

telemetrieTP23, 2021-09-09:
i recompiled alsa-lib and before tried your hint:

configure --libdir=/usr/lib/x86_64-linux-gnu/

so the 6 lines of "new" errors are gone again.

Now I found out some same strange things:

Playing the 11khz arrow_x.wav file with aplay still works:

D: ../../../src/asound/bluealsa-pcm.c:1057: Getting BlueALSA PCM: PLAYBACK AC:20:19:08:92:D9 a2dp
D: ../../../src/asound/bluealsa-pcm.c:921: /org/bluealsa/hci0/dev_AC_20_19_08_92_D9/a2dpsrc/sink: Setting constraints
Wiedergabe: WAVE 'arrow_x.wav' : Unsigned 8 bit, Rate: 11025 Hz, mono
D: ../../../src/asound/bluealsa-pcm.c:443: /org/bluealsa/hci0/dev_AC_20_19_08_92_D9/a2dpsrc/sink: Initializing HW
D: ../../../src/asound/bluealsa-pcm.c:466: /org/bluealsa/hci0/dev_AC_20_19_08_92_D9/a2dpsrc/sink: FIFO buffer size: 1024 frames
D: ../../../src/asound/bluealsa-pcm.c:474: /org/bluealsa/hci0/dev_AC_20_19_08_92_D9/a2dpsrc/sink: Selected HW buffer: 4 periods x 24000 bytes == 96000 bytes
D: ../../../src/asound/bluealsa-pcm.c:489: /org/bluealsa/hci0/dev_AC_20_19_08_92_D9/a2dpsrc/sink: Initializing SW
D: ../../../src/asound/bluealsa-pcm.c:489: /org/bluealsa/hci0/dev_AC_20_19_08_92_D9/a2dpsrc/sink: Initializing SW
D: ../../../src/asound/bluealsa-pcm.c:489: /org/bluealsa/hci0/dev_AC_20_19_08_92_D9/a2dpsrc/sink: Initializing SW
D: ../../../src/asound/bluealsa-pcm.c:489: /org/bluealsa/hci0/dev_AC_20_19_08_92_D9/a2dpsrc/sink: Initializing SW
D: ../../../src/asound/bluealsa-pcm.c:528: /org/bluealsa/hci0/dev_AC_20_19_08_92_D9/a2dpsrc/sink: Prepared
D: ../../../src/asound/bluealsa-pcm.c:489: /org/bluealsa/hci0/dev_AC_20_19_08_92_D9/a2dpsrc/sink: Initializing SW
D: ../../../src/asound/bluealsa-pcm.c:345: /org/bluealsa/hci0/dev_AC_20_19_08_92_D9/a2dpsrc/sink: Starting
D: ../../../src/asound/bluealsa-pcm.c:212: /org/bluealsa/hci0/dev_AC_20_19_08_92_D9/a2dpsrc/sink: Starting IO loop: 8
D: ../../../src/asound/bluealsa-pcm.c:489: /org/bluealsa/hci0/dev_AC_20_19_08_92_D9/a2dpsrc/sink: Initializing SW
D: ../../../src/asound/bluealsa-pcm.c:496: /org/bluealsa/hci0/dev_AC_20_19_08_92_D9/a2dpsrc/sink: Changing SW avail min: 6000 -> 1
D: ../../../src/asound/bluealsa-pcm.c:489: /org/bluealsa/hci0/dev_AC_20_19_08_92_D9/a2dpsrc/sink: Initializing SW
D: ../../../src/asound/bluealsa-pcm.c:496: /org/bluealsa/hci0/dev_AC_20_19_08_92_D9/a2dpsrc/sink: Changing SW avail min: 1 -> 6000
D: ../../../src/asound/bluealsa-pcm.c:217: /org/bluealsa/hci0/dev_AC_20_19_08_92_D9/a2dpsrc/sink: Pausing IO thread
D: ../../../src/asound/bluealsa-pcm.c:381: /org/bluealsa/hci0/dev_AC_20_19_08_92_D9/a2dpsrc/sink: Stopping
D: ../../../src/asound/bluealsa-pcm.c:147: /org/bluealsa/hci0/dev_AC_20_19_08_92_D9/a2dpsrc/sink: IO thread cleanup
D: ../../../src/asound/bluealsa-pcm.c:381: /org/bluealsa/hci0/dev_AC_20_19_08_92_D9/a2dpsrc/sink: Stopping
D: ../../../src/asound/bluealsa-pcm.c:481: /org/bluealsa/hci0/dev_AC_20_19_08_92_D9/a2dpsrc/sink: Freeing HW
D: ../../../src/asound/bluealsa-pcm.c:431: /org/bluealsa/hci0/dev_AC_20_19_08_92_D9/a2dpsrc/sink: Closing

Playing a 48000 khz wav file with aplay dont works:

pi@raspberrypi:~ $ aplay pyo_rec.wav 
D: ../../../src/asound/bluealsa-pcm.c:1057: Getting BlueALSA PCM: PLAYBACK AC:20:19:08:92:D9 a2dp
D: ../../../src/asound/bluealsa-pcm.c:921: /org/bluealsa/hci0/dev_AC_20_19_08_92_D9/a2dpsrc/sink: Setting constraints
Wiedergabe: WAVE 'pyo_rec.wav' : Signed 16 bit Little Endian, Rate: 48000 Hz, stereo
D: ../../../src/asound/bluealsa-pcm.c:443: /org/bluealsa/hci0/dev_AC_20_19_08_92_D9/a2dpsrc/sink: Initializing HW
D: ../../../src/asound/bluealsa-pcm.c:466: /org/bluealsa/hci0/dev_AC_20_19_08_92_D9/a2dpsrc/sink: FIFO buffer size: 1024 frames
D: ../../../src/asound/bluealsa-pcm.c:474: /org/bluealsa/hci0/dev_AC_20_19_08_92_D9/a2dpsrc/sink: Selected HW buffer: 4 periods x 24000 bytes == 96000 bytes
D: ../../../src/asound/bluealsa-pcm.c:489: /org/bluealsa/hci0/dev_AC_20_19_08_92_D9/a2dpsrc/sink: Initializing SW
D: ../../../src/asound/bluealsa-pcm.c:489: /org/bluealsa/hci0/dev_AC_20_19_08_92_D9/a2dpsrc/sink: Initializing SW
D: ../../../src/asound/bluealsa-pcm.c:528: /org/bluealsa/hci0/dev_AC_20_19_08_92_D9/a2dpsrc/sink: Prepared
D: ../../../src/asound/bluealsa-pcm.c:489: /org/bluealsa/hci0/dev_AC_20_19_08_92_D9/a2dpsrc/sink: Initializing SW
D: ../../../src/asound/bluealsa-pcm.c:381: /org/bluealsa/hci0/dev_AC_20_19_08_92_D9/a2dpsrc/sink: Stopping
D: ../../../src/asound/bluealsa-pcm.c:381: /org/bluealsa/hci0/dev_AC_20_19_08_92_D9/a2dpsrc/sink: Stopping
D: ../../../src/asound/bluealsa-pcm.c:481: /org/bluealsa/hci0/dev_AC_20_19_08_92_D9/a2dpsrc/sink: Freeing HW
D: ../../../src/asound/bluealsa-pcm.c:431: /org/bluealsa/hci0/dev_AC_20_19_08_92_D9/a2dpsrc/sink: Closing

Playing the same 48khz file with vlc works:

D: ../../../src/asound/bluealsa-pcm.c:1057: Getting BlueALSA PCM: PLAYBACK AC:20:19:08:92:D9 a2dp
D: ../../../src/asound/bluealsa-pcm.c:921: /org/bluealsa/hci0/dev_AC_20_19_08_92_D9/a2dpsrc/sink: Setting constraints
D: ../../../src/asound/bluealsa-pcm.c:443: /org/bluealsa/hci0/dev_AC_20_19_08_92_D9/a2dpsrc/sink: Initializing HW
D: ../../../src/asound/bluealsa-pcm.c:466: /org/bluealsa/hci0/dev_AC_20_19_08_92_D9/a2dpsrc/sink: FIFO buffer size: 1024 frames
D: ../../../src/asound/bluealsa-pcm.c:474: /org/bluealsa/hci0/dev_AC_20_19_08_92_D9/a2dpsrc/sink: Selected HW buffer: 75 periods x 7680 bytes == 576000 bytes
D: ../../../src/asound/bluealsa-pcm.c:489: /org/bluealsa/hci0/dev_AC_20_19_08_92_D9/a2dpsrc/sink: Initializing SW
[... cut ...]
D: ../../../src/asound/bluealsa-pcm.c:528: /org/bluealsa/hci0/dev_AC_20_19_08_92_D9/a2dpsrc/sink: Prepared
D: ../../../src/asound/bluealsa-pcm.c:381: /org/bluealsa/hci0/dev_AC_20_19_08_92_D9/a2dpsrc/sink: Stopping
D: ../../../src/asound/bluealsa-pcm.c:381: /org/bluealsa/hci0/dev_AC_20_19_08_92_D9/a2dpsrc/sink: Stopping
D: ../../../src/asound/bluealsa-pcm.c:481: /org/bluealsa/hci0/dev_AC_20_19_08_92_D9/a2dpsrc/sink: Freeing HW
D: ../../../src/asound/bluealsa-pcm.c:431: /org/bluealsa/hci0/dev_AC_20_19_08_92_D9/a2dpsrc/sink: Closing

?

borine, 2021-09-09:
> Playing a 48000 khz wav file with aplay dont works:

What if you try
```shell
aplay --start-threshold=1 pyo_rec.wav
```

telemetrieTP23, 2021-09-09:
I take a complete different way, without bluealza.
pulseaudio and compiled the portaudio with pulseaudio host  (https://github.com/illuusio/portaudio).
More Latency but even it is working, thank you for trying to help me....

closed 2021-09-09 by borine