arkq/bluez-alsa #493: Can't get a RPi 4 to act as sink

tommyzat, 2021-09-14:
Hello! I installed bluez-alsa using this command:
`../configure --enable-aac --enable-aptx --enable-aptx-hd --with-libopenaptx --enable-ofono --enable-debug --disable-payloadcheck`
because I Buster currently uses PulseAudio 12.2

I then followed the steps from https://github.com/Arkq/bluez-alsa/wiki/PulseAudio-integration and I'm getting this:

`Connection failed - GDBus.Error:org.bluez.Error: Failed: Protocol not available. Try connecting manually.`

borine, 2021-09-15:
First, check that aptX and/or AAC sink profile is actually available in your bluealsa server:
```shell
$ bluealsa --help
...
Available BT profiles:
...
  - a2dp-sink	Advanced Audio Sink (aptX-HD, aptX, ..., SBC)
...
```

If so, then was aptX and/or AAC active in bluealsa *at the time the android device was paired* with the bluealsa host? Some devices cache the available codecs when paired, and will only offer those later. If your android is such a device, perhaps removing the host from the andoid paired device, and removing the android from the bluealsa host, then pairing again with bluealsa running might help. Also after pairing tell the bluealsa host to "trust" the android, then initiate the connection from the android.

tommyzat, 2021-09-15:
I removed the RPi from paired devices on the android, and in the RPi i did `remove` `pair` `trust` `connect`. Still same result.

bluealsa--help:

```
Available BT profiles:
  - a2dp-source	Advanced Audio Source (aptX-HD, aptX, AAC, SBC)
  - a2dp-sink	Advanced Audio Sink (aptX-HD, aptX, AAC, SBC)
  - hfp-ofono	Hands-Free handled by oFono (hf & ag)
  - hfp-hf	Hands-Free (v1.7)
  - hfp-ag	Hands-Free Audio Gateway (v1.7)
  - hsp-hs	Headset (v1.2)
  - hsp-ag	Headset Audio Gateway (v1.2)

By default only output profiles are enabled, which includes A2DP Source and
HSP/HFP Audio Gateways. If one wants to enable other set of profiles, it is
required to explicitly specify all of them using `-p NAME` options.
```

I also tried what the last line from --help wrote, and I have this

```
pi@raspberrypi:~ $ bluealsa --p a2dp-sink
bluealsa: D: ../../src/bluez.c:482: Creating media endpoint object: /org/bluez/hci0/A2DP/aptXHD/Sink/1
bluealsa: D: ../../src/bluez.c:403: Registering media endpoint: /org/bluez/hci0/A2DP/aptXHD/Sink/1
bluealsa: D: ../../src/bluez.c:482: Creating media endpoint object: /org/bluez/hci0/A2DP/aptXHD/Sink/2
bluealsa: D: ../../src/bluez.c:403: Registering media endpoint: /org/bluez/hci0/A2DP/aptXHD/Sink/2
bluealsa: D: ../../src/bluez.c:482: Creating media endpoint object: /org/bluez/hci0/A2DP/aptX/Sink/1
bluealsa: D: ../../src/bluez.c:403: Registering media endpoint: /org/bluez/hci0/A2DP/aptX/Sink/1
bluealsa: D: ../../src/bluez.c:482: Creating media endpoint object: /org/bluez/hci0/A2DP/aptX/Sink/2
bluealsa: D: ../../src/bluez.c:403: Registering media endpoint: /org/bluez/hci0/A2DP/aptX/Sink/2
bluealsa: D: ../../src/bluez.c:482: Creating media endpoint object: /org/bluez/hci0/A2DP/AAC/Sink/1
bluealsa: D: ../../src/bluez.c:403: Registering media endpoint: /org/bluez/hci0/A2DP/AAC/Sink/1
bluealsa: D: ../../src/bluez.c:482: Creating media endpoint object: /org/bluez/hci0/A2DP/AAC/Sink/2
bluealsa: D: ../../src/bluez.c:403: Registering media endpoint: /org/bluez/hci0/A2DP/AAC/Sink/2
bluealsa: D: ../../src/bluez.c:482: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/Sink/1
bluealsa: D: ../../src/bluez.c:403: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/Sink/1
bluealsa: D: ../../src/bluez.c:482: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/Sink/2
bluealsa: D: ../../src/bluez.c:403: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/Sink/2
bluealsa: D: ../../src/main.c:419: Acquiring D-Bus service name: org.bluealsa
bluealsa: D: ../../src/main.c:424: Starting main dispatching loop
**bluealsa: E: ../../src/main.c:85: Couldn't acquire D-Bus name: org.bluealsa**
bluealsa: D: ../../src/main.c:427: Exiting main loop
```

Don't know if the last 2 messages are relevant errors or not.

borine, 2021-09-15:
> bluealsa: E: ../../src/main.c:85: Couldn't acquire D-Bus name: org.bluealsa

I suspect that is because you are not running `bluealsa` as root.

Other possible causes are that another instance of bluealsa is already running or the bluealsa DBus config file (`/etc/dbus-1/system.d/bluealsa.conf` is either missing or corrupt.

But you previously said that connection was OK, but only SBC codec was available, so please can we go back to that setup so that we can investigate it. I need to know:

- how you are starting bluealsa (systemd file from installation, systemd file from wiki, some other method?);
- debug logs from bluealsa at the time the android device connects

tommyzat, 2021-09-15:
I tried a fresh raspbian installation. Compiled fdk-aac and libopenaptx, and then bluealsa with `../configure --enable-aac --enable-aptx --enable-aptx-hd --with-libopenaptx --enable-ofono --enable-debug --disable-payloadcheck` (because raspbian runs on pulseaudio 12.2 < 13).

I didn't disable any pulse bluetooth audio modules since I'm following the **systemd integration** from https://github.com/Arkq/bluez-alsa/wiki/Systemd-integration.
Bluealsa is started using a service unit file `sudo systemctl enable bluealsa.service`
Bluealsa-aplay.service was not implemented.

systemctl bluetooth status
```
Sep 16 00:46:40 raspberrypi bluetoothd[663]: Endpoint registered: sender=:1.54 path=/org/bluez/hci0/A2DP/SBC/Sink/1
Sep 16 00:46:40 raspberrypi bluetoothd[663]: Endpoint registered: sender=:1.54 path=/org/bluez/hci0/A2DP/SBC/Sink/2
Sep 16 00:46:40 raspberrypi bluetoothd[663]: Endpoint unregistered: sender=:1.54 path=/org/bluez/hci0/A2DP/aptXHD/Sink/1
Sep 16 00:46:40 raspberrypi bluetoothd[663]: Endpoint unregistered: sender=:1.54 path=/org/bluez/hci0/A2DP/aptXHD/Sink/2
Sep 16 00:46:40 raspberrypi bluetoothd[663]: Endpoint unregistered: sender=:1.54 path=/org/bluez/hci0/A2DP/aptX/Sink/1
Sep 16 00:46:40 raspberrypi bluetoothd[663]: Endpoint unregistered: sender=:1.54 path=/org/bluez/hci0/A2DP/aptX/Sink/2
Sep 16 00:46:40 raspberrypi bluetoothd[663]: Endpoint unregistered: sender=:1.54 path=/org/bluez/hci0/A2DP/AAC/Sink/1
Sep 16 00:46:40 raspberrypi bluetoothd[663]: Endpoint unregistered: sender=:1.54 path=/org/bluez/hci0/A2DP/AAC/Sink/2
Sep 16 00:46:40 raspberrypi bluetoothd[663]: Endpoint unregistered: sender=:1.54 path=/org/bluez/hci0/A2DP/SBC/Sink/1
Sep 16 00:46:40 raspberrypi bluetoothd[663]: Endpoint unregistered: sender=:1.54 path=/org/bluez/hci0/A2DP/SBC/Sink/2
```

bluealsa --help
```
Available BT profiles:
  - a2dp-source	Advanced Audio Source (aptX-HD, aptX, AAC, SBC)
```

bluealsa --p a2dp-sink still shows `Couldn't acquire D-Bus name: org.bluealsa`

`/etc/dbus-1/system.d/bluealsa.conf`:
```
  GNU nano 3.2                                           /etc/dbus-1/system.d/bluealsa.conf                                                      

<!-- This configuration file specifies the required security policies
     for BlueALSA core daemon to work. -->

<!DOCTYPE busconfig PUBLIC "-//freedesktop//DTD D-BUS Bus Configuration 1.0//EN"
 "http://www.freedesktop.org/standards/dbus/1.0/busconfig.dtd">
<busconfig>

  <!-- ../system.conf have denied everything, so we just punch some holes -->

  <policy user="root">
    <allow own_prefix="org.bluealsa"/>
    <allow send_destination="org.bluealsa"/>
  </policy>

  <policy group="audio">
    <allow send_destination="org.bluealsa"/>
  </policy>

    <policy user="bluealsa">
        <allow own_prefix="org.bluealsa"/>
        <allow send_destination="org.bluealsa"/>
        <allow send_destination="org.bluez"/>
    </policy>

</busconfig>
```

I also added the policy in system-local.conf: `/etc/dbus-1/system-local.conf`
```
  GNU nano 3.2                                              /etc/dbus-1/system-local.conf                                                        

<!DOCTYPE busconfig PUBLIC
 "-//freedesktop//DTD D-BUS Bus Configuration 1.0//EN"
 "http://www.freedesktop.org/standards/dbus/1.0/busconfig.dtd">

<busconfig>

    <policy user="bluealsa">
        <allow own_prefix="org.bluealsa"/>
        <allow send_destination="org.bluealsa"/>
        <allow send_destination="org.bluez"/>
    </policy>

</busconfig>
```

`/etc/default/bluealsa`:
`OPTIONS="-p a2dp-sink"`

With this new raspberry pi installation, this time SBC doesn't work, so sink isn't working at all. I have the old installation on another SD card but I believe I may have followed other installation practices, so I chose to try this on a new installation.

I'd appreciate it if you could direct me to where bluealsa log is stored.

tommyzat, 2021-09-16:
I tried a fresh raspbian installation. Compiled fdk-aac and libopenaptx, and then bluealsa with `../configure --enable-aac --enable-aptx --enable-aptx-hd --with-libopenaptx --enable-ofono --enable-debug --disable-payloadcheck` (because raspbian runs on pulseaudio 12.2 < 13).

I didn't disable any pulse bluetooth audio modules since I'm following the **systemd integration** from https://github.com/Arkq/bluez-alsa/wiki/Systemd-integration.
Bluealsa is started using a service unit file `sudo systemctl enable bluealsa.service`
Bluealsa-aplay.service was not implemented.

systemctl bluetooth status
```
Sep 16 00:46:40 raspberrypi bluetoothd[663]: Endpoint registered: sender=:1.54 path=/org/bluez/hci0/A2DP/SBC/Sink/1
Sep 16 00:46:40 raspberrypi bluetoothd[663]: Endpoint registered: sender=:1.54 path=/org/bluez/hci0/A2DP/SBC/Sink/2
Sep 16 00:46:40 raspberrypi bluetoothd[663]: Endpoint unregistered: sender=:1.54 path=/org/bluez/hci0/A2DP/aptXHD/Sink/1
Sep 16 00:46:40 raspberrypi bluetoothd[663]: Endpoint unregistered: sender=:1.54 path=/org/bluez/hci0/A2DP/aptXHD/Sink/2
Sep 16 00:46:40 raspberrypi bluetoothd[663]: Endpoint unregistered: sender=:1.54 path=/org/bluez/hci0/A2DP/aptX/Sink/1
Sep 16 00:46:40 raspberrypi bluetoothd[663]: Endpoint unregistered: sender=:1.54 path=/org/bluez/hci0/A2DP/aptX/Sink/2
Sep 16 00:46:40 raspberrypi bluetoothd[663]: Endpoint unregistered: sender=:1.54 path=/org/bluez/hci0/A2DP/AAC/Sink/1
Sep 16 00:46:40 raspberrypi bluetoothd[663]: Endpoint unregistered: sender=:1.54 path=/org/bluez/hci0/A2DP/AAC/Sink/2
Sep 16 00:46:40 raspberrypi bluetoothd[663]: Endpoint unregistered: sender=:1.54 path=/org/bluez/hci0/A2DP/SBC/Sink/1
Sep 16 00:46:40 raspberrypi bluetoothd[663]: Endpoint unregistered: sender=:1.54 path=/org/bluez/hci0/A2DP/SBC/Sink/2
```

bluealsa --help
```
Available BT profiles:
  - a2dp-source	Advanced Audio Source (aptX-HD, aptX, AAC, SBC)
```

sudo bluealsa:
```
...
bluealsa: D: ../../src/main.c:419: Acquiring D-Bus service name: org.bluealsa
bluealsa: D: ../../src/main.c:424: Starting main dispatching loop
bluealsa: E: ../../src/main.c:85: Couldn't acquire D-Bus name: org.bluealsa
bluealsa: D: ../../src/main.c:427: Exiting main loop
```

/etc/dbus-1/system.d/bluealsa.conf:
```
<!-- This configuration file specifies the required security policies
     for BlueALSA core daemon to work. -->

<!DOCTYPE busconfig PUBLIC "-//freedesktop//DTD D-BUS Bus Configuration 1.0//EN"
 "http://www.freedesktop.org/standards/dbus/1.0/busconfig.dtd">
<busconfig>

  <!-- ../system.conf have denied everything, so we just punch some holes -->

  <policy user="root">
    <allow own_prefix="org.bluealsa"/>
    <allow send_destination="org.bluealsa"/>
  </policy>

  <policy group="audio">
    <allow send_destination="org.bluealsa"/>
  </policy>

    <policy user="bluealsa">
        <allow own_prefix="org.bluealsa"/>
        <allow send_destination="org.bluealsa"/>
        <allow send_destination="org.bluez"/>
    </policy>

</busconfig>
```

I also added the policy in system-local.conf: `/etc/dbus-1/system-local.conf`
```
<!DOCTYPE busconfig PUBLIC
 "-//freedesktop//DTD D-BUS Bus Configuration 1.0//EN"
 "http://www.freedesktop.org/standards/dbus/1.0/busconfig.dtd">

<busconfig>

    <policy user="bluealsa">
        <allow own_prefix="org.bluealsa"/>
        <allow send_destination="org.bluealsa"/>
        <allow send_destination="org.bluez"/>
    </policy>

</busconfig>
```

/etc/default/bluealsa:
`OPTIONS="-p a2dp-sink"`

With this new raspberry pi installation, this time SBC doesn't work, so sink isn't working at all, BUT connection is succesful, I do have the AptX, AptX HD and AAC options. There's just no sound being output. I have the old installation on another SD card but I believe I may have followed other installation practices, so I chose to try this on a new installation.

I'd appreciate it if you could direct me to where bluealsa log is stored.

borine, 2021-09-16:
> I didn't disable any pulse bluetooth audio modules 

Then I hope you have made certain that pulseaudio is not running (https://github.com/Arkq/bluez-alsa#troubleshooting item 1)

> I'd appreciate it if you could direct me to where bluealsa log is stored.

bluealsa writes its debug output to its standard error (that is documented in the manual page https://github.com/Arkq/bluez-alsa/blob/master/doc/bluealsa.8.rst). So, assuming it is being started by systemd, as with any other systemd service you should be able to see the logs with:
```
sudo journalctl -xeu bluealsa.service
```
Also assuming you are starting it with systemd, you cannot then also run it from the command line, because the dbus service name "org.bluealsa" can only be owned by one process at a time (that is also documented in the README .https://github.com/Arkq/bluez-alsa#troubleshooting item 3)

BTW, does the android successfully use aptX or AAC when sending audio directly to bluetooth headphones or speaker? - Just to absolutely rule out the (unlikely) possibility of the problem being caused by android.

tommyzat, 2021-09-16:
This is the log from, starting from pairing & trusting the android device to succesful connection, with notification showing aptx hd support.
```
Sep 16 13:58:51 raspberrypi bluealsa[763]: /usr/bin/bluealsa: D: ../../src/bluez.c:1000: Signal: org.freedesktop.DBus.ObjectManager.InterfacesRemoved()
Sep 16 13:59:24 raspberrypi bluealsa[763]: /usr/bin/bluealsa: D: ../../src/bluez.c:906: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
Sep 16 13:59:42 raspberrypi bluealsa[763]: /usr/bin/bluealsa: D: ../../src/bluez.c:906: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
Sep 16 13:59:49 raspberrypi bluealsa[763]: /usr/bin/bluealsa: D: ../../src/dbus.c:59: Called: org.bluez.MediaEndpoint1.SelectConfiguration() on /org/bluez/hci0/A2DP/aptXHD/Sink/1
Sep 16 13:59:49 raspberrypi bluealsa[763]: /usr/bin/bluealsa: D: ../../src/bluez.c:150: A2DP peer capabilities blob [len=11]: d700000024002200000000
Sep 16 13:59:49 raspberrypi bluealsa[763]: /usr/bin/bluealsa: D: ../../src/bluez.c:906: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
Sep 16 13:59:49 raspberrypi bluealsa[763]: /usr/bin/bluealsa: D: ../../src/dbus.c:59: Called: org.bluez.MediaEndpoint1.SetConfiguration() on /org/bluez/hci0/A2DP/aptXHD/Sink/1
Sep 16 13:59:49 raspberrypi bluealsa[763]: /usr/bin/bluealsa: D: ../../src/bluez.c:290: A2DP Sink (aptX HD) configured for device 5C:17:CF:20:68:BE
Sep 16 13:59:49 raspberrypi bluealsa[763]: /usr/bin/bluealsa: D: ../../src/bluez.c:292: A2DP selected configuration blob [len=11]: d700000024002200000000
Sep 16 13:59:49 raspberrypi bluealsa[763]: /usr/bin/bluealsa: D: ../../src/bluez.c:294: PCM configuration: channels: 2, sampling: 44100
Sep 16 13:59:49 raspberrypi bluealsa[763]: /usr/bin/bluealsa: D: ../../src/bluez.c:482: Creating media endpoint object: /org/bluez/hci0/A2DP/aptXHD/Sink/3
Sep 16 13:59:49 raspberrypi bluealsa[763]: /usr/bin/bluealsa: D: ../../src/bluez.c:403: Registering media endpoint: /org/bluez/hci0/A2DP/aptXHD/Sink/3
Sep 16 13:59:50 raspberrypi bluealsa[763]: /usr/bin/bluealsa: D: ../../src/bluez.c:906: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
Sep 16 13:59:51 raspberrypi bluealsa[763]: /usr/bin/bluealsa: D: ../../src/bluez.c:1094: Signal: org.freedesktop.DBus.Properties.PropertiesChanged(): org.bluez.MediaTransport1: State
Sep 16 13:59:51 raspberrypi bluealsa[763]: /usr/bin/bluealsa: D: ../../src/ba-transport.c:566: New A2DP transport: 14
Sep 16 13:59:51 raspberrypi bluealsa[763]: /usr/bin/bluealsa: D: ../../src/ba-transport.c:567: A2DP socket MTU: 14: R:672 W:1008
Sep 16 13:59:51 raspberrypi bluealsa[763]: /usr/bin/bluealsa: D: ../../src/bluez.c:1094: Signal: org.freedesktop.DBus.Properties.PropertiesChanged(): org.bluez.MediaTransport1: State
Sep 16 13:59:51 raspberrypi bluealsa[763]: /usr/bin/bluealsa: D: ../../src/ba-transport.c:1138: Starting transport: A2DP Sink (aptX HD)
Sep 16 13:59:51 raspberrypi bluealsa[763]: /usr/bin/bluealsa: D: ../../src/ba-transport.c:295: Created BT socket duplicate: [14]: 15
Sep 16 13:59:51 raspberrypi bluealsa[763]: /usr/bin/bluealsa: D: ../../src/ba-transport.c:1456: Created new IO thread [ba-a2dp-aptx-hd]: A2DP Sink (aptX HD)
Sep 16 13:59:51 raspberrypi bluealsa[763]: /usr/bin/bluealsa: D: ../../src/a2dp-aptx-hd.c:221: IO loop: START: a2dp_aptx_hd_dec_thread: A2DP Sink (aptX HD)
Sep 16 13:59:54 raspberrypi bluealsa[763]: /usr/bin/bluealsa: D: ../../src/bluez.c:1094: Signal: org.freedesktop.DBus.Properties.PropertiesChanged(): org.bluez.MediaTransport1: State
Sep 16 13:59:54 raspberrypi bluealsa[763]: /usr/bin/bluealsa: D: ../../src/ba-transport.c:307: Closing BT socket duplicate [14]: 15
Sep 16 13:59:54 raspberrypi bluealsa[763]: /usr/bin/bluealsa: D: ../../src/ba-transport.c:620: Closing A2DP transport: 14
Sep 16 13:59:54 raspberrypi bluealsa[763]: /usr/bin/bluealsa: D: ../../src/ba-transport.c:1480: Exiting IO thread [ba-a2dp-aptx-hd]: A2DP Sink (aptX HD)
```

Pulseaudio bluetooth modules aren't running after `apt purge pulseaudio-module-bluetooth`

> BTW, does the android successfully use aptX or AAC 
Sorry, I meant that the options are shown as available, but no sound is coming through the 3.5mm whenever using any of the options, not even SBC for this fresh installation.

borine, 2021-09-16:
Hmm, the logs clearly show that an A2DP connection is established, and the chosen codec is aptX-HD.

So the next question is probably the first I should have asked - what application on the Pi are you using to play the sound? Are you running `bluealsa-aplay` ? If not, I would recommend it. To send bluetooth audio to the ALSA default device, simply type `bluealsa-aplay` at the command line using you normal user account.

tommyzat, 2021-09-16:
It's finally working. What I had to do was completely remove pulseaudio from my system and only after bluealsa-aplay would work. I truly appreciate your patience with me.

As for the first Pi installation with SBC only working, I believe that I also tried to install bluez-alsa using the experimental bluepulse method, and didn't properly clean up after then trying systemd.

PSA for anyone experiencing the same issues as me, just purge pulseaudio.

closed 2021-09-16 by tommyzat