arkq/bluez-alsa #495: AC_PROG_CC_C99 warning and build error

skidoo23, 2021-09-17:
On Debian testing x64, autoconf 2.71, gcc version 10.3.0 (Debian 10.3.0-9)

$ autoreconf --install
configure.ac:14: warning: m4_warn([obsolete],
configure.ac:14: [AC_PROG_CC_C99 is obsolete; use AC_PROG_CC
configure.ac:14: ])dnl
configure.ac:14: AC_REQUIRE(AC_PROG_CC) is obsolete; use AC_PROG_CC
configure.ac:20: warning: ac_ext=c
configure.ac:20: ac_cpp='$CPP $CPPFLAGS'
configure.ac:20: ac_compile='$CC -c $CFLAGS $CPPFLAGS conftest.$ac_ext >&5'
configure.ac:20: ac_link='$CC -o conftest$ac_exeext $CFLAGS $CPPFLAGS $LDFLAGS conftest.$ac_ext $LIBS >&5'
configure.ac:20: ac_compiler_gnu=$ac_cv_c_compiler_gnu
configure.ac:20: if test -n "$ac_tool_prefix"; then
configure.ac:20:   # Extract the first word of "${ac_tool_prefix}gcc", so it can be a program name with args.
configure.ac:20: set dummy ${ac_tool_prefix}gcc; ac_word=$2
configure.ac:20: { printf "%s\n" "$as_me:${as_lineno-$LINENO}: checking for $ac_word" >&5
configure.ac:20: printf %s "checking for $ac_word... " >&6; }
configure.ac:20: if test ${ac_cv_prog_CC+y}
configure.ac:20: then :
configure.ac:20:   printf %s "(cached) " >&6
configure.ac:20: else $as_nop
configure.ac:20:   if test -n "$CC"; then
configure.ac:20:   ac_cv_prog_CC="$CC" # Let the user override the test.
configure.ac:20: else
configure.ac:20: as_save_IFS=$IFS; IFS=$PATH_SEPARATOR
configure.ac:20: for as_dir in $PATH
configure.ac:20: do
configure.ac:20:   IFS=$as_save_IFS
configure.ac:20:   case $as_dir in #(((
configure.ac:20:     '' is m4_require'd but not m4_defun'd
lib/m4sugar/m4sh.m4:692: _AS_IF_ELSE is expanded from...
lib/m4sugar/m4sh.m4:699: AS_IF is expanded from...
./lib/autoconf/general.m4:2249: AC_CACHE_VAL is expanded from...
./lib/autoconf/programs.m4:41: _AC_CHECK_PROG is expanded from...
./lib/autoconf/programs.m4:101: AC_CHECK_PROG is expanded from...
./lib/autoconf/programs.m4:221: AC_CHECK_TOOL is expanded from...
./lib/autoconf/c.m4:452: AC_PROG_CC is expanded from...
configure.ac:20: the top level

$ cd build
me@home:/src/bluez-alsa/build$ ../configure --enable-aac --enable-ofono --enable-rfcomm
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a race-free mkdir -p... /bin/mkdir -p
checking for gawk... no
checking for mawk... mawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether make supports the include directive... yes (GNU style)
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables...
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether the compiler supports GNU C... yes
checking whether gcc accepts -g... yes
checking for gcc option to enable C11 features... none needed
checking whether gcc understands -c and -o together... yes
checking dependency style of gcc... gcc3
checking for stdio.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for strings.h... yes
checking for sys/stat.h... yes
checking for sys/types.h... yes
checking for unistd.h... yes
checking for wchar.h... yes
checking for minix/config.h... no
checking whether it is safe to define __EXTENSIONS__... yes
checking whether _XOPEN_SOURCE should be defined... no
checking for gcc... (cached) gcc
checking whether the compiler supports GNU C... (cached) yes
checking whether gcc accepts -g... (cached) yes
checking for gcc option to enable C11 features... (cached) none needed
checking whether gcc understands -c and -o together... (cached) yes
checking dependency style of gcc... (cached) gcc3
../configure: line 5782: syntax error near unexpected token `newline'
../configure: line 5782: `    '''

borine, 2021-09-17:
This patch should fix it (I don't have autoconf 2.70, so haven't actually tested it.)
@skidoo23 - could you give it a try?
```diff
diff --git a/configure.ac b/configure.ac
index 2247ac2..89978b4 100644
--- a/configure.ac
+++ b/configure.ac
@@ -12,8 +12,7 @@ AC_CONFIG_HEADERS([config.h])
 AC_CONFIG_MACRO_DIR([m4])
 
 AC_USE_SYSTEM_EXTENSIONS
-AC_PROG_CC
-AC_PROG_CC_C99
+m4_version_prereq(2.70, [AC_PROG_CC], [AC_PROG_CC_C99])
 AC_PROG_INSTALL
 AC_PROG_LN_S
 AC_PROG_MKDIR_P
```

skidoo23, 2021-09-17:
Warning is gone and BlueALSA compiles. Thank you!

commit ab823f28 (2021-09-17): Get rid of all warnings and enforce warnings check

Fixes #489, fixes #495
closed 2021-09-17 by Arkq