arkq/bluez-alsa #497: How to change Audio Sink Codec Configured to 48000 Hz

bmt1596, 2021-09-20:
Hi everyone, I am programming an ARM Board that uses Petalinux to receive Audio from Phone and send that Audio A2DP to another Bluetooth speaker.

My problem is that the sound I get is an active sound with frequency 44100 Hz (capture) and the output (playback) is again 48000 Hz.

Is there a way for me to increase the Audio Sink Codec Configured to 48000 Hz. Because bluetooth speaker is fixed at 48000 Hz.

I found out bluealsa offers the option --a2dp-force-audio-cd to force capture und playblack devices to 44100. But I really want both ends to be 48000 Hz.

Is there a way to tune this frequency from the capture side to 48000.

Thank you.

Here is the result of bluealsa-aplay-l

bluealsa-aplay -l
bluealsa: D: ../../src/dbus.c:59: Called: org.bluealsa.Manager1.GetPCMs() on /org/bluealsa
**** List of PLAYBACK Bluetooth Devices ****
hci0: B9:76:59:02:C8:B5 [BT-WUZHI], trusted audio-card
  A2DP (SBC): S16_LE 2 channels 48000 Hz
hci0: 9C:E6:5E:D5:DF:D8 [iPad của Minh Tung], trusted computer
  SCO (CVSD): S16_LE 1 channel 8000 Hz
**** List of CAPTURE Bluetooth Devices ****
hci0: 9C:E6:5E:D5:DF:D8 [iPad của Minh Tung], trusted computer
  A2DP (SBC): S16_LE 2 channels 44100 Hz
  SCO (CVSD): S16_LE 1 channel 8000 Hz

Arkq, 2021-09-20:
Unfortunately, it's not possible to force Bluetooth audio to 48000 (also it's not possible to **force** to 44100). Bluetooth sink (receiving end) shall support both 48kHz and 44.1kHz in order to comply with the Bluetooth A2DP specification. The `--a2dp-force-audio-cd` option is not **force** but more like **try to force** in case where Bluetooth source supports both sampling frequencies (e.g. some Android phones support only 44.1kHz).

In order to resample audio from 44.1kHz to 48kHz you need another piece of software. When using ALSA, you can use standard ALSA `plug` plugin. For more information consult ALSA documentation.

bmt1596, 2021-09-20:
> r
Hi Arkq. 
Thank you for this quick response. Now I can hear the sound very well using the command bluealsa-aplay-vv.

But I am trying to write a program in C++ to read and write Audio signal without using commands like bluealsa-aplay -vv. The goal is to combine with several other components in the project to complete a complete signal processing device. Currently I am using CYW20719 bluetooth chip and as far as I know it has support for both 44.1 kHz and 48 kHz frequencies. To avoid having to resample from 44.1 kHz to 48 kHz again via C++ application, I would like to ask for your help.

Is using alsa plugin type plug the only solution I have??

Arkq, 2021-09-20:
> But I am trying to write a program in C++
> Currently I am using CYW20719 bluetooth chip and as far as I know it has support for both 44.1 kHz and 48 kHz frequencies

If you're writing your own app, then I think there is no issue with writing/using resampling lib. Because, there is no way of avoiding resampling issue. Even though your chip supports 44.1kHz and 48kHz (I'm not sure what you mean by that, though), the sampling frequency is determined by source Bluetooth device. If the source does not support 48kHz you will end up with 44.1kHz no matter what.

bmt1596, 2021-09-20:
OK so I get what you mean. my source bluetooth device is a phone and it seems to only support 44100 Hz. Thank you very much.

closed 2021-09-20 by Arkq