arkq/bluez-alsa #498: --a2dp-volume not always forwarding volume information correctly for A2DP sinks

hifiberry, 2021-09-27:
Forwarding volume information to ALSA seems to work only with selected devices

With iOS 14.8 ALS volume control works with the --a2dp-volume command line argument

With Android 11 ALSA it only works without the --a2dp-volume command line argument

I'm happy to provide further details if needed.

Arkq, 2021-09-27:
The question is whether Android sends volume notification. With iPhone I know that everything works, because I've been testing with it. But with Android (I've got only a device with Android 6) I'm not receiving volume notifications from BlueZ. I have not investigated it thoroughly, though...

hifiberry, 2021-09-27:
It seems to do so. 
Interestingly it works here with an android device, if the --a2dp-volume command line option has been removed.
Another user have been reported the same behavior with a PC running Fedora as the audio source.

Therefore, my guess is that there is a bug in the handling of these messages in bluealsa. But that's just a guess.

Arkq, 2021-09-27:
> Therefore, my guess is that there is a bug in the handling of these messages in bluealsa.

OK, I'll check that.

hifiberry, 2021-09-27:
Let me know if I can provide additional information.

borine, 2021-10-12:
A wild guess at a possible cause to test:
- what version of AVRCP does the android phone use? I *think* that version 1.4 or later is required for the absolute volume control used by Bluez and BlueALSA - although I could be wrong!

borine, 2021-11-02:
I found this post: https://patchwork.kernel.org/project/bluetooth/patch/20210422192253.553048-1-marijns95@gmail.com/ that suggests this is a known problem with Bluez and Android. However the included patch appears to have been ignored. So Bluez at present does not correctly interpret the AVRCP capabilities of android and fails to action the volume change requests. Bluealsa therefore never gets notified of these requests and I think that explains the behaviour you are seeing.

martinmcclure, 2021-12-04:
@borine Thanks for the link to the kernel patch! My Android phone had the problem, and I was preparing to build a kernel with that patch. This morning, though, volume control on the phone started unexpectedly working. This puzzled me until I recalled that last night I'd updated my phone to the LineageOS Nov 30 build of Android 11. Since confirmed with my wife's phone (both are Pixel 3A) -- LineageOS of Nov 23 and earlier has the volume problem; Nov 30 works.
Since I only only started seriously looking into the volume control problem a couple of days ago, this timing is kinda spooky, but very welcome.

borine, 2022-09-15:
This issue has had no updates for 9 months. From the link in comment https://github.com/Arkq/bluez-alsa/issues/498#issuecomment-957222375 I believe this to be a bug in Bluez, whereby volume control does not work correctly from devices using Android 10 or earlier (with the "Flouride" bluetooth stack). It appears that Andorid 11 and later (with the "Gabeldorsche" bluetooth stack) do work correctly with Bluez. So the solution is to either upgrade the phone OS or patch Bluez as per the above link.

This is outside the control of BlueALSA and therefore I believe this issue should be closed.

closed 2022-09-15 by Arkq