arkq/bluez-alsa #513: Galaxy Bluetooth earbuds no sound

OvoNemaSmisla, 2021-11-29:
Hi,

I'm running the latest version of OSMC on raspberry pi 3 model B.
I've been using without problems 2 bluetooth earbuds: ENACFIRE Future and Galaxy Buds+.
After the previous update, Galaxy Buds+ don't work.

I'm able to connect to them:
```
osmc@osmc:~$ bluetoothctl
[NEW] Controller B8:27:EB:A8:15:31 osmc [default]
[NEW] Device 00:00:90:B0:EF:77 ENACFIRE  Future
[NEW] Device 64:03:7F:56:E6:ED Galaxy Buds+ (E6ED)
[Galaxy Buds+ (E6ED)]# info 64:03:7F:56:E6:ED
Device 64:03:7F:56:E6:ED
	Name: Galaxy Buds+ (E6ED)
	Alias: Galaxy Buds+ (E6ED)
	Class: 0x240404
	Icon: audio-card
	Paired: yes
	Trusted: yes
	Blocked: no
	Connected: yes
	LegacyPairing: no
	UUID: Serial Port               (00001101-0000-1000-8000-00805f9b34fb)
	UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
	UUID: Advanced Audio Distribu.. (0000110d-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
	UUID: Handsfree                 (0000111e-0000-1000-8000-00805f9b34fb)
	UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
	UUID: Vendor specific           (a23d00bc-217c-123b-9c00-fc44577136ee)
	UUID: Vendor specific           (b4a9d6a0-b2e3-4e40-976d-a69f167ea895)
	UUID: Vendor specific           (e7ab2241-ca64-4a69-ac02-05f5c6fe2d62)
	UUID: Vendor specific           (f8620674-a1ed-41ab-a8b9-de9ad655729d)
	Modalias: bluetooth:v0075pA012d0001
```
My .asoundrc looks like this:
```
pcm.galaxybuds {
        type plug
        slave.pcm {
        type bluealsa
        device "64:03:7F:56:E6:ED"
        profile "a2dp"
        }
        hint { show on description "Galaxy Buds+"}
}
```
I've tried the follwoing:
```
sudo rm /etc/alsa/conf.d/21-bt-audio-osmc.conf
sudo alsactl nrestore
```
And tested with `speaker-test -D galaxybuds -c 2 -p 1 -t wav -l 3`:
```
speaker-test 1.1.8

Playback device is galaxybuds
Stream parameters are 48000Hz, S16_LE, 2 channels
WAV file(s)
Rate set to 48000Hz (requested 48000Hz)
Buffer size range from 960 to 4194304
Period size range from 480 to 4096
Requested period time 1 us
Periods = 4
Unable to set hw params for playback: Device or resource busy
Setting of hwparams failed: Device or resource busy
```

On the other hand, ENACFIRE Future works without problems.

Arkq, 2021-11-29:
Could you post the result of `bluealsa-aplay -l`. And preferably logs from bluealsa service (with debug enabled during configuration build stage).

graham8, 2021-11-29:
We'll have to build with debug then.

OvoNemaSmisla, 2021-11-29:
Here is the `bluealsa-aplay -l`:
```
osmc@osmc:~$ bluealsa-aplay -l
**** List of PLAYBACK Bluetooth Devices ****
hci0: 64:03:7F:56:E6:ED [Galaxy Buds+ (E6ED)], trusted audio-card
  A2DP (AAC): S16_LE 2 channels 48000 Hz
**** List of CAPTURE Bluetooth Devices ****
```

Arkq, 2021-11-29:
The strange thing is that the HFP/HSP profile is not connected. Or maybe you have it disabled in bluealsa (e.g launching bluealsa only with `-p a2dp-source`)? If you've got `bluealsa-cli` installed, please, try something like this:
```sh
# convert some audio to raw format, e.g.:
# ffmpeg -i <input-audio>  -acodec pcm_s16le -f s16le -ac 2 -ar 48000 output.pcm
bluealsa-cli list-pcms
cat output.pcm | bluealsa-cli open <pcm-path>
```

This should play audio directly via bluealsa service omitting other components (i.e. ALSA).

graham8, 2021-11-29:
User has got it working now.  Heisenbug.  And yes, we don't enable SCO.

graham8, 2021-11-29:
Thanks for the rapid response. The thing is, the only change we made in the latest OSMC release was to add the automatic connection node:
```
osmc@vero4kp2:/usr/share/alsa/alsa.conf.d$ cat 21-bt-audio-osmc.conf
## Do not modify this file - it may be overwritten in a future update
## Use ~/.asoundrc for customisations

pcm.btaudio {
        type plug
        slave.pcm {
                type bluealsa
                device "00:00:00:00:00:00"
                profile "a2dp"
        }
        hint { show on description "Bluetooth Audio"}
}
```
We were already running 3.1.0 with users writing their own .asoundrc.  @OvoNemaSmisla was fine with that. Is there any way using the conf node above differs from using explicit device nodes?

Arkq, 2021-11-29:
> explicit device nodes

What do you mean by that? The node which uses real BT address instead of "00:00:00..."?

graham8, 2021-11-29:
> The node which uses real BT address instead of "00:00:00..."?

Yes.  One node for each device.

Arkq, 2021-11-29:
The only difference is that this configuration will (shall) use lastly connected BT device, that's all. Other than that is should be the same. But maybe there is some bug in there....

graham8, 2021-11-29:
Hmmm. Thanks. Now it seems these buds work fine as long as you just allow them to connect as 'trusted'.  User was attempting connection from the source.

OvoNemaSmisla, 2021-11-30:
I can confirm it works fine if the buds connect automatically.

closed 2021-11-30 by OvoNemaSmisla