arkq/bluez-alsa #514: Trying to record audio from bluetooth

lelemm, 2021-12-04:
I'm trying to use bluealsa to record audio from my headset (and use it to stream audio). basically, i'm trying to use ALEXA SDK from DOCKER but using a bluetooth speaker to communicate with it.

For now, I'm just trying to use the bluetooth to play a wav and record the microphone from the headset.

For now I can play the wav to the speaker using a2dp.
But when I try to record the audio from the mic, the systems hangs and that's it:

arecord -D bluealsa:05:43:54:4C:7E:13,sco capture.wav
D: ../../../src/asound/bluealsa-pcm.c:1312: Getting BlueALSA PCM: CAPTURE 05:43:54:4C:7E:13 sco
D: ../../../src/asound/bluealsa-pcm.c:1062: /org/bluealsa/hci0/dev_05_43_54_4C_7E_13/hspag/source: Setting constraints
Recording WAVE 'capture.wav' : Signed 16 bit Little Endian, Rate 8000 Hz, Mono
D: ../../../src/asound/bluealsa-pcm.c:532: /org/bluealsa/hci0/dev_05_43_54_4C_7E_13/hspag/source: Initializing HW
D: ../../../src/asound/bluealsa-pcm.c:567: /org/bluealsa/hci0/dev_05_43_54_4C_7E_13/hspag/source: FIFO buffer size: 32768 fra                  mes
D: ../../../src/asound/bluealsa-pcm.c:573: /org/bluealsa/hci0/dev_05_43_54_4C_7E_13/hspag/source: Selected HW buffer: 4 perio                  ds x 2000 bytes == 8000 bytes
D: ../../../src/asound/bluealsa-pcm.c:600: /org/bluealsa/hci0/dev_05_43_54_4C_7E_13/hspag/source: Initializing SW
D: ../../../src/asound/bluealsa-pcm.c:600: /org/bluealsa/hci0/dev_05_43_54_4C_7E_13/hspag/source: Initializing SW
D: ../../../src/asound/bluealsa-pcm.c:639: /org/bluealsa/hci0/dev_05_43_54_4C_7E_13/hspag/source: Prepared
D: ../../../src/asound/bluealsa-pcm.c:600: /org/bluealsa/hci0/dev_05_43_54_4C_7E_13/hspag/source: Initializing SW
D: ../../../src/asound/bluealsa-pcm.c:356: /org/bluealsa/hci0/dev_05_43_54_4C_7E_13/hspag/source: Starting
D: ../../../src/asound/bluealsa-pcm.c:226: /org/bluealsa/hci0/dev_05_43_54_4C_7E_13/hspag/source: Starting IO loop: 7

bluealsa: D: ../../src/dbus.c:66: Called: org.bluealsa.PCM1.Open() on /org/bluealsa/hci0/dev_05_43_54_4C_7E_13/hspag/source
bluealsa: D: ../../src/ba-transport.c:710: New SCO link: 05:43:54:4C:7E:13: 28
bluealsa: D: ../../src/hci.c:133: SCO link socket MTU: 28: 64
bluealsa: D: ../../src/hci.c:141: USB adjusted SCO MTU: 28: 48
bluealsa: D: ../../src/ba-transport.c:1145: Starting transport: HSP Audio Gateway
bluealsa: D: ../../src/ba-transport.c:293: Created BT socket duplicate: [28]: 29
bluealsa: D: ../../src/sco.c:229: IO loop: START: sco_cvsd_enc_thread: HSP Audio Gateway
bluealsa: D: ../../src/ba-transport.c:1462: Created new IO thread [ba-sco-enc]: HSP Audio Gateway
bluealsa: D: ../../src/ba-transport.c:293: Created BT socket duplicate: [28]: 30
bluealsa: D: ../../src/ba-transport.c:1462: Created new IO thread [ba-sco-dec]: HSP Audio Gateway
bluealsa: D: ../../src/sco.c:300: IO loop: START: sco_cvsd_dec_thread: HSP Audio Gateway
bluealsa: D: ../../src/ba-transport.c:1377: PCM resumed: 25

Any ideias what's going on?

Thanks!

lelemm, 2021-12-04:
More info:

 uname -a
Linux orangepizero 5.10.60-sunxi #21.08.2 SMP Tue Sep 14 16:28:44 UTC 2021 armv7l armv7l armv7l GNU/Linu

during connection:

bluealsa process log:
bluealsa: D: ../../src/ofono.c:629: Checking oFono service presence
bluealsa: D: ../../src/main.c:440: Starting main dispatching loop
bluealsa: D: ../../src/main.c:89: Acquired D-Bus service name: org.bluealsa
bluealsa: D: ../../src/bluealsa-dbus.c:317: Registering D-Bus manager: /org/bluealsa
bluealsa: W: ../../src/ba-adapter.c:53: Couldn't get HCI version: Network is down
bluealsa: D: ../../src/bluez.c:671: Registering battery provider: /org/bluez/hci0/battery
bluealsa: D: ../../src/bluez.c:691: BlueZ battery provider support not available
bluealsa: D: ../../src/bluez.c:483: Creating media endpoint object: /org/bluez/hci0/A2DP/AAC/source/1
bluealsa: D: ../../src/bluez.c:392: Registering media endpoint: /org/bluez/hci0/A2DP/AAC/source/1
bluealsa: D: ../../src/bluez.c:483: Creating media endpoint object: /org/bluez/hci0/A2DP/AAC/source/2
bluealsa: D: ../../src/bluez.c:392: Registering media endpoint: /org/bluez/hci0/A2DP/AAC/source/2
bluealsa: D: ../../src/bluez.c:483: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/source/1
bluealsa: D: ../../src/bluez.c:392: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/source/1
bluealsa: D: ../../src/bluez.c:483: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/source/2
bluealsa: D: ../../src/bluez.c:392: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/source/2
bluealsa: D: ../../src/bluez.c:920: Creating hands-free profile object: /org/bluez/HSP/AudioGateway
bluealsa: D: ../../src/bluez.c:844: Registering hands-free profile: /org/bluez/HSP/AudioGateway
bluealsa: D: ../../src/bluez.c:920: Creating hands-free profile object: /org/bluez/HFP/AudioGateway
bluealsa: D: ../../src/bluez.c:844: Registering hands-free profile: /org/bluez/HFP/AudioGateway
bluealsa: D: ../../src/dbus.c:66: Called: org.bluez.Profile1.NewConnection() on /org/bluez/HFP/AudioGateway
bluealsa: D: ../../src/ba-rfcomm.c:1013: Starting RFCOMM loop: HFP Audio Gateway (CVSD)
bluealsa: D: ../../src/ba-rfcomm.c:1398: Created new RFCOMM thread [ba-rfcomm]: HFP Audio Gateway (CVSD)
bluealsa: D: ../../src/sco.c:203: Created SCO dispatcher [ba-sco-dispatch]: hci0
bluealsa: D: ../../src/sco.c:77: Starting SCO dispatcher loop: hci0
bluealsa: D: ../../src/bluez.c:759: HFP Audio Gateway (CVSD) configured for device 05:43:54:4C:7E:13
bluealsa: D: ../../src/at.c:162: AT message: SET: command:+BRSF, value:191
bluealsa: D: ../../src/ba-rfcomm.c:108: Sending AT message: RESP: command:+BRSF, value:2272
bluealsa: D: ../../src/ba-rfcomm.c:108: Sending AT message: RESP: command:(null), value:OK
bluealsa: D: ../../src/ba-rfcomm.c:127: RFCOMM: HFP Audio Gateway (CVSD) state transition: 0 -> 2
bluealsa: D: ../../src/at.c:162: AT message: TEST: command:+CIND, value:(null)
bluealsa: D: ../../src/ba-rfcomm.c:108: Sending AT message: RESP: command:+CIND, value:("service",(0-1)),("call",(0,1)),("callsetup",(0-3)),("callheld",(0-2)),("signal",(0-5)),("roam",(0-1)),("battchg",(0-5))
bluealsa: D: ../../src/ba-rfcomm.c:108: Sending AT message: RESP: command:(null), value:OK
bluealsa: D: ../../src/ba-rfcomm.c:127: RFCOMM: HFP Audio Gateway (CVSD) state transition: 2 -> 5
bluealsa: D: ../../src/bluez.c:1049: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
bluealsa: D: ../../src/bluez.c:1137: Adding new Stream End-Point: 05:43:54:4C:7E:13: SBC
bluealsa: D: ../../src/at.c:162: AT message: GET: command:+CIND, value:(null)
bluealsa: D: ../../src/ba-rfcomm.c:108: Sending AT message: RESP: command:+CIND, value:0,0,0,0,0,0,5
bluealsa: D: ../../src/ba-rfcomm.c:108: Sending AT message: RESP: command:(null), value:OK
bluealsa: D: ../../src/ba-rfcomm.c:127: RFCOMM: HFP Audio Gateway (CVSD) state transition: 5 -> 7
bluealsa: D: ../../src/at.c:162: AT message: SET: command:+CMER, value:3, 0, 0, 1
bluealsa: D: ../../src/ba-rfcomm.c:108: Sending AT message: RESP: command:(null), value:OK
bluealsa: D: ../../src/ba-rfcomm.c:127: RFCOMM: HFP Audio Gateway (CVSD) state transition: 7 -> 8
bluealsa: D: ../../src/ba-rfcomm.c:127: RFCOMM: HFP Audio Gateway (CVSD) state transition: 8 -> 9
bluealsa: D: ../../src/dbus.c:66: Called: org.bluez.MediaEndpoint1.SelectConfiguration() on /org/bluez/hci0/A2DP/SBC/source/1
bluealsa: D: ../../src/bluez.c:166: A2DP peer capabilities blob [len=4]: ffff0235
bluealsa: D: ../../src/at.c:162: AT message: SET: command:+VGS, value:14
bluealsa: D: ../../src/ba-rfcomm.c:108: Sending AT message: RESP: command:(null), value:OK
bluealsa: D: ../../src/bluez.c:1049: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
bluealsa: D: ../../src/dbus.c:66: Called: org.bluez.MediaEndpoint1.SetConfiguration() on /org/bluez/hci0/A2DP/SBC/source/1
bluealsa: D: ../../src/a2dp.c:844: Selected A2DP SBC bit-pool range: [2, 53]
bluealsa: D: ../../src/bluez.c:303: A2DP Source (SBC) configured for device 05:43:54:4C:7E:13
bluealsa: D: ../../src/bluez.c:306: A2DP selected configuration blob [len=4]: 11150235
bluealsa: D: ../../src/bluez.c:308: PCM configuration: channels: 2, sampling: 48000
bluealsa: D: ../../src/bluez.c:483: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/source/3
bluealsa: D: ../../src/bluez.c:392: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/source/3
bluealsa: D: ../../src/at.c:162: AT message: GET: command:+BTRH, value:(null)
[... cut ...]
bluealsa: W: ../../src/ba-rfcomm.c:1295: Unsupported AT message: SET: command:+CSRSF, value:0,0,0,1,0,0,0
bluealsa: D: ../../src/ba-rfcomm.c:108: Sending AT message: RESP: command:(null), value:ERROR

bluetoothctl log:
[bluetooth]# power on
[CHG] Controller 00:1A:7D:DA:71:10 Class: 0x00480000
Changing power on succeeded
[CHG] Controller 00:1A:7D:DA:71:10 Powered: yes
[bluetooth]# connect 05:43:54:4C:7E:13
Attempting to connect to 05:43:54:4C:7E:13
[CHG] Device 05:43:54:4C:7E:13 Connected: yes
Connection successful
[CHG] Device 05:43:54:4C:7E:13 ServicesResolved: yes

lelemm, 2021-12-04:
ust had the same problem when trying to play to sco, this time I got this messages:

Message from syslogd@orangepizero at Dec  4 19:44:26 ...
 kernel:[  622.084662] Internal error: Oops: 5 [#1] SMP THUMB2

Message from syslogd@orangepizero at Dec  4 19:44:26 ...
 kernel:[  622.203960] Process ba-sco-enc (pid: 2252, stack limit = 0x65a07e2f)

Message from syslogd@orangepizero at Dec  4 19:44:26 ...
 kernel:[  622.203966] Stack: (0xd36f1e30 to 0xd36f2000)

Message from syslogd@orangepizero at Dec  4 19:44:26 ...
 kernel:[  622.203975] 1e20:                                     c0f04fc8 d36f1ea0 d583b400 c0f04fc8

Message from syslogd@orangepizero at Dec  4 19:44:26 ...
 kernel:[  622.203992] 1e40: d36f1ea0 00000030 00000030 bf9c6015 c04e3ca5 d36de608 00000000 9ea2a9c0

Message from syslogd@orangepizero at Dec  4 19:44:26 ...
 kernel:[  622.219712] 1e60: c91c0c80 d36f1e98 c91c0c80 c91c0c80 d36f1f20 c0f04fc8 00000000 c373a480

Message from syslogd@orangepizero at Dec  4 19:44:26 ...
 kernel:[  622.219723] 1e80: 013b42c8 c0814b3f 00000080 d36f1f08 d36f1eb0 c0814bb9 00000000 00000000

Message from syslogd@orangepizero at Dec  4 19:44:26 ...
 kernel:[  622.219733] 1ea0: 00000005 00000000 00000000 d36f1f08 00000000 c0f04fc8 00000000 00000000

Message from syslogd@orangepizero at Dec  4 19:44:26 ...
 kernel:[  622.219755] 1ec0: 00000000 00000080 d36f1f20 9ea2a9c0 000000c0 c373a480 00000000 c0f04fc8

Message from syslogd@orangepizero at Dec  4 19:44:26 ...
 kernel:[  622.231838] 1ee0: 00000030 00000000 00000000 c0274ae5 00000030 00000000 00000000 000000c0

Message from syslogd@orangepizero at Dec  4 19:44:26 ...
 kernel:[  622.231849] 1f00: b1f005f8 00000030 00000005 00000000 00000030 d36f1f00 00000001 c0f04fc8

Message from syslogd@orangepizero at Dec  4 19:44:26 ...
 kernel:[  622.231859] 1f20: c373a480 00000000 00000000 00000000 00000000 00000000 00000000 00000000

Message from syslogd@orangepizero at Dec  4 19:44:26 ...
 kernel:[  622.231881] 1f40: 00000000 00000000 00000060 9ea2a9c0 00000001 c0f04fc8 c373a481 b28ff890

Message from syslogd@orangepizero at Dec  4 19:44:26 ...
 kernel:[  622.242836] 1f60: 00000004 c373a480 b1f005f8 00000030 013b42c8 c0275aa5 00000000 9ea2a9c0

Message from syslogd@orangepizero at Dec  4 19:44:26 ...
 kernel:[  622.242847] 1f80: d36f1f98 9ea2a9c0 00000030 b1f005f8 b28ff890 00000004 c0100284 d36f0000

Message from syslogd@orangepizero at Dec  4 19:44:26 ...
 kernel:[  622.242857] 1fa0: 00000004 c0100061 00000030 b1f005f8 0000001d b1f005f8 00000030 00000001

Message from syslogd@orangepizero at Dec  4 19:44:26 ...
 kernel:[  622.242880] 1fc0: 00000030 b1f005f8 b28ff890 00000004 00000030 b28feacc 00000004 013b42c8

Message from syslogd@orangepizero at Dec  4 19:44:26 ...
 kernel:[  622.255742] 1fe0: 00000004 b28feaa8 b6beb3ad b6bed4d6 800f0030 0000001d 00000000 00000000

Message from syslogd@orangepizero at Dec  4 19:44:26 ...
 kernel:[  622.256056] [<bf9973f0>] (hci_send_sco [bluetooth]) from [<bf9c6015>] (sco_sock_sendmsg+0x12d/0x1d0 [bluetooth])

Message from syslogd@orangepizero at Dec  4 19:44:26 ...
 kernel:[  622.256195] [<bf9c6015>] (sco_sock_sendmsg [bluetooth]) from [<c0814b3f>] (sock_sendmsg+0x23/0x2c)

Message from syslogd@orangepizero at Dec  4 19:44:26 ...
 kernel:[  622.266529] [<c0814b3f>] (sock_sendmsg) from [<c0814bb9>] (sock_write_iter+0x71/0xa4)

Message from syslogd@orangepizero at Dec  4 19:44:26 ...
 kernel:[  622.266545] [<c0814bb9>] (sock_write_iter) from [<c0274ae5>] (vfs_write+0x2b1/0x300)

Message from syslogd@orangepizero at Dec  4 19:44:26 ...
 kernel:[  622.266574] [<c0274ae5>] (vfs_write) from [<c0275aa5>] (ksys_write+0x85/0x94)

Message from syslogd@orangepizero at Dec  4 19:44:26 ...
 kernel:[  622.279608] [<c0275aa5>] (ksys_write) from [<c0100061>] (ret_fast_syscall+0x1/0x62)

Message from syslogd@orangepizero at Dec  4 19:44:26 ...
 kernel:[  622.279614] Exception stack(0xd36f1fa8 to 0xd36f1ff0)

Message from syslogd@orangepizero at Dec  4 19:44:26 ...
 kernel:[  622.279636] 1fa0:                   00000030 b1f005f8 0000001d b1f005f8 00000030 00000001

Message from syslogd@orangepizero at Dec  4 19:44:26 ...
 kernel:[  622.279650] 1fc0: 00000030 b1f005f8 b28ff890 00000004 00000030 b28feacc 00000004 013b42c8

Message from syslogd@orangepizero at Dec  4 19:44:26 ...
 kernel:[  622.292679] 1fe0: 00000004 b28feaa8 b6beb3ad b6bed4d6

Message from syslogd@orangepizero at Dec  4 19:44:26 ...
 kernel:[  622.292696] Code: f85d eb04 4605 460c (f8d0) 6364

borine, 2021-12-05:
> kernel:[ 622.084662] Internal error: Oops: 5 [#1] SMP THUMB2

A kernel OOPS suggests a system-level problem. If the only failure you are seeing is when using SCO audio I would suggest beginning your debugging efforts with the bluetooth subsystem. Start with the adapter firmware. You don't say what bluetooth adapter you are using, but from the logs it appears to be a USB adapter. So start by searching for issues/bug reports relating to the use of your adapter, and/or the specific bluetooth module it uses, to make sure you have a firmware known to work with the kernel version you have. Look at `dmesg` output for any messages relating to the adapter.

lelemm, 2021-12-06:
this USB adapter is an generic 

during boot:
[   14.373662] Bluetooth: Core ver 2.22
[   14.373872] NET: Registered protocol family 31
[   14.373882] Bluetooth: HCI device and connection manager initialized
[   14.373912] Bluetooth: HCI socket layer initialized
[   14.373925] Bluetooth: L2CAP socket layer initialized
[   14.373956] Bluetooth: SCO socket layer initialized
[   14.377356] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[   14.514422] zram1: detected capacity change from 0 to 52428800
[   14.595186] usbcore: registered new interface driver btusb
[   14.610876] Bluetooth: hci0: CSR: Unbranded CSR clone detected; adding workarounds...

when trying to use SCO:

[172820.242730] Bluetooth: RFCOMM TTY layer initialized
[172820.242773] Bluetooth: RFCOMM socket layer initialized
[172820.242822] Bluetooth: RFCOMM ver 1.11
[172838.844949] input: X13 (AVRCP) as /devices/virtual/input/input0
[172897.409945] Bluetooth: hci0: SCO packet for unknown connection handle 0
[172897.416746] Bluetooth: hci0: SCO packet for unknown connection handle 0
[172897.423798] Bluetooth: hci0: SCO packet for unknown connection handle 0
[172897.430607] Bluetooth: hci0: SCO packet for unknown connection handle 0
[172897.437576] Bluetooth: hci0: SCO packet for unknown connection handle 0
[172897.444382] Bluetooth: hci0: SCO packet for unknown connection handle 0
[172897.451305] Bluetooth: hci0: SCO packet for unknown connection handle 2
[172897.458212] Bluetooth: hci0: SCO packet for unknown connection handle 1
[172897.464989] Bluetooth: hci0: SCO packet for unknown connection handle 1
[172897.471915] Bluetooth: hci0: SCO packet for unknown connection handle 0
[172897.478856] Bluetooth: hci0: SCO packet for unknown connection handle 0
[172897.485656] Bluetooth: hci0: SCO packet for unknown connection handle 2560
[172897.492839] Bluetooth: hci0: SCO packet for unknown connection handle 1072
and goes on with this error until the system freezes

lelemm, 2021-12-06:
Any good bluetooth dongle that I should try?
I will try to use a raspberry pi instead of orange pi to see if there is any change.

Arkq, 2021-12-07:
> [ 14.610876] Bluetooth: hci0: CSR: Unbranded CSR clone detected; adding workarounds...

I think that this might be the issue. It seems that your board is using some chinese crappy BT chip. Please, check Linux kernel mailing list for similar issues.

lelemm, 2021-12-07:
probably is. problem is to find a reliable option to use. any suggestions?

borine, 2021-12-07:
> any suggestions?

I'm not aware of any up-to-date online databases of compatible bluetooth adapters, so really you just have to rely on amateur reviews and posts on various distribution's issue lists or wiki pages to see what is known to work with your particular distribution.

It's been some time since I bought mine, and I have no idea if it will work on others' systems, but FWIW it is an ASUS BT400 bluetooth 4.0 which has a Broadcom BT chip. SCO did not work out-of-the-box, but A2DP is good. To get SCO working with it I had to extract a firmware blob from the Windows driver, although newer revisions may now have that firmware pre-installed, I don't know. I posted the script I used to obtain the firmware in a previous comment here: https://github.com/Arkq/bluez-alsa/issues/347#issuecomment-663083402

I

lelemm, 2021-12-08:
nice. thanks. I will try other dongles i have around, if none works, I will try the one you suggested.

closed 2021-12-08 by lelemm

lelemm, 2021-12-09:
You were right, crap bluetooth dongle was the problem. Got one from ORICO, was detected correctly:
[71711.555969] usb 1-1.4: new full-speed USB device number 5 using dwc_otg
[71711.740578] usb 1-1.4: New USB device found, idVendor=0a12, idProduct=0001, bcdDevice=88.91
[71711.740611] usb 1-1.4: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[71711.740630] usb 1-1.4: Product: CSR8510 A10

And now SCO is working for playing and recording.

Thanks!