arkq/bluez-alsa #515: Max Volume output is Approx 50% of max volume from other devices

Muppet1856, 2021-12-06:
I've been working an a bluezalsa shairport-sync bluetooth bridge on a Pi Zero W 2, and I seem to have run up against something that I figured would be trivial, but hasn't been.

I've been able to play a local wav file using aplay to my BT speaker, but when I do it is about half the volume of what I can get form it from my cell phone when it is paired to the same speaker.  Additionally, when I pair, I lose the ability to have local volume control on the speaker itself.  It seems like the speaker is being set to 50% volume via AVRCP and being stuck there, even when I change the volume on amixer.

Any thoughts on what theheck I'm missing?  How can I correct this?

Arkq, 2021-12-06:
You might try to use AVRCP with bluealsa by disabling software volume for A2DP PCMs (by default bluealsa uses software volume for A2DP and hardware volume for HFP/HSP). Simply add `--a2dp-volume` to bluealsa command line options. Alternatively, you can control this option per PCM with `bluealsa-cli soft-volume <pcm>`. However, AVRCP volume control might require rather new BlueZ version due to a bug in older versions.

borine, 2021-12-12:
could this possibly be a shairport issue? see for example https://github.com/mikebrady/shairport-sync/issues/1179

borine, 2022-01-30:
@Muppet1856 Is this issue still happening? I have been unable to find any similar reports, and am unable to replicate it on my own system. To be able to help, we need enough information to be able to replicate it.

If you still require help, please can to say which version of bluealsa you are using and how you obtained it - from a distribution or self-built? If self built, please also state the configure options used.

What command line options do you use when starting `bluealsa`, and what command line do you use when starting `aplay`; have you added any ALSA configuration entries to `~/.asoundrc` or `/etc/asound.conf`?

igrekster, 2022-07-19:
This has happened to me: the volume was fine until I updated the system (pretty sure was running the same versions of `bluez-alsa`, `bluetoothd`, `alsa-lib`, `alsa-plugins` as before), now the volume is something like 10-20%.

Things I've tried:
* `--a2dp-volume`
* `bluealsa-cli volume /org/bluealsa/hci0/dev_5C_C6_E9_35_BE_63/a2dpsrc/sink` (set at max, `127`)
* `bluetoothd --noplugin=avrcp`
* downgrading `bluez-alsa` from `4.0.0` to `3.1.0`
* Android phone volume is fine

Here are some details:
```
Device 5C:C6:E9:35:BE:63 (public)
        Name: EDIFIER S350DB
        Alias: EDIFIER S350DB
        Class: 0x00240428
        Icon: audio-card
        Paired: yes
        Trusted: yes
        Blocked: no
        Connected: yes
        LegacyPairing: no
        UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
        UUID: Advanced Audio Distribu.. (0000110d-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
```
```
Service: org.bluealsa
Version: v4.0.0
Adapters: hci0
Profiles:
  A2DP-source : SBC MP3 AAC aptX FastStream LDAC
  A2DP-sink   : SBC MP3 AAC aptX FastStream
```

```
pcm.edifier {
        type plug
        slave.pcm {
                type bluealsa
                device "5C:C6:E9:35:BE:63"
                service "org.bluealsa"
                profile "a2dp"
                delay 10000
        }
        hint {
                show on
                description "Edifier"
        }
}

ctl.edifier {
        type bluealsa
        device "5C:C6:E9:35:BE:63"
        service "org.bluealsa"
        hint {
                show on
                description "Edifier Control Device"
        }
}
```

I can rebuild the packages and apply patches if that helps to debug this.

Arkq, 2022-07-19:
@igrekster I'm not sure whether I understand you correctly. Does volume work with version 3.1.0 ? If that's true, and you've got some time for debugging, you can try to do `git bisect` between those versions and check which commit breaks things.

As for `bluetoothd --noplugin=avrcp` I would rather stick with default bluez setup. I doubt that the AVRCP causes the issues here. What I suspect is that the BT headset  remembered the volume for your controller (device on which bluealsa is running) and it is set to ~30%. Then, on top of that you've got software volume scaling done by bluealsa, so effectively you can see volume scaling in range 0-30%. Try something like this: `bluealsa-cli soft-volume /org/bluealsa/hci0/dev_5C_C6_E9_35_BE_63/a2dpsrc/sink n` and then check bluealsa volume control.

igrekster, 2022-07-19:
> What I suspect is that the BT headset remembered the volume for your controller (device on which bluealsa is running) and it is set to ~30%.

Hmm, I fixed the volume issue by re-pairing the speakers (`remove 5C:C6:E9:35:BE:63`, `pair 5C:C6:E9:35:BE:63`, `connect 5C:C6:E9:35:BE:63`).

borine, 2023-02-15:
This issue has had no updates in the last 6 months, and appears to be more likely caused by either Bluez or the characteristics of the specific remote device. I am closing it now.

There is now additional detail in the BlueALSA manual pages covering volume control.

If anyone experiences a similar issue with the latest BlueALSA code, please raise a new issue giving clear instructions on how to reproduce the issue. If we can discover the underlying cause then it may be possible to further improve the documentation to help users avoid or rectify whatever the cause(s) are.

closed 2023-02-15 by borine