arkq/bluez-alsa #517: systemd file for bluealasa-aplay does not work

borine, 2021-12-14:
I've just tried the new `misc/systemd` files for the first time, and found that bluealsa-aplay does not work. The directive
```
PrivateDevices=true 
```
denies access to the `/dev/snd/*` device nodes, so `bluealsa-aplay` is unable to open any CTLs or PCMs on local soundcards. At least that is what I am seeing with RapsberryPiOS bullseye (systemd 247, linux 5.10).

Please remove that line from `misc/systemd/bluealsa-aplay.service.in` only.

hrc-amilcar, 2021-12-14:
I ran into a similar issue with the service file. The restricted address families were also preventing bluealsa from functioning properly.

I removed the entire #Sandboxing section to get everything working on my Yocto [dunfell] based embedded device.

borine, 2021-12-14:
> The restricted address families were also preventing bluealsa from functioning

Hmm, that's strange. I have no issues with the other sandbox settings, and would not expect any. It is recommended practice with systemd to lock down service as much as possible for security (especially as these units run both services under the `root` account). Do you have a note of any specific error messages in your logs relating to these settings?

I think there is also an issue with starting bluealsa-aplay before bluealsa, but that is not related to security settings.

borine, 2021-12-14:
A little more reading, and I think I have to backtrack a little on my previous comment. Re `PrivateNetwork=`, the systemd docs says:
```
for AF_UNIX this has the effect that AF_UNIX sockets in the abstract socket namespace
of the host will become unavailable to the unit's processes (however, those located in the
file system will continue to be accessible).
```

Bluealsa uses `socketpair()` to create its AF_UNIX socket, so I guess that is in the "abstract namespace", and `PrivateNetwork=yes` should indeed block it. But on Rpi with RaspiOS bullseye, it does not.  However. I now agree with @hrc-amilcar that PrivateNetwork is also too restrictive for the upstream service file.

hrc-amilcar, 2021-12-14:
I think that's it. I vaguely remember it failing on `socketpair`.

I can try and put back the `PrivateNetwork` and `RestrictAddressFamilies` lines to recreate the error. If you still need to see it.

borine, 2021-12-14:
@hrc-amilcar any errors caused by `RestrictAddressFamilies=` would be useful to know - searching the codebase I find only BLUETOOTH and UNIX domain sockets used.

@Arkq - [pedant's corner :smile:] hci.c calls `socket(PF_BLUETOOTH, ...)`. I wonder whether technically that should be `socket(AF_BLUETOOTH, ...)`? They are defined to be the same on my systems, but the domain arg is for "Address Family", not "Protocol Family" ?

Arkq, 2021-12-14:
> [pedant's corner smile] hci.c calls socket(PF_BLUETOOTH, ...). I wonder whether technically that should be socket(AF_BLUETOOTH, ...)?

Technically speaking (according to socket() manual) the domain parameter selects protocol family. Quoting from `man socket`:

> The domain argument specifies a communication domain; this selects the protocol family which will be used for communication.

However, 3 lines later we've got a list of domain constants all with `AF_` prefix :D. There is also a note at the bottom which says:

> The manifest constants used under 4.x BSD for protocol families are PF_UNIX, PF_INET, and so on, while AF_UNIX, AF_INET, and so on are used for address families.  However, already the BSD man page promises:  "The
       protocol family generally is the same as the address family", and subsequent standards use AF_* everywhere.

For the sake of unification I think that it would be reasonable to change that to AF_ anyway :) Thanks for pointing that out.

borine, 2021-12-15:
I've gone through the systemd docs for the directives used in the two unit files, and  make the following observations:

- `PrivateNetwork=true` disallows use of AF_UNIX sockets in abstract namespace, so prevents bluealsa from creating a control channel for its clients. Therefore this directive should not be used in either unit file
- `PrivateDevices=true` disallows use of most /dev nodes, and in particular all /dev/snd/ nodes. This is OK for bluealsa but prevents bluealsa-aplay from opening PCMs and CTLs on local soundcards. Therefore this directive should nor be used in  the bluealsa-aplay unit file.
- `RemoveIPC=true` has no effect on services running as root - so is redundant here and should not be used in either unit file.
- `RestrictAddressFamilies=` is appropriate in both unit files, and should remain in place unless clear evidence suggests otherwise.

I recommend the following patch:

```diff
diff --git a/misc/systemd/bluealsa-aplay.service.in b/misc/systemd/bluealsa-aplay.service.in
index a44e458..7783e04 100644
--- a/misc/systemd/bluealsa-aplay.service.in
+++ b/misc/systemd/bluealsa-aplay.service.in
@@ -19,12 +19,9 @@ ExecStart=@bindir@/bluealsa-aplay
 Restart=on-failure
 
 # Sandboxing
-PrivateDevices=true
-PrivateNetwork=true
 PrivateTmp=true
 ProtectHome=true
 ProtectSystem=strict
-RemoveIPC=true
 RestrictAddressFamilies=AF_UNIX
 
 [Install]
diff --git a/misc/systemd/bluealsa.service.in b/misc/systemd/bluealsa.service.in
index d849de5..bfab3ad 100644
--- a/misc/systemd/bluealsa.service.in
+++ b/misc/systemd/bluealsa.service.in
@@ -22,11 +22,9 @@ Restart=on-failure
 
 # Sandboxing
 PrivateDevices=true
-PrivateNetwork=true
 PrivateTmp=true
 ProtectHome=true
 ProtectSystem=strict
-RemoveIPC=true
 RestrictAddressFamilies=AF_UNIX AF_BLUETOOTH
 
 [Install]
```

commit 70541eec (2021-12-17): Fix too restrictive sandboxing in systemd units

Fixes #517
closed 2021-12-17 by Arkq