arkq/bluez-alsa #518: A2DP sink Codec listing and selection

paul-1, 2021-12-27:
My iphone 12 pro max, with IOS 15.2 connects just fine, but only about half of the time bluealsa will show [ unknown ] for the list of available codecs.   The other time it shows   SBC, SBC, AAC, AAC  (each codec is shown twice)  I don't see anything significantly different in debug data printed by bluealsa when the phone connects.

While the phone does select the codec it wants, and is able to play just fine to bluealsa.  Is it possible to force the phone to use a specific codec?

borine, 2021-12-28:
I also see each codec listed twice (some even three times!) with some devices, but not all. I've never seen `[unknown]` for an A2DP device though. This happens both with `bluealsa-cli` and with D-Bus utilities such as `dbus-send`, so its a server-side issue. I can't tell whether its Bluez or BlueALSA that is to blame.
 
> is it possible to force the phone to use a specific codec?

see:
https://github.com/Arkq/bluez-alsa/blob/master/doc/bluealsa-plugins.7.rst#pcm-plugin
and
https://github.com/Arkq/bluez-alsa/blob/master/doc/bluealsa-cli.1.rst

paul-1, 2021-12-28:
> I can't tell whether its Bluez or BlueALSA that is to blame.

The debug info printed by bluealsa shows nearly identical output when the phone connects(when it shows unknown vs a list of codecs).  I see both SBC and AAC profiles signals from org.freedesktop.DBus.ObjectManager.  I wasn't sure which application was responsible for the reporting of available the codecs.

I've had users report AAC issues with devices since I started to compile bluealsa with AAC support.  So I want to give them a way to choose which codec is used......unfortunately, all I have is IOS devices.   I do have a few older android devices, but they only support SBC, so I cannot test them out.

I have been able switch codecs, I'm just not sure if it actually works for the phone.   If I switch my iphone to SBC, it fails to produce audio.  (The iphone thinks it is playing though).    I'm using bluealsa-aplay if that matters.  The process I use,  The phone connects, I switch it to SBC (I see the switch happen in the bluealsa debug info), then I start bluealsa-aplay.

Arkq, 2021-12-28:
>  I wasn't sure which application was responsible for the reporting of available the codecs.

Responsible for reporting available codecs are Bluetooth devices. BlueZ and BlueALSA only pass this information. If you see duplicated codecs in the list, it means that the iPhone AVDTP response contains duplicated entries for available codecs. I thought about making this list unique (based on codec capabilities), but it's in my backlog so to speak.

> If I switch my iphone to SBC, it fails to produce audio.

To be honest I've got the same result (maybe a year ago when I've checked that)... I thought that maybe the SBC capabilities are not valid for iPhone... But to be honest I'm not sure why switching to SBC (from AAC) does not work. Unfortunately, I've got no Android smartphone for comparison (Android device which supports more than one codec).

borine, 2021-12-28:
> Responsible for reporting available codecs are Bluetooth devices

that's interesting. I'm using bluealsa sink (with bluealsa-aplay) and bluealsa source. From a terminal on the source I get:
```
bluealsa-cli codec /org/bluealsa/hci0/dev_5C_F3_70_9B_FB_1A/a2dpsrc/sink
Available codecs: aptX aptX SBC SBC aptX
Selected codec: aptX
```

And from the sink I get:
```
bluealsa-cli codec /org/bluealsa/hci0/dev_58_00_E3_B9_DD_F2/a2dpsnk/source 
Available codecs: aptX SBC SBC aptX aptX SBC
Selected codec: aptX
```

So there are no "real" devices, only bluealsa, but I get each codec reported 3 times.

Also, from the source I can change the codec, but on the sink I get:
```
bluealsa-cli codec /org/bluealsa/hci0/dev_58_00_E3_B9_DD_F2/a2dpsnk/source sbc
bluealsa-cli: E: CMD "codec": Couldn't select BlueALSA PCM Codec: Input/output error
```

Is that expected? If so I think the bluealsa-cli and bluealsa-plugins man pages need updating.

Arkq, 2021-12-28:
> So there are no "real" devices, only bluealsa, but I get each codec reported 3 times.

BlueALSA registers supported codecs in BlueZ, and then BlueZ sends AVDTP message accordingly. It is also a "real" device :) You've got 2x "unused" codec and 3x used codec - BlueALSA keeps at least 2 extra slots all the time.

> Also, from the source I can change the codec, but on the sink I get:

Hmmm... not sure to be honest. It might depend on who has initialized A2DP connection... But as far as I remember codec renegotiation should be possible from sink and from source. I will have to check Bluetooth spec to be sure.

paul-1, 2021-12-28:
>Also, from the source I can change the codec, but on the sink I get:

It works on an iphone, as long as you do it before opening an audio connection from bluealsa-aplay
```
tc@piTouch:~$ sudo bluealsa-cli codec /org/bluealsa/hci0/dev_90_81_58_46_8B_A5/a2dpsnk/source
Available codecs: SBC SBC AAC AAC
Selected codec: AAC
tc@piTouch:~$ sudo bluealsa-cli codec /org/bluealsa/hci0/dev_90_81_58_46_8B_A5/a2dpsnk/source SBC
tc@piTouch:~$ sudo bluealsa-cli codec /org/bluealsa/hci0/dev_90_81_58_46_8B_A5/a2dpsnk/source
Available codecs: SBC SBC AAC AAC
Selected codec: SBC
```
This is the transitioning signals in bluealsa
```luealsa: D: ../../src/bluez.c:1049: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
bluealsa: D: ../../src/bluez.c:1137: Adding new Stream End-Point: 90:81:58:46:8B:A5: SBC
bluealsa: D: ../../src/bluez.c:1049: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
bluealsa: D: ../../src/bluez.c:1137: Adding new Stream End-Point: 90:81:58:46:8B:A5: SBC
bluealsa: D: ../../src/bluez.c:1049: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
bluealsa: D: ../../src/bluez.c:1137: Adding new Stream End-Point: 90:81:58:46:8B:A5: AAC
bluealsa: D: ../../src/bluez.c:1049: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
bluealsa: D: ../../src/bluez.c:1137: Adding new Stream End-Point: 90:81:58:46:8B:A5: AAC
bluealsa: D: ../../src/bluez.c:1049: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
bluealsa: D: ../../src/a2dp.c:227: Unknown vendor codec [len=14]: 4c0000000180008000840084e200
bluealsa: D: ../../src/bluez.c:1137: Adding new Stream End-Point: 90:81:58:46:8B:A5: (null)
bluealsa: D: ../../src/bluez.c:1049: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
bluealsa: D: ../../src/a2dp.c:227: Unknown vendor codec [len=14]: 4c0000000180008000840084e200
bluealsa: D: ../../src/bluez.c:1137: Adding new Stream End-Point: 90:81:58:46:8B:A5: (null)
bluealsa: D: ../../src/bluez.c:1049: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
bluealsa: D: ../../src/bluez.c:1049: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
bluealsa: D: ../../src/dbus.c:66: Called: org.bluez.MediaEndpoint1.SetConfiguration() on /org/bluez/hci0/A2DP/AAC/sink/1
bluealsa: D: ../../src/bluez.c:303: A2DP Sink (AAC) configured for device 90:81:58:46:8B:A5
bluealsa: D: ../../src/bluez.c:306: A2DP selected configuration blob [len=6]: 800104835b60
bluealsa: D: ../../src/bluez.c:308: PCM configuration: channels: 2, sampling: 44100
bluealsa: D: ../../src/dbus.c:66: Called: org.bluealsa.PCM1.Open() on /org/bluealsa/hci0/dev_90_81_58_46_8B_A5/a2dpsnk/source
bluealsa: D: ../../src/dbus.c:66: Called: org.bluealsa.PCM1.GetCodecs() on /org/bluealsa/hci0/dev_90_81_58_46_8B_A5/a2dpsnk/source
bluealsa: D: ../../src/dbus.c:66: Called: org.bluealsa.PCM1.SelectCodec() on /org/bluealsa/hci0/dev_90_81_58_46_8B_A5/a2dpsnk/source
bluealsa: D: ../../src/dbus.c:66: Called: org.bluez.MediaEndpoint1.ClearConfiguration() on /org/bluez/hci0/A2DP/AAC/sink/1
bluealsa: D: ../../src/bluez.c:344: Disconnecting media endpoint: /org/bluez/hci0/A2DP/AAC/sink/1
bluealsa: D: ../../src/ba-transport.c:1424: Closing PCM: 15
bluealsa: D: ../../src/bluez.c:1148: Signal: org.freedesktop.DBus.ObjectManager.InterfacesRemoved()
bluealsa: D: ../../src/ba-transport.c:437: PCM clients check keep-alive: 0 ms
bluealsa: D: ../../src/bluez.c:1049: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
bluealsa: D: ../../src/dbus.c:66: Called: org.bluez.MediaEndpoint1.SetConfiguration() on /org/bluez/hci0/A2DP/SBC/sink/2
bluealsa: D: ../../src/a2dp.c:321: Selected A2DP SBC bit-pool range: [2, 250]
bluealsa: D: ../../src/bluez.c:303: A2DP Sink (SBC) configured for device 90:81:58:46:8B:A5
bluealsa: D: ../../src/bluez.c:306: A2DP selected configuration blob [len=4]: 211502fa
bluealsa: D: ../../src/bluez.c:308: PCM configuration: channels: 2, sampling: 44100
bluealsa: D: ../../src/bluez.c:483: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/sink/3
bluealsa: D: ../../src/bluez.c:392: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/sink/3
bluealsa: D: ../../src/dbus.c:66: Called: org.bluealsa.PCM1.Open() on /org/bluealsa/hci0/dev_90_81_58_46_8B_A5/a2dpsnk/source
bluealsa: D: ../../src/ba-transport.c:873: Freeing transport: A2DP Sink (AAC)
bluealsa: D: ../../src/dbus.c:66: Called: org.bluealsa.PCM1.GetCodecs() on /org/bluealsa/hci0/dev_90_81_58_46_8B_A5/a2dpsnk/source
```

It will act like its playing, then if I pause the music on the phone, bluealsa spits out quite a few errors.  Here is when music starts and when I press pause, that is when the SBC errors occur.

```
bluealsa: D: ../../src/dbus.c:66: Called: org.bluealsa.PCM1.Open() on /org/bluealsa/hci0/dev_90_81_58_46_8B_A5/a2dpsnk/source
bluealsa: D: ../../src/bluez.c:1239: Signal: org.freedesktop.DBus.Properties.PropertiesChanged(): org.bluez.MediaTransport1: State
bluealsa: D: ../../src/ba-transport.c:564: New A2DP transport: 8
bluealsa: D: ../../src/ba-transport.c:565: A2DP socket MTU: 8: R:672 W:1004
bluealsa: D: ../../src/bluez.c:1239: Signal: org.freedesktop.DBus.Properties.PropertiesChanged(): org.bluez.MediaTransport1: State
bluealsa: D: ../../src/ba-transport.c:1145: Starting transport: A2DP Sink (SBC)
bluealsa: D: ../../src/ba-transport.c:293: Created BT socket duplicate: [8]: 11
bluealsa: D: ../../src/a2dp-sbc.c:346: IO loop: START: a2dp_sbc_dec_thread: A2DP Sink (SBC)
[... cut ...]
bluealsa: E: ../../src/a2dp-sbc.c:375: SBC decoding error: No such file or directory
bluealsa: E: ../../src/a2dp-sbc.c:375: SBC decoding error: No such file or directory
bluealsa: D: ../../src/bluez.c:1239: Signal: org.freedesktop.DBus.Properties.PropertiesChanged(): org.bluez.MediaTransport1: State
bluealsa: D: ../../src/ba-transport.c:305: Closing BT socket duplicate [8]: 11
bluealsa: D: ../../src/ba-transport.c:618: Closing A2DP transport: 8
bluealsa: D: ../../src/ba-transport.c:1487: Exiting IO thread [ba-a2dp-sbc]: A2DP Sink (SBC)
```

borine, 2021-12-29:
SBC repotrs bizarre error codes. ENOENT (No such file or directory) indicates that the first byte of an SBC frame was not the correct syncword - this means the frame is corrupted, and most likely results from packet loss within the transport.

@Arkq - I discovered that the bluealsa sink failed to change codecs because I run it on a read-only file system. Remount as RW, restart bluetoothd, and the problem goes away. A simple case of user error - sorry!

paul-1, 2021-12-29:
Does the phone play music after forcing the codec change.   Is this just an iPhone bug?

borine, 2021-12-29:
> Does the phone play music after forcing the codec change. Is this just an iPhone bug?

This is just bluealsa <-> bluealsa. When the codec is changed, the running stream is stopped (alsa applications see ENODEV "No such device" error). But the application can be restarted and the playback continues with the new codec.

I've just experimented with an Android 'phone, but I cannot persuade it to change codecs. From the 'phone settings, I can select AptX, but it ignores my selection and remains on SBC. Using bluealsa-cli I get:
```
$ bluealsa-cli codec /org/bluealsa/hci0/dev_30_0A_C5_2A_74_B6/a2dpsnk/source aptx
bluealsa-cli: E: CMD "codec": Couldn't select BlueALSA PCM Codec: Input/output error
```

Arkq, 2021-12-29:
> SBC repotrs bizarre error codes. ENOENT (No such file or directory) indicates that the first byte of an SBC frame was not the correct syncword - this means the frame is corrupted, and most likely results from packet loss within the transport.

Yes, these error messages are pissing me off as well :D. I think that it will be reasonable to add SBC specific error messages rather than `strerror()` ones.

>  From the 'phone settings, I can select AptX, but it ignores my selection and remains on SBC.

The question is whether the aptX codec is indeed there and is usable. Maybe it is only displayed in the GUI. You will have to check logcat on Android. I remember that once I've got an Android image with aptX codec available (the encoder file was on the file system), but I was not able to use it because some issues with Bluetooth codec manager on Android (there were error messages in Android logs).

borine, 2021-12-29:
> it will be reasonable to add SBC specific error messages rather than strerror() ones.

These are from notes I made last year:
```
libsbc decoder errors:
-1 (EPERM) - frame incomplete
-2 (ENOENT) - invalid syncword
-3 (ESRCH) - crc check failed

libsbc encoder errors:
-1 (EPERM) - incomplete input data
-2 (ENOENT) - sampling frequency not supported
-3 (ESRCH) - invalid frame block count
-4 (EINTR) - invalid frame subband count
-5 (EIO) - invalid frame bitpool
```

Arkq, 2021-12-29:
I've just sketched something like this: https://github.com/Arkq/bluez-alsa/blob/sbc/src/codec-sbc.c#L199

> -3 (ESRCH) - invalid frame block count

Where you've found `-3` in encoder? In libsbc-1.5 I can see only CRC check error reported as `-3`.

paul-1, 2021-12-29:
It appears there are other issues with the swapping of codecs with sinks.  It seems to work just fine with sources......

Anyway, another way to solve the problem of a codec - device not working well.   Would be to have the ability to disable a codec either at runtime of bluealsa, or from a dbus call to unregister the codec endpoint.

borine, 2021-12-30:
>Where you've found -3 in encoder? 

https://github.com/rtc-draper/libsbc/blob/8472d68c7de6db414f94ee3b2ce4905fba49ed7e/sbc/lib/sbc.c#L1088

Arkq, 2021-12-30:
> Anyway, another way to solve the problem of a codec - device not working well. Would be to have the ability to disable a codec either at runtime of bluealsa, or from a dbus call to unregister the codec endpoint.

Yes, it might be a good idea to add an option to fine-tune the subset of available codecs. However, there might be one caveat here. BlueALSA registers A2DP codecs in BlueZ at it's startup. Then BlueZ uses the list of available codecs (not only from BlueALSA, but potentially from other providers) and exposes that list to external Bluetooth device via AVDTP or uses matching codec (the search for codec is done in the order of registration) when establishing connection (when BlueZ is initiator). BlueALSA is configured in such a way, that it allows multiple Bluetooth devices to be connected at the same time (codecs are registered more than once). So now when narrowing the list of used A2DP codecs we could do it also in a global way - at startup BlueALSA will register only "enabled" codecs. Doing that on runtime might result in dropped A2DP connection if particular codec was already used.

Ideally, it would be nice to allow to filter available codecs per remote device by Bluetooth MAC address. E.g.: device 00:33:22:33:11:22 sees all codecs, but device 33:33:33:33:33:33 sees only SBC. However, with current BlueZ capabilities it will not be possible.

paul-1, 2021-12-30:
Command line might be the simplest approach ATM, do it just like profiles get enabled.  --disable_codec=AAC  --disable_codec=aptX....etc.

Arkq, 2021-12-30:
@paul-1 see this draft for codec selection: https://github.com/Arkq/bluez-alsa/pull/519

PR is not ready yet, but it's already functional for A2DP codecs. By default (currently) all available (compiled-in) A2DP codecs are enabled. And you will have two options: 1) create your own set of enabled codes by overriding defaults; 2) modify defaults. E.g.:

1) `bluealsa -c sbc -c aptx`
2) `bluealsa -c -aac -c -lc3plus -c +codec-not-enabled-by-default`

The +/- syntax will be also available for BT profiles.

What do you think?

paul-1, 2021-12-31:
Seems okay, I’ll test it out in the morning.  Should the output of —help show a list of potential choices.  It wasn’t obvious what would happen if a codec on the command line doesn’t match anything.

Arkq, 2021-12-31:
Like I've said, it's only a draft. In the final version there will be a
proper documentation in man page and in help message. Also, HFP codecs will
be tunable.

paul-1, 2021-12-31:
Works as expected.  If I disable AAC, the iphone properly connects with SBC.

Arkq, 2022-01-14:
The feature for codec enable/disable via command line has been just merged into the master branch. In the meantime this issue has got a lot of messages, so as a recap I will state the original question here:

> How to switch iPhone to SBC codec when BlueALSA has SBC and AAC enabled?
> Debug attempt: https://github.com/Arkq/bluez-alsa/issues/518#issuecomment-1002288586

paul-1, 2022-01-15:
Looks good to me, really busy at work for another week.  I've not had a try with the dbus interface part of this but will.

Thanks for implementing.

paul-1, 2022-01-23:
Finally got this built and tested. Everything checked out

closed 2022-01-23 by paul-1