arkq/bluez-alsa #523: Question: br-connection-profile-unavailable unavailable

humarf, 2022-01-11:
Hello,

since upgrading my raspberry pi installation to bluez 5.63 I get an error when connecting my phone via bluetooth (pairing and trusting devices still works). It says:
`Failed to connect: org.bluez.Error.Failed br-connection-profile-unavailable`
I am not sure what this is about. It could also be a bug in bluez. When searching the web for that error one finds the answer that the package pulseaudio-bluetooth should be installed. I guess this is not a proper solution when running alsa and bluealsa.

I wonder if this missing profile has moved from a bluez package to pulseaudio-bluetooth package in latest update. If this is the case shouldn't bluealsa package provide what's missing as well?

Happy to hear your advice.

Best

Arkq, 2022-01-11:
I've just checked current master of bluez-alsa with bluez v5.63 (custom build: `./configure --prefix=/ --enable-debug --disable-android --disable-systemd --disable-client --disable-monitor --disable-cups --enable-deprecated --enable-testing`) and everything seems to be working fine... I think that there are no "extra" requirements for applications which implement audio profiles.

You will have to look into bluealsa service logs. Maybe you will see something there.

humarf, 2022-01-12:
Thanks a lot for testing. Good news that in general this should still work. Bad news that I have no clue what's wrong. Maybe you could point me into the right direction. Based on your logs advice I have seen the following.

I start bluetooth and bluealsa service. It start with no errors. Bluealsa writes this to the log:
`Jan 12 07:27:13 kitchenDS bluetoothd[15169]: Endpoint registered: sender=:1.154 path=/org/bluez/hci0/A2DP/AAC/source/1`
`Jan 12 07:27:13 kitchenDS bluetoothd[15169]: Endpoint registered: sender=:1.154 path=/org/bluez/hci0/A2DP/AAC/source/2`
`Jan 12 07:27:13 kitchenDS bluetoothd[15169]: Endpoint registered: sender=:1.154 path=/org/bluez/hci0/A2DP/SBC/source/1`
`Jan 12 07:27:13 kitchenDS bluetoothd[15169]: Endpoint registered: sender=:1.154 path=/org/bluez/hci0/A2DP/SBC/source/2
`
Now I start bluetoothctl and try to connect my already paired device. Bluetoothctl says this:
`Failed to connect: org.bluez.Error.Failed br-connection-profile-unavailable`

Looking in the logs I find this additional error message:
`Jan 12 07:28:23 kitchenDS bluetoothd[15169]: src/service.c:btd_service_connect() a2dp-source profile connect failed for XX:XX:XX:XX:XX:XX: Protocol not available`

This seems weird since bluealsa reported at startup that A2DP endpoints have been registered.

Do you have any advice for me?

humarf, 2022-01-12:
Got it. a2dp-sink needs to be explicitly activated when starting bluealsa service. In my old installation this was not needed.

So fix is to run `systemctl edit bluealsa` and the add `-p a2dp-sink` as option to bluealsa command.

closed 2022-01-12 by humarf