arkq/bluez-alsa #528: Headset crashes after a few minutes while using mSBC codec

Glaeqen, 2022-01-29:
After 10-15 minutes in HSP/mSBC mode, headset crashes (while being in BT Multipoint mode I can see it being suddenly disconnected both on a phone and a PC). Never seem to happen while having calls on an Android phone.

OS: Arch Linux
Kernel: 5.16.2
bluez: 5.63
Network Card: Intel Corporation Wi-Fi 6 AX200 (rev 1a)
Headset: JLab Go Work

Run with a debug mode:
```
Jan 29 02:20:34 hp-probook-445-g7 pipewire-media-session[1724]: spa.bluez5.sink.sco: failed to write data: -110 (Connection timed out)
Jan 29 02:20:34 hp-probook-445-g7 bluetoothd[89198]: src/adapter.c:dev_disconnected() Device 20:21:05:06:E8:6C disconnected, reason 1
Jan 29 02:20:34 hp-probook-445-g7 bluetoothd[89198]: src/adapter.c:adapter_remove_connection()
Jan 29 02:20:34 hp-probook-445-g7 bluetoothd[89198]: plugins/policy.c:disconnect_cb() reason 1
Jan 29 02:20:34 hp-probook-445-g7 bluetoothd[89198]: plugins/policy.c:disconnect_cb() Device /org/bluez/hci0/dev_20_21_05_06_E8_6C identified for auto-reconnection
Jan 29 02:20:34 hp-probook-445-g7 bluetoothd[89198]: plugins/policy.c:reconnect_set_timer() attempt 1/7 1 seconds
Jan 29 02:20:34 hp-probook-445-g7 bluetoothd[89198]: src/adapter.c:bonding_attempt_complete() hci0 bdaddr 20:21:05:06:E8:6C type 0 status 0xe
Jan 29 02:20:34 hp-probook-445-g7 bluetoothd[89198]: src/device.c:device_bonding_complete() bonding (nil) status 0x0e
Jan 29 02:20:34 hp-probook-445-g7 bluetoothd[89198]: src/device.c:device_bonding_failed() status 14
Jan 29 02:20:34 hp-probook-445-g7 bluetoothd[89198]: src/adapter.c:resume_discovery()
Jan 29 02:20:34 hp-probook-445-g7 bluetoothd[89198]: profiles/audio/avdtp.c:session_cb()
Jan 29 02:20:34 hp-probook-445-g7 bluetoothd[89198]: profiles/audio/avdtp.c:avdtp_ref() 0x562525674c90: ref=3
Jan 29 02:20:34 hp-probook-445-g7 bluetoothd[89198]: profiles/audio/avdtp.c:connection_lost() Disconnected from 20:21:05:06:E8:6C
Jan 29 02:20:34 hp-probook-445-g7 bluetoothd[89198]: profiles/audio/a2dp.c:abort_cfm() Source 0x5625256835b0: Abort_Cfm
Jan 29 02:20:34 hp-probook-445-g7 bluetoothd[89198]: profiles/audio/avdtp.c:avdtp_sep_set_state() stream state changed: OPEN -> IDLE
Jan 29 02:20:34 hp-probook-445-g7 bluetoothd[89198]: profiles/audio/avdtp.c:avdtp_unref() 0x562525674c90: ref=2
Jan 29 02:20:34 hp-probook-445-g7 bluetoothd[89198]: src/service.c:change_state() 0x56252565f4c0: device 20:21:05:06:E8:6C profile a2dp-sink state changed: connected -> disconnected (0)
Jan 29 02:20:34 hp-probook-445-g7 bluetoothd[89198]: plugins/policy.c:policy_disconnect() /org/bluez/hci0/dev_20_21_05_06_E8_6C profile avrcp-controller
Jan 29 02:20:34 hp-probook-445-g7 bluetoothd[89198]: src/service.c:change_state() 0x56252565f390: device 20:21:05:06:E8:6C profile avrcp-controller state changed: connected -> disconnecting (0)
Jan 29 02:20:34 hp-probook-445-g7 bluetoothd[89198]: profiles/audio/avrcp.c:avrcp_disconnect() path /org/bluez/hci0/dev_20_21_05_06_E8_6C
Jan 29 02:20:34 hp-probook-445-g7 bluetoothd[89198]: src/service.c:change_state() 0x562525662630: device 20:21:05:06:E8:6C profile audio-avrcp-target state changed: connected -> disconnected (0)
Jan 29 02:20:34 hp-probook-445-g7 bluetoothd[89198]: src/service.c:change_state() 0x56252565f390: device 20:21:05:06:E8:6C profile avrcp-controller state changed: disconnecting -> disconnected (0)
Jan 29 02:20:34 hp-probook-445-g7 bluetoothd[89198]: profiles/audio/avrcp.c:controller_destroy() 0x562525693200
Jan 29 02:20:34 hp-probook-445-g7 bluetoothd[89198]: profiles/audio/avrcp.c:target_destroy() 0x56252568dee0
Jan 29 02:20:34 hp-probook-445-g7 bluetoothd[89198]: profiles/audio/avctp.c:avctp_set_state() AVCTP Disconnected
Jan 29 02:20:34 hp-probook-445-g7 bluetoothd[89198]: profiles/audio/avctp.c:avctp_disconnected() AVCTP: closing uinput for 20:21:05:06:E8:6C
Jan 29 02:20:34 hp-probook-445-g7 dbus-daemon[1254]: [system] Rejected send message, 0 matched rules; type="method_return", sender=":1.60" (uid=1000 pid=1724 comm="/usr/bin/pipewire-media-session ") interface="(unset)" member="(unset)" error name="(unset)" requested_reply="0" destination=":1.206" (uid=0 pid=89198 comm="/usr/lib/bluetooth/bluetoothd -d ")
Jan 29 02:20:34 hp-probook-445-g7 bluetoothd[89198]: profiles/audio/avdtp.c:avdtp_unref() 0x562525674c90: ref=1
Jan 29 02:20:34 hp-probook-445-g7 bluetoothd[89198]: profiles/audio/sink.c:sink_set_state() State changed /org/bluez/hci0/dev_20_21_05_06_E8_6C: SINK_STATE_CONNECTED -> SINK_STATE_DISCONNECTED
Jan 29 02:20:34 hp-probook-445-g7 bluetoothd[89198]: profiles/audio/a2dp.c:channel_remove() chan 0x56252566e560
Jan 29 02:20:34 hp-probook-445-g7 bluetoothd[89198]: profiles/audio/avdtp.c:avdtp_unref() 0x562525674c90: ref=0
Jan 29 02:20:34 hp-probook-445-g7 bluetoothd[89198]: profiles/audio/avdtp.c:avdtp_free() 0x562525674c90
Jan 29 02:20:34 hp-probook-445-g7 bluetoothd[89198]: src/profile.c:ext_io_disconnected() Unable to get io data for Hands-Free Voice gateway: getpeername: Transport endpoint is not connected (107)
Jan 29 02:20:34 hp-probook-445-g7 bluetoothd[89198]: src/service.c:change_state() 0x562525677520: device 20:21:05:06:E8:6C profile Hands-Free Voice gateway state changed: connected -> disconnected (0)
Jan 29 02:20:34 hp-probook-445-g7 bluetoothd[89198]: src/service.c:btd_service_unref() 0x562525677520: ref=2
Jan 29 02:20:36 hp-probook-445-g7 bluetoothd[89198]: plugins/policy.c:reconnect_timeout() Reconnecting profiles
Jan 29 02:20:36 hp-probook-445-g7 bluetoothd[89198]: profiles/audio/a2dp.c:a2dp_sink_connect() path /org/bluez/hci0/dev_20_21_05_06_E8_6C
Jan 29 02:20:36 hp-probook-445-g7 bluetoothd[89198]: profiles/audio/avdtp.c:avdtp_ref() 0x562525674c90: ref=1
Jan 29 02:20:36 hp-probook-445-g7 bluetoothd[89198]: profiles/audio/avdtp.c:avdtp_ref() 0x562525674c90: ref=2
Jan 29 02:20:36 hp-probook-445-g7 bluetoothd[89198]: profiles/audio/a2dp.c:setup_ref() 0x562525685360: ref=1
Jan 29 02:20:36 hp-probook-445-g7 bluetoothd[89198]: profiles/audio/sink.c:sink_set_state() State changed /org/bluez/hci0/dev_20_21_05_06_E8_6C: SINK_STATE_DISCONNECTED -> SINK_STATE_CONNECTING
Jan 29 02:20:36 hp-probook-445-g7 bluetoothd[89198]: profiles/audio/sink.c:sink_connect() stream creation in progress
Jan 29 02:20:36 hp-probook-445-g7 bluetoothd[89198]: src/service.c:change_state() 0x56252565f4c0: device 20:21:05:06:E8:6C profile a2dp-sink state changed: disconnected -> connecting (0)
Jan 29 02:20:41 hp-probook-445-g7 bluetoothd[89198]: profiles/audio/avdtp.c:avdtp_connect_cb() connect to 20:21:05:06:E8:6C: Host is down (112)
Jan 29 02:20:41 hp-probook-445-g7 bluetoothd[89198]: profiles/audio/avdtp.c:avdtp_ref() 0x562525674c90: ref=3
Jan 29 02:20:41 hp-probook-445-g7 bluetoothd[89198]: profiles/audio/avdtp.c:connection_lost() Disconnected from 20:21:05:06:E8:6C
Jan 29 02:20:41 hp-probook-445-g7 bluetoothd[89198]: profiles/audio/a2dp.c:discover_cb() version 0x0103 err 0x7fff76b02890
Jan 29 02:20:41 hp-probook-445-g7 bluetoothd[89198]: profiles/audio/avdtp.c:avdtp_unref() 0x562525674c90: ref=2
Jan 29 02:20:41 hp-probook-445-g7 bluetoothd[89198]: src/service.c:change_state() 0x56252565f4c0: device 20:21:05:06:E8:6C profile a2dp-sink state changed: connecting -> disconnected (-112)
Jan 29 02:20:41 hp-probook-445-g7 bluetoothd[89198]: src/device.c:device_profile_connected() a2dp-sink Host is down (112)
Jan 29 02:20:41 hp-probook-445-g7 bluetoothd[89198]: profiles/audio/a2dp.c:setup_unref() 0x562525685360: ref=0
Jan 29 02:20:41 hp-probook-445-g7 bluetoothd[89198]: profiles/audio/a2dp.c:setup_free() 0x562525685360
Jan 29 02:20:41 hp-probook-445-g7 bluetoothd[89198]: profiles/audio/avdtp.c:avdtp_unref() 0x562525674c90: ref=1
Jan 29 02:20:41 hp-probook-445-g7 bluetoothd[89198]: profiles/audio/sink.c:sink_set_state() State changed /org/bluez/hci0/dev_20_21_05_06_E8_6C: SINK_STATE_CONNECTING -> SINK_STATE_DISCONNECTED
Jan 29 02:20:41 hp-probook-445-g7 bluetoothd[89198]: profiles/audio/a2dp.c:channel_remove() chan 0x5625256895a0
Jan 29 02:20:41 hp-probook-445-g7 bluetoothd[89198]: profiles/audio/avdtp.c:avdtp_unref() 0x562525674c90: ref=0
Jan 29 02:20:41 hp-probook-445-g7 bluetoothd[89198]: profiles/audio/avdtp.c:avdtp_free() 0x562525674c90
Jan 29 02:20:41 hp-probook-445-g7 bluetoothd[89198]: src/adapter.c:connect_failed_callback() hci0 20:21:05:06:E8:6C status 4
Jan 29 02:20:41 hp-probook-445-g7 bluetoothd[89198]: plugins/policy.c:conn_fail_cb() status 4
Jan 29 02:20:41 hp-probook-445-g7 bluetoothd[89198]: plugins/policy.c:reconnect_set_timer() attempt 2/7 2 seconds
[... 73 lines cut ...]
[... cut ...]
Jan 29 01:26:52 hp-probook-445-g7 pipewire-media-session[1724]: spa.bluez5.native: RFCOMM receive command before SLC completed: AT+IPHONEACCEV=1,1,6
Jan 29 01:26:52 hp-probook-445-g7 systemd-logind[1256]: Watching system buttons on /dev/input/event21 (JLab GO Work (AVRCP))
Jan 29 01:26:52 hp-probook-445-g7 bluetoothd[1253]: src/profile.c:ext_io_disconnected() Unable to get io data for Hands-Free Voice gateway: getpeername: Transport endpoint is not connected>
Jan 29 01:26:56 hp-probook-445-g7 bluetoothd[1253]: /org/bluez/hci0/dev_20_21_05_06_E8_6C/fd4: fd(32) ready
```

Thanks a lot! Really appreciate the project :)

borine, 2022-01-30:
> `Jan 25 08:43:02 hp-probook-445-g7 pipewire-media-session[1848]: spa.bluez5.sink.sco: failed to write data: -110 (Connection timed out)`

That is a message from the pipwire bluetooth module. Bluez does not allow two programs to both register the same Profile - so you  cannot successfully run Pipewire with bluetooth enabled and BlueALSA at the same time.

Glaeqen, 2022-01-30:
I realized I probably don't run any code from `bluez-alsa` project. Audio & BT stack is so confusing. I replaced `pipewire-media-session` with `wireplumber` now, I wonder if it'll anyhow help

borine, 2022-01-30:
Not BlueALSA issue. closing.

closed 2022-01-30 by borine