arkq/bluez-alsa #530: Is something holding up 3.2.0?

gearhead, 2022-01-30:
No particular reason, but is there something holding this up? I use 3.1.0 and am investigating borine's dbus monitor and it requires 3.2.0 which is not available? I guess the git version reports as 3.2.0?

Arkq, 2022-01-30:
There are no particular reasons not to release a new version right now. However, I'd like to finish my work with LC3plus before making a new release. And the new release will be most likely 4.0.0 because of backward incompatible changes in command like options for bluealsa.

Alternatively, I will restore removed options... and the second release from now will be 4.0.0. I will have to think about that :D

borine, 2022-01-31:
> borine's dbus monitor ... requires 3.2.0

Apologies, that was a wild guess at the next version identifier; I did not intend to cause any confusion! I'll change the wording next time I do an update to that project.

Arkq, 2022-02-01:
> Apologies, that was a wild guess at the next version identifier;

There is nothing to apologize for. The next version number is unknown even for me. And after all, it's only an identifier :D The important thing is that at the current stage of master branch the new version will be released sooner than later.

Arkq, 2022-05-19:
I still have got one item which needs to be  merged into the master before releasing a new version. However, right now I'm waiting for some external tests. I hope that within week or two new version will be released.

closed 2022-06-03 by Arkq