arkq/bluez-alsa #531: không thể gửi lệnh rfcomm tới thiết bị Bluetooth

bmt1596, 2022-02-03:
Hi everyone, I am currently using Bluealsa to configure my device as an HFP using Ofono. In our ARM embedded device, echo cancellation has been performed, but when transmitting the processed signal, the sound quality is very poor. We found the cause, it's because the Phone also has a noise cancellation mode. see here: [](https://pub.dega-akustik.de/DAGA_2011/data/articles/000416.pdf)

To disable this mode on phone I need to send an AT command AT+NREC=0 to Phone and use RFCOMM. But when I try to send it I get below error:

```
root@urser: bluealsa-rfcomm /org/bluealsa/hci0/dev_C4_86_E9_37_7E_91
bluealsa-rfcomm: E: ../../../utils/rfcomm/rfcomm.c:174: Couldn't open RFCOMM: No such interface 'org.bluealsa.RFCOMM1' on object at path /org/bluealsa/hci0/dev_C4_86_E9_37_7E_91/rfcomm
```
Can you guys explain to me what causes it to not exist RFCOMM in DBUS.

Thank you.

Arkq, 2022-02-03:
If you're using oFono, then you will have to use it for rfcomm communication as well. The bluealsa-rfcomm will work only for built-in HFP in bluealsa.

bmt1596, 2022-02-03:
Thanks for your very quick reply. I tried to start BlueALSA with the following command. But after connecting to the phone I only get the list of PCMs for A2DP, also the SCO-PCMs for HFP are not displayed. SCO-PCM is only displayed when I enable it via Ofono. The PulseAudio profiles I turned off completely I think BlueALSA did its job very well. Do you know where the cause lies? I party using BlueZ Version 5.46 and BlueALSA V.3.1.

I have BlueALSA configured with the following config:
`../configure  --enable-ofono --enable-debug --disable-payloadcheck --enable-cli --enable-rfcomm`

Start BlueALSA with HFP:
 `bluealsa -i hci0 --syslog -p a2dp-sink -p hfp-hf &`
 and get Debug included:
```
bluealsa: D: ../../src/ofono.c:644: Checking oFono service presence
bluealsa: W: ../../src/main.c:366: Disabling native HFP support due to oFono service presence
bluealsa: D: ../../src/bluez.c:503: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/Sink/1
bluealsa: D: ../../src/bluez.c:419: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/Sink/1
bluealsa: D: ../../src/bluez.c:503: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/Sink/2
bluealsa: D: ../../src/bluez.c:419: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/Sink/2
bluealsa: D: ../../src/main.c:396: Acquiring D-Bus service name: org.bluealsa
bluealsa: D: ../../src/main.c:401: Starting main dispatching loop
```
 
 Start BlueALSA with Ofono:
  `bluealsa -i hci0 --syslog -p a2dp-sink -p hfp-ofono &`
```
bluealsa: D: ../../src/bluez.c:503: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/Sink/1
bluealsa: D: ../../src/bluez.c:419: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/Sink/1
bluealsa: D: ../../src/bluez.c:503: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/Sink/2
bluealsa: D: ../../src/bluez.c:419: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/Sink/2
bluealsa: D: ../../src/ofono.c:487: Registering oFono audio agent: /org/bluez/HFP/oFono
bluealsa: D: ../../src/main.c:396: Acquiring D-Bus service name: org.bluealsa
bluealsa: D: ../../src/main.c:401: Starting main dispatching loop
```

List of PCM with HFP Option:
```
root@user:~ # bluealsa-aplay -l
**** List of PLAYBACK Bluetooth Devices ****
**** List of CAPTURE Bluetooth Devices ****
hci0: F8:C3:CC:92:08:6A [Minh Tung], trusted phone
A2DP (SBC): S16_LE 2 channels 44100 Hz
```

List of PCM with Ofono Option:
```
root@user:~ # bluealsa-aplay -l
**** List of PLAYBACK Bluetooth Devices ****
hci0: F8:C3:CC:92:08:6A [Minh Tung], trusted phone
  SCO (CVSD): S16_LE 1 channel 8000 Hz
**** List of CAPTURE Bluetooth Devices ****
hci0: F8:C3:CC:92:08:6A [Minh Tung], trusted phone
  A2DP (SBC): S16_LE 2 channels 44100 Hz
  SCO (CVSD): S16_LE 1 channel 8000 Hz
```

borine, 2022-02-03:
Ofono always registers support for HFP profile with Bluez, so if BlueALSA also tries to register support there would be a conflict. For this reason BlueALSA *always* disables HFP support when Ofono is running at the same time, irrespective of the command-line options. So to use bluealsa-rfcomm it is necessary to prevent Ofono from running at the same time as bluealsa.

bmt1596, 2022-02-04:
thank you, it seems to work. I have reconfigured bluealsa with the following options and got pretty good results:
`../configure  --enable-debug --disable-payloadcheck --enable-cli --enable-rfcomm`

after booting bluealsa with  `bluealsa -i hci0 --syslog -p a2dp-sink -p hfp-hfp &` i get the following Debug:
```
bluealsa: D: ../../src/bluez.c:503: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/Sink/1
bluealsa: D: ../../src/bluez.c:419: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/Sink/1
bluealsa: D: ../../src/bluez.c:503: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/Sink/2
bluealsa: D: ../../src/bluez.c:419: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/Sink/2
bluealsa: D: ../../src/bluez.c:798: Creating hands-free profile object: /org/bluez/HFP/HandsFree
bluealsa: D: ../../src/bluez.c:732: Registering hands-free profile: /org/bluez/HFP/HandsFree
bluealsa: D: ../../src/main.c:396: Acquiring D-Bus service name: org.bluealsa
bluealsa: D: ../../src/main.c:401: Starting main dispatching loop

```
After successfully connecting the phone:
```
bluealsa: D: ../../src/dbus.c:63: Called: org.bluez.Profile1.NewConnection() on /org/bluez/HFP/HandsFree
bluealsa: D: ../../src/ba-rfcomm.c:1286: Created new RFCOMM thread [ba-rfcomm]: HFP Hands-Free (CVSD)
bluealsa: D: ../../src/sco.c:204: Created SCO dispatcher [ba-sco-dispatch]: hci0
bluealsa: D: ../../src/bluez.c:624: HFP Hands-Free (CVSD) configured for device 18:F6:43:60:22:FA
bluealsa: D: ../../src/ba-transport.c:670: Starting transport: HFP Hands-Free (CVSD)
bluealsa: D: ../../src/ba-transport.c:1055: Created new transport thread [ba-sco]: HFP Hands-Free (CVSD)
bluealsa: D: ../../src/sco.c:78: Starting SCO dispatcher loop: hci0
bluealsa: D: ../../src/ba-rfcomm.c:901: Starting RFCOMM loop: HFP Hands-Free (CVSD)
bluealsa: D: ../../src/ba-rfcomm.c:108: Sending AT message: SET: command:+BRSF, value:628
bluealsa: D: ../../src/sco.c:248: IO loop: START: sco_thread: HFP Hands-Free (CVSD)
bluealsa: D: ../../src/at.c:161: AT message: RESP: command:+BRSF, value:1519
bluealsa: D: ../../src/ba-rfcomm.c:127: RFCOMM: HFP Hands-Free (CVSD) state transition: 0 -> 1
bluealsa: D: ../../src/at.c:161: AT message: RESP: command:, value:OK
bluealsa: D: ../../src/ba-rfcomm.c:127: RFCOMM: HFP Hands-Free (CVSD) state transition: 1 -> 2
bluealsa: D: ../../src/ba-rfcomm.c:108: Sending AT message: TEST: command:+CIND, value:(null)
bluealsa: D: ../../src/at.c:161: AT message: RESP: command:+CIND, value: ("service",(0-1)),("call",(0-1)),("callsetup",(0-3)),("battchg",(0-5)),("signal",(0-5)),("roam",(0-1)),("callheld",(0-2))
bluealsa: D: ../../src/ba-rfcomm.c:127: RFCOMM: HFP Hands-Free (CVSD) state transition: 2 -> 4
bluealsa: D: ../../src/at.c:161: AT message: RESP: command:, value:OK
bluealsa: D: ../../src/ba-rfcomm.c:127: RFCOMM: HFP Hands-Free (CVSD) state transition: 4 -> 5
bluealsa: D: ../../src/ba-rfcomm.c:108: Sending AT message: GET: command:+CIND, value:(null)
bluealsa: D: ../../src/at.c:161: AT message: RESP: command:+CIND, value: 1,0,0,2,2,0,0
bluealsa: D: ../../src/ba-rfcomm.c:127: RFCOMM: HFP Hands-Free (CVSD) state transition: 5 -> 6
bluealsa: D: ../../src/at.c:161: AT message: RESP: command:, value:OK
bluealsa: D: ../../src/ba-rfcomm.c:127: RFCOMM: HFP Hands-Free (CVSD) state transition: 6 -> 7
bluealsa: D: ../../src/ba-rfcomm.c:108: Sending AT message: SET: command:+CMER, value:3,0,0,1,0
bluealsa: D: ../../src/at.c:161: AT message: RESP: command:, value:OK
bluealsa: D: ../../src/ba-rfcomm.c:127: RFCOMM: HFP Hands-Free (CVSD) state transition: 7 -> 8
bluealsa: D: ../../src/ba-rfcomm.c:127: RFCOMM: HFP Hands-Free (CVSD) state transition: 8 -> 9
bluealsa: D: ../../src/dbus.c:63: Called: org.bluez.MediaEndpoint1.SelectConfiguration() on /org/bluez/hci0/A2DP/SBC/Sink/1
bluealsa: D: ../../src/bluez.c:928: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
bluealsa: D: ../../src/dbus.c:63: Called: org.bluez.MediaEndpoint1.SetConfiguration() on /org/bluez/hci0/A2DP/SBC/Sink/1
bluealsa: D: ../../src/a2dp.c:724: Selected A2DP SBC bit-pool range: [2, 250]
bluealsa: D: ../../src/bluez.c:299: A2DP Sink (SBC) configured for device 18:F6:43:60:22:FA
bluealsa: D: ../../src/bluez.c:301: Configuration: channels: 2, sampling: 44100
bluealsa: D: ../../src/bluez.c:306: Send BLUEALSA-EVENT: 0 to BRAN
bluealsa: D: ../../src/bluez.c:503: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/Sink/3
bluealsa: D: ../../src/bluez.c:419: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/Sink/3
bluealsa: D: ../../src/bluez.c:928: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
bluealsa: D: ../../src/bluez.c:928: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
bluealsa: D: ../../src/bluez.c:928: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
bluealsa: D: ../../src/bluez.c:1122: Signal: org.freedesktop.DBus.Properties.PropertiesChanged(): org.bluez.MediaTransport1: Volume
bluealsa: D: ../../src/bluez.c:1143: Updating A2DP volume: 95 [-4.18 dB]
bluealsa: D: ../../src/bluez.c:928: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
bluealsa: D: ../../src/bluez.c:928: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
bluealsa: D: ../../src/bluez.c:928: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
bluealsa: D: ../../src/ba-rfcomm.c:1106: RFCOMM poll timeout
bluealsa: D: ../../src/ba-rfcomm.c:810: Updating microphone gain: 15
[... cut ...]
bluealsa: D: ../../src/at.c:161: AT message: RESP: command:+VGS, value:8
bluealsa: D: ../../src/at.c:161: AT message: RESP: command:+VGS, value:7
bluealsa: D: ../../src/at.c:161: AT message: RESP: command:+VGS, value:8
bluealsa: D: ../../src/at.c:161: AT message: RESP: command:+VGS, value:9
bluealsa: D: ../../src/at.c:161: AT message: RESP: command:+VGS, value:8
```
`dbus-send --system --type=method_call --print-reply --dest=org.bluealsa /org/bluealsa/hci0/dev_18_F6_43_60_22_FA/hfphf/sink org.bluealsa.PCM1  string:Volume variant:uint16:10`

borine, 2022-02-04:
`Volume` is not a method on the `org.bluealsa.PCM1` interface, it is a **property**. You need to use the `org.freedesktop.DBus.Properties` interface to change the volume. Also, only the high byte of the uint16 value is used for mono stream volume - so you have to left-shift your desired value by 8 bits. See the  BlueALSA D-Bus API is documentation https://github.com/Arkq/bluez-alsa/blob/816cf14aca9bcc1bcf4bf03b9827d0daee7bf97e/doc/bluealsa-api.txt for more information.

If you are using the `bash` shell, try
```
dbus-send --system --type=method_call --dest=org.bluealsa --print-reply /org/bluealsa/hci0/dev_18_F6_43_60_22_FA/hfphf/sink org.freedesktop.DBus.Properties.Set string:org.bluealsa.PCM1 string:Volume variant:uint16:$(( 10 << 8 ))
```

bmt1596, 2022-02-05:
thank you very much, everything is working fine so far, with bluealsa-cli i can also reset the volume when a call comes in. Thank you so much team for bluealsa. 👍

borine, 2022-02-05:
Just for completeness, I should also point out that `amixer` and `alsamixer` can also be used. For example:
```
amixer -D bluealsa:18:F6:43:60:22:FA sset SCO 10
```
or
```
alsamixer -D bluealsa:18:F6:43:60:22:FA
```

bmt1596, 2022-02-07:
Thank you for the suggestions, after I tried to change the volume via the commands you suggested. it looks like it had different notifications about changing the volume. But when I decrease to both 0 and increase to 15. The volume remains completely unchanged. 

Hier ist the debug when i control the volum with alsamixer:

bluealsa: D: ../../src/dbus.c:63: Called: org.bluealsa.Manager1.GetPCMs() on /org/bluealsa
```
bluealsa: D: ../../src/bluealsa-dbus.c:824: Setting volume: 4 [-19.06 dB] <> 0 [-96.00 dB]
bluealsa: D: ../../src/at.c:163: AT message: RESP: command:+VGS, value:5
bluealsa: D: ../../src/dbus.c:63: Called: org.bluealsa.Manager1.GetPCMs() on /org/bluealsa
bluealsa: D: ../../src/bluealsa-dbus.c:824: Setting volume: 5 [-15.84 dB] <> 0 [-96.00 dB]
bluealsa: D: ../../src/at.c:163: AT message: RESP: command:+VGS, value:6
bluealsa: D: ../../src/dbus.c:63: Called: org.bluealsa.Manager1.GetPCMs() on /org/bluealsa
bluealsa: D: ../../src/bluealsa-dbus.c:824: Setting volume: 6 [-13.21 dB] <> 0 [-96.00 dB]
bluealsa: D: ../../src/at.c:163: AT message: RESP: command:+VGS, value:7
bluealsa: D: ../../src/dbus.c:63: Called: org.bluealsa.Manager1.GetPCMs() on /org/bluealsa
bluealsa: D: ../../src/bluealsa-dbus.c:824: Setting volume: 7 [-10.99 dB] <> 0 [-96.00 dB]
bluealsa: D: ../../src/at.c:163: AT message: RESP: command:+VGS, value:8
bluealsa: D: ../../src/dbus.c:63: Called: org.bluealsa.Manager1.GetPCMs() on /org/bluealsa
bluealsa: D: ../../src/bluealsa-dbus.c:824: Setting volume: 8 [-9.06 dB] <> 0 [-96.00 dB]
bluealsa: D: ../../src/at.c:163: AT message: RESP: command:+VGS, value:9
bluealsa: D: ../../src/dbus.c:63: Called: org.bluealsa.Manager1.GetPCMs() on /org/bluealsa
bluealsa: D: ../../src/bluealsa-dbus.c:824: Setting volume: 9 [-7.36 dB] <> 0 [-96.00 dB]
bluealsa: D: ../../src/at.c:163: AT message: RESP: command:+VGS, value:10
bluealsa: D: ../../src/dbus.c:63: Called: org.bluealsa.Manager1.GetPCMs() on /org/bluealsa
bluealsa: D: ../../src/bluealsa-dbus.c:824: Setting volume: 10 [-5.84 dB] <> 0 [-96.00 dB]
bluealsa: D: ../../src/at.c:163: AT message: RESP: command:+VGS, value:11
bluealsa: D: ../../src/dbus.c:63: Called: org.bluealsa.Manager1.GetPCMs() on /org/bluealsa
bluealsa: D: ../../src/bluealsa-dbus.c:824: Setting volume: 11 [-4.47 dB] <> 0 [-96.00 dB]
bluealsa: D: ../../src/at.c:163: AT message: RESP: command:+VGS, value:12
bluealsa: D: ../../src/dbus.c:63: Called: org.bluealsa.Manager1.GetPCMs() on /org/bluealsa
bluealsa: D: ../../src/bluealsa-dbus.c:824: Setting volume: 12 [-3.21 dB] <> 0 [-96.00 dB]
bluealsa: D: ../../src/at.c:163: AT message: RESP: command:+VGS, value:13
bluealsa: D: ../../src/dbus.c:63: Called: org.bluealsa.Manager1.GetPCMs() on /org/bluealsa
bluealsa: D: ../../src/bluealsa-dbus.c:824: Setting volume: 13 [-2.06 dB] <> 0 [-96.00 dB]
bluealsa: D: ../../src/at.c:163: AT message: RESP: command:+VGS, value:14
bluealsa: D: ../../src/dbus.c:63: Called: org.bluealsa.Manager1.GetPCMs() on /org/bluealsa
bluealsa: D: ../../src/bluealsa-dbus.c:824: Setting volume: 14 [-0.99 dB] <> 0 [-96.00 dB]
bluealsa: D: ../../src/at.c:163: AT message: RESP: command:+VGS, value:15
bluealsa: D: ../../src/dbus.c:63: Called: org.bluealsa.Manager1.GetPCMs() on /org/bluealsa
bluealsa: D: ../../src/bluealsa-dbus.c:824: Setting volume: 15 [0.00 dB] <> 0 [-96.00 dB]
```
and hier is the debug when i record voice from phone:
`arecord -f S16_LE -D bluealsa:PROFILE=sco sco_iphone.wav`
```
D: ../../../src/asound/bluealsa-pcm.c:1053: Getting BlueALSA PCM: CAPTURE 00:00:00:00:00:00 sco
D: ../../../src/asound/bluealsa-pcm.c:921: /org/bluealsa/hci0/dev_18_F6_43_60_22_FA/hfphf/source: Setting constraints
Recording WAVE 'sco_iphone.wav' : Signed 16 bit Little Endian, Rate 8000 Hz, Mono
D: ../../../src/asound/bluealsa-pcm.c:443: /org/bluealsa/hci0/dev_18_F6_43_60_22_FA/hfphf/source: Initializing HW
D: ../../../src/asound/bluealsa-pcm.c:466: /org/bluealsa/hci0/dev_18_F6_43_60_22_FA/hfphf/source: FIFO buffer size: 32768 frames
D: ../../../src/asound/bluealsa-pcm.c:474: /org/bluealsa/hci0/dev_18_F6_43_60_22_FA/hfphf/source: Selected HW buffer: 4 periods x 2000 bytes == 8000 bytes
D: ../../../src/asound/bluealsa-pcm.c:489: /org/bluealsa/hci0/dev_18_F6_43_60_22_FA/hfphf/source: Initializing SW
D: ../../../src/asound/bluealsa-pcm.c:489: /org/bluealsa/hci0/dev_18_F6_43_60_22_FA/hfphf/source: Initializing SW
D: ../../../src/asound/bluealsa-pcm.c:528: /org/bluealsa/hci0/dev_18_F6_43_60_22_FA/hfphf/source: Prepared
D: ../../../src/asound/bluealsa-pcm.c:489: /org/bluealsa/hci0/dev_18_F6_43_60_22_FA/hfphf/source: Initializing SW
D: ../../../src/asound/bluealsa-pcm.c:345: /org/bluealsa/hci0/dev_18_F6_43_60_22_FA/hfphf/source: Starting
D: ../../../src/asound/bluealsa-pcm.c:212: /org/bluealsa/hci0/dev_18_F6_43_60_22_FA/hfphf/source: Starting IO loop: 7
^CAborted by signal Interrupt...
arecord: pcm_read:2122: read error: Interrupted system call
D: ../../../src/asound/bluealsa-pcm.c:381: /org/bluealsa/hci0/dev_18_F6_43_60_22_FA/hfphf/source: Stopping
D: ../../../src/asound/bluealsa-pcm.c:147: /org/bluealsa/hci0/dev_18_F6_43_60_22_FA/hfphf/source: IO thread cleanup
D: ../../../src/asound/bluealsa-pcm.c:481: /org/bluealsa/hci0/dev_18_F6_43_60_22_FA/hfphf/source: Freeing HW
D: ../../../src/asound/bluealsa-pcm.c:431: /org/bluealsa/hci0/dev_18_F6_43_60_22_FA/hfphf/source: Closing
FILE=sco sco_iphone.wavecord -f S16_LE -D bluealsa:SRV=org.bluealsa,DEV=18:F6:43:60:22:FA,PROF
[... cut ...]
D: ../../../src/asound/bluealsa-pcm.c:528: /org/bluealsa/hci0/dev_18_F6_43_60_22_FA/hfphf/source: Prepared
D: ../../../src/asound/bluealsa-pcm.c:489: /org/bluealsa/hci0/dev_18_F6_43_60_22_FA/hfphf/source: Initializing SW
D: ../../../src/asound/bluealsa-pcm.c:345: /org/bluealsa/hci0/dev_18_F6_43_60_22_FA/hfphf/source: Starting
D: ../../../src/asound/bluealsa-pcm.c:212: /org/bluealsa/hci0/dev_18_F6_43_60_22_FA/hfphf/source: Starting IO loop: 7
```
Looks like it's having some problem with volume control.

borine, 2022-02-07:
> Is it because I turned off all pulse audio?

No - do not use bluealsa and pulseaudio at the same time; only 1 application can register as provider of HFP. 

Please can you explain what you expect to happen when the HFP volume is changed? I suspect that it is working, but perhaps not quite as you wish. For example `arecord` never takes any notice of volume change requests, so you would not see any effect of volume change with that application. Do you always record the incoming audio, or do you perhaps use some other application to send it to a local sound card?

bmt1596, 2022-02-07:
Thanks you for reply!! 

OK let me explain a little bit more clearly. I start my device with HFP-HF and connect a phone to it. In this case, the device using `bluealsa ` is (HF) and the phone is (AG). I use Echo-Effect for calls at 48000 kHz and pass a virtuelle Soundcard to transmit the signal back to the Phone.

I wish that I could control the volume of the HF device when a call comes in. For example, you are using your phone to receive a call, but when the speakerphone is loud and you want to turn the volume down.

Is the problem in virtuell soundcard?

borine, 2022-02-07:
Ok, that sounds rather more complicated than anything I've done myself, but hopefully I have understood. I think you need to decide where is the best place to *operate* the volume (perhaps the phone volume switch ?) and where is the best place to *implement* it (perhaps in the virtual soundcard, or perhaps in `bluealsa` ?).

If you wish the soundcard to apply the volume changes then you will need an application that can listen for D-Bus "PropertiesChanged" signals from the org.bluealsa.PCM1 interface to know when a volume change request has been received by `bluealsa`. (`bluealsa-aplay` is one example, or you could use the bluealsa CTL device within an ALSA application)

If you wish `bluealsa` to actually apply the volume changes to the audio signal, you can do that by enabling the "SoftVolume" property. There are several ways to enable bluealsa SoftVolume for a PCM, these examples apply it to the playback PCM, but you could equally use the capture PCM instead:

    - use `bluealsa-cli`
        `bluealsa-cli /org/bluealsa/hci0/dev_18_F6_43_60_22_FA/hfphf/sink y`
        (or use dbus-send to change the `org.bluealsa.PCM1.SoftVolume` property)

    - set it when the ALSA PCM device is opened using the SOFTVOL parameter
        `aplay -D bluealsa:DEV=18:F6:43:60:22:FA,PROFILE=sco,SOFTVOL=y ...`

bmt1596, 2022-02-08:
thank you so much for the hint with `softvolume ` it really worked. i installed `soft-volume` and adjusted the `volume `via `bluealsa-cli` command and indeed it worked. Your suggestion about listening to Dbus to catch Events about the volume change is a great idea. Tks for BlueALSA ! 👍 💯 🥇