arkq/bluez-alsa #532: Integration of echo-canceling

dr-ni, 2022-02-07:
Do you think you can integrate an echo-canceling option for hfp ofono?

There is already a very well working speexdsp example for mono raw audio files:
https://github.com/xiph/speexdsp/blob/master/libspeexdsp/testecho.c

```./testecho ~/local_mic_recording.raw ~/incomming_bluealsa.raw ~/outgoing_to_bluealsa.raw```
is working very good, incomming_bluealsa is completely removed from the recording mixture.
However, smaller stream frames for realtime processing will not remove all - but should be possible.

```
#ifdef HAVE_CONFIG_H
#include "config.h"
#endif

#include "speex/speex_echo.h"
#include "speex/speex_preprocess.h"
#include <stdio.h>
#include <stdlib.h>
#include <sys/types.h>
#include <sys/stat.h>
#include <fcntl.h>

#define NN 128
#define TAIL 1024

int main(int argc, char **argv)
{
   FILE *echo_fd, *ref_fd, *e_fd;
   short echo_buf[NN], ref_buf[NN], e_buf[NN];
   SpeexEchoState *st;
   SpeexPreprocessState *den;
   int sampleRate = 8000;

   if (argc != 4)
   {
      fprintf(stderr, "testecho mic_signal.sw speaker_signal.sw output.sw\n");
      exit(1);
   }
   echo_fd = fopen(argv[2], "rb");
   ref_fd  = fopen(argv[1],  "rb");
   e_fd    = fopen(argv[3], "wb");

   st = speex_echo_state_init(NN, TAIL);
   den = speex_preprocess_state_init(NN, sampleRate);
   speex_echo_ctl(st, SPEEX_ECHO_SET_SAMPLING_RATE, &sampleRate);
   speex_preprocess_ctl(den, SPEEX_PREPROCESS_SET_ECHO_STATE, st);

   while (!feof(ref_fd) && !feof(echo_fd))
   {
      fread(ref_buf, sizeof(short), NN, ref_fd);
      fread(echo_buf, sizeof(short), NN, echo_fd);
      speex_echo_cancellation(st, ref_buf, echo_buf, e_buf);
      speex_preprocess_run(den, e_buf);
      fwrite(e_buf, sizeof(short), NN, e_fd);
   }
   speex_echo_state_destroy(st);
   speex_preprocess_state_destroy(den);
   fclose(e_fd);
   fclose(echo_fd);
   fclose(ref_fd);
   return 0;
}
```

Arkq, 2022-02-08:
To be honest I'm not sure whether AEC should be a part of bluealsa itself. The case is that the BlueALSA service should be treated as a sound card with many PCM devices (playback and capture). It is not a telephony application. One can build a telephony-like application on top of bluealsa - using it as an audio backend for mic and speaker (in the same way a standard audio card would be used). Do you know of any sound card which has build-in echo cancelation? If there are sound cards like that, maybe we can try to mimic such solution in BlueALSA service. But for now, I'm not sure how it should work from the user point of view - BlueALSA service might have many capture PCMs and many playback PCMs...

Arkq, 2022-02-09:
> I think echo cancellation is a function that is applied to the stream from the local microphone, not the remote bluetooth one. 

It all depends how BlueALSA is used - what kind of setup we are talking about. Because remote Bluetooth device might be a local device, e.g. BT speaker. So, echo cancelation will be done on a "local" BT device, and audio from the BT speaker mic will be transferred over VoIP. Anyway, I do agree that it's not a BlueALSA job to perform echo cancelation, but rather an application which is using PCM audio device.

dr-ni, 2022-02-09:
Hi again, and sorry for my late answer.
Of course `AEC` is a local sound problem. And you are totally right that it is not directly related to `bluez-alsa`. However, I have now managed to hack in a little prove of concept tool - and it hasn't been showing so much effort to build. So may be an option for you to include such a small tool?

```
/*
** A straight forward AEC proove of concept based on speexdsp
** MIT License Copyright (c) 2022 U. Niethammer
*/
#ifdef HAVE_CONFIG_H
#include "config.h"
#endif

#include "speex/speex_echo.h"
#include "speex/speex_preprocess.h"
#include <fcntl.h>
#include <stdio.h>
#include <stdlib.h>
#include <string.h>
#include <sys/stat.h>
#include <sys/types.h>

#define MONO_FRAME_SIZE 256
#define STEREO_FRAME_SIZE 512
#define FILTER_SIZE 1024

int main(int argc, char **argv) {

  short input[STEREO_FRAME_SIZE];
  short output[MONO_FRAME_SIZE];
  short rec[MONO_FRAME_SIZE];
  short ref[MONO_FRAME_SIZE];

  size_t input_size = 0;
  size_t output_size = 0;
  int counter = 0;

  SpeexEchoState *st;
  SpeexPreprocessState *den;
  int sampleRate = 16000;
  
  // init speexdsp
  st = speex_echo_state_init(MONO_FRAME_SIZE, FILTER_SIZE);
  den = speex_preprocess_state_init(MONO_FRAME_SIZE, sampleRate);
  speex_echo_ctl(st, SPEEX_ECHO_SET_SAMPLING_RATE, &sampleRate);
  speex_preprocess_ctl(den, SPEEX_PREPROCESS_SET_ECHO_STATE, st);

  // read the first frame...
  memset(input, 0, sizeof(input));
  memset(output, 0, sizeof(output));
  input_size = fread(input, 1, sizeof(input), stdin);
  do {
    // extract interleaved stereo frame S16_LE to separated mono frames
    // rec and ref
    for (counter = 0; counter < STEREO_FRAME_SIZE; counter += 2) {
      rec[counter / 2] = input[counter];
      ref[counter / 2] = input[counter + 1];
    }

    // do the echo-cancelling
    speex_echo_cancellation(st, rec, ref, output);
    speex_preprocess_run(den, output);

    //output the processed frame
    output_size = fwrite(output, 1, sizeof(output), stdout);
    if (!output_size) {
      fprintf(stderr, "\nError writing frame\n");
      break;
      // no data writable
    }
    if (input_size < sizeof(input)) {
      break;
      // no data readable
    }
    // Put the next frame in buffer
    memset(input, 0, sizeof(input));
    input_size = fread(input, 1, sizeof(input), stdin);
  } while (input_size == sizeof(input));

  // all done
  speex_echo_state_destroy(st);
  speex_preprocess_state_destroy(den);
  return 0;
}
```

Such tool canbe easily be used for files:

`aec_cmdline_chain_tool < infile.raw > outfile.raw`

or in combination with arecord and aplay:

`arecord -f s16_le -D bluealsa | aplay -D mloopplay -f S16_LE -r 16000`

and

`arecord -D aec_internal -f S16_LE -c 2 -r 16000 | aec_cmdline_chain_tool | aplay -D bluealsa`

The tool is reading a stereo track with the local recording on left and the parasitic reference signal (here bluealsa remote) on the right.
This stereo track is beeing supplied from a Loopback device as described in:

https://github.com/SaneBow/alsa-aec.git

dr-ni, 2022-02-11:
Update:
Now for 8000 Hz mono_mloopplay and stereo_mloopplay support (prior 16 kHz stereo mloopplay only, see my new 50-aec.conf)

So

`arecord -f S16_LE -D bluealsa | aplay -D mono_mloopplay -f S16_LE -r 8000`

and

`arecord -D aec_internal -f S16_LE -c 2 -r 8000 | aec_cmdline | aplay -D bluealsa`

should now be working with ofono