arkq/bluez-alsa #533: How to add msbc codec to hfp profile

bmt1596, 2022-02-08:
Hi everybody. First I want to thank for bluealsa. Currently I am using Bluealsa to support device (HF) for A2DP-Sink and HFP and using phone (AG) to connect and transmit Audio signal to  embedded device.

My embedded device uses BlueZ version 5.63. Unexpectedly after installing Bluealsa on the device it seems to have worked quite well.

I installed for BlueALSA including the following options:

`../configure --enable-debug --disable-payloadcheck --enable-cli --enable-rfcomm`

It works when I start bluealsa with the command:
`bluealsa -i hci0 --syslog -p a2dp-sink -p hfp-hfp &`

The displayed result of bluealsa-aplay-l is:

```
root@user:~ # bluealsa-aplay -l
**** List of PLAYBACK Bluetooth Devices ****
hci0: F8:C3:CC:92:08:6A [Minh Tung], trusted phone
  SCO (CVSD): S16_LE 1 channel 8000 Hz
**** List of CAPTURE Bluetooth Devices ****
hci0: F8:C3:CC:92:08:6A [Minh Tung], trusted phone
  A2DP (SBC): S16_LE 2 channels 44100 Hz
  SCO (CVSD): S16_LE 1 channel 8000 Hz
```

So it can be said that BlueALSA has done its job well, but is there any other way to upgrade the sampling rate for voice channels? I have consulted in Bluealsa's issues and found that there are some cases where WBS is enabled via `--enable-msbc` but when I try it with my Linux Kernel version, it shows the following results:

```
root@user:~ # bluealsa-aplay -l
**** List of PLAYBACK Bluetooth Devices ****
hci0: F8:C3:CC:92:08:6A [Minh Tung], trusted phone
  SCO (<null>): S16_LE 1 channel 0 Hz
**** List of CAPTURE Bluetooth Devices ****
hci0: F8:C3:CC:92:08:6A [Minh Tung], trusted phone
  SCO (<null>): S16_LE 1 channel 0 Hz
  A2DP (SBC): S16_LE 2 channels 44100 Hz
```

Looks like the sampling rate for SCO has not been configured. Do I need to upgrade  the Linux-kernel version? and what is the minimum linux-kernel version for Bluealsa to update the mSBC-Codec. I am using petalinux 2018.3 and linux-kernel version is 4.14.0 but it seems that it still doesn't support for WBS(eSBC).

Hope to receive your reply as soon. Have a nice day!!

Arkq, 2022-02-09:
> Looks like the sampling rate for SCO has not been configured. 

When mSBC is enabled, the sampling rate is not know prior to codec selection procedure. Codec selection might be performed just before first audio connection. So, in other words, you might see sampling rate as 0 Hz, which will change to 8000 or 16000 after first audio connection.

In most cases codec selection is performed by the AG, so in case of Android HFP codec is selected before first audio transfer. When BlueALSA acts as an AG, codec selection is performed just after connection establishment, so PCM client will know sampling rate before starting audio. I know that this might be inconvenient for BlueALSA in HF mode, however, I'm not sure how to solve this issue. I might set codec to CVSD as a sane default, and it will be updated by the AG before audio transfer. However, this will most likely mean that prior to first audio transfer the PCM configuration will be incorrect (which right now is indicated by 0 Hz sampling rate).

bmt1596, 2022-02-09:
thanks for your suggestion, indeed it worked as soon as I opened and configured 2 PCMs for SCO. The displayed result of `bluealsa-aplay-l `is:

```
bluealsa-aplay -l
**** List of PLAYBACK Bluetooth Devices ****
hci0: 18:F6:43:60:22:FA [18-F6-43-60-22-FA], trusted phone
  SCO (mSBC): S16_LE 1 channel 16000 Hz
**** List of CAPTURE Bluetooth Devices ****
hci0: 18:F6:43:60:22:FA [18-F6-43-60-22-FA], trusted phone
  SCO (mSBC): S16_LE 1 channel 16000 Hz
  A2DP (SBC): S16_LE 2 channels 44100 Hz
```

when i run the `arecord `command to record the voice signal, everything works fine, but when i send the recorded voice signal from the microphone to the phone with `aplay`, it shows the following error, that it's mSBC packets is missing. Do you know where the cause lies?

```
bluealsa: W: ../../src/sco.c:285: Couldn't decode mSBC: No such process
bluealsa: W: ../../src/codec-msbc.c:152: Missing mSBC packet: 2 != 1
bluealsa: W: ../../src/codec-msbc.c:152: Missing mSBC packet: 2 != 0
bluealsa: W: ../../src/codec-msbc.c:152: Missing mSBC packet: 1 != 0
bluealsa: W: ../../src/codec-msbc.c:152: Missing mSBC packet: 2 != 1
bluealsa: W: ../../src/codec-msbc.c:152: Missing mSBC packet: 3 != 2
bluealsa: W: ../../src/sco.c:285: Couldn't decode mSBC: No such process
bluealsa: W: ../../src/codec-msbc.c:152: Missing mSBC packet: 0 != 3
bluealsa: W: ../../src/codec-msbc.c:152: Missing mSBC packet: 2 != 1
bluealsa: W: ../../src/codec-msbc.c:152: Missing mSBC packet: 2 != 1
bluealsa: W: ../../src/codec-msbc.c:152: Missing mSBC packet: 0 != 3
bluealsa: W: ../../src/codec-msbc.c:152: Missing mSBC packet: 2 != 1
bluealsa: D: ../../src/at.c:166: AT message: RESP: command:+CIEV, value: 5,3
bluealsa: W: ../../src/codec-msbc.c:152: Missing mSBC packet: 2 != 1
bluealsa: W: ../../src/sco.c:285: Couldn't decode mSBC: No such process
bluealsa: W: ../../src/codec-msbc.c:152: Missing mSBC packet: 3 != 2
bluealsa: W: ../../src/sco.c:285: Couldn't decode mSBC: No such file or directory
bluealsa: W: ../../src/codec-msbc.c:152: Missing mSBC packet: 2 != 1
bluealsa: W: ../../src/codec-msbc.c:152: Missing mSBC packet: 0 != 3
bluealsa: D: ../../src/at.c:166: AT message: RESP: command:+CIEV, value: 5,4
bluealsa: W: ../../src/codec-msbc.c:152: Missing mSBC packet: 3 != 2
bluealsa: W: ../../src/codec-msbc.c:152: Missing mSBC packet: 2 != 1
bluealsa: W: ../../src/codec-msbc.c:152: Missing mSBC packet: 3 != 2
bluealsa: W: ../../src/codec-msbc.c:152: Missing mSBC packet: 2 != 1
bluealsa: W: ../../src/codec-msbc.c:152: Missing mSBC packet: 0 != 3
bluealsa: W: ../../src/codec-msbc.c:152: Missing mSBC packet: 2 != 1
bluealsa: W: ../../src/codec-msbc.c:152: Missing mSBC packet: 2 != 1
bluealsa: W: ../../src/codec-msbc.c:152: Missing mSBC packet: 0 != 3
bluealsa: W: ../../src/codec-msbc.c:152: Missing mSBC packet: 1 != 0
bluealsa: W: ../../src/codec-msbc.c:152: Missing mSBC packet: 3 != 2
bluealsa: W: ../../src/codec-msbc.c:152: Missing mSBC packet: 1 != 0
bluealsa: W: ../../src/codec-msbc.c:152: Missing mSBC packet: 3 != 2
bluealsa: W: ../../src/codec-msbc.c:152: Missing mSBC packet: 1 != 0
bluealsa: W: ../../src/sco.c:285: Couldn't decode mSBC: No such file or directory
```

borine, 2022-02-09:
> Playing WAVE 'voicemail-3.wav' : Signed 16 bit Little Endian, Rate 8000 Hz, Mono

Is `voicemail-3.wav` a recording of the mSBC audio from the phone? If so an mSBC recording should be 16000Hz, so I think your `arecord` command line needs to include `-r 16000`. That may make some improvement, but I'm not sure that it is the complete answer. Give it a try and report back the result.

bmt1596, 2022-02-10:
Tks @borine for reply!!! Looks like things are getting worse. Right now the phonedown recording is no longer possible. I tried to try recording phonedown again with the following command for 16 kHz:

```
 arecord -d 10 -f S16_LE -r 16000 -D bluealsa:PROFILE=sco Record_PhoneCall.wav
D: ../../../src/asound/bluealsa-pcm.c:1053: Getting BlueALSA PCM: CAPTURE 00:00:00:00:00:00 sco
bluealsa: D: ../../src/dbus.c:63: Called: org.bluealsa.Manager1.GetPCMs() on /org/bluealsa
D: ../../../src/asound/bluealsa-pcm.c:921: /org/bluealsa/hci0/dev_18_F6_43_60_22_FA/hfphf/source: Setting constraints
Recording WAVE 'Record_PhoneCall.wav' : Signed 16 bit Little Endian, Rate 16000 Hz, Mono
D: ../../../src/asound/bluealsa-pcm.c:443: /org/bluealsa/hci0/dev_18_F6_43_60_22_FA/hfphf/source: Initializing HW
bluealsa: D: ../../src/dbus.c:63: Called: org.bluealsa.PCM1.Open() on /org/bluealsa/hci0/dev_18_F6_43_60_22_FA/hfphf/source
D: ../../../src/asound/bluealsa-pcm.c:466: /org/bluealsa/hci0/dev_18_F6_43_60_22_FA/hfphf/source: FIFO buffer size: 32768 frames
D: ../../../src/asound/bluealsa-pcm.c:474: /org/bluealsa/hci0/dev_18_F6_43_60_22_FA/hfphf/source: Selected HW buffer: 4 periods x 4000 bytes == 16000 bytes
D: ../../../src/asound/bluealsa-pcm.c:489: /org/bluealsa/hci0/dev_18_F6_43_60_22_FA/hfphf/source: Initializing SW
D: ../../../src/asound/bluealsa-pcm.c:489: /org/bluealsa/hci0/dev_18_F6_43_60_22_FA/hfphf/source: Initializing SW
D: ../../../src/asound/bluealsa-pcm.c:528: /org/bluealsa/hci0/dev_18_F6_43_60_22_FA/hfphf/source: Prepared
D: ../../../src/asound/bluealsa-pcm.c:489: /org/bluealsa/hci0/dev_18_F6_43_60_22_FA/hfphf/source: Initializing SW
D: ../../../src/asound/bluealsa-pcm.c:345: /org/bluealsa/hci0/dev_18_F6_43_60_22_FA/hfphf/source: Starting
bluealsa: D: ../../src/ba-transport.c:779: PCM resumed: 23
D: ../../../src/asound/bluealsa-pcm.c:212: /org/bluealsa/hci0/dev_18_F6_43_60_22_FA/hfphf/source: Starting IO loop: 7
bluealsa: W: ../../src/codec-msbc.c:152: Missing mSBC packet: 2 != 1
bluealsa: W: ../../src/codec-msbc.c:152: Missing mSBC packet: 0 != 3
bluealsa: W: ../../src/codec-msbc.c:152: Missing mSBC packet: 0 != 3
bluealsa: W: ../../src/codec-msbc.c:152: Missing mSBC packet: 2 != 1
bluealsa: W: ../../src/codec-msbc.c:152: Missing mSBC packet: 1 != 0
D: ../../../src/asound/bluealsa-pcm.c:381: /org/bluealsa/hci0/dev_18_F6_43_60_22_FA/hfphf/source: Stopping
D: ../../../src/asound/bluealsa-pcm.c:147: /org/bluealsa/hci0/dev_18_F6_43_60_22_FA/hfphf/source: IO thread cleanup
D: ../../../src/asound/bluealsa-pcm.c:481: /org/bluealsa/hci0/dev_18_F6_43_60_22_FA/hfphf/source: Freeing HW
D: ../../../src/asound/bluealsa-pcm.c:431: /org/bluealsa/hci0/dev_18_F6_43_60_22_FA/hfphf/source: Closing
bluealsa: D: ../../src/ba-transport.c:1025: Closing PCM: 23
```

an try to record with 8 kHz for 10 seconds:
```
arecord -d 10 -f S16_LE -r 8000 -D bluealsa:PROFILE=sco Record_PhoneCall.wav
D: ../../../src/asound/bluealsa-pcm.c:1053: Getting BlueALSA PCM: CAPTURE 00:00:00:00:00:00 sco
bluealsa: D: ../../src/dbus.c:63: Called: org.bluealsa.Manager1.GetPCMs() on /org/bluealsa
D: ../../../src/asound/bluealsa-pcm.c:921: /org/bluealsa/hci0/dev_18_F6_43_60_22_FA/hfphf/source: Setting constraints
Recording WAVE 'Record_PhoneCall.wav' : Signed 16 bit Little Endian, Rate 8000 Hz, Mono
D: ../../../src/asound/bluealsa-pcm.c:443: /org/bluealsa/hci0/dev_18_F6_43_60_22_FA/hfphf/source: Initializing HW
bluealsa: D: ../../src/dbus.c:63: Called: org.bluealsa.PCM1.Open() on /org/bluealsa/hci0/dev_18_F6_43_60_22_FA/hfphf/source
D: ../../../src/asound/bluealsa-pcm.c:466: /org/bluealsa/hci0/dev_18_F6_43_60_22_FA/hfphf/source: FIFO buffer size: 32768 frames
D: ../../../src/asound/bluealsa-pcm.c:474: /org/bluealsa/hci0/dev_18_F6_43_60_22_FA/hfphf/source: Selected HW buffer: 4 periods x 4000 bytes == 16000 bytes
D: ../../../src/asound/bluealsa-pcm.c:489: /org/bluealsa/hci0/dev_18_F6_43_60_22_FA/hfphf/source: Initializing SW
D: ../../../src/asound/bluealsa-pcm.c:489: /org/bluealsa/hci0/dev_18_F6_43_60_22_FA/hfphf/source: Initializing SW
D: ../../../src/asound/bluealsa-pcm.c:489: /org/bluealsa/hci0/dev_18_F6_43_60_22_FA/hfphf/source: Initializing SW
D: ../../../src/asound/bluealsa-pcm.c:528: /org/bluealsa/hci0/dev_18_F6_43_60_22_FA/hfphf/source: Prepared
D: ../../../src/asound/bluealsa-pcm.c:489: /org/bluealsa/hci0/dev_18_F6_43_60_22_FA/hfphf/source: Initializing SW
D: ../../../src/asound/bluealsa-pcm.c:345: /org/bluealsa/hci0/dev_18_F6_43_60_22_FA/hfphf/source: Starting
bluealsa: D: ../../src/ba-transport.c:779: PCM resumed: 23
D: ../../../src/asound/bluealsa-pcm.c:212: /org/bluealsa/hci0/dev_18_F6_43_60_22_FA/hfphf/source: Starting IO loop: 7
bluealsa: W: ../../src/sco.c:285: Couldn't decode mSBC: No such file or directory
bluealsa: W: ../../src/codec-msbc.c:152: Missing mSBC packet: 0 != 3
bluealsa: W: ../../src/codec-msbc.c:152: Missing mSBC packet: 2 != 1
bluealsa: W: ../../src/codec-msbc.c:152: Missing mSBC packet: 1 != 0
bluealsa: W: ../../src/sco.c:285: Couldn't decode mSBC: No such process
bluealsa: W: ../../src/sco.c:285: Couldn't decode mSBC: No such process
D: ../../../src/asound/bluealsa-pcm.c:381: /org/bluealsa/hci0/dev_18_F6_43_60_22_FA/hfphf/source: Stopping
D: ../../../src/asound/bluealsa-pcm.c:147: /org/bluealsa/hci0/dev_18_F6_43_60_22_FA/hfphf/source: IO thread cleanup
D: ../../../src/asound/bluealsa-pcm.c:481: /org/bluealsa/hci0/dev_18_F6_43_60_22_FA/hfphf/source: Freeing HW
bluealsa: D: ../../src/ba-transport.c:1025: Closing PCM: 23
[... cut ...]
bluealsa: W: ../../src/sco.c:285: Couldn't decode mSBC: No such file or directory
bluealsa: W: ../../src/codec-msbc.c:152: Missing mSBC packet: 2 != 1
bluealsa: W: ../../src/codec-msbc.c:152: Missing mSBC packet: 2 != 1
bluealsa: W: ../../src/codec-msbc.c:152: Missing mSBC packet: 2 != 1
bluealsa: W: ../../src/codec-msbc.c:152: Missing mSBC packet: 2 != 1
bluealsa: W: ../../src/codec-msbc.c:152: Missing mSBC packet: 2 != 1
```

whether my Bluealsa version has some error for mSBC decoding, i use bluealsa v3.1.0?

bmt1596, 2022-02-10:
Thank you very much. Maybe I can also figure out where the cause lies and may need to find a patch for `btusb`. But when I switch to using UART will things be better for WBS ?? Although the transfer speed is slower but maybe it will have less WBS support errors. I will also find out where the problem lies in the usb-driver and if found I will update you so that bluealsa can improve the SCO transmission with WBS for this Bluetooth Modul of  Cypress.

borine, 2022-02-10:
There is an important change to `btusb` in Linux kernel 5.15 regarding mSBC support, so you may like to try that kernel version (or a more recent one) first (see https://github.com/torvalds/linux/commit/55981d3541812234e687062926ff199c83f79a39).

UART should work fine, provided the serial device config is correct. Its been a long time since a last used mSBC with a UART adapter though. The SCO issues that I am aware of all relate specifically to USB.

borine, 2022-02-11:
> PCM returns 23 bytes
> aplay writes 22 bytes

No. those numbers (22 and 23) are the file descriptors of the SCO sockets, and have nothing to do with the number of bytes written. To understand the required size of each write() you need to know how USB isochronous endpoints work, and their relationship to the USB interface bAlternateSetting value. This is discussed in https://github.com/Arkq/bluez-alsa/issues/400#issuecomment-736934135 - the socket layer code will add a USB header of 3 bytes, so you need to subtract 3 from the URB sizes calculated there to arrive at the MTU from the application's perspective. So in the end we arrive at:
Alt-1 : 24 bytes
Alt-2 : 48 bytes
Alt-3 : 72 bytes (i think, but not sure about that one)
Alt-6 : 60 bytes

The problem for the application is that it does not know which Alternate setting is in use, and the socket reports wrong MTU values (e.g. in your case 64, which is wrong for any Alt setting).

Bluealsa simply chooses to write 24 bytes in all cases, since this has proved to be a "safe" value that works in most cases (to date the majority of adapters seem to use Alt-1). Writing too many bytes can cause lockups in the system that require a hard reset, so it is not advisable to experiment with random values.

bmt1596, 2022-02-11:
Thank you very much for the suggestions. I will try patching `btusb.c ` in bluetooth-drivers and update the default alt value to 1 and will update the results again.

bmt1596, 2022-02-11:
Very helpful! I was really surprised when I changed Alt to 1. It work now for me to play PhoneUp with `aplay`. 

```
bluealsa: W: ../../src/codec-msbc.c:152: Missing mSBC packet: 2 != 1
bluealsa: W: ../../src/codec-msbc.c:152: Missing mSBC packet: 2 != 1
```
but with `arecord `it seems to be a bit slow to read which leads to some loss of SCO pakets, is there any way to fix it via `buffer_size` or `buffer_time` via Bluealsa.

borine, 2022-02-11:
If the client (`arecord`) runs too slowly, then bluealsa's internal buffers can become full and then BT packets will be dropped.

This can happen when the client is writing to another device with its own clock, especially if that involves resampling to a different rate, because the resampler may introduce rounding error. However, if `arecord` is writing to a file there should be no problem (unless the file system is very, *very*, slow).

The latest bluealsa source uses a revised SCO design, so that may help. But if the problem is in the USB subsystem, then there is nothing that bluealsa can do.

Rather than forcing the driver to use Alt-1, if you can find out which Alt setting it uses natively then you could instead change the packet size used by bluealsa to the correct value for that setting. That might help get better throughput.

bmt1596, 2022-02-13:
Thank you very much for Bluealsa, up to this point can say Bluealsa works both ways quite well for mSBC. It's true that the buffer is a bit small, but since the audio signal is recorded directly to the file, it shouldn't have too much impact on the quality.

bmt1596, 2022-02-13:
thank you very much, up to this point can say Bluealsa works both ways quite well for mSBC. It's true that the buffer is a bit small, but since the audio signal is recorded directly to the file, it shouldn't have too much impact on the quality. 

```
bluealsa-aplay -l
**** List of PLAYBACK Bluetooth Devices ****
hci0: 18:F6:43:60:22:FA [18-F6-43-60-22-FA], trusted phone
  SCO (mSBC): S16_LE 1 channel 16000 Hz
**** List of CAPTURE Bluetooth Devices ****
hci0: 18:F6:43:60:22:FA [18-F6-43-60-22-FA], trusted phone
  SCO (mSBC): S16_LE 1 channel 16000 Hz
  A2DP (SBC): S16_LE 2 channels 44100 Hz
```

is there any way to make bluealsa default setting sco_codec as msbc as soon as bluetooth device is connected. Because I sometimes have some problems like the following when making calls in HF mode:

As A explained, when there is an incoming call bluealsa will initiate the mSBC by itself but I make the call as soon as the device is connected bluealsa will hang in after a short time it will end completely:

```
bluealsa: D: ../../src/ba-rfcomm.c:108: Sending AT message: SET: command:+BCS, value:2
bluealsa: D: ../../src/sco.c:103: New incoming SCO link: F8:C3:CC:92:08:6A: 21
bluealsa: D: ../../src/ba-rfcomm.c:1106: RFCOMM poll timeout
bluealsa: D: ../../src/ba-rfcomm.c:1106: RFCOMM poll timeout
bluealsa: D: ../../src/ba-rfcomm.c:1106: RFCOMM poll timeout
bluealsa: D: ../../src/ba-rfcomm.c:1106: RFCOMM poll timeout
bluealsa: D: ../../src/ba-rfcomm.c:1106: RFCOMM poll timeout
bluealsa: D: ../../src/ba-rfcomm.c:1106: RFCOMM poll timeout
bluealsa: D: ../../src/ba-rfcomm.c:1106: RFCOMM poll timeout
bluealsa: D: ../../src/ba-rfcomm.c:1106: RFCOMM poll timeout
bluealsa: D: ../../src/ba-rfcomm.c:1106: RFCOMM poll timeout
bluealsa: D: ../../src/ba-rfcomm.c:1106: RFCOMM poll timeout
bluealsa: D: ../../src/ba-rfcomm.c:1106: RFCOMM poll timeout
bluealsa: D: ../../src/ba-rfcomm.c:1106: RFCOMM poll timeout
```

borine, 2022-02-14:
> bluealsa: W: ../../src/hci.c:126: Couldn't get SCO socket options: Transport endpoint is not connected
> bluealsa: D: ../../src/hci.c:132: SCO link socket MTU: 21: 0

hmm, you may have found a bug here. it seems the socket is not fully configured at that point, so the call to getsockopt() is failing. As a result we have an MTU of 0.

@Arkq is the getsockopt() failure expected? before ac5706f8c83395471604668b29b5a7e689c95e84 the MTU would simply be set to 48, but since that commit it is left as 0. If the getsockopt() failure is normal, then a method of determining the MTU later is required; if the failure is not expected, maybe the message should be changed to error() and the transport code should abandon the link?

Arkq, 2022-02-14:
I think that is should/shall not fail in any of the `getsockopt()` calls, otherwise we will end up with wrong MTU (not initialized). Warning could be changed to errors, but that will not resolve the problem. I think that we will have to know why the transport is not connected... It's strange, because we've got incoming connection, so why it's not connected? Maybe it was closed between `accept()` (sco.c file) and `hci_sco_get_mtu()` (hci.c file)...

borine, 2022-02-15:
Looking at the kernel source in net/bluetooth/sco.c, I think I've understood the getsockopt() failure now.

In order for SCO_OPTIONS to succeed, the socket must be in state BT_CONNECTED or BT_CONNECT2. Any other state results in ENOTCONN.

The accept() call puts the socket into state BT_CONNECT2. 

The read(fd, &voice, 1) sends a HCI request to the controller, sets the socket state to BT_CONFIG, **but does not wait for controller to respond**

Some time later, the controller responds, and the socket state is changed to BT_CONNECTED

So we are in a race. If our call to getsockopt() occurs before the controller has responded, we get ENOTCONN; if it happens later we get success.

Since it is not obvious how to wait for the controller response, I suggest that we make the  hci_sco_get_mtu() call **before** read(fd, &voice, 1). Then we should be guaranteed a succesful return.

borine, 2022-02-15:
... following on from my previous comment, io_bt_write() in io.c treats ENOTCONN as equivalent to ECONNABORTED, ECONNRESET and ETIMEDOUT. I think that is wrong. ENOTCONN means that the connection protocol is not complete, or the local process has called close(). Since we should know that we have not called close() here, I think that we should treat it as equivalent to EAGAIN, i.e. poll() for readiness to write; does that sound sensible?

borine, 2022-02-16:
OK, scratch my above solution, initial testing shows it does not work. I'll look into it in more depth, possibly at the weekend.

borine, 2022-02-16:
So, I now think this is probably not a bluealsa bug, at least not in the latest code. The debug logs given above appear to be from release 3.1.0. There has been a major re-design of SCO I/O in bluealsa since then, so before looking any deeper, we need to establish whether this behaviour persists in the latest code. I haven't been able to reproduce in simple tests, and I don't have time to set up a clean test system at the moment. @bmt1596 is it possible for you to build from the latest master branch and re-test?

bmt1596, 2022-02-16:
@borine  no problem. i will try with the latest version of Bluealsa. Anyway, thank you very much for your help.

closed 2022-02-16 by bmt1596

Arkq, 2022-02-20:
@bmt1596, since you are experimenting with mSBC support, maybe you could try this PR #539 ? It adds PLC for received mSBC packets, so it should improve overall user experience when using mSBC codec. I've got no real test setup to check this PR properly, so I thought that maybe you could use this new code :).

bmt1596, 2022-02-21:
@Arkq  Hi, thank you very much for this help. I'm also having the problem of SBC packet loss for SCO but overall it doesn't affect too much. I try this new version with PLC support and update you with the results as soon as possible.