arkq/bluez-alsa #534: clicks instead of sound at first connect

gearhead, 2022-02-09:
I recently started using bluealsa with my headphones. I am playing music with mpd to the bluealsa output. At first connect from startup of the RPi, I get clicks through the headset. If I disconnect and reconnect, it plays music. Bluealsa is running, bluealsa-aplay is running as is bluetooth. In the journal, there are errors which look like the headset is trying to autoconnect but bluetooth does not allow it.:
```
Feb 09 08:08:48 rune64 bluetoothd[302]: Authentication attempt without agent
Feb 09 08:08:48 rune64 bluetoothd[302]: src/profile.c:ext_auth() Hands-Free Voice gateway rejected XX:XX:XX:XX:XX:XX: org.bluez.Error.Rejected
Feb 09 08:08:49 rune64 bluetoothd[302]: Authentication attempt without agent
Feb 09 08:08:49 rune64 bluetoothd[302]: profiles/audio/a2dp.c:auth_cb() Access denied: org.bluez.Error.Rejected
Feb 09 08:08:53 rune64 systemd[2283]: Reached target Bluetooth.
Feb 09 08:09:00 rune64 bluetoothd[302]: Authentication attempt without agent
Feb 09 08:09:00 rune64 bluetoothd[302]: src/profile.c:ext_auth() Hands-Free Voice gateway rejected XX:XX:XX:XX:XX:XX: org.bluez.Error.Rejected
Feb 09 08:09:00 rune64 bluetoothd[302]: Authentication attempt without agent
Feb 09 08:09:00 rune64 bluetoothd[302]: profiles/audio/a2dp.c:auth_cb() Access denied: org.bluez.Error.Rejected
Feb 09 08:09:07 rune64 bluetoothd[302]: src/profile.c:ext_connect() Headset Voice gateway failed connect to XX:XX:XX:XX:XX:XX: Connection refused (111)
Feb 09 08:09:08 rune64 bluetoothd[302]: Endpoint registered: sender=:1.10 path=/org/bluez/hci0/A2DP/SBC/source/3
Feb 09 08:09:08 rune64 bluealsa[405]: /usr/bin/bluealsa: W: Unsupported AT message: SET: command:+CLIP, value:1
Feb 09 08:09:08 rune64 kernel: input: Jabra Evolve2 65 (AVRCP) as /devices/virtual/input/input1
Feb 09 08:09:08 rune64 systemd-logind[2277]: Watching system buttons on /dev/input/event1 (Jabra Evolve2 65 (AVRCP))
```
I manually connect it with `bluetoothctl connect XX:XX:XX:XX:XX:XX` and start mpd (the output is configured in the mpd.conf). All I head are clicks. If I disconnect and reconnect, it plays. If I stop bluealsa and restart it then connect, it plays. How can I get it to play at startup each time?

borine, 2022-02-09:
> Feb 09 08:08:53 rune64 systemd[2283]: Reached target Bluetooth.

The messages before this indicate that the headphones are attempting to connect before systemd has completed its startup of all the bluetooth components. The messages after that suggest that either Bluez does not "trust" the headphones, or that bluealsa is still not ready when the headphones try again to connect HFP and A2DP. The fact that the AVRCP connection succeeds suggests that it is bluealsa which is not yet ready.

I have no idea what is causing the "clicks"; some debug logs from bluealsa might help determine if they are caused by bluealsa.

gearhead, 2022-02-11:
Well, this AM, MPD was off for about 24 hrs. I tried to start playing and I got clicks again and I started/stopped/started mpd a couple times then it started playing. No errors in the journal. I will turn on verbose logging for bluealsa and see if something shows up.

gearhead, 2022-02-17:
what I find is that if I restart mpd, it can be resolved. If I connect then restart mod, I can get clicks sometimes really faint. Then if I restart mod, it can get to a choppy sound output. If I just stop and start the stream, it can resolve.

gearhead, 2022-02-27:
I have seen this a number of times. I have only seen it happen after a reboot. It is like the stream does not sync in order to play correctly. If I re-start mpd it will usually play. Sometimes it takes 2x of restarting before it will play properly form mpd. Once it starts playing properly it continues to play when I stop and start mpd.

gearhead, 2022-09-15:
borine, thanks for the work! I appreciate it. I have dropped the ball on this. It may be that I am too impatient when testing to let everything come up and be ready. If I run into it again. I will be more responsive to getting logs. Thanks!

closed 2022-09-15 by borine