arkq/bluez-alsa #535: There is no sound for some reason

BlueSkyKDM, 2022-02-13:
Can anyone give me some advice

OS : Ubuntu 20.10
BluezAlsa : 3.0

D: bluealsa-pcm.c:895: Getting BlueALSA PCM: PLAYBACK BE:22:4A:E4:CB:58 a2dp                                                                                                                                                                 D: bluealsa-pcm.c:769: /org/bluealsa/hci0/dev_BE_22_4A_E4_CB_58/a2dpsrc/sink: Setting constraints                                                                                                                                            D: bluealsa-pcm.c:396: /org/bluealsa/hci0/dev_BE_22_4A_E4_CB_58/a2dpsrc/sink: Initializing HW                                                                                                                                                D: bluealsa-pcm.c:416: /org/bluealsa/hci0/dev_BE_22_4A_E4_CB_58/a2dpsrc/sink: FIFO buffer size: 4096                                                                                                                                         D: bluealsa-pcm.c:422: /org/bluealsa/hci0/dev_BE_22_4A_E4_CB_58/a2dpsrc/sink: Selected HW buffer: 4 periods x 4320 bytes == 17280 bytes                                                                                                      
D: bluealsa-pcm.c:440: /org/bluealsa/hci0/dev_BE_22_4A_E4_CB_58/a2dpsrc/sink: Initializing SW                                                                                                                                                D: bluealsa-pcm.c:440: /org/bluealsa/hci0/dev_BE_22_4A_E4_CB_58/a2dpsrc/sink: Initializing SW                                                                                                                                                D: bluealsa-pcm.c:440: /org/bluealsa/hci0/dev_BE_22_4A_E4_CB_58/a2dpsrc/sink: Initializing SW                                                                                                                                                D: bluealsa-pcm.c:477: /org/bluealsa/hci0/dev_BE_22_4A_E4_CB_58/a2dpsrc/sink: Prepared                                                                                                                                                       D: bluealsa-pcm.c:440: /org/bluealsa/hci0/dev_BE_22_4A_E4_CB_58/a2dpsrc/sink: Initializing SW                                                                                                                                                D: bluealsa-pcm.c:477: /org/bluealsa/hci0/dev_BE_22_4A_E4_CB_58/a2dpsrc/sink: Prepared                                                                                                                                                       D: bluealsa-pcm.c:316: /org/bluealsa/hci0/dev_BE_22_4A_E4_CB_58/a2dpsrc/sink: Starting                                                                                                                                                       D: bluealsa-pcm.c:176: /org/bluealsa/hci0/dev_BE_22_4A_E4_CB_58/a2dpsrc/sink: Waiting for first period of frames                                                                                                                             D: bluealsa-pcm.c:183: /org/bluealsa/hci0/dev_BE_22_4A_E4_CB_58/a2dpsrc/sink: Starting IO loop: 20                                                                                                                                           D: bluealsa-pcm.c:349: /org/bluealsa/hci0/dev_BE_22_4A_E4_CB_58/a2dpsrc/sink: Stopping                                                                                                                                                       D: bluealsa-pcm.c:129: /org/bluealsa/hci0/dev_BE_22_4A_E4_CB_58/a2dpsrc/sink: IO thread cleanup                                                                                                                                              D: bluealsa-pcm.c:349: /org/bluealsa/hci0/dev_BE_22_4A_E4_CB_58/a2dpsrc/sink: Stopping                                                                                                                                                       D: bluealsa-pcm.c:432: /org/bluealsa/hci0/dev_BE_22_4A_E4_CB_58/a2dpsrc/sink: Freeing HW                                                                                                                                                     D: bluealsa-pcm.c:385: /org/bluealsa/hci0/dev_BE_22_4A_E4_CB_58/a2dpsrc/sink: Closing

Arkq, 2022-02-14:
First of all try to use current master code - maybe it will solve issue you've got. If that will not help, see the troubleshooting section 2 (https://github.com/Arkq/bluez-alsa#troubleshooting). I do not remember which version of alsa-lib is available in Ubuntu 20.04, but if I recall correctly such workaround is required for this LTS.

BlueSkyKDM, 2022-02-17:
I could do it according to the troubleshooting section 2.
Thank you.

closed 2022-02-17 by BlueSkyKDM