arkq/bluez-alsa #537: arecord not working on ARM target Qualcomm BT chipset

ssirig714, 2022-02-15:
Hi ,

HFP headphone connection success with bluealsa.
arecord is not working writing into test.wav file for qualcomm BT UART adaptor for HFP SCO .
hciconfig -a o/p sco count is showing zero  when arecord.
Could you please help is it simialr issue or any fix we need to add for qualcomm like broadcomm route the sco?
from the bluealsa debugs did not see any error.

i think similar issue #347  CONFIG_USB_EHCI_TT_NEWSCHED=y kernal config set and tried no luck . once set this box is keeps reloading.

logs are :

uname -a
Linux 4.9.232-1.19 https://github.com/Arkq/bluez-alsa/issues/1 SMP Tue Feb 1 23:20:16 UTC 2022 armv7l GNU/Linux

hciconfig -a
hci0: Type: Primary Bus: UART
BD Address: 61:47:AA:32:44:08 ACL MTU: 1024:7 SCO MTU: 60:8
UP RUNNING
RX bytes:54325 acl:404 sco:0 events:2229 errors:0
TX bytes:32952 acl:413 sco:0 commands:1180 errors:0
Features: 0xff 0xfe 0x8f 0xfe 0xd8 0x3f 0x5b 0x87
Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3
Link policy: RSWITCH HOLD SNIFF
Link mode: SLAVE ACCEPT
Name: 'BlueZ 5.54'
Class: 0x480000
Service Classes: Capturing, Telephony
Device Class: Miscellaneous,
HCI Version: 4.2 (0x8) Revision: 0x0
LMP Version: 4.2 (0x8) Subversion: 0x25a
Manufacturer: Qualcomm (29)

#arecord -D bluealsa:DEV=11:11:22:33:80:72,PROFILE=sco /tmp/test.wav
D: ../../../git/src/asound/bluealsa-pcm.c:1312: Getting BlueALSA PCM: CAPTURE 11:11:22:33:80:72 sco
D: ../../../git/src/asound/bluealsa-pcm.c:1062: /org/bluealsa/hci0/dev_11_11_22_33_80_72/hfpag/source: Setting constraints
Recording WAVE '/tmp/test.wav' : Unsigned 8 bit, Rate 8000 Hz, Mono
D: ../../../git/src/asound/bluealsa-pcm.c:532: /org/bluealsa/hci0/dev_11_11_22_33_80_72/hfpag/source: Initializing HW
bluealsa: D: ../../git/src/dbus.c:66: Called: org.bluealsa.PCM1.Open() on /org/bluealsa/hci0/dev_11_11_22_33_80_72/hfpag/source
bluealsa: D: ../../git/src/ba-transport.c:711: New SCO link: 11:11:22:33:80:72: 28
bluealsa: D: ../../git/src/hci.c:133: SCO link socket MTU: 28: 60
bluealsa: D: ../../git/src/ba-transport.c:1151: Starting transport: HFP Audio Gateway (CVSD)
bluealsa: D: ../../git/src/ba-transport.c:294: Created BT socket duplicate: [28]: 29
bluealsa: D: ../../git/src/ba-transport.c:1472: Created new IO thread [ba-sco-enc]: HFP Audio Gateway (CVSD)
bluealsa: D: ../../git/src/ba-transport.c:294: Created BT socket duplicate: [28]: 30
bluealsa: D: ../../git/src/sco.c:230: IO loop: START: sco_cvsd_enc_thread: HFP Audio Gateway (CVSD)
bluealsa: D: ../../git/src/ba-transport.c:1472: Created new IO thread [ba-sco-dec]: HFP Audio Gateway (CVSD)
bluealsa: D: ../../git/src/sco.c:301: IO loop: START: sco_cvsd_dec_thread: HFP Audio Gateway (CVSD)
D: ../../../git/src/asound/bluealsa-pcm.c:567: /org/bluealsa/hci0/dev_11_11_22_33_80_72/hfpag/source: FIFO buffer size: 32768 frames
D: ../../../git/src/asound/bluealsa-pcm.c:573: /org/bluealsa/hci0/dev_11_11_22_33_80_72/hfpag/source: Selected HW buffer: 4 periods x 2000 bytes == 8000 bytes
D: ../../../git/src/asound/bluealsa-pcm.c:600: /org/bluealsa/hci0/dev_11_11_22_33_80_72/hfpag/source: Initializing SW
D: ../../../git/src/asound/bluealsa-pcm.c:600: /org/bluealsa/hci0/dev_11_11_22_33_80_72/hfpag/source: Initializing SW
D: ../../../git/src/asound/bluealsa-pcm.c:600: /org/bluealsa/hci0/dev_11_11_22_33_80_72/hfpag/source: Initializing SW
D: ../../../git/src/asound/bluealsa-pcm.c:639: /org/bluealsa/hci0/dev_11_11_22_33_80_72/hfpag/source: Prepared
D: ../../../git/src/asound/bluealsa-pcm.c:600: /org/bluealsa/hci0/dev_11_11_22_33_80_72/hfpag/source: Initializing SW
D: ../../../git/src/asound/bluealsa-pcm.c:356: /org/bluealsa/hci0/dev_11_11_22_33_80_72/hfpag/source: Starting
bluealsa: D: ../../git/src/ba-transport.c:1387: PCM resumed: 25
D: ../../../git/src/asound/bluealsa-pcm.c:226: /org/bluealsa/hci0/dev_11_11_22_33_80_72/hfpag/source: Starting IO loop: 7
[ 851.338036] >>Open gpK7 (pid:14781 "RF4CE")
[ 852.289070] >>Close gpK7 (pid:14740 "controlMgr")
[ 858.584298] >>Open gpK7 (pid:15691 "RF4CE")
[ 859.534209] >>Close gpK7 (pid:15631 "controlMgr")
[ 865.731508] >>Open gpK7 (pid:16353 "RF4CE")
[ 866.668530] >>Close gpK7 (pid:16351 "controlMgr")
[ 872.925503] >>Open gpK7 (pid:19552 "RF4CE")
^CAborted by signal Interrupt...
arecord: pcm_read:2151: read error: Interrupted system call
D: ../../../git/src/asound/bluealsa-pcm.c:392: /org/bluealsa/hci0/dev_11_11_22_33_80_72/hfpag/source: Stopping
D: ../../../git/src/asound/bluealsa-pcm.c:161: /org/bluealsa/hci0/dev_11_11_22_33_80_72/hfpag/source: IO thread cleanup
D: ../../../git/src/asound/bluealsa-pcm.c:592: /org/bluealsa/hci0/dev_11_11_22_33_80_72/hfpag/source: Freeing HW
D: ../../../git/src/asound/bluealsa-pcm.c:443: /org/bluealsa/hci0/dev_11_11_22_33_80_72/hfpag/source: Closing
bluealsa: D: ../../git/src/ba-transport.c:1434: Closing PCM: 25
bluealsa: D: ../../git/src/ba-transport.c:438: PCM clients check keep-alive: 0 ms
bluealsa: D: ../../git/src/ba-transport.c:387: Stopping transport: No PCM clients
bluealsa: D: ../../git/src/ba-transport.c:306: Closing BT socket duplicate [28]: 29
bluealsa: D: ../../git/src/ba-transport.c:726: Releasing SCO link: 28
bluealsa: D: ../../git/src/ba-transport.c:1497: Exiting IO thread [ba-sco-enc]: HFP Audio Gateway (CVSD)
bluealsa: D: ../../git/src/ba-transport.c:306: Closing BT socket duplicate [-1]: 30
bluealsa: D: ../../git/src/ba-transport.c:1497: Exiting IO thread [ba-sco-dec]: HFP Audio Gateway (CVSD)

ls -alr /tmp/test.wav
-rw-r--r-- 1 root root 44 Feb 14 12:21 /tmp/test.wav
root@tchxi6:/opt#
file size is 44 .        Same i have tried to aplay its not working . no soud

borine, 2022-02-15:
CONFIG_USB_EHCI_TT_NEWSCHED is for the USB subsystem, it has nothing to do with UART.

>RX bytes:54325 acl:404 sco:0 events:2229 errors:0
TX bytes:32952 acl:413 sco:0 commands:1180 errors:0

Looks like there is no SCO traffic over the HCI, so my guess is that the adapter is routing SCO through some other interface. For QCA adapters SCO routing is usually set by the firmware, and to change it is generally necessary to use a different firmware. I recommend that you search more generally for your specific adapter model, to see if anyone else has found a working firmware. If not, perhaps you need to contact the vendor or manufacturer.

Arkq, 2022-06-21:
Since no more information was provided for the last 4 months, I assume that the issue was resolved.

closed 2022-06-21 by Arkq