arkq/bluez-alsa #538: Which codes can I enable legally?

createcandle, 2022-02-20:
By default the Raspberry Pi offers BlueAlsa version 1.4 in  its repositories. That's seems to be quite an old version.

I'd like to incorporate a newer version of BlueAlsa in a Raspberry Pi disk image that I'm working on.

In the build instructions I see a lot of options for codecs, but it's not clear to me which ones I can safely enable if I want to respect copyright. I've been trying to look this up one by one, but haven't been very succesful.

Do you perhaps have suggestions here? Maybe the build instructions could engrain some small hints on his topic? I'd be happy to create a pull request proposal if you tell me which codecs are 'safe'.

Arkq, 2022-02-20:
> In the build instructions I see a lot of options for codecs, but it's not clear to me which ones I can safely enable if I want to respect copyright. I've been trying to look this up one by one, but haven't been very succesful.

It's rather hard to answer that question. It all depends what you mean by "respect copyright". As far as I know all codecs which are publicly available (there is a public download page) can be used privately. However, if you intent to create a RPi image which can be used for commercial usage, you might hit some issues. As far as I know AAC codec for commercial usage requires a fee. The same case is with LDAC. There might be some issues with apt-X, however, ffmpeg provides aptX and it seems that the patent has expired (this shall be confirmed), so maybe it can be safely used. I think that FastStream might be used without any issues, because in fact it is SBC codec (which is royalty free). Also, I think that MP3 patent is also expired, so this codec can be used, but I remember that few years ago some Linux distributions have had MP3 codecs in non-free repositories (but maybe it is not longer the case). And lastly, LC3plus. I think that it is free to use (but I might be wrong).

So as a summary. You can safely use SBC, FastStream and mSBC (for HFP). I think you can use LC3plus and maybe MP3. Other codecs might need some fees for commercial usage.

In order to be sure, you will have to search for all this codecs websites and check whether there are some conditions for personal/commercial usage.

Please not that BlueALSA only provides a support for all these codes. The responsibility to respect the low is on the user side. If you gather some info regarding these codes, please add a comment in here or create a PR with adequate info in the README file.

createcandle, 2022-02-20:
Great explanation, thanks!

If I learn anything new I will be sure to add it here.

Would you like me to create a pull request for documentation that incorporates this info?

Arkq, 2022-02-20:
> Would you like me to create a pull request for documentation that incorporates this info?

Yes, you can create a PR, or alternatively add a page to the bluez-alsa wiki which will summarize this kind of info. Then we can add a link for this page into the README file.

createcandle, 2022-02-20:
This Wikipedia page has some information about certain flavours of AptX patents being expired or revoked:
https://en.wikipedia.org/wiki/AptX

Is your AptX version an implementation of the 'basic' AptX version? The HD version seems to still have a license.

createcandle, 2022-02-20:
I'm not sure LC3+ is available freely yet:
https://www.iis.fraunhofer.de/en/ff/amm/communication/lc3.html

The Wikipedia page doesn't mention anything on the topic:
https://en.wikipedia.org/wiki/LC3_(codec)

createcandle, 2022-02-20:
The MP3 patents seem expired:
https://www.tunequest.org/a-big-list-of-mp3-patents/20070226/

createcandle, 2022-02-20:
According to this Wikipedia page the Fasttream patent is expired:
https://en.wikipedia.org/wiki/SBC_%28codec%29

createcandle, 2022-02-20:
In short: `../configure —-enable-faststream --enable-mp3lame` seems to be the safest option

createcandle, 2022-02-20:
The original AAC was released in 1997. You'd think a version of that would be available now. But according to the Wikipedia page you're right, it's still under license:
https://en.wikipedia.org/wiki/Advanced_Audio_Coding#Licensing_and_patents

Arkq, 2022-02-20:
> Is your AptX version an implementation of the 'basic' AptX version? The HD version seems to still have a license.

BlueALSA has both, "basic" (`--enable-aptx`) and HD version (`--enable-aptx-hd`). When using aptX it is also advised to add `--with-libopenaptx`, however, please make sure that this: https://github.com/pali/libopenaptx/commit/811bc18586d634042618d633727ac0281d4170b8#commitcomment-54153707 does not apply to bluez-alsa, becasue bluez-alsa is MIT, not GPL (I'm not license "conflicts" specialist :D).

flatsiedatsie, 2022-02-21:
I've created an initial wiki page: https://github.com/Arkq/bluez-alsa/wiki/Patents-and-copyright-relating-to-Bluetooth-codecs