arkq/bluez-alsa #542: Bluez 5.62 - can't register sink anymore

hifiberry, 2022-02-25:
After upgrading (not only) bluetooth to 5.62, bluealsa can't register the service anymore:

 /usr/bin/bluealsa: W: Couldn't register media endpoint: GDBus.Error:org.freedesktop.DBus.Error.UnknownMethod: Method "RegisterEndpoint" with signature "oa{sv}" on interface "org.bluez.Media1" doesn't exist
 
 This probably isn't caused by bluealsa, but I don't know anymore where to look for the root cause. 
 
 DBUS runs in system mode, access lists seem to be ok. Any idea?

borine, 2022-02-25:
What is the output of:
```shell
dbus-send --system --dest=org.bluez --type=method_call --print-reply /org/bluez/hci0 org.freedesktop.DBus.Introspectable.Introspect
```

hifiberry, 2022-02-25:
`# dbus-send --system --dest=org.bluez --type=method_call --print-reply /org/bluez/hci0 org.freedesktop.DBus.Introspectable.Introspect
method return time=1645812074.886044 sender=:1.10 -> destination=:1.20 serial=53 reply_serial=2
   string "<!DOCTYPE node PUBLIC "-//freedesktop//DTD D-BUS Object Introspection 1.0//EN"
"http://www.freedesktop.org/standards/dbus/1.0/introspect.dtd">
<node><interface name="org.freedesktop.DBus.Introspectable"><method name="Introspect"><arg name="xml" type="s" direction="out"/>
</method></interface><interface name="org.bluez.Adapter1"><method name="StartDiscovery"></method><method name="SetDiscoveryFilter"><arg name="properties" type="a{sv}" direction="in"/>
</method><method name="StopDiscovery"></method><method name="RemoveDevice"><arg name="device" type="o" direction="in"/>
</method><method name="GetDiscoveryFilters"><arg name="filters" type="as" direction="out"/>
</method><property name="Address" type="s" access="read"></property><property name="AddressType" type="s" access="read"></property><property name="Name" type="s" access="read"></property><property name="Alias" type="s" access="readwrite"></property><property name="Class" type="u" access="read"></property><property name="Powered" type="b" access="readwrite"></property><property name="Discoverable" type="b" access="readwrite"></property><property name="DiscoverableTimeout" type="u" access="readwrite"></property><property name="Pairable" type="b" access="readwrite"></property><property name="PairableTimeout" type="u" access="readwrite"></property><property name="Discovering" type="b" access="read"></property><property name="UUIDs" type="as" access="read"></property><property name="Modalias" type="s" access="read"></property><property name="Roles" type="as" access="read"></property><property name="ExperimentalFeatures" type="as" access="read"></property></interface><interface name="org.freedesktop.DBus.Properties"><method name="Get"><arg name="interface" type="s" direction="in"/>
<arg name="name" type="s" direction="in"/>
<arg name="value" type="v" direction="out"/>
</method><method name="Set"><arg name="interface" type="s" direction="in"/>
<arg name="name" type="s" direction="in"/>
<arg name="value" type="v" direction="in"/>
</method><method name="GetAll"><arg name="interface" type="s" direction="in"/>
<arg name="properties" type="a{sv}" direction="out"/>
</method><signal name="PropertiesChanged"><arg name="interface" type="s"/>
<arg name="changed_properties" type="a{sv}"/>
<arg name="invalidated_properties" type="as"/>
</signal>
</interface><interface name="org.bluez.GattManager1"><method name="RegisterApplication"><arg name="application" type="o" direction="in"/>
<arg name="options" type="a{sv}" direction="in"/>
</method><method name="UnregisterApplication"><arg name="application" type="o" direction="in"/>
</method></interface><interface name="org.bluez.LEAdvertisingManager1"><method name="RegisterAdvertisement"><arg name="advertisement" type="o" direction="in"/>
<arg name="options" type="a{sv}" direction="in"/>
</method><method name="UnregisterAdvertisement"><arg name="service" type="o" direction="in"/>
</method><property name="ActiveInstances" type="y" access="read"></property><property name="SupportedInstances" type="y" access="read"></property><property name="SupportedIncludes" type="as" access="read"></property><property name="SupportedSecondaryChannels" type="as" access="read"></property></interface><interface name="org.bluez.NetworkServer1"><method name="Register"><arg name="uuid" type="s" direction="in"/>
<arg name="bridge" type="s" direction="in"/>
</method><method name="Unregister"><arg name="uuid" type="s" direction="in"/>
</method></interface></node>"`

borine, 2022-02-25:
bizarre - your bluetooth installation is completely lacking the `org.bluez.Media1` interface. I don't know what can cause that - perhaps some build configuration option for bluez, or maybe some bluetooth runtime configuration error? Without that interface it is impossible for `bluealsa` to register A2DP services.

hifiberry, 2022-02-25:
I think that helps. That helped! Just noticed that Buildroot had added a new configuration option to enable A2DP. This had to be enabled now, otherwise this bluez was compiled with "--disable-a2dp --disable-avrcp". 
Fixed this, now it seems to work again.
Thank you!

closed 2022-02-25 by hifiberry