arkq/bluez-alsa #546: Simpler mixing option?

createcandle, 2022-02-28:
I'm tried implementing the loopback method of merging multiple audio signals into a single stream to feed too BlueAlsa. Howewer, the loopback modprobe threw a spanner in my existing systems which relied on getting lists and Alsa devices (they immediately disappeared from `aplay -L`). I'd like my users to still be able to easily switch to headphone jack audio output, for example.

For my use case, it would be fine if only the last connected bluetooth speaker could be streamed into.

Is there really no simpler way to merge audio streams than through the loopback method? Or is there a way to make the loopback method less "destructive"?

Arkq, 2022-02-28:
Please look at @borine's fork of bluez-alsa: https://github.com/borine/bluez-alsa It contains audio mixer which allows to stream audio to BlueALSA from many applications simultaneously. For now it's in a separate fork, but I hope that one day it will be available in "upstream" (this) repo :)

borine, 2022-02-28:
@Arkq I am happy to submit my multi-client code as a PR if you think it would be beneficial. However, beware that I have very limited time available at present, so would not be able to respond to any review comments in a timely manner. Right now, that fork does everything I need it to do, so there is little incentive for me re-work it; and from experience we both know that my coding skills are somewhat out-of-date so that it is likely to need a lot of effort to make it fit for upstream. That is the real reason I have not yet proposed it.
@createcandle If you are able to test it and give feedback and possibly coding effort to a PR then it may be possible to at least use it as a basis for some future enhancement to bluealsa.

> they immediately disappeared from aplay -L

If you can provide more details I may be able to help with that. It's some time since I worked on the wiki article for bluealsa with dmix, but hopefully I can remember enough detail to troubleshoot it. Also note that the same fork also has a new utility `bluealsa-autoconfig` for dynamically adding bluealsa PCMs to the ALSA config, with a "default" definition that includes a fallback to system default when no bluealsa devices are connected. I use this very successfully with `kodi` and `chromium-browser`, for example.

Arkq, 2022-03-01:
> @Arkq I am happy to submit my multi-client code as a PR if you think it would be beneficial.

@borine Of course it will be beneficial. However, you don't need to create a PR right away, because I've still got some things to do in the core of bluealsa . These are some minor things, I hope, but before adding a new big feature I would like to be sure that there in no big refactoring around the corner.

createcandle, 2022-03-06:
I'm giving it a try now. 

`../configure --enable-msbc --enable-mp3lame --enable-faststream` lead to an error:

```
configure: error: Package requirements (spandsp >= 0.0.6) were not met:

No package 'spandsp' found

Consider adjusting the PKG_CONFIG_PATH environment variable if you
installed software in a non-standard prefix.

Alternatively, you may set the environment variables SPANDSP_CFLAGS
and SPANDSP_LIBS to avoid the need to call pkg-config.
```

Some searching lead to me `libspandsp-dev`. Although that seems to mostly install... things related to images?

`libdeflate-dev libjbig-dev libjpeg-dev libjpeg62-turbo-dev liblzma-dev libspandsp2 libtiff-dev libtiffxx5`

In any case, the configure command seems to work now. Installation also seems to have worked, and I'm now playing one audio stream without an issue.

To quote the Ghostbusters: how do I merge the streams? I'm seeing some new options from `aplay -L`:

```
null
    Discard all samples (playback) or generate zero samples (capture)
bluealsa
    Bluetooth Audio Hub
loop_playback_out
    ALSA playback loop out
loop_playback_in_mix
    Loopback in mixer device, outputs to loop_playback_out
```

- Should I send audio to `loop_playback_in_mix` ?
- Do I need to do anything with `loop_playback_out`?

borine, 2022-03-06:
> loop_playback_out
    ALSA playback loop out
loop_playback_in_mix
    Loopback in mixer device, outputs to loop_playback_out

I don't know where these come from, they are nothing to do with BlueALSA, neither this repository nor my fork. My guess is you have some entries in ~/.asoundrc from previous experiments. If you are using the fork from https://github.com/borine/bluez-alsa then the bluealsa pcm can be opened by multiple applications at the same time, provided the `bluealsa` server is started with the `--multi-client` option. There is no need for any separate mix, snoop, loop etc plugins.

To avoid confusion for other readers here it might be better to raise an issue in my repository at the above link for debugging/troubleshooting that fork. Discussion can move back to this repository if/when a PR is raised to merge the multi-client function here.

createcandle, 2022-03-07:
> My guess is you have some entries in ~/.asoundrc from previous experiments.

You were right.

I didn't know about the `--multi-client` option, I'm trying that now.

IT WORKS! Brilliant!

closed 2022-03-07 by createcandle