arkq/bluez-alsa #553: Couldn't get BlueALSA PCM: PCM not found

MarkusPint, 2022-03-17:
Hi, i'm having trouble getting audio from a BT device.

My outputs:

`bluetoothctl info DC:F7:56:4A:9B:81
Device DC:F7:56:4A:9B:81 (public)
	Name: Galaxy A40
	Alias: Galaxy A40
	Class: 0x005a020c
	Icon: phone
	Paired: yes
	Trusted: yes
	Blocked: no
	Connected: yes
	LegacyPairing: no
	UUID: OBEX Object Push          (00001105-0000-1000-8000-00805f9b34fb)
	UUID: Audio Source              (0000110a-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
	UUID: Advanced Audio Distribu.. (0000110d-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
	UUID: Headset AG                (00001112-0000-1000-8000-00805f9b34fb)
	UUID: PANU                      (00001115-0000-1000-8000-00805f9b34fb)
	UUID: NAP                       (00001116-0000-1000-8000-00805f9b34fb)
	UUID: Handsfree Audio Gateway   (0000111f-0000-1000-8000-00805f9b34fb)
	UUID: SIM Access                (0000112d-0000-1000-8000-00805f9b34fb)
	UUID: Phonebook Access Server   (0000112f-0000-1000-8000-00805f9b34fb)
	UUID: Message Access Server     (00001132-0000-1000-8000-00805f9b34fb)
	UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
	UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
	UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
	UUID: Vendor specific           (7c7bcc5e-27a2-11e9-ab14-d663bd873d93)
	Modalias: bluetooth:v0075p0100d0201
`

`bluealsa-aplay -l
**** List of PLAYBACK Bluetooth Devices ****
**** List of CAPTURE Bluetooth Devices ****`

`cat ~/.asoundrc
defaults.bluealsa.service "org.bluealsa"
defaults.bluealsa.device "DC:F7:56:4A:9B:81"
defaults.bluealsa.profile "a2dp"
defaults.bluealsa.delay 10000`

`sudo bluealsa -p a2dp-sink -p a2dp-source
bluealsa: D: ../../src/ofono.c:629: Checking oFono service presence
bluealsa: D: ../../src/main.c:580: Starting main dispatching loop
bluealsa: D: ../../src/main.c:112: Acquired D-Bus service name: org.bluealsa
bluealsa: D: ../../src/bluealsa-dbus.c:426: Registering D-Bus manager: /org/bluealsa
bluealsa: D: ../../src/bluez.c:679: Registering battery provider: /org/bluez/hci0/battery
bluealsa: D: ../../src/bluez.c:699: BlueZ battery provider support not available
bluealsa: D: ../../src/bluez.c:493: Creating media endpoint object: /org/bluez/hci0/A2DP/AAC/source/1
bluealsa: D: ../../src/bluez.c:402: Registering media endpoint: /org/bluez/hci0/A2DP/AAC/source/1
bluealsa: D: ../../src/bluez.c:493: Creating media endpoint object: /org/bluez/hci0/A2DP/AAC/source/2
bluealsa: D: ../../src/bluez.c:402: Registering media endpoint: /org/bluez/hci0/A2DP/AAC/source/2
bluealsa: D: ../../src/bluez.c:493: Creating media endpoint object: /org/bluez/hci0/A2DP/AAC/sink/1
bluealsa: D: ../../src/bluez.c:402: Registering media endpoint: /org/bluez/hci0/A2DP/AAC/sink/1
bluealsa: D: ../../src/bluez.c:493: Creating media endpoint object: /org/bluez/hci0/A2DP/AAC/sink/2
bluealsa: D: ../../src/bluez.c:402: Registering media endpoint: /org/bluez/hci0/A2DP/AAC/sink/2
bluealsa: D: ../../src/bluez.c:493: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/source/1
bluealsa: D: ../../src/bluez.c:402: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/source/1
bluealsa: D: ../../src/bluez.c:493: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/source/2
bluealsa: D: ../../src/bluez.c:402: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/source/2
bluealsa: D: ../../src/bluez.c:493: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/sink/1
bluealsa: D: ../../src/bluez.c:402: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/sink/1
bluealsa: D: ../../src/bluez.c:493: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/sink/2
bluealsa: D: ../../src/bluez.c:402: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/sink/2
bluealsa: D: ../../src/bluez.c:679: Registering battery provider: /org/bluez/hci1/battery
bluealsa: D: ../../src/bluez.c:699: BlueZ battery provider support not available
bluealsa: D: ../../src/bluez.c:493: Creating media endpoint object: /org/bluez/hci1/A2DP/AAC/source/1
bluealsa: D: ../../src/bluez.c:402: Registering media endpoint: /org/bluez/hci1/A2DP/AAC/source/1
bluealsa: D: ../../src/bluez.c:493: Creating media endpoint object: /org/bluez/hci1/A2DP/AAC/source/2
bluealsa: D: ../../src/bluez.c:402: Registering media endpoint: /org/bluez/hci1/A2DP/AAC/source/2
bluealsa: D: ../../src/bluez.c:493: Creating media endpoint object: /org/bluez/hci1/A2DP/AAC/sink/1
bluealsa: D: ../../src/bluez.c:402: Registering media endpoint: /org/bluez/hci1/A2DP/AAC/sink/1
bluealsa: D: ../../src/bluez.c:493: Creating media endpoint object: /org/bluez/hci1/A2DP/AAC/sink/2
bluealsa: D: ../../src/bluez.c:402: Registering media endpoint: /org/bluez/hci1/A2DP/AAC/sink/2
bluealsa: D: ../../src/bluez.c:493: Creating media endpoint object: /org/bluez/hci1/A2DP/SBC/source/1
bluealsa: D: ../../src/bluez.c:402: Registering media endpoint: /org/bluez/hci1/A2DP/SBC/source/1
bluealsa: D: ../../src/bluez.c:493: Creating media endpoint object: /org/bluez/hci1/A2DP/SBC/source/2
bluealsa: D: ../../src/bluez.c:402: Registering media endpoint: /org/bluez/hci1/A2DP/SBC/source/2
bluealsa: D: ../../src/bluez.c:493: Creating media endpoint object: /org/bluez/hci1/A2DP/SBC/sink/1
bluealsa: D: ../../src/bluez.c:402: Registering media endpoint: /org/bluez/hci1/A2DP/SBC/sink/1
bluealsa: D: ../../src/bluez.c:493: Creating media endpoint object: /org/bluez/hci1/A2DP/SBC/sink/2
bluealsa: D: ../../src/bluez.c:402: Registering media endpoint: /org/bluez/hci1/A2DP/SBC/sink/2
bluealsa: D: ../../src/bluez.c:928: Creating hands-free profile object: /org/bluez/HSP/AudioGateway
bluealsa: D: ../../src/bluez.c:852: Registering hands-free profile: /org/bluez/HSP/AudioGateway
bluealsa: D: ../../src/bluez.c:928: Creating hands-free profile object: /org/bluez/HFP/AudioGateway
bluealsa: D: ../../src/bluez.c:852: Registering hands-free profile: /org/bluez/HFP/AudioGateway
bluealsa: D: ../../src/bluez.c:1049: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
bluealsa: D: ../../src/bluez.c:1145: Adding new Stream End-Point: DC:F7:56:4A:9B:81: SBC
bluealsa: D: ../../src/bluez.c:1049: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
bluealsa: D: ../../src/bluez.c:1145: Adding new Stream End-Point: DC:F7:56:4A:9B:81: AAC
bluealsa: D: ../../src/bluez.c:1049: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
bluealsa: D: ../../src/bluez.c:1145: Adding new Stream End-Point: DC:F7:56:4A:9B:81: aptX
bluealsa: D: ../../src/bluez.c:1049: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
bluealsa: D: ../../src/bluez.c:1145: Adding new Stream End-Point: DC:F7:56:4A:9B:81: LDAC
bluealsa: D: ../../src/bluez.c:1049: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
bluealsa: D: ../../src/bluez.c:1145: Adding new Stream End-Point: DC:F7:56:4A:9B:81: samsung-SC
bluealsa: D: ../../src/bluez.c:1049: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
bluealsa: D: ../../src/bluez.c:1049: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()`

Arkq, 2022-03-23:
Looking at the output from the "bluealsa -p a2dp-sink -p a2dp-source" command I assume that BlueALSA is not the only Bluetooth Audio service which is registered in BlueZ. Since the device is connected but there is no such info in the BlueALSA log (we can see only new endpoints but not SetConfiguration call) I assume that you are running PulseAudio or PipeWire on your system as well. See https://github.com/Arkq/bluez-alsa#troubleshooting section 1.

Arkq, 2022-06-21:
Since no more information was provided, I assume that the issue was resolved.

closed 2022-06-21 by Arkq