arkq/bluez-alsa #554: Output only works from SCO. A2DP profile returns PCM error.

reitermiller, 2022-03-31:
Hi, I'm experiencing problems outputting to Airpods Pro(s) over bluez-alsa v. v3.1.0-106-g177e163. I am able to output over SCO ```aplay -D ass2 example.wav ```, but A2DP output, ```aplay -D ass example.wav``` returns the following:

```
eros@lycia /tmp                                                                                                                                                                    [10:20:04]
> $ aplay -D ass example.wav                                                                                                                                                            
D: ../../../src/asound/bluealsa-pcm.c:1309: Getting BlueALSA PCM: PLAYBACK 74:65:0C:8E:5B:9D a2dp
ALSA lib ../../../src/asound/bluealsa-pcm.c:1313:(_snd_pcm_bluealsa_open) Couldn't get BlueALSA PCM: PCM not found
aplay: main:852: audio open error: No such device
```

**(working) output of sco playback**
```
eros@lycia /tmp                                                                                                                                                                    [10:23:48]
> $ aplay -D ass2 example.wav                                                                                                                                                                
D: ../../../src/asound/bluealsa-pcm.c:1309: Getting BlueALSA PCM: PLAYBACK 74:65:0C:8E:5B:9D sco
D: ../../../src/asound/bluealsa-pcm.c:1062: /org/bluealsa/hci0/dev_74_65_0C_8E_5B_9D/hfpag/sink: Setting constraints
Playing WAVE 'example.wav' : Signed 16 bit Little Endian, Rate 44100 Hz, Stereo
D: ../../../src/asound/bluealsa-pcm.c:532: /org/bluealsa/hci0/dev_74_65_0C_8E_5B_9D/hfpag/sink: Initializing HW
D: ../../../src/asound/bluealsa-pcm.c:567: /org/bluealsa/hci0/dev_74_65_0C_8E_5B_9D/hfpag/sink: FIFO buffer size: 2048 frames
D: ../../../src/asound/bluealsa-pcm.c:573: /org/bluealsa/hci0/dev_74_65_0C_8E_5B_9D/hfpag/sink: Selected HW buffer: 4 periods x 2000 bytes == 8000 bytes
D: ../../../src/asound/bluealsa-pcm.c:600: /org/bluealsa/hci0/dev_74_65_0C_8E_5B_9D/hfpag/sink: Initializing SW
D: ../../../src/asound/bluealsa-pcm.c:600: /org/bluealsa/hci0/dev_74_65_0C_8E_5B_9D/hfpag/sink: Initializing SW
D: ../../../src/asound/bluealsa-pcm.c:600: /org/bluealsa/hci0/dev_74_65_0C_8E_5B_9D/hfpag/sink: Initializing SW
D: ../../../src/asound/bluealsa-pcm.c:600: /org/bluealsa/hci0/dev_74_65_0C_8E_5B_9D/hfpag/sink: Initializing SW
D: ../../../src/asound/bluealsa-pcm.c:639: /org/bluealsa/hci0/dev_74_65_0C_8E_5B_9D/hfpag/sink: Prepared
D: ../../../src/asound/bluealsa-pcm.c:600: /org/bluealsa/hci0/dev_74_65_0C_8E_5B_9D/hfpag/sink: Initializing SW
D: ../../../src/asound/bluealsa-pcm.c:356: /org/bluealsa/hci0/dev_74_65_0C_8E_5B_9D/hfpag/sink: Starting
D: ../../../src/asound/bluealsa-pcm.c:226: /org/bluealsa/hci0/dev_74_65_0C_8E_5B_9D/hfpag/sink: Starting IO loop: 8
```

## Hardware config:

XPS 15 7590
Pop_OS! 20.04 Focal
Airpods Pro 

**bluetoothctl info 74:65:0C:8E:5B:9D**   

```
Device 74:65:0C:8E:5B:9D (public)
	Name: milfpods
	Alias: milfpods
	Class: 0x00240418
	Icon: audio-card
	Paired: yes
	Trusted: yes
	Blocked: no
	Connected: yes
	LegacyPairing: no
	UUID: Service Discovery Serve.. (00001000-0000-1000-8000-00805f9b34fb)
	UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
	UUID: Handsfree                 (0000111e-0000-1000-8000-00805f9b34fb)
	UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
	UUID: Vendor specific           (74ec2172-0bad-4d01-8f77-997b2be0722a)
	Modalias: bluetooth:v004Cp200Ed88A5

```

**cat ~/.asoundrc**

```
pcm.!default {
    type plug
    slave { pcm "jack" }
}

pcm.jack {
    type jack
[... 131 lines cut ...]
bluealsa: D: ../../src/ba-transport.c:297: Created BT socket duplicate: [22]: 24
bluealsa: D: ../../src/sco.c:232: IO loop: START: sco_cvsd_enc_thread: HFP Audio Gateway (CVSD)
bluealsa: D: ../../src/ba-transport.c:1487: Created new IO thread [ba-sco-dec]: HFP Audio Gateway (CVSD)
bluealsa: D: ../../src/sco.c:303: IO loop: START: sco_cvsd_dec_thread: HFP Audio Gateway (CVSD)
bluealsa: D: ../../src/ba-transport.c:1402: PCM resumed: 8
bluealsa: D: ../../src/ba-transport.c:1434: PCM dropped: 8
bluealsa: D: ../../src/ba-transport.c:1449: Closing PCM: 8
bluealsa: D: ../../src/ba-transport.c:441: PCM clients check keep-alive: 0 ms
bluealsa: D: ../../src/ba-transport.c:390: Stopping transport: No PCM clients
bluealsa: D: ../../src/ba-transport.c:309: Closing BT socket duplicate [22]: 23
bluealsa: D: ../../src/ba-transport.c:730: Releasing SCO link: 22
bluealsa: D: ../../src/ba-transport.c:1512: Exiting IO thread [ba-sco-enc]: HFP Audio Gateway (CVSD)
bluealsa: D: ../../src/ba-transport.c:309: Closing BT socket duplicate [-1]: 24
bluealsa: D: ../../src/ba-transport.c:1512: Exiting IO thread [ba-sco-dec]: HFP Audio Gateway (CVSD)
bluealsa: D: ../../src/bluez.c:1049: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
bluealsa: D: ../../src/bluez.c:1049: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
bluealsa: D: ../../src/bluez.c:1156: Signal: org.freedesktop.DBus.ObjectManager.InterfacesRemoved()
bluealsa: D: ../../src/bluez.c:1156: Signal: org.freedesktop.DBus.ObjectManager.InterfacesRemoved()
bluealsa: D: ../../src/bluez.c:1156: Signal: org.freedesktop.DBus.ObjectManager.InterfacesRemoved()
bluealsa: D: ../../src/bluez.c:1156: Signal: org.freedesktop.DBus.ObjectManager.InterfacesRemoved()
bluealsa: D: ../../src/bluez.c:1156: Signal: org.freedesktop.DBus.ObjectManager.InterfacesRemoved()
bluealsa: D: ../../src/bluez.c:1049: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
bluealsa: D: ../../src/bluez.c:1049: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()

```

borine, 2022-04-13:
I suspect this is an issue with the Bluez cache. I've seen similar issues when I changed the codec(s) used by `bluealsa` **after** I've already paired the headphones. This *may* be fixed in very recent `bluez` code (see for example https://github.com/bluez/bluez/issues/313).

In the meantime, you could try manually deleting the cache; however this is not an officially supported procedure for Bluez!. I fixed my similar issue this way. Suppose your host BT adapter is AA:AA:AA:AA:AA:AA, and your remote headphone device is DD:DD:DD:DD:DD:DD. You need to unpair your headphones first:
`bluetoothctl remove DD:DD:DD:DD:DD:DD`

This is is how I cleared its cache:
```
# start a root shell
sudo -s

# stop bluealsa
systemctl stop bluealsa

# make absolutely sure bluetoothd is not running:
systemctl mask bluetooth
systemctl stop bluetooth

# navigate to the bluetooth cache
cd /var/lib/bluetooth/AA:AA:AA:AA:AA:AA/cache

# remove the cache file for your headphones
rm DD:DD:DD:DD:DD:DD

# restart bluetoothd
systemctl unmask bluetooth
systemctl start bluetooth

# restart bluealsa
systemctl start bluealsa

# close the root shell
exit
```
Now re-pair the headphones and hopefully they will now attempt to connect the correct codec.

borine, 2022-07-30:
I believe Bluez has now fixed this bug in bluez release 5.65

borine, 2022-11-09:
I've been setting up a new device with BlueALSA, and got the symptoms reported here, but now clearing the Bluez cache did not help. It turns out that it is now necessary for both devices to trust each other for A2DP connections to succeed. So I fixed it using bluetoothctl telling Bluez to trust the remote device. Now connecting creates transports for both A2DP and HFP.

borine, 2023-02-16:
Seems that the symptoms reported here are all actually faults in Bluez. PopOS! 200.04 uses bluez 5.53 which is known to have problems with pairing and conneceting when using the NoInputNoOutput capablilty, and also has the cache management bug. The NoInputNoOutput issue appears to have been fixed in Bluez 5.61 (it is certainly OK in Bluez 5.64) and the cache problem is fixed in Bluez 5.65.

borine, 2023-03-24:
As this issue is believed to be caused by Bluez, not BlueALSA, and there have been no further comments to suggest otherwise for nearly a year, I am closing now.

closed 2023-03-24 by borine