arkq/bluez-alsa #559: Does not compile

IT-Spezialist-des-Asozialen, 2022-05-10:
```
user@user-Latitude-E5550:~/Programme/Carla usw./Bluezalsa/bluez-alsa$ autoreconf --install

aclocal: warning: couldn't open directory 'm4': No such file or directory
libtoolize: putting auxiliary files in '.'.
libtoolize: copying file './ltmain.sh'
libtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'm4'.
libtoolize: copying file 'm4/libtool.m4'
libtoolize: copying file 'm4/ltoptions.m4'
libtoolize: copying file 'm4/ltsugar.m4'
libtoolize: copying file 'm4/ltversion.m4'
libtoolize: copying file 'm4/lt~obsolete.m4'
configure.ac:19: installing './ar-lib'
configure.ac:14: installing './compile'
configure.ac:21: installing './config.guess'
configure.ac:21: installing './config.sub'
configure.ac:8: installing './install-sh'
configure.ac:8: installing './missing'
src/Makefile.am: installing './depcomp'
parallel-tests: installing './test-driver'
user@user-Latitude-E5550:~/Programme/Carla usw./Bluezalsa/bluez-alsa$ mkdir build && cd build
user@user-Latitude-E5550:~/Programme/Carla usw./Bluezalsa/bluez-alsa/build$ ../configure --enable-aac --enable-ofono --enable-debug
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether make supports the include directive... yes (GNU style)
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking dependency style of gcc... gcc3
checking how to run the C preprocessor... gcc -E
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking minix/config.h usability... no
checking minix/config.h presence... no
checking for minix/config.h... no
checking whether it is safe to define __EXTENSIONS__... yes
checking for gcc option to accept ISO C99... none needed
checking whether ln -s works... yes
checking for ar... ar
checking the archiver (ar) interface... ar
configure: WARNING: Libtool does not cope well with whitespace in `pwd`
checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-pc-linux-gnu
checking how to print strings... printf
checking for a sed that does not truncate output... /usr/bin/sed
checking for fgrep... /usr/bin/grep -F
checking for ld used by gcc... /usr/bin/ld
checking if the linker (/usr/bin/ld) is GNU ld... yes
checking for BSD- or MS-compatible name lister (nm)... /usr/bin/nm -B
checking the name lister (/usr/bin/nm -B) interface... BSD nm
checking the maximum length of command line arguments... 1572864
checking how to convert x86_64-pc-linux-gnu file names to x86_64-pc-linux-gnu format... func_convert_file_noop
checking how to convert x86_64-pc-linux-gnu file names to toolchain format... func_convert_file_noop
checking for /usr/bin/ld option to reload object files... -r
checking for objdump... objdump
checking how to recognize dependent libraries... pass_all
checking for dlltool... no
checking how to associate runtime and link libraries... printf %s\n
checking for archiver @FILE support... @
checking for strip... strip
checking for ranlib... ranlib
checking command to parse /usr/bin/nm -B output from gcc object... ok
checking for sysroot... no
checking for a working dd... /usr/bin/dd
checking how to truncate binary pipes... /usr/bin/dd bs=4096 count=1
checking for mt... mt
checking if mt is a manifest tool... no
checking for dlfcn.h... yes
checking for objdir... .libs
checking if gcc supports -fno-rtti -fno-exceptions... no
checking for gcc option to produce PIC... -fPIC -DPIC
checking if gcc PIC flag -fPIC -DPIC works... yes
checking if gcc static flag -static works... yes
checking if gcc supports -c -o file.o... yes
checking if gcc supports -c -o file.o... (cached) yes
checking whether the gcc linker (/usr/bin/ld -m elf_x86_64) supports shared libraries... yes
checking whether -lc should be explicitly linked in... no
checking dynamic linker characteristics... GNU/Linux ld.so
checking how to hardcode library paths into programs... immediate
checking whether stripping libraries is possible... yes
checking if libtool supports shared libraries... yes
checking whether to build shared libraries... yes
checking whether to build static libraries... yes
checking pkg-config m4 macros... yes
checking for backtrace in -lSegFault... yes
checking for pkg-config... /usr/bin/pkg-config
checking pkg-config is at least version 0.9.0... yes
checking execinfo.h usability... yes
checking execinfo.h presence... yes
checking for execinfo.h... yes
checking stdatomic.h usability... yes
checking stdatomic.h presence... yes
checking for stdatomic.h... yes
checking for eventfd... yes
checking for pipe2... yes
checking for splice... yes
checking for library containing clock_gettime... none required
checking for library containing pow... -lm
checking for library containing pthread_create... -lpthread
checking for ALSA... no
configure: error: Package requirements (alsa) were not met:

No package 'alsa' found

Consider adjusting the PKG_CONFIG_PATH environment variable if you
installed software in a non-standard prefix.

Alternatively, you may set the environment variables ALSA_CFLAGS
and ALSA_LIBS to avoid the need to call pkg-config.
See the pkg-config man page for more details.
```
Then I tried to install the alsa packages (I looked up the names from https://www.alsa-project.org/wiki/Download and tried to install it with apt.
```
user@user-Latitude-E5550:~/Programme/Carla usw./Bluezalsa/bluez-alsa/build$  apt list --installed | grep alsa

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

alsa-base/focal,focal,now 1.0.25+dfsg-0ubuntu5 all [installed]
alsa-firmware-loaders/focal,now 1.1.7-1ubuntu1 amd64 [installed,automatic]
alsa-firmware/bionic,now 1.0.20-1kxstudio1 amd64 [installed]
alsa-oss/focal,now 1.1.8-1 amd64 [installed]
alsa-tools/focal,now 1.1.7-1ubuntu1 amd64 [installed]
alsa-topology-conf/focal,focal,now 1.2.2-1 all [installed]
alsa-ucm-conf/focal-updates,focal-updates,now 1.2.2-1ubuntu0.13 all [installed]
alsa-utils/focal-updates,now 1.2.2-1ubuntu2.1 amd64 [installed]
gstreamer1.0-alsa/focal-updates,focal-security,now 1.16.2-4ubuntu0.1 amd64 [installed]
libsox-fmt-alsa/focal,now 14.4.2+git20190427-2 amd64 [installed,automatic]
user@user-Latitude-E5550:~/Programme/Carla usw./Bluezalsa/bluez-alsa/build$  sudo apt install alsa-driver alsa-lib alsa-plugins pyalsa
[sudo] password for user:         
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package alsa-driver
E: Unable to locate package alsa-lib
E: Unable to locate package alsa-plugins
E: Unable to locate package pyalsa
```
OS: Linux Mint
What can I do?

borine, 2022-05-11:
> What can I do?

Read the wiki article: https://github.com/Arkq/bluez-alsa/wiki/Installation-from-source

As Mint is derived from Ubuntu, I would guess that most of the package names will be the same as for Ubuntu/Debian

IT-Spezialist-des-Asozialen, 2022-05-16:
Thanks, I've installed it know
But I've some other problems now. Should I aks it right here or open a new issue?

Arkq, 2022-05-16:
If it's related to this issue, ask here :)

IT-Spezialist-des-Asozialen, 2022-05-17:
Ok, then I'll open a new one

closed 2022-05-17 by IT-Spezialist-des-Asozialen