arkq/bluez-alsa #560: Bluez-Alsa doen't work on Mint

IT-Spezialist-des-Asozialen, 2022-05-17:
Hello,
I've installed Bluez-Alsa, but it seems that it doesn't work. Always if I try to run it, it appears that:
```
user@user-Latitude-E5550:~$ zita-a2j -j bt_headphones -d bluealsa:C9:F0:F4:F3:FD:E5 -p 4096 -n 3 -c 2 -L
ALSA lib ../../../src/asound/bluealsa-pcm.c:1313:(_snd_pcm_bluealsa_open) Couldn't get BlueALSA PCM: The name org.bluealsa was not provided by any .service files
Can't open ALSA capture device 'bluealsa:C9:F0:F4:F3:FD:E5'.
```
I thought the reason could be the way I've installed Bluez-Alsa, so I've removed everything and used all possibilites from [here](https://github.com/Arkq/bluez-alsa/wiki/Installation-from-source#installation-options).

In the .asoundrc.txt file I've already added it:
```
pcm.!default {
    type plug
    slave { pcm "pulse" }
}

pcm.pulse {
    type pulse
}

ctl.mixer0 {
    type hw
    card 0
}

pcm.bt-headphones {
	type bluealsa
	device C9:F0:F4:F3:FD:E5
	profile a2dp
}
ctl.bt-headphones {
	type bluealsa
}
```
Maybe you've some other idea.
Many thanks in advance.

Arkq, 2022-05-17:
> The name org.bluealsa was not provided by any .service files

BlueALSA requires system service for managing PCM streams. If your system uses systemd, please refer to this wiki: https://github.com/Arkq/bluez-alsa/wiki/Systemd-integration Otherwise, you will have to run `bluealsa` service using upstart/init.d or whatever Mint is using. Alternatively, you can start `bluealsa` manually.

Also see this:
https://github.com/Arkq/bluez-alsa/blob/master/doc/bluealsa.8.rst#example

IT-Spezialist-des-Asozialen, 2022-05-20:
```
user@user-Latitude-E5550:~$ sudo systemctl start bluealsa.service
Job for bluealsa.service failed because the control process exited with error code.
See "systemctl status bluealsa.service" and "journalctl -xe" for details.
user@user-Latitude-E5550:~$ systemctl status bluealsa.service
● bluealsa.service - Bluealsa daemon
     Loaded: loaded (/lib/systemd/system/bluealsa.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Fri 2022-05-20 17:11:11 CEST; 46s ago
       Docs: https://github.com/Arkq/bluez-alsa/
    Process: 63078 ExecStart=/usr/bin/bluealsa $OPTIONS (code=exited, status=1/FAILURE)
   Main PID: 63078 (code=exited, status=1/FAILURE)

Mai 20 17:11:11 user-Latitude-E5550 systemd[1]: bluealsa.service: Scheduled restart job, restart counter is at 5.
Mai 20 17:11:11 user-Latitude-E5550 systemd[1]: Stopped Bluealsa daemon.
Mai 20 17:11:11 user-Latitude-E5550 systemd[1]: bluealsa.service: Start request repeated too quickly.
Mai 20 17:11:11 user-Latitude-E5550 systemd[1]: bluealsa.service: Failed with result 'exit-code'.
Mai 20 17:11:11 user-Latitude-E5550 systemd[1]: Failed to start Bluealsa daemon.
```
Why does this appear? Why can't it start?

borine, 2022-05-20:
```
sudo journalctl -xeu bluealsa.service
```
will print out the error messages from the bluealsa daemon

IT-Spezialist-des-Asozialen, 2022-05-20:
```
user@user-Latitude-E5550:~$ sudo journalctl -xeu bluealsa.service
-- 
-- The job identifier is 80154.
Mai 20 17:13:16 user-Latitude-E5550 bluealsa[63406]: /usr/bin/bluealsa: E: It is required to enabled at least one BT profile
Mai 20 17:13:16 user-Latitude-E5550 bluealsa[63406]: Try '/usr/bin/bluealsa --help' for more information.
Mai 20 17:13:16 user-Latitude-E5550 systemd[1]: bluealsa.service: Main process exited, code=exited, status=1/FAILURE
-- Subject: Unit process exited
-- Defined-By: systemd
-- Support: http://www.ubuntu.com/support
-- 
-- An ExecStart= process belonging to unit bluealsa.service has exited.
-- 
-- The process' exit code is 'exited' and its exit status is 1.
Mai 20 17:13:16 user-Latitude-E5550 systemd[1]: bluealsa.service: Failed with result 'exit-code'.
-- Subject: Unit failed
-- Defined-By: systemd
-- Support: http://www.ubuntu.com/support
-- 
-- The unit bluealsa.service has entered the 'failed' state with result 'exit-code'.
Mai 20 17:13:16 user-Latitude-E5550 systemd[1]: Failed to start Bluealsa daemon.
-- Subject: A start job for unit bluealsa.service has failed
-- Defined-By: systemd
-- Support: http://www.ubuntu.com/support
-- 
-- A start job for unit bluealsa.service has finished with a failure.
-- 
-- The job identifier is 80154 and the job result is failed.
Mai 20 17:13:16 user-Latitude-E5550 systemd[1]: bluealsa.service: Scheduled restart job, restart counter is at 5.
-- Subject: Automatic restarting of a unit has been scheduled
-- Defined-By: systemd
-- Support: http://www.ubuntu.com/support
-- 
-- Automatic restarting of the unit bluealsa.service has been scheduled, as the result for
-- the configured Restart= setting for the unit.
Mai 20 17:13:16 user-Latitude-E5550 systemd[1]: Stopped Bluealsa daemon.
-- Subject: A stop job for unit bluealsa.service has finished
-- Defined-By: systemd
-- Support: http://www.ubuntu.com/support
-- 
-- A stop job for unit bluealsa.service has finished.
-- 
-- The job identifier is 80241 and the job result is done.
Mai 20 17:13:16 user-Latitude-E5550 systemd[1]: bluealsa.service: Start request repeated too quickly.
Mai 20 17:13:16 user-Latitude-E5550 systemd[1]: bluealsa.service: Failed with result 'exit-code'.
-- Subject: Unit failed
-- Defined-By: systemd
-- Support: http://www.ubuntu.com/support
-- 
-- The unit bluealsa.service has entered the 'failed' state with result 'exit-code'.
Mai 20 17:13:16 user-Latitude-E5550 systemd[1]: Failed to start Bluealsa daemon.
-- Subject: A start job for unit bluealsa.service has failed
-- Defined-By: systemd
-- Support: http://www.ubuntu.com/support
-- 
-- A start job for unit bluealsa.service has finished with a failure.
-- 
-- The job identifier is 80241 and the job result is failed.
lines 1697-1752/1752 (END)
```

borine, 2022-05-20:
OK, so the wiki systemd guide is out-of-date; with the latest bluealsa sources it is necessary to specify at least one profile on the command line. See the manual page for more info: https://github.com/Arkq/bluez-alsa/blob/master/doc/bluealsa.8.rst

I recommend starting bluealsa manually from the command line first to experiment with different arguments until you have it working correctly, then apply the command line you have used to the systemd config.

IT-Spezialist-des-Asozialen, 2022-05-20:
```
user@user-Latitude-E5550:~$ bluealsa -p hsp-hs -B C9:F0:F4:F3:FD:E5

(process:81802): GLib-GIO-CRITICAL **: 20:00:45.447: g_bus_own_name_on_connection: assertion 'g_dbus_is_name (name) && !g_dbus_is_unique_name (name)' failed
^C
user@user-Latitude-E5550:~$ bluealsa -p a2dp-sink -B C9:F0:F4:F3:FD:E5

(process:82043): GLib-GIO-CRITICAL **: 20:01:41.530: g_bus_own_name_on_connection: assertion 'g_dbus_is_name (name) && !g_dbus_is_unique_name (name)' failed
```
In the next line the cursor twinkles but it doesn't connect, even if I click the connect button in the bluetooth menu.
So I've interrupt it with Strg+C and try again with another BT profile but it doen't work again.

borine, 2022-05-20:
omit the `-B C9:F0:F4:F3:FD:E5` option - it has nothing to do with bluetooth addresses, and is not needed except in special cases (see the EXAMPLE section of the manual page if you wish to know more about this option). If you can give a brief description of what you want bluealsa to do on your system, I may be able to help with choosing an appropriate configuration.

@Arkq - the `GLib-GIO-CRITICAL ` message is not nice, and not helpful for the user, perhaps bluealsa should sanitize this option before passing to glib ?

IT-Spezialist-des-Asozialen, 2022-05-20:
I want to use [Carla](https://kx.studio/Applications) after I've installed that and activated the bridges I've released that my bluetooth boxes and my bluetooth headset don't work with that.
In another forum somebody suggests me bluez-alsa.

IT-Spezialist-des-Asozialen, 2022-05-20:
If I didn't use the `-B C9:F0:F4:F3:FD:E5` option, an error appears.
```
user@user-Latitude-E5550:~$ bluealsa -p a2dp-sink
bluealsa: E: Couldn't acquire D-Bus name. Please check D-Bus configuration. Requested name: org.bluealsa
```

borine, 2022-05-20:
That error is because your user does not have permission in D-Bus to acquire the name "org.bluealsa". Run it as root:
```
sudo bluealsa -p a2dp-sink
```

IT-Spezialist-des-Asozialen, 2022-05-20:
Nothing changes if I run as root.

borine, 2022-05-20:
OK, looks like your installation is broken - you appear to be missing at least one critical file (`/etc/dbus-1/system.d/bluealsa.conf`). I am out of time on this for today, will look at it again soon.

IT-Spezialist-des-Asozialen, 2022-05-20:
I've the file you mentioned:
```
<!-- This configuration file specifies the required security policies
     for BlueALSA core daemon to work. -->

<!DOCTYPE busconfig PUBLIC "-//freedesktop//DTD D-BUS Bus Configuration 1.0//EN"
 "http://www.freedesktop.org/standards/dbus/1.0/busconfig.dtd">
<busconfig>

  <!-- ../system.conf have denied everything, so we just punch some holes -->

  <policy user="root">
    <allow own_prefix="org.bluealsa"/>
    <allow send_destination="org.bluealsa"/>
  </policy>

  <policy group="audio">
    <allow send_destination="org.bluealsa"/>
  </policy>

</busconfig>
```
Hear from you soon, thanks for your help until this time.

borine, 2022-05-21:
> Nothing changes if I run it as root.

Please can you post the output from `sudo bliuealsa -p a2dp-sink`

IT-Spezialist-des-Asozialen, 2022-05-21:
In the next line the cursor twinkles and nothing else happens…

borine, 2022-05-21:
That's good !! bluealsa is running - in another terminal type
```
ps -ef | grep bluealsa
```
you should see it running

IT-Spezialist-des-Asozialen, 2022-05-21:
```
user@user-Latitude-E5550:~$ ps -ef | grep bluealsa
root      128824  128791  0 16:46 pts/0    00:00:00 sudo bluealsa -p a2dp-sink
root      128825  128824  0 16:46 pts/0    00:00:00 bluealsa -p a2dp-sink
user   129639  129633  0 16:58 pts/1    00:00:00 grep bluealsa
```

IT-Spezialist-des-Asozialen, 2022-05-21:
And how can I connect then my Bluetooth device?

borine, 2022-05-21:
Well, now we go back to a previous question - does your bluetooth device act as a A2DP source (ie A2DP microphone)? Right now that is the only profile that is available. Also, are you aiming to achieve music capture/playback or just voice?

IT-Spezialist-des-Asozialen, 2022-05-21:
The Blueooth device I want to connect first is only an output device.
But I've a Bluetooth headset too what I want to connect later.

Arkq, 2022-05-21:
I would like to ask some other question. Do you really want to use BlueALSA? If you are using Mint Linux distro, probably you are using it as a desktop, so.... maybe you should choose PulseAudio or PipeWire as a viable solution for Blueooth audio? BlueALSA is not a "plug-and-play" solution so to speak. In simple cases it should work out of the box, but if you want something more elaborate, the setup might become very complicated (in terms of ALSA setup) :D

> And how can I connect then my Bluetooth device?

Use Bluetooth manager provided by your Linux distribution (e.g. in Gnome there is a Bluetooth setup tab, you can use it for discovering and connecting BT devices), or `bluetoothctl` for command line experience. When the device connects, you should see it in the output of `bluealsa-cli list-pcms`.

IT-Spezialist-des-Asozialen, 2022-05-21:
As I've already wrote [here](https://github.com/Arkq/bluez-alsa/issues/560#issuecomment-1133216683) it was the only way I knew to get Carla managed with Bluetooth devices.

borine, 2022-05-21:
> it was the only way I knew to get Carla managed with Bluetooth devices.

Actually, you/we don't know that (yet). If Carla does not work with PulseAudio bluetooth, it is likely that it will not work with bluealsa either. Once you remove pulseaudio from the picture, you then have to do all the ALSA configuration yourself, you have to workaround the problem of many desktop applications being configured specifically to use PulseAudio and therefore unable to use ALSA directly without reconfiguration/recompiling, and the possibililty that you are the first person ever to use the combination of Carla, Bluealsa and Mint. I believe that Bluealsa will work as well on Mint as it does on Ubuntu, but just like with Ubuntu you will compromise much of the Desktop by trying to force applications to use ALSA when they were configured to use PulseAudio or Pipewire.

Bluealsa is really an excellent bluetooth solution for embedded devices, dedicated audio servers, or Kiosk-type systems where the desktop is essentially a single application. For integrated desktop systems you really do need an audio server such as pulseaudio or pipewire, or alternatively Jack if you are building a professional music creation system. Both pulseaudio and pipewire have their own internal bluetooth implementation, so there is no need for bluealsa with them. Jack can be used with bluealsa. But since it can also be used with pulseaudio it does seem more logical to troubleshoot the pulseaudio problems you have experienced rather than add another layer of complexity by adding Bluealsa on top of all that. However, if you are convinced you need to do this, then I will try to answer any bluealsa-specific questions as best I can - I am intrigued to know how this works out !

borine, 2022-05-23:
Hmm, humble pie! I just read https://jackaudio.org/faq/pulseaudio_and_jack.html, and so now I realize that Jack and PulseAudio together is not a good option at all. So you are right, BlueALSA is possibly the only way to have Jack route to a bluetooth audio device; but beware the limitations as noted in https://github.com/Arkq/bluez-alsa/wiki/Using-BlueALSA-with-the-JACK-Audio-Connection-Kit

IT-Spezialist-des-Asozialen, 2022-05-27:
@borine You've aks me something: https://github.com/Arkq/bluez-alsa/issues/560#issuecomment-1133649766
I've already answered:
https://github.com/Arkq/bluez-alsa/issues/560#issuecomment-1133749391

So with this device it doesn't matter which profile I use?

borine, 2022-05-27:
> So with this device it doesn't matter which profile I use?

Well, for music quality audio you need A2DP profile, and I guess that is fine for your output only device. SCO profiles (HFP, HSP) are voice quality only, and most headsets support only SCO for the microphone but both SCO and A2DP for the speakers. However, I've never seen a headset that supports A2DP and SCO at the same time: normally when a A2DP stream is playing the microphone is disabled. So if you want your system  to be able to both play music to a bluetooth device, *and* to support 2-way voice comms, then you will need  `-p a2dp-source` and at least one of `-p hfp-ag` or `-p hsp-ag` profiles, possibly both.

I also recommend that you carefully choose the A2DP codecs you want to use, because of a bug in Bluez which sometimes makes it difficult to change the codecs later (see #554). Explicitly enable each codec with the '-c' option, eg: `-c aac -c aptx -c aptx-hd` etc.

christian179717, 2022-05-28:
> Explicitly enable each codec with the '-c' option, eg: -c aac -c aptx -c aptx-hd etc.

This I've to do while installing?

borine, 2022-05-28:
Sorry, I wasn't very clear: I mean when starting the bluealsa daemon at runtime. So for example in my case I use only SBC and aptX codecs with A2DP profile, and CVSD and mSBC codecs with HFP profile. So my bluealsa command line is:
```
sudo bluealsa -p a2dp-source -p hfp-ag -p hsp-ag -c sbc -c aptx -c cvsd -c msbc
```
I think that it is not essential to mention sbc or cvsd as they are included by default anyway, but I like to add them on my systems for clarity.

christian179717, 2022-05-28:
```
sudo bluealsa -p a2dp-source -p hfp-ag -p hsp-ag -c sbc -c aptx -c cvsd -c msbc
bluealsa: E: Invalid BT codec name: aptx
```

borine, 2022-05-28:
If you type
```
bluealsa --help
```
that will tell you the codecs that are compiled into your daemon. I get:
```
...
Available BT audio codecs:
  a2dp-source:	SBC, AAC, aptX, FastStream
  a2dp-sink:	SBC, AAC, aptX, FastStream
  hfp-*:	CVSD, mSBC
```

If yours does not mention aptX then I guess you did not include that in your configure options when building it.

christian179717, 2022-05-28:
```
Available BT profiles:
  - a2dp-source	Advanced Audio Source (v1.3)
  - a2dp-sink	Advanced Audio Sink (v1.3)
  - hfp-ag	Hands-Free Audio Gateway (v1.7)
  - hfp-hf	Hands-Free (v1.7)
  - hsp-ag	Headset Audio Gateway (v1.2)
  - hsp-hs	Headset (v1.2)

Available BT audio codecs:
  a2dp-source:	SBC
  a2dp-sink:	SBC
  hfp-*:	CVSD
```
So I've to install it with including it?
Can I easy delete the directory and start from zero or do I have to mind something?

borine, 2022-05-28:
The guidance for adding optional codecs is all included in the wiki page https://github.com/Arkq/bluez-alsa/wiki/Installation-from-source

christian179717, 2022-05-28:
I know, but I've already installed. So I don't know if this is possible afterwards…

borine, 2022-05-28:
bluez-alsa is just like any other autotools based project. You run `configure` with the options you want in the build, then `make`, then `make install`. If you later change your mind about the build options, you run `make clean`, run `configure` with the options you want this time, then `make` and `make install`. Autotools has been like this for 25 years.

I can't give general advice on using `configure` or `make` - that would be well beyond the scope of this project.

christian179717, 2022-05-28:
Ok, thanks

christian179717, 2022-05-28:
What would do you suggest me for `configure`? Resp. what do I need all?

borine, 2022-05-28:
> What would do you suggest me for configure

Look for the codecs supported by your bluetooth devices, and pick those. My only real experience is with apt-X and SBC for A2DP, I've done some brief experiments with a few others, but not enough to be confident for giving advice.

Since SBC is mandatory for all A2DP devices, perhaps it would be simpler for you to just start with the build you already have and get your devices paired, connected, and streaming correctly with SBC before looking at re-building with other codecs. Perhaps I was over-enthusiastic to suggest getting all codecs working before you even have the basics in place. So start bluealsa as:
```
sudo bluealsa -p a2dp-source -p hfp-ag -p hsp-ag
```
Then look at pairing and connecting your output device. I never use GUI bluetooth tools, so I'm not familiar with the Mint interface for this; I always use bluetoothctl from the command line. The problem I have with GUIs is that they tend to confuse and combine the pairing, trusting, and connecting operations which makes it hard to diagnose and correct any errors. But the upside, I guess, is that if they "just work" then that's one less headache to deal with.

christian179717, 2022-05-29:
I've connected my Bluetooth device, but Carla doens't work. I've aks already there and will then proceed here for the bluetooth profiles.
Now already many thanks

christian179717, 2022-06-05:
Bluez-Alsa doesn't create any sink. Do you know why? Is this because of Bluez-Alsa or because of something else?

christian179717, 2022-06-07:
```
bluealsa:SRV=org.bluealsa,DEV=C9:F0:F4:F3:FD:E5,PROFILE=a2dp
    BT-200, trusted audio-card, playback
    A2DP (SBC): S16_LE 2 channels 48000 Hz
```

borine, 2022-06-07:
OK, so you *do* have a sink, it is `bluealsa:DEV=C9:F0:F4:F3:FD:E5,PROFILE=a2dp` You can test that with any alsa application, eg:
```
aplay -D bluealsa:DEV=C9:F0:F4:F3:FD:E5,PROFILE=a2dp /usr/share/sounds/alsa/Front_Center.wav
```
What are you trying to with that sink? At the start of this issue you mentioned zita-a2j - that is for jack sources. To create a jack sink you need zita-j2a, for example something like:
```
zita-j2a -j BT-200 -d bluealsa:DEV=C9:F0:F4:F3:FD:E5,PROFILE=a2dp -p 1024 -n 3 -c 2 -L
```
See the wiki page https://github.com/Arkq/bluez-alsa/wiki/Using-BlueALSA-with-the-JACK-Audio-Connection-Kit for more info.

christian179717, 2022-06-07:
Aaah, ok thanks

christian179717, 2022-06-09:
So the bluetoothboxes work now.
Which profiles would you improve me to install for the headset?

borine, 2022-06-11:
For profiles:
If you are only interested in music quality audio, then use `-p a2dp-source -p a2dp-sink` which will allow bluealsa to both send and receive a2dp audio. If you also require voice-quality audio, then you should probably choose either gateway role (to communicate with headphones and speakers) *or* handsfree/headset role (to communicate with mobile phones etc). It may be possible to have both roles enabled at the same time, but I have never tried it and it may well confuse some devices. For gateway role use `-p hsp-ag -p hfp-ag` (in addition to the a2dp options), for handsfree/headset role use `-p hsp-hs -p hfp-hf`.

Note that many HFP devices require support for telephony control commands which are not implemented by bluealsa, so you may have variable success with them.

For codecs:
You need to consider the devices you will be connecting to. Look at their documentation to see which codecs they support. You will need to enable them individually in the configuration step when building bluealsa *and* enable them individually on the command line when running bluealsa.

christian179717, 2022-06-11:
Where is the difference between music and voice quality?

borine, 2022-06-11:
A2DP emphasizes audio quality over low latency and is therefore best suited to music listening. A typical A2DP codec uses a sample rate of 48kHz 16-bit stereo. HFP/HSP emphasize low latency and synchronicity over quality and are therefore best suited to voice (ie telephone conversations). Depending on the codec, you get either 8kHz or 16kHz 16-bit mono. Think of your mobile phone. When you stream Spotify to your headphones, the phone uses A2DP; when you make a phone call using your headphones the phone uses HFP (or HSP if HFP is not available)

borine, 2022-09-15:
@christian179717 this issue has been idle for a long time now. Can this issue be closed?

christian179717, 2022-09-15:
Yes, i'll do it

closed 2022-09-15 by borine