arkq/bluez-alsa #569: a2dp volume with softvol alsa plugin

mczerski, 2022-06-13:
Currently it is not possible to control softvol alsa plugin with aplay because aplay uses snd_mixer_selem_set_playback_dB_all and snd_mixer_selem_get_playback_dB functions which require mixer control to accept values in decibels and that is not the case for softvol plugin. Also softvol plugin does not have mute capability. To control volume snd_mixer_selem_get_playback_volume and snd_mixer_selem_set_playback_volume_all  function may be used which works with softvolume. Also treating error while getting/setting mute as not an error would allow aplay to work nicely with softvol plugin.

borine, 2022-06-14:
Please can you clarify which products you are referring to:
- is `aplay` the utility `bluealsa-aplay` (https://github.com/Arkq/bluez-alsa/blob/master/doc/bluealsa-aplay.1.rst) ?
- is `softvol` the alsa-lib plugin type (https://www.alsa-project.org/alsa-doc/alsa-lib/pcm_plugins.html#pcm_plugins_softvol) ?

Arkq, 2022-06-14:
> Decibel handling is not a capability, I think it is supposed to be implemented by all simple volume controls

I'm not sure to be hones. Decibel handling requires TLV support which was not present in early versions of control spec: https://github.com/alsa-project/alsa-lib/blob/master/include/control_external.h#L138

> @Arkq I think bluealsa-aplay should check the return from ...

Yes, error handling should be more thorough :/

mczerski, 2022-06-14:
according to amixer manual:
"
set or sset <SCONTROL> <PARAMETER> ...
    Sets the simple mixer control contents. The parameter can be the volume either as a percentage from 0% to 100% with % suffix, a dB gain with dB suffix (like -12.5dB), or an exact hardware value. **The dB gain can be used only for the mixer elements with available dB information.** When plus(+) or minus(-) letter is appended after volume value, the volume is incremented or decremented from the current value, respectively. 
"

also below is the amixer sget output of volume control for two cases softvol and hardware volume:
SOFTVOL:
```
Simple mixer control 'Master',0
  Capabilities: volume
  Playback channels: Front Left - Front Right
  Capture channels: Front Left - Front Right
  Limits: 0 - 255
  Front Left: 76 [30%]
  Front Right: 76 [30%]
```

HARDWARE:
```
Simple mixer control 'Line Out',0
  Capabilities: pvolume pvolume-joined pswitch
  Playback channels: Front Left - Front Right
  Limits: Playback 0 - 31
  Mono:
  Front Left: Playback 9 [29%] [-33.00dB] [on]
  Front Right: Playback 9 [29%] [-33.00dB] [on]
```
You can see that output for softvol does not contain volume information in db.
I don't know how amixer is related to alsa c api but I would take manual for amixer as source of truth for db "capability" of the volume control.

borine, 2022-06-14:
I was wrong about `snd_mixer_selem_get_playback_dB()`, clearly some card drivers do not provide the `tlv` data required by alsa-lib to provide dB mapping. So applications do have to check for dB range availability and fallback to raw scaling if necessary. `alsamixer` implements this here: https://github.com/alsa-project/alsa-utils/blob/a566f8a0ed6d7ca5fd6ae2d224f3f28654a2f3be/alsamixer/volume_mapping.c#L120

(note that file is *not* GPL so could legitimately be included here, or extracts used with appropriate attribution).

However, the `softvol` PCM plugin **does** implement a dB range when it creates a user control; so I guess you are not seeing dB data here because softvol is actually passing the API calls directly through to the slave card, and it is that card which is lacking tlv data.

mczerski, 2022-06-14:
That would make sense since I use softvol on top of i2s dac that does not have any volume control.

borine, 2022-06-14:
> does not have any volume control

That confuses me. If the card has *no* control, then I assume "Master" must actually be a user control and therefore should have the default dB range implemented by the softol plugin?

mczerski, 2022-06-14:
I don't know, I'm not familiar with alsa internals. Below is my asound.conf to be more precise about my setup:
```
pcm.softvol {
    type softvol;
    slave.pcm "dmix:0,0";
    [control.name](http://control.name/) "Master";
    control.card 0;
    min_dB -50.0;
    max_dB 0.0;
}
```
and aplay -l:
```
card 0: I2Smaster [I2S-master], device 0: 1c22000.i2s-pcm5102a-hifi pcm5102a-hifi-0 [1c22000.i2s-pcm5102a-hifi pcm5102a-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```
As You can see I even tried to set min_dB and max_dB but without any success.

borine, 2022-06-14:
Try this instead:
```
pcm.softvol {
	type softvol
	slave.pcm {
		type softvol
		slave.pcm "dmix:0,0"
		control {
			name "Master Playback Switch"
			card 0
		}
		resolution 2
	}
	control {
		name "Master Playback Volume"
		card 0
	}
}
```
You need to close alsamixer and any other mixer appications, then instantiate the new definition: for example
```
aplay -D softvol -t raw /dev/null
```

Then open a mixer on card 0 to show the scontrols, for example
```
amixer -c 0 scontents
```

Hopefully you will see a simple mixer with capabilites pvolume and pswitch and dB values for the channels.

If the dB values are still missing, then I suspect your card *does* have a control called "Master PCM Volume", in which case I think you will have to choose a different name instead of "Master" to force the creation of a user control.

mczerski, 2022-06-14:
It does not seem to work, there is just another control for muting but Volume control still does not have muting capability. I restarted the machine to be sure new config is used.
```
amixer -c 0 scontents
Simple mixer control 'Master',0
  Capabilities: volume
  Playback channels: Front Left - Front Right
  Capture channels: Front Left - Front Right
  Limits: 0 - 255
  Front Left: 94 [37%]
  Front Right: 94 [37%]
Simple mixer control 'Mute',0
  Capabilities: pswitch
  Playback channels: Front Left - Front Right
  Mono:
  Front Left: Playback [on]
  Front Right: Playback [on]
```

borine, 2022-06-15:
OK, so I've had a think about this, and can now reproduce your original problem. When you create a user control called "Master", the alsa-lib simple control code does not know how to deal with this and you end up with a simple control with volume scale and no dB range. This is not well documented, but a softvol pcm creates a "primitive" control, and alsa-lib tries to create a simple control from this by appyling heuristics to the primitive name. To get a simple volume control with dB range you need to append " Playback Volume" to your control name in the softvol config, so that `control.name "Master Playback Volume" would have given you dB range data.

Now we hit the problem of how to remove user controls from the config, and that is not trivial. Most Linux systems have some way of writing out the card control states to a file at shutdown, then reload them at boot. This preserves volume settings across boots. So we need to:
1 prevent the system from saving this broken config so that it does not overwrite any manual changes to the state file
2 edit (or remove) the state file
3 reboot to clear the current state from kernel memory and load the amended state file

How to do this depends on your system. If it is derived from Debian then it will be the same as my Ubuntu system; other systems may vary.

I cannot explain the appearance of a control called "Mute", and am unable to reproduce that here. But I can reproduce your original user control, and replace it with a new Master with switch and dB range, using my config above. If possible, I would like to remove your alsa state file to get a clean start after reviewing your ~/.asoundrc. That means your card will go back to its state when you first booted linux, so please only do this if you did not need to do a lot of setup on the card to get it working.

If you are happy to try it, please post your OS distribution and version, and your current ~/.asoundrc.

mczerski, 2022-06-15:
ok, i'll try that when I find some time.

mczerski, 2022-06-15:
so after:
1. setting names to "Master Playback Switch" and "Master Playback Volume"
2. removing /var/lib/alsa/asound.state
3. renaming /usr/sbin/alsactl + rebooting
I got volume control with dB range and mute capability :) Thanks !

So, remaining issue is that not all cards have mute capability thus controlling it should be optional.

Arkq, 2022-06-16:
@mczerski  please check branch https://github.com/Arkq/bluez-alsa/tree/mixer-no-switch whether it will resolve the mute issues for you.

borine, 2022-06-19:
@Arkq there is also the issue of volume control failing when there is no decibel range info. In the absence of any immediate fix for this, I think the documentation needs report to it as a limitation. Here is a draft patch:
```diff
diff --git a/doc/bluealsa-aplay.1.rst b/doc/bluealsa-aplay.1.rst
index 9541268..4f5e25e 100644
--- a/doc/bluealsa-aplay.1.rst
+++ b/doc/bluealsa-aplay.1.rst
@@ -104,12 +104,20 @@ OPTIONS
     The default is ``default``.
 
 --mixer-name=NAME
-    Set the name of the mixer element.
+    Set the name of the ALSA "simple" mixer control to use.
     The default is ``Master``.
+    To work with ``bluealsa-aplay`` this simple control must provide decibel
+    scaling information for the volume control. Most, but not all, modern sound
+    cards do provide this information. ``bluealsa-aplay`` is also unable apply
+    the mute setting if the selected simple control does not include a mute
+    switch.
 
 --mixer-index=NUM
-    Set the index of the mixer channel.
+    Set the index of the ALSA simple mixer control.
     The default is ``0``.
+    This is required only if the simple mixer control name applies to multiple
+    simple controls on the same card. This is most common with HDMI devices
+    which may have many IEC958 playback ports.
 
 --profile-a2dp
     Use A2DP profile (default).
```

Arkq, 2022-06-19:
Yes, if this issue does not require anything else than mute switch fix, I'd stay with the manual update for now. I will commit your diff.

Arkq, 2022-06-20:
I've updated the mixer-no-switch branch with doc update and slightly reworked my previous patch (@borine your comment about "is also unable apply the mute setting" has forced me into thinking more abut that, so right now mute will be possible via alsa-lib silence function).

mczerski, 2022-10-11:
I don't know, I'm not familiar with alsa internals. Below is my asound.conf
to be more precise about my setup:
```
pcm.softvol {
    type softvol;
    slave.pcm "dmix:0,0";
    control.name "Master";
    control.card 0;
    min_dB -50.0;
    max_dB 0.0;
}
```
and aplay -l:
```
card 0: I2Smaster [I2S-master], device 0: 1c22000.i2s-pcm5102a-hifi
pcm5102a-hifi-0 [1c22000.i2s-pcm5102a-hifi pcm5102a-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```
As You can see I even tried to set min_dB and max_dB but without any
success.

commit 48572d79 (2022-06-20): Do not fail if mixer does not support mute switch

Fixes #569