arkq/bluez-alsa #570: bluealsa-aplay -l lists no devices

danboid, 2022-06-14:
I'm running a Ubuntu 22.04 based install of Armbian 22.08 on a X96 Max+ TV box. I have read the bluez-alsa docs and given it my best shot at getting it configured but `bluealsa-aplay -l` doesn't list any devices so I'm not doing something right and the most likely candidate is my ALSA config as I really don't understand ALSA config files very well. They seem to have their own scripting language that I do not know.

What I'm hoping to ultimately achieve is to have bluez-alsa running alongside JACK on the same ALSA device. I want to use bluez-alsa as an a2dp sink so that I can stream audio from a phone or tablet to my TV box via BT, my TV box acting as a BT speaker and have that audio played back via JACK. The docs seem to hint this should be possible and I thought I might increase my chances of getting it configured if I keep JACK out of the equation at first and try to just get bluez-alsa working by itself. I have not currently installed JACK or Pulseaudio, only ALSA.

My current attempt at setting it up went something like this:

```
apt install bluez-alsa-utils
```

I'm using the `bluez-alsa-utils` package in the Ubuntu 22.04 repos. That should work?

Then I edited `/etc/systemd/system/bluetooth.target.wants/bluez-alsa.service` and changed the `ExecStart` line to:

```
ExecStart=/usr/bin/bluealsa -p a2dp-source -p a2dp-sink
```

My X96 Max+ TV box has an integrated sound card that is partially supported by ALSA but only HDMI audio seems to work so I have to attach a USB audio device to get a working audio jack, so I have to tell ALSA to use ALSA device 1 as the default device. Here is my obviously broken attempt at trying to configure bluez-alsa to use ALSA device 1 (not 0) in `/etc/asound.conf`:

```
defaults.pcm.card 1
defaults.ctl.card 1

pcm.!default {
	capture.pcm cards.pcm.default
	playback.pcm bluealsa
}
```

On top of that I have successfully paired a phone but when I run `bluealsa-aplay -l` it lists no playback or capture devices.

Thanks

Arkq, 2022-06-14:
> On top of that I have successfully *paired* a phone

You have to connect the phone, not only pair it :)

> Jun 14 10:55:59 armbian bluealsa[1649]: /usr/bin/bluealsa: D: bluez.c:406: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/Sink/2
> Jun 14 10:55:59 armbian bluealsa[1649]: /usr/bin/bluealsa: D: main.c:390: Acquiring D-Bus service name: org.bluealsa
> Jun 14 10:55:59 armbian bluealsa[1649]: /usr/bin/bluealsa: D: main.c:395: Starting main dispatching loop
> Jun 14 10:55:59 armbian systemd[1]: Started Bluealsa daemon.
> Jun 14 10:56:39 armbian bluealsa[1649]: /usr/bin/bluealsa: D: dbus.c:59: Called: org.bluealsa.Manager1.GetPCMs() on /org/bluealsa

After "Started Bluealsa daemon." but before "Called: org.bluealsa.Manager1.GetPCMs()" you should have some messages regarding phone being connected. Without such log messages, bluealsa does not see any BT device. Hence, the output of `bluealsa-aplay -L` is empty.

danboid, 2022-06-14:
I don't have pipewire installed either, no. Armbian s9xxx is a very minimal install without a desktop, X/Wayland etc by default.

I have run:

`bluetoothctrl connect MYDEVICESADDRESS`

but still not devices are listed.

Does my asound.conf look correct to you? Should that correctly configure ALSA to use bluesalsa on ALSA device 1 for pcm playback? That's what I think I need to tell it to do, but maybe extra ALSA config is required?

The only lines I need in my asound.conf when I'm not trying to use bluealsa are:

```
defaults.pcm.card 1
defaults.ctl.card 1
```

danboid, 2022-06-14:
I've got a bit closer to getting it working as I get this now:

```
root@armbian:/home/dan# bluealsa-aplay -L
bluealsa:SRV=org.bluealsa,DEV=XX:XX:XX:XX:XX:XX,PROFILE=a2dp
    A phone, capture
    A2DP (SBC): S16_LE 2 channels 44100 Hz
root@armbian:/home/dan# bluealsa-aplay -l
**** List of PLAYBACK Bluetooth Devices ****
**** List of CAPTURE Bluetooth Devices ****
hci0: XX:XX:XX:XX:XX:XX [A], phone
  A2DP (SBC): S16_LE 2 channels 44100 Hz
```

but sound didn't start coming out of my USB ALSA device when I started playing stuff on my phone. I had to unpair and re-pair my phone to get it recognised as an audio device. I tried running:

```
bluealsa-aplay XX:XX:XX:XX:XX:XX
```

only to get the error:

```
ALSA lib pcm.c:2533:(snd_pcm_open_conf) type is not defined
bluealsa-aplay: W: Couldn't open PCM: Open PCM: No such file or directory
```
over and over.

I'm also a bit concerned that I'm seeing

```
Bluetooth: hci0: unexpected event for opcode 0x2005
```

In my kernel messages. Do you think that is anything to worry about?

Arkq, 2022-06-14:
> but still no devices are listed when I run bluealsa-aplay -L or bluealsa-aplay -l

Hmm, strange.

> Does my asound.conf look correct to you?

Checking available BT PCM does not require anything in the asound.conf file.

> bluetoothctrl connect MYDEVICESADDRESS

And you are 100% that the device is indeed connected? For simplicity run bluealsa only with `a2dp-sink` profile (in your case source is not required). But firstly, stop bluealsa service and check whether `bluetoothctl show` does not list anything related to Advanced Audio profile. Then, start bluealsa service and such a profile should be available. Otherwise, you've got something else on the system which is registered in BlueZ and acts as a A2DP provider. Or the worst case scenario is that you will not see A2DP in the `show` output even after starting bluealsa service... :/

Arkq, 2022-06-14:
OK, so you've got connected BT device, great :)

> ALSA lib pcm.c:2533:(snd_pcm_open_conf) type is not defined

To be honest I'm not sure what is the cause of that, but when running `bluealsa-aplay` you should think of it as running `aplay`, but instead of file you are using BT device as a source. So, you have to specify output ALSA device. Maybe there is no `default` on your system. Firstly check whether `aplay` (aplay from ALSA) works.

danboid, 2022-06-14:
> > Does my asound.conf look correct to you?
> 
> Checking available BT PCM does not require anything in the asound.conf file.

Huh? This makes no sense to me. I thought BT PCM devices would only get printed by bluealsa-aplay if they were correctly configured in the asound.conf.

What command(s) should I be using to verify bluealsa is correctly configured.

Yes, ALSA, aplay, moc etc works fine if I use the ALSA config I have already posted ie:

```
defaults.pcm.card 1
defaults.ctl.card 1
```

but that won't suffice to get my ALSA device working a BT speaker will it?

danboid, 2022-06-14:
I would love a step by step idiots guide to getting my sound working then we (I) can add this to the wiki and it should work for nearly all Ubuntu and Debian users using a USB sound card, hopefully.

Arkq, 2022-06-14:
> Huh? This makes no sense to me. 

OK, I can explain.

The `bluealsa-aplay` tool is there to "simplify" Bluetooth speaker setup. Normally, you will have bluealsa PCM capture device which you will have to "link" with playback device of your choice, e.g.: `arecord -D bluealsa | aplay -D output-pcm`. Such approach is OK, but it involves pipe, ALSA latency in the capture device, and lack of "advanced" features. This can be replaced with `belualsa-aply -D output-pcm`.

The name of the PCM device can contain some options. You can specify them on command line, or store them in the .asoundrc config file. For that reason, there is no requirement of any .asoundrc file for `bluealsa-aplay` to print available PCMs. These PCMs require only `20-belualsa.conf` file which shall be installed when installing bluez-alsa (if you would like to use them, `-L` option will work even without properly installed `20-bluealsa.conf`). For example, this is the device name: `bluealsa:SRV=org.bluealsa,DEV=XX:XX:XX:XX:XX:XX,PROFILE=a2dp` you can use it with the `-D` option of `arecord`.

If `aplay` works for you, check which playback PCM it uses and try to use the same device with `bluealsa-aplay`: `bluealsa-aplay -D <pcm>`

danboid, 2022-06-14:
You say "try to use the same device with bluealsa-aplay: bluealsa-aplay -D <pcm>"

What is <pcm> in this context? How do I discern the correct value to use? Is it something I can extract from `aplay`'s output?

This is also not covered here https://github.com/Arkq/bluez-alsa/blob/master/doc/bluealsa-aplay.1.rst

"

-D NAME, --pcm=NAME
--
  | Select ALSA playback PCM device to use for audio output. The default is default. bluealsa-aplay does not perform any mixing of streams. If multiple devices are connected it opens a new connection to the ALSA PCM device for each stream. Therefore the PCM NAME must itself allow multiple open connections and mix the streams together. See option --single-audio to change this behavior. Similarly, bluealsa-aplay does not apply any transformations to the stream. For this reason it is often necessary to use the ALSA dmix and plug plugins in the NAME PCM."

How does the user know what are the valid values here? An example would help.

Arkq, 2022-06-14:
> How does the user know what are the valid values here? An example would help.

I think that there is some basic knowledge required here :D Project is named bluez-alsa, so a prerequisite is a basic knowledge of BlueZ (how to connect device) and ALSA (ALSA devices, mixers).

You can open `man aplay`, I think that there will be no much more explanation either....

The ALSA PCM device name is a broad topic. It can be something as simple as `default`, `hdmi`, or as complicated as `bluealsa:SRV=org.bluealsa,DEV=XX:XX:XX:XX:XX:XX,PROFILE=a2dp`. However, the simple case in most cases in an alias for more complicated one. The best way to get available PCMs on your system is to run `aplay -L`. When using `aplay` to play audio use `--verbose` option to see more info.

danboid, 2022-06-14:
OK looks like we're very nearly there now!

```
aplay -D front test.wav
```

Works fine and so it seems the command I was looking for is

```
bluealsa-aplay -D front
```

but if I run that I get the error:

```
bluealsa-aplay: W: Couldn't open PCM: Set HW params: Set sampling rate: Inavalid argument: 44100
```

Arkq, 2022-06-14:
> The wav file I used to test the aplay command is 44.1 KHz and that plays fine.

That's kinda strange. Which version of bluez-alsa you are using? In old version there was a bug related to setting HW params, maybe that's the case.... Either way, I think that you can use `plug` ALSA plugin to automatically resample data. And now you will have to create .asoundrc configuration file.

Add something like this to `~/.asoundrc`:
```sh
pcm.front-plug {
    type plug
    slave.pcm "front"
}
```

And now instead of "front" use "front-plug" as a device name.

danboid, 2022-06-14:
Excellent!

I've got it working now!

With `aplay`, I was able to use the short pcm device name but I had to use the full name of the device in my asound.conf for it to work, so the final config looks like this:

```
pcm.front-plug {
    type plug
    slave.pcm "front:CARD=UACDemoV10,DEV=0"
}
```

The next step is to get it working with JACK, then I'll write a guide for the wiki.

Thanks!

closed 2022-06-14 by danboid

danboid, 2022-06-16:
Here's the guide I promised. Let's keep it simple and beginner friendly please!

https://github.com/Arkq/bluez-alsa/wiki/Setting-up-a-USB-audio-device-as-a-Bluetooth-speaker-under-Ubuntu