arkq/bluez-alsa #573: bluealsa 4.0.0: alsa-lib 1.2.7.1 issues

macmpi, 2022-06-21:
Hi,

@borine I tried to update my system (Alpine 3.16 with `bluealsa 4.0.0`) with latest `alsa-lib 1.2.7.1` 
(basic usecase is essentially ` mpd 0.23.7` playing internet streams on default audio card and bluealsa speakers).

As soon as I connect a bluetotoh speaker, mpd reports many alsa xrun errors, and sound gets very choppy is on both outputs:
`daemon.warn mpd: alsa_output: Decoder is too slow; playing silence to avoid xrun`
Without bluetooth speaker, mpd playback is fine on audio card.

The only moving part from previously working setup was using `alsa-lib 1.2.6.1`
I'm unsure if it is related to https://github.com/Arkq/bluez-alsa/commit/7532af52ee0c48720cd516ed279d39c6a80f6b66#commitcomment-75771387 or something else.
Ironically `alsa-lib 1.2.7.1` has many changes related to xrun situations.

Thanks for any thoughts

macmpi, 2022-06-21:
I just rebuild with suggested commit.
It does improve situation a bit (sound out is more frequent), bur there are still many xruns.
Isn't this constant 5sec period puzzling?...
```
Jun 21 12:48:11 tiny-radiocd daemon.info bluetoothd[2294]: /org/bluez/hci0/dev_XX_XX_XX_XX_XX_XX/sep1/fd0: fd(26) ready
Jun 21 12:48:47 tiny-radiocd daemon.warn mpd: alsa_output: Decoder is too slow; playing silence to avoid xrun
Jun 21 12:48:52 tiny-radiocd daemon.warn mpd: alsa_output: Decoder is too slow; playing silence to avoid xrun
Jun 21 12:48:57 tiny-radiocd daemon.warn mpd: alsa_output: Decoder is too slow; playing silence to avoid xrun
Jun 21 12:49:03 tiny-radiocd daemon.warn mpd: alsa_output: Decoder is too slow; playing silence to avoid xrun
Jun 21 12:49:08 tiny-radiocd daemon.warn mpd: alsa_output: Decoder is too slow; playing silence to avoid xrun
Jun 21 12:49:13 tiny-radiocd daemon.warn mpd: alsa_output: Decoder is too slow; playing silence to avoid xrun
Jun 21 12:49:18 tiny-radiocd daemon.warn mpd: alsa_output: Decoder is too slow; playing silence to avoid xrun
Jun 21 12:49:23 tiny-radiocd daemon.warn mpd: alsa_output: Decoder is too slow; playing silence to avoid xrun
Jun 21 12:49:29 tiny-radiocd daemon.warn mpd: alsa_output: Decoder is too slow; playing silence to avoid xrun
Jun 21 12:49:37 tiny-radiocd daemon.warn mpd: alsa_output: Decoder is too slow; playing silence to avoid xrun
Jun 21 12:49:42 tiny-radiocd daemon.warn mpd: alsa_output: Decoder is too slow; playing silence to avoid xrun
Jun 21 12:49:47 tiny-radiocd daemon.warn mpd: alsa_output: Decoder is too slow; playing silence to avoid xrun
Jun 21 12:49:52 tiny-radiocd daemon.warn mpd: alsa_output: Decoder is too slow; playing silence to avoid xrun
Jun 21 12:49:57 tiny-radiocd daemon.warn mpd: alsa_output: Decoder is too slow; playing silence to avoid xrun
Jun 21 12:50:03 tiny-radiocd daemon.warn mpd: alsa_output: Decoder is too slow; playing silence to avoid xrun
```

borine, 2022-06-21:
> Isn't this constant 5sec period puzzling?...

That is MPD code here:
https://github.com/MusicPlayerDaemon/MPD/blob/dcf39ee44e8c7c1314ef34c97c90eb32d25f357f/src/output/plugins/AlsaOutputPlugin.cxx#L1347

It appears to issue the warning only every 5 seconds, but it *may* be padding its internal buffer with silence more often than that.

I need to check with some other applications to see how they behave

borine, 2022-06-21:
To help me reproduce this, please can you tell the the frame rate of the internet stream and of the bluetooth speaker? If they are both 48000 then the rate plugin should not be used.

Also please could you post the `audio_output` stanza for your bluetooth speaker from the mpd.conf file ?

Thanks

borine, 2022-06-21:
> I'm not using rate plugin.

The rate plugin is built-into the definition of the bluealsa pcm, unless you have redefined `pcm.bluealsa` yourself in the alsa config. So it is used if MPD sends a stream of, say 44.1kHz to a bluetooth speaker which operates at 48kHz. `bluealsa-aplay -L` will tell you the frame rate of your bluetooth speaker.

macmpi, 2022-06-21:
I made no special config on bluealsa side: just using default `/etc/alsa/conf.d/20-bluealsa.conf` file customized
with `defaults.bluealsa.device` set with my bluetooth speaker address

```
$bluealsa-aplay -L
bluealsa:SRV=org.bluealsa,DEV=20:74:CF:4C:9E:A5,PROFILE=a2dp
    Aeropex by AfterShokz, trusted audio-headphones, playback
    A2DP (SBC): S16_LE 2 channels 48000 Hz

```

borine, 2022-06-21:
OK, thanks. So MPD is using the rate plugin with bluealsa, so its possible your issue is related to known rate plugin issues. However ...

I'm now testing MPD v0.23.7-7-g8bd1b5228+ with bluealsa v4.0.0 and alsa-lib 1.2.7.1 with your internet stream links and i am not seeing any errors whatsoever. This is on an Intel laptop running Ubuntu 22.04. I'll have to try to set up a raspberry pi to see if that's any different.

macmpi, 2022-06-21:
I'm currently on a small Intel x86-64 laptop too, using Alpine 3.16 (baseline `alpine-setup`) augmented with my setup install as per https://github.com/macmpi/tiny-radioCD/tree/dev (archive to be pulled from dev branch).
Alpine is based on `musl 1.2`, not `glibc`: hoping this is not a tricky bug with that... (was fine with previous `alsa-lib`)

macmpi, 2022-06-21:
hum...yes, same binary, not rebuilt...

borine, 2022-06-21:
OK, that is not necessarily the cause of this issue, but I will try it on my system just to see what happens.

borine, 2022-06-23:
I've re-done my bluealsa v4.0.0 build against alsa-lib 1.2.6 using a completely clean build environment, then installed it onto a machine with alsa-lib 1.2.7.1. This time it works perfectly with MPD and mpg123 using your internet streams running for 30 minutes each. So my previous failure was caused by a corrupted build environment. I'm afraid I am simply unable to reproduce the errors reported here.

borine, 2022-06-23:
mpg123 requires you to run bluealsa with the `--keep-alive` option. I use `--keep-alive=1`. This is because it opens and closes the PCM several times in quick succession before it starts to play, not leaving enough time for bluealsa to release the transport before each retry, resulting in `device busy` errors. `--keep-alive` avoids releasing the transport immediately. I use
```
mpg123 -o alsa -a bluealsa url-of-mpg-audio-stream 
```
Note that this is not a recommendation of mpg123, its just an mp3 player I happened to have installed on the test machine at the time.

I use `mpd` with bluealsa on my main music system at home every day and don't see any playback issues with it.

borine, 2022-06-23:
By the way, you could test the possiblilty that this issue is caused by the alsa rate plugin by changing your mpd.conf to:

```
audio_output {
        type            "alsa"
        name            "bluealsa"
        device          "bluealsa"
        mixer_type      "software"
        auto_resample   "no"
        auto_format     "no"
}
```
This forces mpd to use its own internal resampler instead of the alsa-lib rate plugin.

I also recommend using mixer type "software" with bluealsa because otherwise mpd can trigger an abort in libasound if a bluetooth device disconnects.

borine, 2022-06-23:
By the way, you could test the possiblilty that this issue is caused by the alsa rate plugin by changing your mpd.conf to:

audio_output {
        type            "alsa"
        name            "bluealsa"
        device          "bluealsa"
        mixer_type      "software"
        auto_resample   "no"
        auto_format     "no"
}

This forces mpd to use its own internal resampler instead of the alsa-lib rate plugin.

I also recommend using mixer type "software" with bluealsa because otherwise mpd can trigger an abort in libasound if a bluetooth device disconnects.

macmpi, 2022-06-23:
That change seemed to fix the issue for few minutes (big hopes).
Unfortunately this popped-up again, making the headless PC very unresponsive under ssh...
I then did the same for 2 outputs (default & bluealsa), restarted, but sound is hashed a lot as long as bluetooth speaker is connected. Sound is fine if no bluetooth speaker is connected.

[MPD doc](https://mpd.readthedocs.io/en/latest/plugins.html) says:
`If set to no, then libasound will not attempt to resample. In this case, the user is responsible for ensuring that the requested sample rate can be produced natively by the device, otherwise an error will occur.`

Did not find mention of MPD internal re-sampler: is there any optional plugin to be added then?

macmpi, 2022-06-23:
and with mpg123:
```
$ mpg123 -o alsa -a bluealsa http://direct.franceinter.fr/live/franceinter-midfi.mp3
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
	version 1.29.3; written and copyright by Michael Hipp and others
	free software (LGPL) without any warranty but with best wishes

Directory: http://direct.franceinter.fr/live/

Terminal control enabled, press 'h' for listing of keys and functions.

Playing MPEG stream 1 of 1: franceinter-midfi.mp3 ...
ICY-NAME: franceinter-midfi.mp3

MPEG 1.0 L III cbr128 48000 j-s
ALSA lib pcm.c:8568:(snd_pcm_recover) underrun occurred
ALSA lib pcm.c:8568:(snd_pcm_recover) underrun occurred
ALSA lib pcm.c:8568:(snd_pcm_recover) underrun occurred
ALSA lib pcm.c:8568:(snd_pcm_recover) underrun occurred
ALSA lib pcm.c:8568:(snd_pcm_recover) underrun occurred
ALSA lib pcm.c:8568:(snd_pcm_recover) underrun occurred
ALSA lib pcm.c:8568:(snd_pcm_recover) underrun occurred
ALSA lib pcm.c:8568:(snd_pcm_recover) underrun occurred

```

borine, 2022-06-23:
That text is referring to the mpd ***alsa input*** plugin (eg for streaming from a phone through mpd). See the [input plugins](https://mpd.readthedocs.io/en/stable/plugins.html#alsa-plugin) and [resampler plugins](https://mpd.readthedocs.io/en/stable/plugins.html#resampler-plugins) manual sections for output options.

borine, 2022-06-23:
The mpg123 output seems to indicate that the issue is not with MPD. Looks like there is a definite issue here relating to bluealsa, but I just can't reproduce it here. I have no experience with musl to be able to guess if that could be a contributing factor.

macmpi, 2022-06-23:
Added:
```
resampler {
  plugin "internal"
}
```
Does not seem to help.
Anyway,  unsure other resampler (sox, etc) will fix the issue if it's not an `alsa-lib` resampling issue as per below then...

> The mpg123 output seems to indicate that the issue is not with MPD. Looks like there is a definite issue here relating to bluealsa, 

I thought it signed an `alsa-lib` issue actually...

macmpi, 2022-06-23:
I could make a small image for use on FAT32 USB disk and boot a x86_64 machine with Alpine and my setup: it could help reproducing the issue.
Other option is to create your own [Alpine setup](https://wiki.alpinelinux.org/wiki/Installation) and eventually install my app (std packages+configs & scripts) by just executing a single install script.

There's a bit of learning curve on the basics of Alpine distribution (apk package tools, etc), but it's quite easy and standard...
But I understand if it's too circumvented and have other priorities.

macmpi, 2022-06-23:
BTW, one [must read ](https://musl.libc.org/time64.html) about musl 1.2, that caused hidden troubles hard to debunk in several components over the past months (ie. bluez had to remove some duplicated older header files lately for this)

borine, 2022-06-24:
I've had a re-think about the symptoms described here. `mpd` runs each output in a separate thread and makes no attempt to synchronize them. So even if bluealsa is skipping frames, or running too fast for some other reason, then the default audio card should not be affected. The fact that both outputs are broken when bluetooth is in use suggests to me that this may in fact be interference between bluetooth and wifi.

You could test this theory with an mp3 audio file on local storage. Use `mpd` to play an internet stream to the default audio card only,  and `mpg123` to play the local file through bluealsa. Does the default card now sound OK when bluealsa is in use? If not,  then it is most likely *not* bluealsa that is causing the problem. The underruns are not caused by bluealsa running too fast, but by wifi being too slow.

macmpi, 2022-06-24:
Great catch!...
I actually took a more radical approach, by pulling-in a good-old ethernet cable to the PC, and disabling `wlan0` interface: it works mint in the 2 previous scenarii (`mpd` or `mpg123`) with updated `alsa-lib 1.2.7.1` !!

Now, I'm curious as to why this issue appears with `alsa-lib 1.2.7.1` !...
And more importantly I have no much clue on how to resolve the BT-Wifi coexistence issue, which I did not have previously.

macmpi, 2022-06-24:
> The underruns are not caused by bluealsa running too fast, but by wifi being too slow.

hum...if it's really a strict BT/Wifi coexistence issue, that would be some sort of bandwidth bottleneck/congestion between the two, right?
Stream acquired by wifi network is always the same bandwidth, whatever alsa-lib version used (and It was acquired and processed fine under previous alsa-lib).
Stream is processed/converted and then sent over BT: would it be using more BT bandwith with recent alsa-lib then, so that it would now harm wifi signal for stream acquisition ?
Really strange...

Wouldn't rather be some sort of in/out buffer contentions in the alsa-lib / bluealsa / Bluez chain processing?

Let me know if you want me to `strace` some specific things, or give access to the PC over ssh (through Teamviewer)

Arkq, 2022-06-24:
If the network stream is the same and the BT speaker is the same, therebut is really no room for changed bandwidth. However, bluez-alsa project has a tool hcitop which displays Bluetooth traffic, you can see whether there is a significant change in rx/tx

borine, 2022-06-24:
I can guarantee that the co-existence issue has nothing to do with alsa-lib. You need to look at kernel version and patches, kernel config, wireless card firmware version and driver module params. Something in there is changed with your OS upgrade, not just alsa-lib. I think you are more likely to find expert help with this from your OS or platform support groups.

macmpi, 2022-06-24:
> bluez-alsa project has a tool hcitop which displays Bluetooth traffic

Thanks @Arkq : I'm going to build it and check-out what it says in both situations.

macmpi, 2022-06-25:
While I'm looking more closely at build messages, I noticed many similar warnings:
```
In file included from /usr/include/bsd/sys/cdefs.h:51,
                 from /usr/include/bsd/sys/time.h:41,
                 from ../../../src/shared/rt.h:24,
                 from ../../../src/asound/../shared/rt.c:11:
/usr/include/sys/cdefs.h:1:2: warning: #warning usage of non-standard #include <sys/cdefs.h> is deprecated [-Wcpp]
    1 | #warning usage of non-standard #include <sys/cdefs.h> is deprecated
      |  ^~~~~~~
```
as mentioned previously, musl 1.2 is very sensitive to recent [time64 definitions](https://musl.libc.org/time64.html): not sure these warnings are hints to some consequences, but just a heads-up in case.

borine, 2022-06-25:
re `libbsd` dependency - the wiki installation doc is out-of-date here. `libbsd` is no longer a requirement for the bluealsa core (although it will be used if available). So building without it may be advisable with musl libc. `libbsd` is still required for hcitop though. I will change the wiki page.

I wonder, is it possible to tell `configure` not to use libbsd for `bluealsa`, even though it is installed for `hcitop` ?

macmpi, 2022-06-25:
Ok, on the root problem: it now looks like it might be a network management issue in Alpine when several interfaces are at play.
I just commented-out `eth0` entries in `/etc/network/interfaces`, and sound is all fine now over wifi/bluetooth...
No clue why this suddenly emerged in my setup with this `alsa-lib` change.
I'll investigate further separately: apologies for the noise.

Current `bluez-alsa` package in Alpine is built without all tools (only `aplay` and `cli`), and I checked older build logs, and they do not show warnings, which is consistent with your explanation.
The test package I'm building with all tools had me include `libbsd` dependency, and hence generated the reported warnings.

Not sure I can use different dependencies for different components, on a single APKBUILD (which generate all `bluez-alsa` stuff, including a sub-package with tools).
Actually, I'm wondering if it's really desirable to have those shared components (`rt.c`, `io.c` & al.) compiled with different sets of dependencies: this could make debugging/maintenance tricky...

Are those deprecated BSD headers worth such gymnastics, or could this be adapted with more current definitions for `hcitop`?

Arkq, 2022-06-25:
> I wonder, is it possible to tell configure not to use libbsd for bluealsa, even though it is installed for hcitop?

It's not required to change configure script. Simply one can remove it from `rt.h` file in here: https://github.com/Arkq/bluez-alsa/blob/master/src/shared/rt.h#L23 The "ifdef" here is as a reminder that the definitions of `timespecadd` and `timespecsub` are taken from BSD version of `time.h`. However, I doubt that the inclusion of `bsd/sys/time.h` causes such a havoc. This project is not the only one which includes `bsd/...` headers.

> Are those deprecated BSD headers worth such gymnastics, or could this be adapted with more current definitions for hcitop?

As far as I can tell from the warning message, the `bsd/sys/time.h` is not deprecated, but the `sys/cdefs.h` which is (unfortunately) included by the `bsd/sys/cdefs.h`.

macmpi, 2022-06-29:
Thanks @Arkq .
Then attached patch did the trick and suppressed warnings for the most part.
Only have warnings for `hcitop` build obviously, but as it's an optional tool, one can live with that.

I'll probably push it to Alpine, unless anything comes-up.
Thanks again for your kind support.

[0001-remove-bsd-sys-time.h-include.patch.txt](https://github.com/Arkq/bluez-alsa/files/9010134/0001-remove-bsd-sys-time.h-include.patch.txt)

closed 2022-07-12 by macmpi