arkq/bluez-alsa #574: systemd wiki page

borine, 2022-07-02:
I've started putting together some notes regarding the bluez-alsa systemd unit files, with the intention of updating the wiki page once I have a clearer picture of what would be generally useful to include in there. I think a section on the sandboxing directives in the files might be useful, and while looking at this again I see that the issues with the "ProtectClock" directive have been fixed in systemd now. So I think that, on balance, it should now be included in both the original files. Do you agree?

For now I will add `ProtectClock=true` as a suggested change in the wiki. The only other changes I can find that will improve the security rating from `systemd-analyze security` are to run both services as non-root users. So I will include a section on how to do that together with recommended additional sandbox directives that would then be applicable.

I will also add a very brief section on how to amend the files after installation, but pointing readers to the systemd docs for detail.

There is also the question of what to do with the old wiki page. My initial thought is just to remove it and replace with the new one, would that be OK?

Arkq, 2022-07-02:
> I see that the issues with the "ProtectClock" directive have been fixed in systemd now. So I think that, on balance, it should now be included in both the original files. Do you agree?

Hmm... I think that it depends whether such change will not break systemd service on common Linux distros.

> run both services as non-root users

Yes, I think that that's a good idea. Also, it might be worth to add configure option for specifying service user, so systemd unit files and dbus config could be configured. Such change might be useful for yocto based distros.

> There is also the question of what to do with the old wiki page. My initial thought is just to remove it and replace with the new one, would that be OK?

Yes, I think that it's OK to keep wiki pages more or less in sync with master. And for sure anything which is not relevant to current release (v4.0.0) might be removed.

graham8, 2022-07-03:
> I see that the issues with the "ProtectClock" directive have been fixed in systemd now.

If your advice/unit files depend on a particular version of systemd/alsa/whatever, please tell us which. (We are still running Debian Buster, for example).

> There is also the question of what to do with the old wiki page. My initial thought is just to remove it and replace with the new one, would that be OK?

Integrators will not be pulling the latest release maybe for some months after it's official (if it ain't broke ....). And only the brave or those with an issue to solve will be pulling the latest and greatest.  IMHO you should base the wiki on the latest **release** with a note against any feature introduced since that release and if possible retain somehow advice which applies to previous releases.

borine, 2022-07-03:
> advice/unit files depend on a particular version of systemd/alsa/whatever

The systemd wiki page is (as far as I know) agnostic as far as systemd versions go. The referenced problem with ProtectClock was that enabling it actually increased the exposure score for bluealsa. This was a bug specifically in systemd 245 (the first to include ProtectClock, if I recall correctly), and appears to be fixed in systemd 246 and later. Nothing was actually broken on bluealsa by it, I was just trying to find the sandbox config that produced the lowest possible exposure score as measured `systemd-analyze security`. So I think `ProtectClock=true` should be included by default, as it is recommended in the systemd docs for any service that does not need to change the system clock. BTW systemd ignores any directives it does not understand, so there is no problem in using very old versions from before our chosen directives were introduced.

> you should base the wiki on the latest release

Thats exactly what @Arkq said :smile: The new systemd wiki page (assuming I ever get it finished !) will relate to bluez-alsa v4.0.0 and later. The problem with the existing systemd page is that it advises a method of setting command-line arguments that contradicts the advice by systemd developers, and is difficult to apply with the new unit files now included in the bluealsa v4.0.0 installation. So I think its time to remove it.

There is also a wiki page on using the ALSA plugins which is now redundant given the new manual page. Maybe that one should now be removed too. I would welcome your thoughts on that. too.

graham8, 2022-07-03:
> I would welcome your thoughts on that. too.

There's a lot of duplication certainly.

borine, 2022-07-09:
I've moved the old ALSA plugin wiki page to an "Archive" section, and propose to do the same with the old systemd wiki page when the new one is ready. The mandatory "Pages" side-panel is a pain here - I've prefixed the the archived page name(s) with a pipe symbol just to force them to the end of the page list. Let me know if you don't like this, or if you know a cleaner way of achieving this.

Arkq, 2022-07-09:
Maybe this will help: https://docs.github.com/en/communities/documenting-your-project-with-wikis/creating-a-footer-or-sidebar-for-your-wiki
I've never created any fancy wiki on GitHub, so I'm not sure whether it will work...

borine, 2022-07-09:
The "Pages" sidebar on github wikis is forced on all wiki pages, and cannot (as far as I know) be removed. Creating a new sidebar, as per the link you gave, merely adds a second sidebar below the mandatory one. There is also no way to "hide" a page from the "Pages" list nor to change the simple lexical ordering it uses. Sub-directories make no difference either, the tree is flattened before sorting by Pages. Like I said, it is a real pain. Very "big brother knows best" and, I suspect, universally hated for that reason.

borine, 2022-07-10:
I accidentally broke README.md when I renamed the plugins wiki page. Here's a fix:
```diff
diff --git a/README.md b/README.md
index 1911926..0fc1dd7 100644
--- a/README.md
+++ b/README.md
@@ -229,9 +229,8 @@ wiki page.
 
    Additionally, the cause of the "PCM not found" error might be an incorrect ALSA PCM name. Run
    `bluealsa-aplay --list-pcms` for the list of currently available ALSA PCM names - it might give
-   you a hint what is wrong with your `.asoundrc` entry. Also, take a look at the "[Using the
-   bluealsa ALSA pcm plugin](https://github.com/Arkq/bluez-alsa/wiki/Using-the-bluealsa-ALSA-pcm-plugin)"
-   bluez-alsa wiki page.
+   you a hint what is wrong with your `.asoundrc` entry. Also, take a look at the
+   [bluealsa-plugins man page](./doc/bluealsa-plugins.7.rst).
 
 ## Resources
 ```