arkq/bluez-alsa #576: bluealsa 4.0.0: buffer overrun giving a choppy sound with echo on RPi

breezerider, 2022-07-05:
I have a Raspberry Pi Zero W system with a USB bluetooth dongle (CSR8510 A10).

My setup is based on the alsa loopback driver and the bmix script from the [wiki](https://github.com/Arkq/bluez-alsa/wiki/Using-BlueALSA-with-dmix) with some tweaks to make it work with BlueAlsa 4.0.0.

I experience really poor audio quality (jitter, echo and huge latency) on my bluetooth speaker. After adjusting the BMIX_LATENCY to no avail as it either caused no sound at all or made the issue even worse, I rebuilt bluealsa with debug flag and captured the log below.
Below is excerpt from the syslog related to bluetooth. When I run an `aplay -D...` command the lines with 'overrun ...' show up:

```
bluetoothd[252]: Bluetooth daemon 5.64
bluetoothd[252]: Starting SDP server
bluetoothd[252]: Ignoring (cli) hostname
bluetoothd[252]: Ignoring (cli) wiimote
bluetoothd[252]: Ignoring (cli) autopair
bluetoothd[252]: Ignoring (cli) policy
bluetoothd[252]: Ignoring (cli) avrcp
bluetoothd[252]: Ignoring (cli) network
bluetoothd[252]: Ignoring (cli) input
bluetoothd[252]: Ignoring (cli) hog
systemd[1]: Starting OpenSSH server daemon...
bluetoothd[252]: Ignoring (cli) gap
bluetoothd[252]: Ignoring (cli) scanparam
bluetoothd[252]: Ignoring (cli) deviceinfo
bluetoothd[252]: Excluding (cli) battery
bluetoothd[252]: Ignoring (cli) sixaxis
bluetoothd[252]: Bluetooth management interface 1.18 initialized
sdptool[266]: Serial Port service registered
systemd[1]: Started Bluetooth service.
systemd[1]: Starting BlueALSA service...
systemd[1]: systemd-rfkill.service: Deactivated successfully.
bluealsa[269]: /usr/bin/bluealsa: D: /var/tmp/portage/media-sound/bluez-alsa-4.0.0/work/bluez-alsa-4.0.0/src/main.c:585: Starting main dispatching loop
systemd[1]: Started BlueALSA service.
bluealsa[269]: /usr/bin/bluealsa: D: /var/tmp/portage/media-sound/bluez-alsa-4.0.0/work/bluez-alsa-4.0.0/src/main.c:112: Acquired D-Bus service name: org.bluealsa
bluealsa[269]: /usr/bin/bluealsa: D: /var/tmp/portage/media-sound/bluez-alsa-4.0.0/work/bluez-alsa-4.0.0/src/bluealsa-dbus.c:426: Registering D-Bus manager: /org/bluealsa
systemd[1]: Started Bluealsa Dmix daemon.
systemd[1]: Reached target Bluetooth Support.
bluealsa[269]: /usr/bin/bluealsa: W: /var/tmp/portage/media-sound/bluez-alsa-4.0.0/work/bluez-alsa-4.0.0/src/ba-adapter.c:53: Couldn't get HCI version: Network is down
bluealsa[269]: /usr/bin/bluealsa: D: /var/tmp/portage/media-sound/bluez-alsa-4.0.0/work/bluez-alsa-4.0.0/src/bluez.c:679: Registering battery provider: /org/bluez/hci0/battery
systemd[1]: Startup finished in 4.314s (kernel) + 23.097s (userspace) = 27.411s.
bluealsa[269]: /usr/bin/bluealsa: D: /var/tmp/portage/media-sound/bluez-alsa-4.0.0/work/bluez-alsa-4.0.0/src/bluez.c:699: BlueZ battery provider support not available
bluealsa[269]: /usr/bin/bluealsa: D: /var/tmp/portage/media-sound/bluez-alsa-4.0.0/work/bluez-alsa-4.0.0/src/bluez.c:493: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/source/1
bluealsa[269]: /usr/bin/bluealsa: D: /var/tmp/portage/media-sound/bluez-alsa-4.0.0/work/bluez-alsa-4.0.0/src/bluez.c:402: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/source/1
bluetoothd[252]: Endpoint registered: sender=:1.6 path=/org/bluez/hci0/A2DP/SBC/source/1
bluealsa[269]: /usr/bin/bluealsa: D: /var/tmp/portage/media-sound/bluez-alsa-4.0.0/work/bluez-alsa-4.0.0/src/bluez.c:493: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/source/2
bluealsa[269]: /usr/bin/bluealsa: D: /var/tmp/portage/media-sound/bluez-alsa-4.0.0/work/bluez-alsa-4.0.0/src/bluez.c:402: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/source/2
bluetoothd[252]: Endpoint registered: sender=:1.6 path=/org/bluez/hci0/A2DP/SBC/source/2
bmixd[274]: ServiceRunning org.bluealsa
bmixd[274]: ServiceRunning org.bluealsa
systemd[283]: Reached target Bluetooth.
bluealsa[269]: /usr/bin/bluealsa: D: /var/tmp/portage/media-sound/bluez-alsa-4.0.0/work/bluez-alsa-4.0.0/src/bluez.c:1049: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
bluealsa[269]: /usr/bin/bluealsa: D: /var/tmp/portage/media-sound/bluez-alsa-4.0.0/work/bluez-alsa-4.0.0/src/bluez.c:1145: Adding new Stream End-Point: 00:42:79:48:F7:0C: SBC
bluealsa[269]: /usr/bin/bluealsa: D: /var/tmp/portage/media-sound/bluez-alsa-4.0.0/work/bluez-alsa-4.0.0/src/bluez.c:1049: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
bluealsa[269]: /usr/bin/bluealsa: D: /var/tmp/portage/media-sound/bluez-alsa-4.0.0/work/bluez-alsa-4.0.0/src/bluez.c:1145: Adding new Stream End-Point: 00:42:79:48:F7:0C: AAC
bluealsa[269]: /usr/bin/bluealsa: D: /var/tmp/portage/media-sound/bluez-alsa-4.0.0/work/bluez-alsa-4.0.0/src/dbus.c:66: Called: org.bluez.MediaEndpoint1.SelectConfiguration() on /org/bluez/hci0/A2DP/SBC/source/1
bluealsa[269]: /usr/bin/bluealsa: D: /var/tmp/portage/media-sound/bluez-alsa-4.0.0/work/bluez-alsa-4.0.0/src/bluez.c:178: A2DP peer capabilities blob [len=4]: 33350235
bluealsa[269]: /usr/bin/bluealsa: D: /var/tmp/portage/media-sound/bluez-alsa-4.0.0/work/bluez-alsa-4.0.0/src/bluez.c:1049: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
bluealsa[269]: /usr/bin/bluealsa: D: /var/tmp/portage/media-sound/bluez-alsa-4.0.0/work/bluez-alsa-4.0.0/src/dbus.c:66: Called: org.bluez.MediaEndpoint1.SetConfiguration() on /org/bluez/hci0/A2DP/SBC/source/1
bluealsa[269]: /usr/bin/bluealsa: D: /var/tmp/portage/media-sound/bluez-alsa-4.0.0/work/bluez-alsa-4.0.0/src/a2dp.c:356: Selected A2DP SBC bit-pool range: [2, 53]
bluealsa[269]: /usr/bin/bluealsa: D: /var/tmp/portage/media-sound/bluez-alsa-4.0.0/work/bluez-alsa-4.0.0/src/bluez.c:313: A2DP Source (SBC) configured for device 00:42:79:48:F7:0C
bluealsa[269]: /usr/bin/bluealsa: D: /var/tmp/portage/media-sound/bluez-alsa-4.0.0/work/bluez-alsa-4.0.0/src/bluez.c:316: A2DP selected configuration blob [len=4]: 11150235
bluealsa[269]: /usr/bin/bluealsa: D: /var/tmp/portage/media-sound/bluez-alsa-4.0.0/work/bluez-alsa-4.0.0/src/bluez.c:318: PCM configuration: channels: 2, sampling: 48000
bluealsa[269]: /usr/bin/bluealsa: D: /var/tmp/portage/media-sound/bluez-alsa-4.0.0/work/bluez-alsa-4.0.0/src/bluez.c:493: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/source/3
bluealsa[269]: /usr/bin/bluealsa: D: /var/tmp/portage/media-sound/bluez-alsa-4.0.0/work/bluez-alsa-4.0.0/src/bluez.c:402: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/source/3
bmixd[274]: PCMAdded /org/bluealsa/hci0/dev_00_42_79_48_F7_0C/a2dpsrc/sink
bmixd[274]: PCMAdded* /org/bluealsa/hci0/dev_00_42_79_48_F7_0C/a2dpsrc/sink 00:42:79:48:F7:0C
bmixd[274]: Device: /org/bluez/hci0/dev_00_42_79_48_F7_0C
bmixd[274]: Device: /org/bluez/hci0/dev_00_42_79_48_F7_0C /org/bluealsa/hci0/dev_00_42_79_48_F7_0C/a2dpsrc/sink /org/bluez/hci0/dev_00_42_79_48_F7_0C 00:42:79:48:F7:0C
bmixd[274]: Sequence: 0
bmixd[274]: Sequence: 0 /org/bluealsa/hci0/dev_00_42_79_48_F7_0C/a2dpsrc/sink 0 /org/bluez/hci0/dev_00_42_79_48_F7_0C 00:42:79:48:F7:0C
bmixd[274]: Transport: A2DP-source
bmixd[274]: Transport: A2DP-source /org/bluealsa/hci0/dev_00_42_79_48_F7_0C/a2dpsrc/sink A2DP 0 /org/bluez/hci0/dev_00_42_79_48_F7_0C 00:42:79:48:F7:0C
[... 62 lines cut ...]
bluealsa[269]: /usr/bin/bluealsa: D: /var/tmp/portage/media-sound/bluez-alsa-4.0.0/work/bluez-alsa-4.0.0/src/bluez.c:1247: Signal: org.freedesktop.DBus.Properties.PropertiesChanged(): org.bluez.MediaTransport1: State
```

When I am using the same dongle on my PC with the same bluetooth speaker via pulseaudio the quality is fine. 
What could be the reason for this? Is there somethin I could do to debug this situation?

Also, DWC OTG driver is in use:

```
kernel: dwc_otg 20980000.usb: DWC OTG Controller
kernel: dwc_otg 20980000.usb: new USB bus registered, assigned bus number 1
kernel: dwc_otg 20980000.usb: irq 56, io mem 0x00000000
kernel: Init: Port Power? op_state=1
kernel: Init: Power Port (0)
kernel: usb usb1: New USB device found, idVendor=1d6b, idProduct=0002, bcdDevice= 5.10
kernel: usb usb1: New USB device strings: Mfr=3, Product=2, SerialNumber=1
kernel: usb usb1: Product: DWC OTG Controller
kernel: usb usb1: Manufacturer: Linux 5.10.63 dwc_otg_hcd
kernel: usb usb1: SerialNumber: 20980000.usb
kernel: hub 1-0:1.0: USB hub found
kernel: hub 1-0:1.0: 1 port detected
kernel: usb 1-1: New USB device found, idVendor=0a12, idProduct=0001, bcdDevice=88.91
kernel: usb 1-1: New USB device strings: Mfr=0, Product=2, SerialNumber=0
kernel: usb 1-1: Product: CSR8510 A10
```

borine, 2022-07-06:
I've just setup "bmix" as per the wiki page on a RpI Zero W running RaspberryPiOS  Lite "bullseye" with my own build of bluealsa v4.0.0 and its working fine here ***using the on-board UART bluetooth adapter***. I've set BMIX_LATENCY=500000, but otherwise everything else is copy&pasted from the wiki page.

Incidentally - I noticed while doing this that the wiki page referred to a bluealsa default PCM called "bluetiith". This is no longer included in the bmix solution, so I have now removed that paragraph from the wiki page. But otherwise I am using it exactly as written.

Have you tried using bluealsa without dmix first? We need to be sure that the bluealsa build and setup are working before looking for errors in bmix.

You mention making some "tweaks" for v4.0.0 - can you please give details, I didn't need to modify anything.

Also, to help try to repoduce your issue, can you state the options you gave to `configure` when building bluealsa and the command-line arguments you give to bluealsa when starting it?

Thanks.

breezerider, 2022-07-07:
Thanks for looking into this issue, @borine . Following in your footsteps, I setup a bullseye system with a simple static bluealsa configuration and used the built-in UART bluetooth adapter. And it works great with my speaker! The, some googling brought up threads about poor performance of dwc_otg and dwc2 with the same USB BT adapter, see [here](https://forums.raspberrypi.com/viewtopic.php?t=60766). So, in the end this seems like it is not a bluealsa issue after all.

Funnily, the quality was better with AAC codec rather than SBC, which make me think it may be an issue with available bandwidth. I will just just stick to using the UART BT for now.

Now, regarding the bmix "tweaks": I had to change a line in the main loop as otherwise the BT device was not appended to the dmix configuration, consider this excerpt:
```
# now listen for dbus signals
while read -u $MONITOR_FD
do
	case "$REPLY" in
		PCMAdded*)
			new_pcm[id]=${REPLY#PCMAdded }
			addr=${REPLY#*dev_}
			addr=${addr%%/*}
			new_pcm[addr]=${addr//_/:}
			;;
		\}) # <- this was never reached
			[[ "${new_pcm[mode]}" = sink ]] && add_pcm
			new_pcm=()
			;;
		PCMRemoved*)
			[[ "${new_pcm[mode]}" = sink ]] && remove_pcm "${REPLY#PCMAdded }"
			;;
		ServiceStopped*)
			new_pcm=()
			handle_service_stopped_event
			;;
		*)
			set_property "$REPLY"
			;;
	esac
done
```
I changed the marked line to `Muted*)` as this was the last thing that was output upon connecting the speaker. Other than that, I added some more `echo` statement for debug output. Please let me know if there is an issue with bmix script or if that is expected with bluealsa 4.0.0. Thanks!

breezerider, 2022-07-10:
Thank a lot for updating the wiki page! I will close the issue now as it is related to the USB driver and not bluealsa itself.

closed 2022-07-10 by breezerider