arkq/bluez-alsa #580: Validating BT audio over i2s on Elkhart Lake

mjhammel, 2022-07-12:
I could use some help in validating BT I2S audio transfers on a new board using bluez-alsa.  I am able to transfer data from a phone to the board over a2dp but am having problems validating HFP.  Any pointers on what else I can look for would be appreciated.

### Hardware setup
- Board: Intel Elkhart Lake (Atom) CPU without a codec chip
- BT: TI WL1831MOD WiFi/BT connected to cpu using 2 channel I2S and a UART.

### Software versions
- Kernel: 5.15.21
- bluez-alsa: 3.1.0
- ofono: 1.34
- alsa-lib: 1.0.28

### Bluez-alsa configure
`./configure \
  --with-dbusconfdir=/etc/dbus-1/system.d \
  --with-libunwind \    
  --build=`uname -m` \  
  --host=$($1_ARCH) \   
  --enable-aac \ 
  --enable-debug \      
  --enable-mp3lame \
  --enable-mpg123 \     
  --enable-msbc \       
  --enable-ofono
`
### Test setup
1. Setup a [Jitsi meeting room](https://meet.jit.si/newboardtest)
2. Dial into the meeting room with my phone.
3. Setup the BT HFP connection between the board and the phone.
4. Play audio from the board using aplay, which should end up in the meeting room:  _aplay -D bluealsa:xx:xx:xx:xx:xx:xx,sco /bootfs/wifi/bt.wav_
5. Play audio from the meeting room which should be recordable on the board:  _arecord -D bluealsa:xx:xx:xx:xx:xx:xx,sco /tmp/capture.wav_

### Test results
1. -
2. -
3. Phone and board are paired and connected.
4. No audio heard in Jitsi
5. No audio recorded on board

### Notes on test
Playing audio on the board through the BT interface changes the SCO/byte counts on the interface stats.  However, interface stats do no change when trying to record from the interface.

> $ hciconfig
hci0:   Type: Primary  Bus: UART
        BD Address: A4:D5:78:11:60:E6  ACL MTU: 1021:6  SCO MTU: 180:4
        UP RUNNING PSCAN 
        RX bytes:8609 acl:76 sco:0 events:173 errors:0
        TX bytes:186871 acl:66 sco:6561 commands:61 errors:0
$ aplay -D bluealsa:D8:CF:BF:D5:AE:BB,sco bt.wav
Playing WAVE 'bt.wav' : Signed 16 bit Little Endian, Rate 44100 Hz, Stereo
$ hciconfig
hci0:   Type: Primary  Bus: UART
        BD Address: A4:D5:78:11:60:E6  ACL MTU: 1021:6  SCO MTU: 180:4
        UP RUNNING PSCAN 
        RX bytes:8611 acl:76 sco:0 events:173 errors:0
        TX bytes:370581 acl:66 sco:13122 commands:61 errors:0

Enabled debug on ofono.  After making the call to jitsi and then setting up the BT connection I can see the following debug messages:

>2022-01-06T08:00:31 daemon.debug ofonod: src/voicecall.c:ofono_voicecall_notify() Got a voicecall event, status: active (0), id: 1, number: 15126471431 called_number: , called_name
2022-01-06T08:00:31 daemon.debug ofonod: src/voicecall.c:ofono_voicecall_notify() Did not find a call with id: 1
2022-01-06T08:00:31 daemon.debug ofonod: src/voicecall.c:ofono_voicecall_mpty_hint() ids: 0
2022-01-06T08:00:31 daemon.debug ofonod: src/siri.c:siri_remove() atom: 0xf610b0
2022-01-06T08:00:31 daemon.debug ofonod: src/network.c:current_operator_callback() 0xf55d90, (nil)
2022-01-06T08:00:31 daemon.debug ofonod: drivers/hfpmodem/network-registration.c:signal_strength_cb() signal_strength_cb: 5
2022-01-06T08:00:31 daemon.debug ofonod: src/network.c:ofono_netreg_strength_notify() strength 100
2022-01-06T08:00:31 daemon.debug ofonod: drivers/hfpmodem/network-registration.c:signal_strength_cb() signal_strength_cb: 5
2022-01-06T08:00:31 daemon.debug ofonod: src/handsfree-audio.c:sco_accept() SCO connection setup between local: A4:D5:78:11:60:E6 and remote: D8:CF:BF:D5:AE:BB
2022-01-06T08:00:31 daemon.debug ofonod: src/handsfree-audio.c:send_new_connection() 0xf66700, fd: 11, codec: 2

The first thing to note is that ofono knows the number I dialed on the phone (line 1), BEFORE the BT connection was established.  This means that configuration messaging is happening between the phone and board.  Also, the SCO connection is established between the phone and board (line 9).  So the wireless connection is working.  What isn’t working is audio transfer.

As stated earlier, an A2DP connection between the phone and board works.  Playing audio on the phone can be recorded properly on the board to a wav file.

### Analysis
I'm not clear on if I need a dummy codec driver for this test setup to work.  I believe I have all the SND and SOF drivers required for this CPU loaded but I don't really have a clear list of those either.  Is there some way to verify that I2S is working using this or similar setup?

borine, 2022-07-12:
Something that stands out here is that, as you mentioned, you have outgoing SCO packets on the HCI, but no incoming. SCO is a synchronous protocol, outgoing and incoming packets alternate on successive link slots. So you would expect to ***always*** see SCO Rx bytes on the HCI when an HFP/HSP connection is established, but Tx bytes only when a application is sending data. This looks like your TI chip does not, by default, route SCO packets over the HCI. Looking at the spec sheet https://www.ti.com/document-viewer/WL1831MOD/datasheet/GUID-B546648E-ADF6-4A4B-9FE3-42B34D76557B#TITLE-SWRS152TOCWRAPPER_TERMINAL_CONFIGURATION_AND_FUNCTIONS I see that pin 56 and pin 57 are labelled "BT_AUD_IN" and "BT_AUD_OUT" respectively; so SCO audio data is almost certainly routed through these pins and not through pins 52 and 53 which are the HCI pins.

I think I recall (from looking at another issue 2 years ago) that TI do support switching the SCO route using a custom vendor HCI command, but don't have a record of it now, unfortunately. If it cannot be done by a command from the host then there may be an alternative firmware, sometimes vendors (or hobbyists) produce firmwares with specific operating parameters changed.

mjhammel, 2022-07-12:
Excellent tip.  Thanks @borine .  I found two references to this problem on TI's forums.

- https://e2e.ti.com/support/wireless-connectivity/wi-fi-group/wifi/f/wi-fi-forum/705119/wl1835mod-vendor-specific-hci-commands
- https://e2e.ti.com/support/wireless-connectivity/wi-fi-group/wifi/f/wi-fi-forum/1058978/wl1831mod-hci-commands-for-pcm-hsp-audio

There was one other post but it just pointed to the latter.

The latter suggests a format for the command, but it's radically different than the former.  I tried the former just for grins, using btmon to see if there were problems.  The command succeeds and btmon shows activity on the interface when using aplay but there is still no audio on the remote end.  

I'll fiddle with the latter post - it has additional information about the command format and some suggestions about using matching channel_edge configurations.  

A couple more questions: when using bluez-alsa/ofono, are SCO packets required over HCI?  Is that why I need to look for how to route them from BT_AUD_<in/out> to the HCI pins?  If so, are there any performance issues I should be aware of with HCI instead of I2S?  Also, does that imply our board layout using I2S for 2ch audio is superfluous for HFP?  I don't think that would be a problem, but I need to be able to justify why I'm doing it this way.  

Again, many thanks for the tip.  I think it's just what I needed.

borine, 2022-07-13:
> when using bluez-alsa/ofono, are SCO packets required over HCI?

Bluez, and therefore bluealsa, only process data coming from the HCI, so yes, to use bluealsa for SCO audio it must come over HCI. (but of course you do not have to use bluealsa, I assume there must be some solution for handling the PCM streams over i2s in linux, but I just don't know what that solution is). Note that if the TI board supports transparent encoding for SCO, then it will be possible to use mSBC codec, and that will always come over the HCI because it needs to be decoded by the host. However if you choose to support only mSBC and not CVSD then you will not be Bluetooth spec compliant.

> are there any performance issues I should be aware of with HCI?

None that I am aware of. The controller must encapsulate the PCM into HCI frames, which will require a small amount of buffering, but I don't recall that being raised as an issue anywhere.

> does that imply our board layout using I2S for 2ch audio is superfluous for HFP?

Yes, unless you can find an alternative solution for CVSD and only use bluealsa for mSBC. I am not an electronics engineer, so at at the limit of my knowledge here, and take this with a pinch of salt, but I assume that it is feasible to have an ALSA driver that presents the TI board i2s interface as a sound card to the kernel, or alternatively you could route the PCM directly to an ADC/DAC without needing the kernel at all?

borine, 2022-07-13:
Referring to this link: https://e2e.ti.com/cfs-file/__key/communityserver-discussions-components-files/538/47427.CC256x-VS-HCI-Commands-_2D00_-Texas-Instruments-Wiki.pdf

I believe the HCI custom command you need is HCI_VS_Write_SCO_Configuration (0xFE10). Applying the parameters given in that document, I get the hcitool command:
```
hcitool cmd 0x3f 0x0210 0x01 0x00 0x00 0x00 0x00
```

I don't have a TI device to test with, so please could you try that one for me, after power-cycling the system to get a fresh start?

mjhammel, 2022-07-13:
Sorry to take so long to reply.  My test setup/process had to be adjusted to do things in the right order.  After brining up the hci interface (but before pairing/connecting), I applied the VS command as you listed.  I then paired and connected.  I used aplay on the board to play a wav file through the bluealsa card and it worked!  I could hear that audio on my Jitsi audio and it sounded just fine.  Also, hciconfig shows both TX and RX data going through now.

> $ hciconfig
hci0:   Type: Primary  Bus: UART
        BD Address: xx:xx:xx:xx:xx:xx  ACL MTU: 1021:6  SCO MTU: 180:4
        UP RUNNING PSCAN
        RX bytes:7254506 acl:77 sco:113216 events:172 errors:0
        TX bytes:300807 acl:63 sco:10622 commands:65 errors:0

FWIW, I get an error from aplay if the phone is not dialed out somewhere (like to Jitsi).  I suspect that's expected.  The only problem I have now is that recordings don't playback any audio.  I used bluealsa-aplay to get the setup:

>$ bluealsa-aplay -L
bluealsa:SRV=org.bluealsa,DEV=xx:xx:xx:xx:xx:xx,PROFILE=sco
    moto g play (2021), trusted phone, playback
    SCO (mSBC): S16_LE 1 channel 16000 Hz
bluealsa:SRV=org.bluealsa,DEV=xx:xx:xx:xx:xx:xx,PROFILE=sco
    moto g play (2021), trusted phone, capture
    SCO (mSBC): S16_LE 1 channel 16000 Hz

And use this arecord command:
> arecord -f S16_LE -r 16000 -D bluealsa:xx:xx:xx:xx:xx:xx,sco /tmp/capture.wav

When transferred to my desktop and played with mpv the audio is blank - no sound.  

>$ mpv capture.wav 
[ffmpeg/demuxer] wav: Estimating duration from bitrate, this may be inaccurate
 (+) Audio --aid=1 (pcm_s16le 1ch 16000Hz)
AO: [pulse] 16000Hz mono 1ch s16
A: 00:00:05 / 00:00:07 (80%)
[ffmpeg/demuxer] wav: Packet corrupt (stream = 0, dts = NOPTS).
A: 00:00:07 / 00:00:07 (96%)

I verified that the Jitsi audio is working by disconnecting BT on the phone and making sure I could hear it there (I could).  

Your suggestion of using an ALSA driver to present a sound card for the TI I2S interface is interesting.  I may have to look into it.  We don't have an ADC/DAC on the board that I know of.  To my knowledge the I2S bus from the TI chip runs direct to the CPU.  I don't know what the user space devs were planning on doing with that, or what their expectations are for it.  I'm still digging around to understand that better.  I also don't know if I2S use is required yet - the guy who might know is out this week.

borine, 2022-07-13:
OK, thanks for that. My own agenda here is to find an HCI SCO routing command that reliably works for TI controllers so that it can potentially be called directly within bluealsa (in the same way as bluealsa currently does for Broadcom controllers).

@Arkq - would you be willing to accept such a change, assuming we can gather enough evidence to demonstrate that it is both "safe" and reliable?

The recorded file issue needs more investigation, as that looks as though it should have worked ok. Also I notice that you are in fact using mSBC codec, and I would not have expected that to be routed through the PCM interface anyway - it is not a PCM stream that passes from host to controller. So there are too many doubts here to convince me that we have proved the HCI command yet, but I hope you can keep this issue updated with progress until you have SCO working as you wish. The information is useful to this project, certainly for the purposes of improving the documentation and also potentially for enhancing the functionality.

Arkq, 2022-07-13:
That's fine for me :) We can add support for other chips as long as there will be no side effects (e.g. this vendor HCI command will not cause issues for other chips of the same vendor).

mjhammel, 2022-07-13:
I can work on this for awhile - it's not an MVP requirement but I don't have many other pressing issues (yet - they pop up all the time).  

Recording works - my testing has my phone and a Jitsi web session running here at my desk and when I was testing audio input between them (without BT) it was actually coming from my phone and not the web session.  Once I found that I needed to use Chrome (instead of Firefox) on Jitsi to select the proper mic everything worked as expected.  Talking to the Jitsi mic that passed through the phone to the board was recorded with proper audio.  So, caveat testing.  I will say I modifed the recording command slightly to match the characteristics of the wav file I use for sending from the board through the phone to Jitsi:

>arecord -D bluealsa:D8:CF:BF:D5:AE:BB,sco -r 44100 -f S16_LE -c 2 /tmp/capture.wav

I'm trying to match up your hcitool cmd to that doc.   Can you explain the ogf and ocf arguments to hcitool?  How do these map to HCI_VS_Write_SCO_Configuration (0xFE10)?

The cmd you gave is the same as the one on [this forum post](https://e2e.ti.com/support/wireless-connectivity/wi-fi-group/wifi/f/wi-fi-forum/705119/wl1835mod-vendor-specific-hci-commands).  I had tried that previously but it didn't work.  I now know I have to establish the hci interface first, then run the VS cmd BEFORE I pair/connect.  I was doing it after the pair/connect.  Not sure if that helps you any.

So this config, with your VS command, works with the hfp-ofono profile in bluealsa on the TI WL1831MOD as long as we're happy with HCI audio and don't need the BT_AUD_<IN/OUT> pins.  I've yet to establish this is actually what we want for this device, but that's a different problem.

Is there anything else I can help test for you that validates this command and/or setup?  Would you like me to document something on a wiki or elsewhere with what we found when using the WL1831MOD?

borine, 2022-07-14:
> Can you explain the ogf and ocf arguments to hcitool?

the HCI opcode is a 16-bit field within an HCI frame, and consists of 2 parts: a 6-bit "ogf" and a 10-bit "ocf". For whatever reason hcitool requires these as separate arguments, where the ogf has to be represented as a 8-bit integer and ocf as a 16-bit integer. So the formula is:

OGF = ((OPCODE & 0xFD00)>>10)
OCF = (OPCODE & 0x03ff)

The remaining arguments must be single 8-bit bytes representing the opcode parameter values. Multi-byte parameters (eg the 16-bit "TX max buffer latency" in this example) must be presented in little-endian order (ie least significant byte first).

> The cmd you gave is the same as the one...

Yes, the difference is that one sets explicit values for all the parameters, whereas mine says "change only the SCO route to Host, leave all other PCM config values unchanged". Given that my objective it to find a generic command that can always be invoked by bluealsa on a TI controller I thought it safer this way.

> Is there anything else I can help test for you that validates this command and/or setup?

I've already coded this HCI command addition to bluealsa, so I may you ask to to test that it then enables your board to work without requiring the `hcitool` step. I'll push it to my fork at some point, but I'm not quite ready to do it yet.

> Would you like me to document something on a wiki or elsewhere with what we found when using the WL1831MOD?

Yes please. SCO documentation is in short supply, and documentation of real oFono usage with bluealsa is like gold dust! Anything that I can link to from the bluealsa HFP/HSP wiki page https://github.com/Arkq/bluez-alsa/wiki/Using-BlueALSA-with-HFP-and-HSP-Devices would be very useful. Or you could extend or correct that page yourself as you think best.

mjhammel, 2022-07-16:
Okay, I'll take a look at the wiki and see what I can add.  Maybe a use-case example with the TI chip.

mjhammel, 2022-07-21:
FWIW, I added a use-case example of using bluealsa and ofono to test HFP to [Using BlueALSA with HFP and HSP Devices](https://github.com/Arkq/bluez-alsa/wiki/Using-BlueALSA-with-HFP-and-HSP-Devices).  Let me know if this needs clarification, simplification, expansion, etc.

I'm closing this issue since I'm now up and running.  Many thanks for the help @borine !

closed 2022-07-21 by mjhammel