arkq/bluez-alsa #581: make with enable ldac, but only enabled in a2dp source not sink.

18271211, 2022-07-21:
I just make with enable ldac, but the bin only enabled in a2dp source not sink.
It should be supported from 3.1.0,  is any package missed?
hope some one can have any idea on this. Thank you

Available BT profiles:
  - a2dp-source Advanced Audio Source (v1.3)
  - a2dp-sink   Advanced Audio Sink (v1.3)
  - hfp-ofono   Hands-Free AG/HF handled by oFono
  - hfp-ag      Hands-Free Audio Gateway (v1.7)
  - hfp-hf      Hands-Free (v1.7)
  - hsp-ag      Headset Audio Gateway (v1.2)
  - hsp-hs      Headset (v1.2)

Available BT audio codecs:
  a2dp-source:  SBC, AAC, aptX, aptX-HD, LDAC
  a2dp-sink:    SBC, AAC, aptX, aptX-HD
  hfp-*:        CVSD

libldacbt-abr-dev/stable,now 2.0.2.3+git20200429+ed310a0-4 arm64 [installed]
libldacbt-abr-dev/stable 2.0.2.3+git20200429+ed310a0-4 armhf
libldacbt-enc-dev/stable,now 2.0.2.3+git20200429+ed310a0-4 arm64 [installed]
libldacbt-enc-dev/stable 2.0.2.3+git20200429+ed310a0-4 armhf

bluez-alsa v3.1.0 (2021-06-01)
==============================

- higher PCM bit depth for apt-X HD (24-bit) and LDAC (32-bit)
- support for A2DP Sink with apt-X (HD) if decoder is available
- support for A2DP Sink with LDAC codec if decoder is available

Arkq, 2022-07-21:
> It should be supported from 3.1.0, is any package missed?

You need LDAC library which supports encoding and decoding. If you've got such, see https://github.com/Arkq/bluez-alsa/wiki/Installation-from-source At the end of "Additional Dependencies" you've got example how to use compile bluez-alsa with such library.

Also you may look at this: https://github.com/Arkq/bluez-alsa/issues/482

18271211, 2022-07-21:
Thank you Arkq
I just seen this in installation guide.
So I just need to remove the package provided from Debian and build from source completely.
Then it will support decoding, right?

(**) - ldacBT is available since Debian 11 (bullseye) and does not provide decoding support.

Arkq, 2022-07-21:
> So I just need to remove the package provided from Debian and build from source completely.

Yes, assuming that you have such library (the source code).

18271211, 2022-07-21:
yes, you are totally correct, I have no library.
I just make one of decoder in github but seems not working...

closed 2022-07-21 by 18271211