arkq/bluez-alsa #582: How to setup both a blue-alsa audio device and sound card for audio playback?

StonedEdge, 2022-07-24:
Hi all,

I've managed to get audio working through my AirPods via the inbuilt bluetooth module but I am struggling to get both my sound card and audio working out of bluetooth simultaneously. I'm currently not using pulseaudio (only ALSA, via BlueAlsa - not using pulseaudio as I find it slows down RetroPie significantly) - on a Raspberry Pi 4 running latest kernel. 

My current .asoundrc file looks like this:

```
pcm.!default {
  type asym
  playback.pcm {
    type plug
    slave.pcm "output"
  }
  capture.pcm {
    type plug
    slave.pcm "input"
  }
}

pcm.output {
  type hw
  card 0
}

ctl.!default {
  type hw
  card 0
}

pcm.!default {
        type plug
        slave {
                pcm {
                        type bluealsa
                        device BC:B8:63:45:65:C0
                        profile "a2dp"
                }
        }
        hint {
                show on
                description "Apple AirPods"
        }
}
ctl.!default {
        type bluealsa
}
```

Is there any way to get both audio playing out of both devices, ideally whereby one audio device switches off (i.e the internal audio card - default device) when the bluetooth speakers are connected and vice versa? I think I need to shift the pcm.!default & ctl.!default contents somewhere else, but this doesn't seem to work for both audio cards (i.e sound only comes out of the bluetooth speaker with the above configurations) at the moment. Sorry for the nooby question, any help would be sincerely appreciated! 

Thank you.

borine, 2022-07-24:
ALSA uses only 1 definition for a given pcm name, the **last** one it sees when parsing the config. So in your case the pcm "default" has the `bluealsa` definition, and the first `asym` definiton is completely ignored. I'm afraid there isn't any easy answer for what you are trying to do when using only ALSA - this is a generic ALSA limitation and is one of the reasons that audio servers such as PulseAudio were introduced in the first place. There are some ideas relating to the use of BlueALSA as `pcm.default`, with a fallback option to soundcard, in the wiki article https://github.com/Arkq/bluez-alsa/wiki/Using-BlueALSA-as-default-ALSA-PCM. I'm not sure that is exactly what you want, and it does require some co-operation from the music player application, but its probably worth reading to get some idea of what can and cannot be done without having an audio server between the applications and ALSA.

borine, 2022-10-17:
This issue appears to be abandoned. Closing.

closed 2022-10-17 by borine