arkq/bluez-alsa #583: How do I install this correctly?

MelleNi, 2022-07-26:
I'm trying to install this on an rpi 4, but I keep running into missing packages. Most I can find out how to install online, but I cannot find out how to install 'sbc'. I tried following the guide on their page, but it doesn't work either. Does anyone know how I can install this?

Arkq, 2022-07-26:
Please check this wiki page: https://github.com/Arkq/bluez-alsa/wiki/Installation-from-source
It's a very comprehensive bluez-alsa installation guide for the most common Linux distributions.

MelleNi, 2022-07-26:
> Please check this wiki page: [Arkq/bluez-alsa/wiki/Installation-from-source](https://github.com/Arkq/bluez-alsa/wiki/Installation-from-source) It's a very comprehensive bluez-alsa installation guide for the most common Linux distributions.

Thank you, it mostly worked, but trying to install it with AAC doesn't seem to work.

```
libfaac-dev is already the newest version (1.29.9.2-2).
0 upgraded, 0 newly installed, 0 to remove and 157 not upgraded.
pi@raspberrypi:~/bluez-alsa/build $ ../configure --enable-aac --enable-mp3lame

...

checking for AAC... no
configure: error: Package requirements (fdk-aac >= 0.1.1) were not met:

No package 'fdk-aac' found

Consider adjusting the PKG_CONFIG_PATH environment variable if you
installed software in a non-standard prefix.

Alternatively, you may set the environment variables AAC_CFLAGS
and AAC_LIBS to avoid the need to call pkg-config.
See the pkg-config man page for more details.
```

Not sure what I'm doing wrong here

borine, 2022-07-26:
For AAC BlueALSA requires the Fraunhofer fdk-aac library, it does not work with libfaac. RaspberryPiOS does not include a pre-built version of libfdk-aac in its repository, so you will have to build it from source. There is a link to the source on the wiki page.

MelleNi, 2022-07-26:
> For AAC BlueALSA requires the Fraunhofer fdk-aac library, it does not work with libfaac. RaspberryPiOS does not include a pre-built version of libfdk-aac in its repository, so you will have to build it from source. There is a link to the source on the wiki page.

Is there any info on how to build that from the source? I have no experience doing this

borine, 2022-07-26:
```
git clone https://github.com/mstorsjo/fdk-aac.git
cd fdk-aac
./autogen.sh
./configure
make -j4
sudo make install
```

borine, 2022-10-17:
@MelleNi Have you had any success with this now? I would like to close this issue if it is no longer being worked on.

MelleNi, 2022-10-21:
I found some workaround, can't remember what it was, but no, it didn't work. I'm ditching rpi

closed 2022-10-21 by borine