arkq/bluez-alsa #584: Switching bluetooth speakers with mpd

macmpi, 2022-07-29:
Hi,

I'm using `mpd` player with `bluealsa` to play on bluetooth speakers, using default configuration as set by `/etc/alsa/conf.d/20-bluealsa.conf`. `mpd` config is as follows:
```
audio_output {
	type		"alsa"
	name		"bluealsa"
	device		"bluealsa"
	mixer_type	"software"
}

```
I manage speakers with some logic changing `defaults.bluealsa.device` MAC address depending on speaker I want to use.

I noticed the following.
If  speaker defined into `/etc/alsa/conf.d/20-bluealsa.conf` comes & go, just doing a `mpd` refresh (`mpc toggle; mpc toggle`) restores sound operation: sweet.

However, when I change speaker (updating `/etc/alsa/conf.d/20-bluealsa.conf` ), previous refresh does not help: I have to restart `mpd` service altogether, so that sounds plays.

I've read through [this](https://github.com/Arkq/bluez-alsa/wiki/Using-BlueALSA-as-default-ALSA-PCM) : is that an illustration of:
>It is important that the application closes then re-opens the default device for this to work - as noted above not all applications do this.

Would you recommend a softer alternative than restarting `mpd` service after switching speakers?
Thanks!

borine, 2022-07-29:
mpd does close its ALSA pcm handle when paused, so `toggle` should be suffcient to cause the default device to be re-opened. The issue you are having is with caching of the ALSA config, so mpd does not see the change to `defaults.bluealsa.device`. You need to force mpd to re-load that part of the configuration every time it opens a pcm. For that, use the environment variable `ALSA_CONFIG_PATH`. I would recommend having a file with just that one definition in it, say `~/asoundrc-bluealsa`. Then run mpd with the environment variable `ALSA_CONFIG_PATH=/usr/share/alsa/alsa.conf:$HOME/asoundrc-bluealsa`. Choose any path you like for the file, so long as mpd can read it.

macmpi, 2022-07-29:
Thanks for the hint, I'll try.
No sure I understand the `~/asoundrc-bluealsa` part and what it should contain: as of now, I only use `/etc/alsa/conf.d/20-bluealsa.conf` (app-specific embedded linux setup, no multi-user).

While I was typing it, I realized `mpd` has also a [`mpc` command to enable/disable/toggle outputs](https://www.musicpd.org/doc/mpc/html/#output-commands): maybe that would work too.

borine, 2022-07-29:
What you need to achieve is for mpd to re-load the definition of `defaults.bluealsa.device` every time you change it. By default, an alsa-lib application will read the config into memory only when a pcm is first opened, then apply that same config from memory cache after that. So changing the definition of any config key in a file makes no diffference until you re-start the application.  However, alsa-lib *does* check the timestamp of every file listed in `ALSA_CONFIG_PATH` (or of the file `/usr/share/alsa/alsa.conf` if that env var is not defined) then re-loads that file if it has changed since the last pcm open. So my suggestion is to have a file that contains ***only*** the definition of `defaults.bluealsa.pcm` and add that file to `ALSA_CONFIG_PATH`. That way `mpd` will re-load the definiton of `defaults.bluealsa.device` every time it opens a pcm and therefore should open the correct device,

BTW, if you are sure that only 1 bluetooth device is connected at any time, then the in-built default device `00:00:00:00:00:00` will always open that device without needing to refresh the ALSA config,

macmpi, 2022-07-30:
> However, alsa-lib _does_ check the timestamp of every file listed in `ALSA_CONFIG_PATH` (or of the file `/usr/share/alsa/alsa.conf` if that env var is not defined)

Thanks a lot for the very detailed explanation!
This led me to just do (after new speaker is connected and `defaults.bluealsa.device` entry modified in `/etc/alsa/conf.d/20-bluealsa.conf`):
```
touch /usr/share/alsa/alsa.conf
mpc -q toggle; mpc -q toggle
```
...and `mpd` now plays onto new speaker.
Thanks again for your prompt and detailed explanations!

borine, 2022-07-30:
Ok, that's great you have a working solution. For any other readers of this, I would still recommend the separate file approach, because:

1. Editing 20-bluealsa.conf is modifying a system file which is just generally bad practice
2. touching /usr/share/alsa.conf causes alsa-lib to completely re-parse every one of its configuration files every time a pcm is opened, which is just a waste of time and resources for the system, again bad practice, especially on embedded systems

However, it's your system, so only you can say what is best for it!

borine, 2022-07-30:
Hmm, this question just prompted me to do a quick test, and now I've just found out that alsa-lib re-reads all its config files if any one of the files in ALSA_CONFIG_PATH has changed - so ignore point 2 above, perhaps yours is the best solution on balance after all

closed 2022-07-30 by macmpi