arkq/bluez-alsa #585: SCO transport canceled just after start / unwanted bufferrized CANCEL_THREADS command

ubicore, 2022-08-01:
Version : v4.0.0

**Description :**

In ba-transport.c layer when no transport was already started (no thread_enc and thread_dec threads was created), if there is a new incoming SCO link in sco_dispatcher_thread() in sco.c function :
- the call to ba_transport_stop() write in the thread_manager_pipe[1] a BA_TRANSPORT_THREAD_MANAGER_CANCEL_THREADS command  although no thread (thread_enc or thread_dec) was created.

Then, in sco_dispatcher_thread(), the ba_transport_start() function is called to create the thread_enc and thread_dec threads.
This both threads are executed first by the scheduler, but as soon as they are waiting with an io_poll_something call the scheduler will resume the transport_thread_manager() thread that immediately poll the pipe to get the buffeurized BA_TRANSPORT_THREAD_MANAGER_CANCEL_THREADS cmd.

=> That why an "unwanted" call to transport_threads_cancel() is executed. 

**To reproduce**

1. start bluez-alsa : bluealsa -p a2dp-sink -p hsp-hs
2. connecte a Android phone throught BT for audio and call
3. make a call from the phone
4. monitor with btmon
5. A soon as the SCO link is open, it is closed by the bluez-alsa side, re-open, re-closed....

**Traces:**
bluez-alsa in debug :
```
Jun 10 09:58:26 localhost user.debug /usr/bin/bluealsa: ../../../../../../../workspace/sources/bluez-alsa/src/sco.c:105: New incoming SCO link: DC:F7:56:83:2D:53: 25
Jun 10 09:58:26 localhost user.debug /usr/bin/bluealsa: ../../../../../../../workspace/sources/bluez-alsa/src/hci.c:133: SCO link socket MTU: 25: 64
Jun 10 09:58:26 localhost user.debug /usr/bin/bluealsa: ../../../../../../../workspace/sources/bluez-alsa/src/ba-transport.c:1174: Starting transport: HSP Headset
Jun 10 09:58:26 localhost user.debug /usr/bin/bluealsa: ../../../../../../../workspace/sources/bluez-alsa/src/ba-transport.c:297: Created BT socket duplicate: [25]: 26
Jun 10 09:58:26 localhost user.debug /usr/bin/bluealsa: ../../../../../../../workspace/sources/bluez-alsa/src/ba-transport.c:1494: Created new IO thread [ba-sco-enc]: HSP Headset
Jun 10 09:58:26 localhost user.debug /usr/bin/bluealsa: ../../../../../../../workspace/sources/bluez-alsa/src/ba-transport.c:297: Created BT socket duplicate: [25]: 27
Jun 10 09:58:26 localhost user.debug /usr/bin/bluealsa: ../../../../../../../workspace/sources/bluez-alsa/src/ba-transport.c:1494: Created new IO thread [ba-sco-dec]: HSP Headset
Jun 10 09:58:26 localhost user.debug /usr/bin/bluealsa: ../../../../../../../workspace/sources/bluez-alsa/src/sco.c:236: IO loop: START: sco_cvsd_enc_thread: HSP Headset
Jun 10 09:58:26 localhost user.debug /usr/bin/bluealsa: ../../../../../../../workspace/sources/bluez-alsa/src/ba-transport.c:309: Closing BT socket duplicate [25]: 26
Jun 10 09:58:26 localhost user.debug /usr/bin/bluealsa: ../../../../../../../workspace/sources/bluez-alsa/src/ba-transport.c:737: Releasing SCO link: 25
Jun 10 09:58:26 localhost user.debug /usr/bin/bluealsa: ../../../../../../../workspace/sources/bluez-alsa/src/ba-transport.c:1519: Exiting IO thread [ba-sco-enc]: HSP Headset
Jun 10 09:58:26 localhost user.debug /usr/bin/bluealsa: ../../../../../../../workspace/sources/bluez-alsa/src/sco.c:307: IO loop: START: sco_cvsd_dec_thread: HSP Headset
Jun 10 09:58:26 localhost user.debug /usr/bin/bluealsa: ../../../../../../../workspace/sources/bluez-alsa/src/ba-transport.c:309: Closing BT socket duplicate [-1]: 27
Jun 10 09:58:26 localhost user.debug /usr/bin/bluealsa: ../../../../../../../workspace/sources/bluez-alsa/src/ba-transport.c:1519: Exiting IO thread [ba-sco-dec]: HSP Headset
Jun 10 09:58:26 localhost user.debug /usr/bin/bluealsa: ../../../../../../../workspace/sources/bluez-alsa/src/at.c:162: AT message: RESP: command:+VGS, value: 15
```

btmon:
```
> HCI Event: Connect Request (0x04) plen 10               #179 [hci0] 25.269374
        Address: DC:F7:56:83:2D:53 (Samsung Electronics Co.,Ltd)
        Class: 0x5a020c
          Major class: Phone (cellular, cordless, payphone, modem)
          Minor class: Smart phone
          Networking (LAN, Ad hoc)
          Capturing (Scanner, Microphone)
          Object Transfer (v-Inbox, v-Folder)
          Telephony (Cordless telephony, Modem, Headset)
        Link type: eSCO (0x02)
< HCI Command: Accept Synchronou.. (0x01|0x0029) plen 21  #180 [hci0] 25.269595
        Address: DC:F7:56:83:2D:53 (Samsung Electronics Co.,Ltd)
        Transmit bandwidth: 8000
        Receive bandwidth: 8000
        Max latency: 65535
        Setting: 0x0060
          Input Coding: Linear
          Input Data Format: 2's complement
          Input Sample Size: 16-bit
          # of bits padding at MSB: 0
          Air Coding Format: CVSD
        Retransmission effort: Don't care (0xff)
        Packet type: 0x003f
          HV1 may be used
          HV2 may be used
          HV3 may be used
          EV3 may be used
          EV4 may be used
          EV5 may be used
> HCI Event: Command Status (0x0f) plen 4                 #181 [hci0] 25.270374
      Accept Synchronous Connection Request (0x01|0x0029) ncmd 1
        Status: Success (0x00)
> HCI Event: Synchronous Connect Complete (0x2c) plen 17  #182 [hci0] 25.291663
        Status: Success (0x00)
        Handle: 6
        Address: DC:F7:56:83:2D:53 (Samsung Electronics Co.,Ltd)
        Link type: eSCO (0x02)
        Transmission interval: 0x0c
        Retransmission window: 0x02
        RX packet length: 60
        TX packet length: 60
        Air mode: CVSD (0x02)
> SCO Data RX: Handle 6 flags 0x00 dlen 60                #183 [hci0] 25.302135
> SCO Data RX: Handle 6 flags 0x00 dlen 60                #184 [hci0] 25.302286
< HCI Command: Disconnect (0x01|0x0006) plen 3            #185 [hci0] 25.305034
        Handle: 6
        Reason: Remote User Terminated Connection (0x13)
> HCI Event: Command Status (0x0f) plen 4                 #186 [hci0] 25.307962
      Disconnect (0x01|0x0006) ncmd 1
        Status: Success (0x00)
> HCI Event: Disconnect Complete (0x05) plen 4            #187 [hci0] 25.334185
        Status: Success (0x00)
        Handle: 6
        Reason: Connection Terminated By Local Host (0x16)
```

**Proposition to fix :**
Test if there an active transport layer exist in ba_transport_stop()
and if not, do not send the BA_TRANSPORT_THREAD_MANAGER_CANCEL_THREADS command

```
int ba_transport_stop(struct ba_transport *t) {

	if (pthread_equal(t->thread_enc.id, config.main_thread) &&
			pthread_equal(t->thread_dec.id, config.main_thread)) {
		debug("no active transport layer");
		return 0;
	}

	transport_thread_manager_send_command(t, BA_TRANSPORT_THREAD_MANAGER_CANCEL_THREADS);
	transport_thread_cancel_wait(&t->thread_enc);
	transport_thread_cancel_wait(&t->thread_dec);
	return 0;
}
```

Arkq, 2022-08-01:
OK, I'll look at that maybe tomorrow (but realistically, during coming weekend).

borine, 2023-01-26:
@Arkq This issue appears to have been forgotten. Although I haven't been able to reproduce the failure in my own system, I think there is clearly a race condition here that still exists in the current master branch.

The problem is that transport_thread_cancel_wait() is a no-op when the t->thread_{enc,dec} == config.main_thread. So the sco dispatcher thread does not wait for the thread manager thread to process the BA_TRANSPORT_THREAD_MANAGER_CANCEL_THREADS command before creating the new dec/enc threads.

The solution proposed by @ubicore does not lock the thread mutexes while checking the thread ids, so may still lose the race in some circumstances, but with that addition it would seem to be correct.

arkq, 2023-01-30:
Hmm... indeed it seems that I've forgotten about that issue... Unfortunately, right now I'm not able to test it, because, due to unexpected chain of events, I've fried my laptop and RPi :D Now I'm waiting for new hardware but the shipment is delayed... I hope that I will be able to test things (on real hardware) again in the next week...

arkq, 2023-02-08:
I've done some refactoring of transport threads management. I hope that it will solve this unwanted cancellation issue. The code is pushed to https://github.com/arkq/bluez-alsa/tree/state-management branch. @ubicore if you are still using this project, could you verify whether it will work for you? I've checked it on my own and it seems to work. However, there is still an issue with not terminated encoder thread, but it will be fixed in the follow-up commit based on the state management refactor.

ubicore, 2023-02-13:
thanks, I try to test it this week.

arkq, 2023-02-13:
I've just merged the `state-management` branch (with some fixes) into the master branch. According to my testing, all changes to transport state management are stable enough (I hope). Anyway, please test it on your own and give here some feedback :)

ubicore, 2023-02-28:
@arkq
Thanks for your work, but unfortunately the issue is pretty alway the same, except that now that there is also an assert in the transport_thread_cancel function, and that the app now crashes... 

I added some debug trace in  ba-transport.c : 

[debug_trace_patch.txt](https://github.com/arkq/bluez-alsa/files/10851815/debug_trace_patch.txt)

My result log :

[debug_trace.txt](https://github.com/arkq/bluez-alsa/files/10851782/debug_trace.txt)

ubicore, 2023-02-28:
To be able to work on I added a check in ba_transport_stop() to see if thread are alreaddy in BA_TRANSPORT_THREAD_STATE_TERMINATED sate (patch hand-tampered ):

[check_not_terminated.patch.txt](https://github.com/arkq/bluez-alsa/files/10851976/check_not_terminated.patch.txt)

It seem to do the job for my test

arkq, 2023-02-28:
What is the platform that you are running it on? Currently, for testing I'm using RPi4, an on that board I've got something like this without a crash. For the record, I'm testing it with a call to a non-existing number, so I've got a quick start/stop events (call is terminated after 2 seconds). 
```
bluealsa: 122.218813576: [40065] D: ../../src/sco.c:116: New incoming SCO link: D0:16:B4:25:BB:AE: 24
bluealsa: 122.228131261: [40065] D: ../../src/hci.c:134: SCO link socket MTU: 24: 64
bluealsa: 122.228308742: [40065] D: ../../src/ba-transport.c:1470: Starting transport: HSP Headset
bluealsa: 122.228438687: [40065] D: ../../src/ba-transport.c:418: Created BT socket duplicate: [24]: 27
bluealsa: 122.228798464: [40314] D: ../../src/sco.c:313: IO loop: START: sco_cvsd_dec_thread: HSP Headset
bluealsa: 122.228864353: [40065] D: ../../src/ba-transport.c:1844: Created new IO thread [ba-sco-dec]: HSP Headset
bluealsa: 122.229213853: [40065] D: ../../src/ba-transport.c:418: Created BT socket duplicate: [24]: 28
bluealsa: 122.229794779: [40065] D: ../../src/ba-transport.c:1844: Created new IO thread [ba-sco-enc]: HSP Headset
bluealsa: 122.229807205: [40315] D: ../../src/sco.c:244: IO loop: START: sco_cvsd_enc_thread: HSP Headset
bluealsa: 124.323108426: [40314] E: ../../src/io.c:49: BT socket disconnected: Connection reset by peer
bluealsa: 124.323277222: [40314] D: ../../src/ba-transport.c:432: Closing BT socket duplicate [24]: 27
bluealsa: 124.323406222: [40314] D: ../../src/sco.c:342: IO loop: EXIT: sco_cvsd_dec_thread: HSP Headset
bluealsa: 124.324131611: [40314] D: ../../src/ba-transport.c:877: Releasing SCO link: 24
bluealsa: 124.324303760: [40314] D: ../../src/ba-transport.c:1878: Exiting IO thread [ba-sco-dec]: HSP Headset
bluealsa: 124.326337463: [40315] D: ../../src/ba-transport.c:432: Closing BT socket duplicate [-1]: 28
bluealsa: 124.326835315: [40315] D: ../../src/ba-transport.c:1878: Exiting IO thread [ba-sco-enc]: HSP Headset
```

I'll try to put delay in few places, maybe I will be able to reproduce it somehow....

ubicore, 2023-03-01:
It run on a stm32mp1, less powerful than the Rpi4 and with only 2 cores.
Maybe the speed of the RPi4 is masking the race condition.

Anyway, this really seems to be a design issue:
As long as the `sco_dispatcher_thread` will call the `ba_transport_stop` which put a `BA_TRANSPORT_THREAD_MANAGER_CANCEL_THREADS` in a pipe, there will be a risk that the processing of this command will be delayed (in fact every time there is no core of processor immediately available to run the `transport_thread_manager`).

To fix it:
- either the `ba_transport_stop` should not send the `BA_TRANSPORT_THREAD_MANAGER_CANCEL_THREADS` if the threads are already in the `BA_TRANSPORT_THREAD_STATE_TERMINATED` state, 
- either the `ba_transport_thread_state_wait_terminated` should replaced by a synchronization mechanism (dedicated semaphore) to be sure that the `transport_thread_manager` has processed the cmd in pipe before exiting `ba_transport_stop`

To reproduce it:
- maybe a delay in `transport_thread_manager` between the `read(fds[0].fd, &cmd, sizeof(cmd)) != sizeof(cmd)`
and the `transport_threads_cancel(t);` will help you to reproduce it

arkq, 2023-03-01:
> It run on a stm32mp1, less powerful than the Rpi4 and with only 2 cores.

Yes, that might be the case. I'll try to add delay in a place you've suggested.

> either the ba_transport_stop should not send the BA_TRANSPORT_THREAD_MANAGER_CANCEL_THREADS if the threads are already in the BA_TRANSPORT_THREAD_STATE_TERMINATED state,

Yes, it might work but only for that case. It will not cover a case where two threads are calling `ba_transport_stop()` from a valid state, so we will end up with 2 termination requests in a queue anyway. I'm more inclined to fix it using a "cookie". IO thread will have a unique cookie which will be incremented on state transition to TERMINATED. That cookie will be sent to a thread manager (in an action request) which will be compared with the current value in the thread structure. This will guarantee that all extra requests will be discarded.

closed 2023-03-15 by arkq

arkq, 2023-03-15:
> To reproduce it: maybe a delay in transport_thread_manager between [...]

Yeap, that worked like a charm :)

After some consideration and thinking of fixing the design of thread termination, I've decided to move forward with your suggestion of how to fix that particular issue. It's indeed the simplest way to avoid queueing up messages in the transport thread manager.

@ubicore , once again really sorry for not taking your suggestion more seriously at the first time... I hope that this time this issue is indeed resolved :D

ubicore, 2023-03-17:
@ark
I tested it yesterday and it seem to work correctly.
Thanks for your work !

commit 1f8e8fc3 (2023-02-08): Refactor transport thread state management

Instead of using pthread thread ID for checking if the thread is
running, use dedicated transport thread state variable.

Also, in order to cancel IO threads in a more reliable way, transport
thread manager will cancel the IO thread only if it was marked for
stopping. This will prevent unintended thread cancellation caused by
buffered commands in the transport thread manager queue.

Fixes #585
commit 19b7ca5e (2023-02-09): Refactor transport thread state management

Instead of using pthread thread ID for checking if the thread is
running, use dedicated transport thread state variable.

Also, in order to cancel IO threads in a more reliable way, transport
thread manager will cancel the IO thread only if it was marked for
stopping. This will prevent unintended thread cancellation caused by
buffered commands in the transport thread manager queue.

Fixes #585
commit 07701805 (2023-02-13): Refactor transport thread state management

Instead of using pthread thread ID for checking if the thread is
running, use dedicated transport thread state variable.

Also, in order to cancel IO threads in a more reliable way, transport
thread manager will cancel the IO thread only if it was marked for
stopping. This will prevent unintended thread cancellation caused by
buffered commands in the transport thread manager queue.

Fixes #585
commit 5466efc7 (2023-03-15): Fix transport stopping for terminated threads

In case when IO threads are already terminated, we can not send
cancel threads request to thread manager and wait for state change
of IO threads. In that case state transition from terminated to
stopping is not valid, so the wait function will not block, and
we will leave cancel threads request in the thread manager queue.
This can lead to unexpected IO threads termination in the nearest
future.

Fixes #585