arkq/bluez-alsa #588: LHDC support

anonymix007, 2022-08-12:
There are both decoder and encoder ARM libraries for [AOSP](https://gitlab.com/savitech-lhdc/savitech-lhdc-codec-aosp-q100r20/-/tree/lhdc_aosp_q100r20_main/external).
AFAIK they work for V2, V3 and V4. There's also [AOSP 12.1](https://gitlab.com/savitech-lhdc/savitech-lhdc-codec-aosp-12.1.0r1/-/tree/lhdc_aosp_1210r1_main) repo with V5, but libraries are not there.

yuv418, 2022-08-29:
`liblhdc.so` from that repository is only compiled for armv7 and armv8, which would create restrictions.

anonymix007, 2022-08-29:
One could always reverse engineer them or simply use QEMU

yuv418, 2022-08-29:
Reverse engineering should theoretically not be difficult—the libraries have not stripped their debugging symbols. I have worked with calling ARM libraries through QEMU, but it is not a great solution in my opinion.

andreadaoud, 2022-10-25:
https://gitlab.com/savitech-lhdc/savitech-lhdc-codec-aosp-12.1.0r1/-/blob/lhdc_aosp_1210r1_main/external/liblhdc/src/lhdcBT_enc.c

https://gitlab.com/savitech-lhdc/savitech-lhdc-codec-aosp-12.1.0r1/-/blob/lhdc_aosp_1210r1_main/external/liblhdcdec/src/lhdcBT_dec.c

https://gitlab.com/savitech-lhdc/savitech-lhdc-codec-aosp-12.1.0r1/-/blob/lhdc_aosp_1210r1_main/external/liblhdcv5/src/lhdcv5BT_enc.c

https://gitlab.com/savitech-lhdc/savitech-lhdc-codec-aosp-12.1.0r1/-/blob/lhdc_aosp_1210r1_main/external/liblhdcv5dec/src/lhdcv5BT_dec.c

This seems to be the source code? no reverse engineer is required.

zhubinsheng, 2023-06-29:
u want [this](https://dumps.tadiphone.dev/dumps/nothing/spacewar/-/blob/qssi-user-13-TKQ1.220915.002-1671077980-release-keys/system/system/lib/liblhdcv5.so) ?

anonymix007, 2023-06-29:
@zhubinsheng that's an old version, I've already got latest v2/3/4 and v5 libs from 1+11 and slowly reverse engineering them. I haven't yet touched v5 as there's certainly a newer version (according to gitlab), but it wasn't used by any smartphone manufacturers.

anonymix007, 2023-09-19:
@arkq I have reverse engineered LHDC encoder library. I don't want to publish poorely tested code (and I'm sure it's broken at least in some cases). 
Currently my library supports 2/3/4 encoding and exposes same API as library from gitlab, some "Extra API" functions are stubs (mostly AR; JAS and META should work, but they don't seem to do much anyway). AR is not supported and probably never will be. It would be easier to just directly use [SAF](https://github.com/leomccormack/Spatial_Audio_Framework) for Spatial Audio (which is what I would've done if I've ever wanted to get feature-complete liblhdc). Handling Spatial Audio has to be done on the level of BlueALSA anyway as head tracking data needs to be acquired via Bluetooth. I have OnePlus Buds Pro 2 with this feature, so can test it if you'll ever decide to work on that.
LLAC is currently not supported. Decoding of versions 2/3/4 will probably be supported later.

PR #672 (2024-11-14): Add support for LHDC V3 A2DP source and sink
Closes #588

~~Source does not work, only some buzzing noise is produced instead of music. I've tried saving pcm.data as raw pcm and it seems to be incorrect already.
Library expects interleaved 24 or 16 bit data depending on selected bit depth, so probably related to the use of BA_TRANSPORT_PCM_FORMAT_S24_4LE.~~ It seems to be somewhat working now, but not yet perfect.
closed 2024-11-14 by arkq