arkq/bluez-alsa #592: Seemingly random test timeout failures

borine, 2022-09-08:
Sometimes the test-alsa-pcm test suite fails with a timeout when closing a pcm. This patch appears to fix it:
```diff
diff --git a/test/bluealsa-mock.c b/test/bluealsa-mock.c
index 6082cbe..ef4683e 100644
--- a/test/bluealsa-mock.c
+++ b/test/bluealsa-mock.c
@@ -205,6 +205,10 @@ static void *mock_a2dp_dec(struct ba_transport_thread *th) {
 		if (io_pcm_write(t_a2dp_pcm, buffer, samples) == -1)
 			error("FIFO write error: %s", strerror(errno));
 
+		/* If io_pcm_write() gets an EAGAIN error is will return with thread
+		* cancellation disabled. So we make sure it is re-enabled here. */
+		pthread_setcancelstate(PTHREAD_CANCEL_ENABLE, NULL);
+
 		/* maintain constant speed */
 		asrsync_sync(&asrs, frames);
```

Arkq, 2022-09-08:
> Sometimes the test-alsa-pcm test suite fails with a timeout when closing a pcm

Yeap... unfortunately, I've got no time to check what causes it. It seems that you are right that the problem might be here. So, when the timeout happens the mock server hangs in the `poll()` on line 180 [here](https://github.com/Arkq/bluez-alsa/blob/cf52310c9e453bf841cab468ec02ac8c764b2d9e/test/bluealsa-mock.c#L180)? If that's true, I think the the proper fix is to wrap `poll()` with PTHREAD_CANCEL_ENABLE/DISABLE. The `io_pcm_write()` should *always* return with cancellation disabled. Also, `mock_a2dp_dec()` should have cancellation disabled at the very begging like every other transport thread function. I'll try to reproduce it locally... but the problem is that as far as I remember, running PCM test in a loop (like this: `while ./test/test-alsa-pcm; do true; done`) on my local PC does not trigger this timeout...

borine, 2022-09-08:
> The io_pcm_write() should always return with cancellation disabled

Actually it returns with the thread cancellation state unchanged, except when the FIFO is full in which case it sets it to PTHREAD_CANCEL_DISABLE

> mock_a2dp_dec() should have cancellation disabled at the very begging like every other transport thread 

I have a reproducible test here (working on some changes to ALSA plugin capture logic) and will try this suggestion later today.

> my local PC does not trigger this timeout

You need to specifically ensure that the FIFO becomes full, which depends on the relative timing of server and client, so is dependent on CPU load, scheduler, codec frame size, client period size, etc etc ...

Arkq, 2022-09-08:
> Actually it returns with the thread cancellation state unchanged

Yes, of course you are right :) But the (my) assumption is that the cancellation is disabled globally and enabled only in "critical" sections. This should ensure that there is no memory leakage when cancellation happens. Before the io.c file was created, transport thread could fail with "undefined" cancellation state. For that reason, before running `pthread_cleanup_pop()` functions, the cancellation is explicitly disabled. However, I think that right now it should be safe to remove this, or maybe add assert that the previous state was disabled.

> so is dependent on CPU load, scheduler, codec frame size, client period size, etc etc 

Yeap... but when I'm running test in a loop I'm using my PC (it's not idle), and test works for hours... But on GitHub it fails every N checks where N < 10 :D So, that's why I have said that my local PC does not trigger it ;p

borine, 2022-09-08:
Try this patch instead - it uses your global disable suggestion, and it also fixes my alsa-capture test failure:
```diff
diff --git a/test/bluealsa-mock.c b/test/bluealsa-mock.c
index 6082cbe..29fa00a 100644
--- a/test/bluealsa-mock.c
+++ b/test/bluealsa-mock.c
@@ -152,6 +152,7 @@ void bluez_battery_provider_update(struct ba_device *device) {
 
 static void *mock_a2dp_dec(struct ba_transport_thread *th) {
 
+	pthread_setcancelstate(PTHREAD_CANCEL_DISABLE, NULL);
 	pthread_cleanup_push(PTHREAD_CLEANUP(ba_transport_thread_cleanup), th);
 
 	struct ba_transport *t = th->t;
@@ -177,6 +178,7 @@ static void *mock_a2dp_dec(struct ba_transport_thread *th) {
 		if (!ba_transport_pcm_is_active(t_a2dp_pcm))
 			timeout = -1;
 
+		pthread_setcancelstate(PTHREAD_CANCEL_ENABLE, NULL);
 		if (poll(fds, ARRAYSIZE(fds), timeout) == 1 &&
 				fds[0].revents & POLLIN) {
 			/* dispatch incoming event */
@@ -191,6 +193,7 @@ static void *mock_a2dp_dec(struct ba_transport_thread *th) {
 				continue;
 			}
 		}
+		pthread_setcancelstate(PTHREAD_CANCEL_DISABLE, NULL);
 
 		fprintf(stderr, ".");
```
(apologies for submitting patches this way, I am not able to create PR right now)

commit 3b4fc5fe (2022-09-08): Fix seemingly random PCM test timeout failures

Closes #592
closed 2022-09-08 by Arkq