arkq/bluez-alsa #594: Couldn't open PCM: Acquire transport: Connection timed out

bmt1596, 2022-09-22:
i everyone, I am currently trying to connect my funk-radio to an Embedded device configured as Headset, on which Bluealsa is installed. I initialized Bluealsa as Headet AG and got the following Debug:

```
bluealsa: D: ../../src/main.c:586: Acquiring D-Bus service name: org.bluealsa
bluealsa: D: ../../src/main.c:595: Starting main dispatching loop
bluealsa: D: ../../src/main.c:115: Acquired D-Bus service name: org.bluealsa
bluealsa: D: ../../src/bluealsa-dbus.c:426: Registering D-Bus manager: /org/bluealsa
bluealsa: D: ../../src/bluez.c:688: Registering battery provider: /org/bluez/hci0/battery
bluealsa: D: ../../src/bluez.c:708: BlueZ battery provider support not available
bluealsa: D: ../../src/bluez.c:502: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/source/1
bluealsa: D: ../../src/bluez.c:411: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/source/1
bluealsa: D: ../../src/bluez.c:502: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/source/2
bluealsa: D: ../../src/bluez.c:411: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/source/2
bluealsa: D: ../../src/bluez.c:502: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/sink/1
bluealsa: D: ../../src/bluez.c:411: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/sink/1
bluealsa: D: ../../src/bluez.c:502: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/sink/2
bluealsa: D: ../../src/bluez.c:411: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/sink/2
bluealsa: D: ../../src/bluez.c:937: Creating hands-free profile object: /org/bluez/HSP/Headset
bluealsa: D: ../../src/bluez.c:861: Registering hands-free profile: /org/bluez/HSP/Headset
bluealsa: D: ../../src/bluez.c:937: Creating hands-free profile object: /org/bluez/HSP/AudioGateway
bluealsa: D: ../../src/bluez.c:861: Registering hands-free profile: /org/bluez/HSP/AudioGateway
bluealsa: D: ../../src/bluez.c:937: Creating hands-free profile object: /org/bluez/HFP/HandsFree
bluealsa: D: ../../src/bluez.c:861: Registering hands-free profile: /org/bluez/HFP/HandsFree
bluealsa: D: ../../src/bluez.c:937: Creating hands-free profile object: /org/bluez/HFP/AudioGateway
bluealsa: D: ../../src/bluez.c:861: Registering hands-free profile: /org/bluez/HFP/AudioGateway
```
I tried connecting funk-radio to the device and it shows the following messages:
```
bluealsa: D: ../../src/dbus.c:69: Called: org.bluez.Profile1.NewConnection() on /org/bluez/HSP/AudioGateway
bluealsa: D: ../../src/ba-rfcomm.c:1478: Created new RFCOMM thread [ba-rfcomm]: HSP Audio Gateway
bluealsa: D: ../../src/sco.c:221: Created SCO dispatcher [ba-sco-dispatch]: hci0
bluealsa: D: ../../src/bluez.c:778: HSP Audio Gateway configured for device 4C:CC:34:4B:12:0F
bluealsa: D: ../../src/ba-rfcomm.c:1066: Starting RFCOMM loop: HSP Audio Gateway
bluealsa: D: ../../src/ba-rfcomm.c:135: RFCOMM: HSP Audio Gateway state transition: 0 -> 9
bluealsa: D: ../../src/sco.c:87: Starting SCO dispatcher loop: hci0
```
Output of "bluealsa-aplay -lL"
```
bluealsa:SRV=org.bluealsa,DEV=4C:CC:34:4B:12:0F,PROFILE=sco
    KuNo 1114, trusted phone, capture
    SCO (mSBC): S16_LE 1 channel 16000 Hz
bluealsa:SRV=org.bluealsa,DEV=4C:CC:34:4B:12:0F,PROFILE=sco
    KuNo 1114, trusted phone, playback
    SCO (mSBC): S16_LE 1 channel 16000 Hz
**** List of PLAYBACK Bluetooth Devices ****
hci0: 4C:CC:34:4B:12:0F [KuNo 1114], trusted phone
  SCO (mSBC): S16_LE 1 channel 16000 Hz
**** List of CAPTURE Bluetooth Devices ****
hci0: 4C:CC:34:4B:12:0F [KuNo 1114], trusted phone
  SCO (mSBC): S16_LE 1 channel 16000 Hz
```
Everything seems to be configured successfully, but when I try to record the conversation using arecord, it gives me the following error:
```
arecord -f cd -D bluealsa:PROFILE=sco Record_PhoneCall.wav

bluealsa: D: ../../src/at.c:162: AT message: RESP: command:, value:RING
D: ../../../src/asound/bluealsa-pcm.c:1309: Getting BlueALSA PCM: CAPTURE 00:00:00:00:00:00 sco
D: ../../../src/asound/bluealsa-pcm.c:1062: /org/bluealsa/hci0/dev_4C_CC_34_4B_12_0F/hspag/source: Setting constraints
Recording WAVE 'Record_PhoneCall.wav' : Signed 16 bit Little Endian, Rate 44100 Hz, Stereo
D: ../../../src/asound/bluealsa-pcm.c:532: /org/bluealsa/hci0/dev_4C_CC_34_4B_12_0F/hspag/source: Initializing HW
bluealsa: D: ../../src/dbus.c:69: Called: org.bluealsa.PCM1.Open() on /org/bluealsa/hci0/dev_4C_CC_34_4B_12_0F/hspag/source
bluealsa: E: ../../src/ba-transport.c:717: Couldn't establish SCO link: Connection timed out
D: ../../../src/asound/bluealsa-pcm.c:551: /org/bluealsa/hci0/dev_4C_CC_34_4B_12_0F/hspag/source: Couldn't open PCM: Acquire transport: Connection timed out
arecord: set_params:1382: Unable to install hw params:
ACCESS:  RW_INTERLEAVED
FORMAT:  S16_LE
SUBFORMAT:  STD
SAMPLE_BITS: 16
FRAME_BITS: 32
CHANNELS: 2
RATE: 44100
PERIOD_TIME: 125000
PERIOD_SIZE: (5512 5513)
PERIOD_BYTES: (22048 22052)
PERIODS: (3 5)
BUFFER_TIME: 500000
BUFFER_SIZE: 22050
BUFFER_BYTES: 88200
TICK_TIME: 0
D: ../../../src/asound/bluealsa-pcm.c:443: /org/bluealsa/hci0/dev_4C_CC_34_4B_12_0F/hspag/source: Closing
bluealsa: D: ../../src/ba-rfcomm.c:1425: RFCOMM disconnected: Connection reset by peer
bluealsa: D: ../../src/ba-rfcomm.c:1022: Closing RFCOMM: 25
bluealsa: D: ../../src/ba-rfcomm.c:1038: RFCOMM link lost quirk: Destroying SCO transport
bluealsa: D: ../../src/ba-transport.c:892: Freeing transport: HSP Audio Gateway
bluealsa: D: ../../src/ba-device.c:135: Freeing device: 4C:CC:34:4B:12:0F
```
Can someone explain to me why the arecord command is not working here? or am i missing the setting at some step? Thank you very much.

borine, 2022-09-23:
You have configured bluealsa as both HSP-AG and HFP-AG (which is fine and normal), so the Bluetooth system has selected HFP (which again is normal). However, HFP devices typically require the AG to initiate a call session before they enable the audio streams. For that to happen some addtional RFCOMM messages are required, but bluealsa does not implement any HFP call handling logic; so you will have to provide these RFCOMM messages by some other means. See the wiki page https://github.com/Arkq/bluez-alsa/wiki/Using-BlueALSA-with-HFP-and-HSP-Devices for more information.

borine, 2022-10-17:
I've just had another look at this, and I realize I read it completely wrong before (sorry). Your funk radio is in fact connecting the HSP profile, so call setup RFCOMM messages here are not relevant at all.

As an alternative theory, this may be due to the "button press" requirement of HSP. When an AG attempts to open an audio connection with the HS, the HS typically requires the user to press a button, or take some equivalent action, to "accept" the connection. For some devices, the audio connection is accepted immediately but the microphone is not enabled until the button press. In those cases a BlueALSA capture client will record silence initially, but a BlueALSA playback device will be able to send audio. For other devices the audio connection is not accepted so no audio can pass in either direction until the button is pressed. The Linux bluetooth layer will timeout the audio connection request after a while and I think this is what you are experiencing. I'm not familiar with the "funk radio" but maybe it flashes an LED or makes a sound when an HSP audio connection is received and the user is required to acknowledge that in some way to accept it?

borine, 2023-02-16:
@bmt1596 Is this issue still occurring with the latest BlueALSA code?  Commit 9d853ba fixes a bug that may be relevant here, as I see that the funk radio is connecting using HSP (so the codec should be CVSD) but your `bluealsa-aplay -lL` output shows that the mSBC codec is used (which can only happen with HFP). So I'm wondering whether the funk radio is actually trying to connect both HFP and HSP, a situation that caused failures in BlueALSA before that commit.

borine, 2023-03-17:
This issue appears to have been abandoned. Closing

closed 2023-03-17 by borine