arkq/bluez-alsa #599: Error compiling with LC3plus

Hunter-Phipps, 2022-11-11:
In file included from ../../src/a2dp-sbc.c:31:
../../src/utils.h:54:30: error: unknown type name ‘LC3PLUS_Error’
   54 | const char *lc3plus_strerror(LC3PLUS_Error err);
      |                              ^~~~~~~~~~~~~
make[3]: *** [Makefile:736: a2dp-sbc.o] Error 1

Arkq, 2022-11-14:
Do you have LC3plus (not LC3) library installed? The lc3.h from lc3plus defines `typdef enum { ... } LC3PLUS_Error;`.

Hunter-Phipps, 2022-11-14:
I compiled libLC3plus.so using the makefile changes in https://github.com/Arkq/bluez-alsa/issues/479 and copied it into the /usr/lib/ folder.

Arkq, 2022-11-15:
Please, check from where the `lc3.h` is included with command like this: `echo '#include <lc3.h>' | gcc -H -x c -` Then, check whether the included `lc3.h` file has LC3PLUS_Error definition.

Hunter-Phipps, 2022-11-15:
I copied lc3.h and lc3.c from the ETSI floating point source code to /usr/include/ and it looks like it compiled / installed without any issues but I still don't see LC3 in the pipewire compile.

Arkq, 2022-11-15:
> but I still don't see LC3 in the pipewire compile

Hmm... you might be right, because this repo is not for pipewire project :D

But if you meant that in pipewire LC3plus (not LC3) is not activated when connecting with bluealsa, make sure that LC3plus is enabled in bluealsa via command line options, i.e.: `bluealsa -p a2dp-source -c lc3plus`.

arkq, 2023-01-01:
Since there are no more replies, I understand that the issues is solved. Otherwise, please reopen it and provide more details about your setup.

closed 2023-01-01 by arkq