arkq/bluez-alsa #600: transport reference counting problem

borine, 2022-11-14:
There is a problem with commit 9d853baa73703c4f5db9cb9917f33426f737dd21 in that if a SCO device disconnects then on re-connect all SCO transports are rejected.

I've traced this to reference counting of the transport object - its reference count never seems to drop below 3 so the cleanup code in  ba_transport_unref() is never invoked. As a result, the transports hash table retains  the entry into the next connection, which is therefore rejected.

If this is intended behavior for the transport ref_count then perhaps that commit should be reverted until an alternative method for detecting multiple SCO transports can be devised.

Arkq, 2022-11-14:
> If this is intended behavior for the transport ref_count 

If I understand you correctly, no, that's not the intended behavior. The count should drop to 0. When the device disconnects the lifetime of the transport object should be over. If the refcount is not 0 by then something is definitely not right with transport lifetime management...

I can try to debug this as well. What is your test case? You are connecting sink (?) BT device to bluealsa with all (?) profiles (A2DP and e.g. HFP), then you are disconnecting BT device entirely or only HFP profile?

borine, 2022-11-14:
Testing with bluealsa<->bluealsa

On HF machine (this machine is set to trust AG in bluetoothctl):
```
bluealsa -p hfp-hf
```

On AG machine:
```
bluealsa -p hfp-ag
```
Then on AG machine:
```shell
$ bluetoothctl connect B8:27:EB:2B:8D:C3
Attempting to connect to B8:27:EB:2B:8D:C3
[CHG] Device B8:27:EB:2B:8D:C3 Connected: yes
Connection successful

$ bluealsa-cli list-pcms 
/org/bluealsa/hci0/dev_B8_27_EB_2B_8D_C3/hspag/sink
/org/bluealsa/hci0/dev_B8_27_EB_2B_8D_C3/hspag/source

$ bluetoothctl disconnect B8:27:EB:2B:8D:C3
Attempting to disconnect from B8:27:EB:2B:8D:C3
[CHG] Device B8:27:EB:2B:8D:C3 ServicesResolved: no
Successful disconnected

$ bluealsa-cli list-pcms

$ bluetoothctl connect B8:27:EB:2B:8D:C3
Attempting to connect to B8:27:EB:2B:8D:C3
[CHG] Device B8:27:EB:2B:8D:C3 Connected: yes
Failed to connect: org.bluez.Error.Failed br-connection-refused

$ bluealsa-cli list-pcms

$
```

borine, 2022-11-14:
Hmm - hold off on this one - the HF machine was not running master branch after all. Looks like a bug in one of my own experiments :(

borine, 2022-11-14:
Yes, sorry, this is not an issue with master branch.

closed 2022-11-14 by borine