arkq/bluez-alsa #611: Using bluealsa with jack on RPi4

anupam-git, 2023-01-13:
Hello,

I am trying to make bluetooth audio work with bluealsa and jack on a RPi 4.
For that I have tried connecting a bluetooth headphone after installing bluealsa which went fine. 

Now when I try to expose the bluealsa device as playback port with `alsa_out` I am getting the following error :
```
# alsa_out -d bluealsa
Unable to set hw params for playback: Invalid argument
Setting of hwparams failed: Invalid argument
```

Some more information :
```
# bluealsa-aplay --version
v4.0.0-88-g8d2c94d
```

```
# aplay -L
bluealsa
    Bluetooth Audio
```

```
# bluealsa-aplay -l
**** List of PLAYBACK Bluetooth Devices ****
hci0: 00:05:18:0C:18:83 [Rockerz 255 Pro+], trusted audio-card
  A2DP (AAC): S16_LE 2 channels 48000 Hz
**** List of CAPTURE Bluetooth Devices ****

# bluealsa-aplay -L
bluealsa:DEV=00:05:18:0C:18:83,PROFILE=a2dp,SRV=org.bluealsa
    Rockerz 255 Pro+, trusted audio-card, playback
    A2DP (AAC): S16_LE 2 channels 48000 Hz
```

```
# bluetoothctl info 00:05:18:0C:18:83
Device 00:05:18:0C:18:83 (public)
        Name: Rockerz 255 Pro+
        Alias: Rockerz 255 Pro+
        Class: 0x00240418
        Icon: audio-card
        Paired: yes
        Trusted: yes
        Blocked: no
        Connected: yes
        LegacyPairing: no
        UUID: Headset                   (00001108-0000-1000-8000-00805f9b34fb)
        UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
        UUID: Handsfree                 (0000111e-0000-1000-8000-00805f9b34fb)
        UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
        Modalias: bluetooth:v000ApFFFFdFFFF
```

```
Kernel : 5.10.103-v7l+
```

borine, 2023-01-13:
What version of ALSA are you using? I suspect you may be hitting this alsa-lib bug https://github.com/alsa-project/alsa-lib/commit/0222f45d11e8b71bf651b985b00fdb0addbf3eed - the linked fix appears in alsa-lib 1.2.8

In any case, I would recommend using `zita-j2a` rather than `alsa-out` if at all possible, it is much more robust code (it is from the same jack-example-tools repository as alsa-out).

There are some hints on using jack with bluealsa in the wiki https://github.com/arkq/bluez-alsa/wiki/Using-BlueALSA-with-the-JACK-Audio-Connection-Kit

anupam-git, 2023-01-13:
Thank you for the quick reply. I did try using zita-a2j but didn't quite get the arguments right and hence did not work.

Now with the above commands, zita-a2j seems to create 2 new jack ports although I dont hear any sound on the bluetooth headphone. Do you know any ways to debug that?

anupam-git, 2023-01-13:
Closing the issue as it is solved with correct `zita-a2j` command. Thanks again.

closed 2023-01-13 by anupam-git