arkq/bluez-alsa #612: I can't control volume of my device.

Jakes9, 2023-01-24:
Hi I have problem wit volume control am I don know how can I configure this. I have Shairport-sync connected with bluealsa and shairport set volume of my speaker in alsa mixer but in device volume is not changing. Is it possible to solve this problem ?

Pozwolę sobie tez opisać to po polsku. Używam bluealsa żeby połączyć shairport-sync z moimi głośnikami Bluetooth. Głośniki mają swoje sterowanie głośnością. Kiedy na iPhone zmieniam głośność, zmienia się ona tez w alsamixer ale nie na głośniku ( mam na nim wskaźnik głośności ). Udało mi się też uruchomić sterowanie głośnością w bluealsa tzn. Kiedy zmieniam głośność na głośniku to dostaje informacje zwrotne w konsoli. Jednak nie potrafię połączyć tego by sterowanie głośnością przyciskami na głośniku zmieniało poziom głośności w alsamixer. 
Arku czy możesz mi podpowiedzieć co robię nie tak ?

Mam uruchomione:
bluealsa --profile=a2dp-source -- a2dp-volume 

Pozdrawiam serdecznie

borine, 2023-01-24:
Please can you post the command-line used to start shairport-sync. You need to specify the volume control as well as the pcm device to tell shairport-sync where to send the audio and how to control its volume.

Jakes9, 2023-01-24:
Hi Borine thank You for your response. 
I have configured shairport in shairport-sync.conf 

output_device = "bluealsa";
mixer_control_name = name of my Bluetooth device

Shairport controls this device: when I changing volume on iPhone -> the volume of PCM is changing in alsamixer

But when I change volume on my Bluetooth speaker it's not changing on alsamixer.

When I change volume on speaker i have response in terminal

borine, 2023-01-24:
For your conf file, try:
```
alsa =
{
	output_device = "bluealsa"
	mixer_device = "bluealsa:00:00:00:00:00:00"
	mixer_control_name = "A2DP"
};
```
> When I change volume on speaker i have response in terminal

Please can you explain this a little more? do you mean in the debug output of bluealsa? or some other program?

Jakes9, 2023-01-24:
Yes, I have response in debug of bluealsa. I mean when I start bluealsa in terminal with command: 
sudo bluealsa -- profile=a2dp-source --a2dp-volume 
I have in this tab of terminal output of bluealsa and I see response for my volume changing on Bluetooth speaker.

Thank You for advice. I will try and see if this solve my problem.

borine, 2023-01-27:
> Yes, I have response in debug of bluealsa.

OK that's good, so the bluetooth device is exchanging volume control data with bluealsa, and therefore I believe that my suggested shairport-sync config above will enable shairport-sync to control the device volume (assuming that you only have one bluetooth device connected).

ALSA volume control by the bluealsa plugin is documented in the manual page https://github.com/arkq/bluez-alsa/blob/master/doc/bluealsa-plugins.7.rst#ctl-plugin

Using "00:00:00:00:00:00" enables single-device mode which simplifies control naming, so that the control is simply "A2DP" and does not include the device alias.

If you find that shairport-sync fails when the device disconnects, you could try changing the mixer_device to:
```
mixer_device = "bluealsa:DEV=00:00:00:00:00:00,DYN=no";
```
I've never used shairport-sync so I don't know whether this is necessary or not.

borine, 2023-01-27:
I should mention that for the "DYN=no" config you need to be using the latest master branch sources; that option was added after the last release (ie it is not in bluealsa 4.0.0 or earlier)

Jakes9, 2023-01-27:
Thank You Borine for support.
I tryed your solutions but it doesn't work.

Maybe it will be easier if we try to connect alsamixer volume control with Bluetooth speaker? 🤔

This is my problem.

Shairport-sync can set and change volume in alsamixer.

But in alsamixer I don't see changing when I set volume on my speaker with volume control buttons.
When I try setting in alsamixer by hand I don't see any changes on my speaker to.

So I think is probably problem wit connection between alsamixer and Bluetooth speaker volume control.

Maybe i configured something wrong but I don't know how to test and see is it ok.

borine, 2023-01-27:
With the bluetooth device connected, please can you post the output of:
```
amixer -D bluealsa:00:00:00:00:00:00 scontents
```
Now change the volume using the switch on the bluetooth device and then post the output of that same command again. I would like to see what controls are being created by bluealsa, and how they are affected when the volume is changed on the device.

Jakes9, 2023-01-30:
Hi i have what you ask for.

Homebridge@Dom:~$ amixer -D bluealsa:F4:4E:FD:AC:5B:C9 scontents
Simple mixer control 'A2DP',0
  Capabilities: pvolume pswitch
  Playback channels: Front Left - Front Right
  Limits: Playback 0 - 127
  Mono:
  Front Left: Playback 13 [10%] [-32.77dB] [on]
  Front Right: Playback 13 [10%] [-32.77dB] [on]
Homebridge@Dom:~$ amixer -D bluealsa:F4:4E:FD:AC:5B:C9 scontents
Simple mixer control 'A2DP',0
  Capabilities: pvolume pswitch
  Playback channels: Front Left - Front Right
  Limits: Playback 0 - 127
  Mono:
  Front Left: Playback 13 [10%] [-32.77dB] [on]
  Front Right: Playback 13 [10%] [-32.77dB] [on]
Homebridge@Dom:~$

Jakes9, 2023-01-30:
and this when i change volume on my iPhone 

amixer -D bluealsa:F4:4E:FD:AC:5B:C9 scontents
Simple mixer control 'A2DP',0
  Capabilities: pvolume pswitch
  Playback channels: Front Left - Front Right
  Limits: Playback 0 - 127
  Mono:
  Front Left: Playback 27 [21%] [-22.48dB] [on]
  Front Right: Playback 27 [21%] [-22.48dB] [on]
Homebridge@Dom:~$

Jakes9, 2023-01-30:
This is my log when i run bluealsa in terminal

Homebridge@Dom:~$ sudo bluealsa --profile=a2dp-source --a2dp-volume
[sudo] hasło użytkownika Homebridge: 
bluealsa: [1373] D: ../../src/main.c:590: Starting main dispatching loop
bluealsa: [1373] D: ../../src/main.c:114: Acquired D-Bus service name: org.bluealsa
bluealsa: [1373] D: ../../src/bluealsa-dbus.c:372: Registering D-Bus manager: /org/bluealsa
bluealsa: [1373] D: ../../src/bluez.c:770: Registering battery provider: /org/bluez/hci0/battery
bluealsa: [1373] D: ../../src/bluez.c:790: BlueZ battery provider support not available
bluealsa: [1373] D: ../../src/bluez.c:584: Creating media endpoint object: /org/bluez/hci0/A2DP/AAC/source/1
bluealsa: [1373] D: ../../src/bluez.c:495: Registering media endpoint: /org/bluez/hci0/A2DP/AAC/source/1
bluealsa: [1373] D: ../../src/bluez.c:584: Creating media endpoint object: /org/bluez/hci0/A2DP/AAC/source/2
bluealsa: [1373] D: ../../src/bluez.c:495: Registering media endpoint: /org/bluez/hci0/A2DP/AAC/source/2
bluealsa: [1373] D: ../../src/bluez.c:584: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/source/1
bluealsa: [1373] D: ../../src/bluez.c:495: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/source/1
bluealsa: [1373] D: ../../src/bluez.c:584: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/source/2
bluealsa: [1373] D: ../../src/bluez.c:495: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/source/2
bluealsa: [1373] D: ../../src/bluez.c:1136: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
bluealsa: [1373] D: ../../src/bluez.c:1232: Adding new Stream End-Point: F4:4E:FD:AC:5B:C9: SBC
bluealsa: [1373] D: ../../src/bluez.c:1136: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
bluealsa: [1373] D: ../../src/bluez.c:1232: Adding new Stream End-Point: F4:4E:FD:AC:5B:C9: SBC
bluealsa: [1373] D: ../../src/dbus.c:66: Called: org.bluez.MediaEndpoint1.SelectConfiguration() on /org/bluez/hci0/A2DP/SBC/source/1
bluealsa: [1373] D: ../../src/bluez.c:271: A2DP peer capabilities blob [len=4]: ffff0223
bluealsa: [1373] D: ../../src/bluez.c:1136: Signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded()
bluealsa: [1373] D: ../../src/dbus.c:66: Called: org.bluez.MediaEndpoint1.SetConfiguration() on /org/bluez/hci0/A2DP/SBC/source/1
bluealsa: [1373] D: ../../src/a2dp.c:369: Selected A2DP SBC bit-pool range: [2, 35]
bluealsa: [1373] D: ../../src/storage.c:102: Loading storage: /var/lib/bluealsa/F4:4E:FD:AC:5B:C9
bluealsa: [1373] D: ../../src/bluez.c:406: A2DP Source (SBC) configured for device F4:4E:FD:AC:5B:C9
bluealsa: [1373] D: ../../src/bluez.c:409: A2DP selected configuration blob [len=4]: 11150223
bluealsa: [1373] D: ../../src/bluez.c:411: PCM configuration: channels: 2, sampling: 48000
bluealsa: [1373] D: ../../src/bluez.c:584: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/source/3
bluealsa: [1373] D: ../../src/bluez.c:495: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/source/3
bluealsa: [1373] D: ../../src/bluez.c:1343: Signal: org.freedesktop.DBus.Properties.PropertiesChanged(): org.bluez.MediaTransport1: Volume
bluealsa: [1373] D: ../../src/bluez.c:1361: Skipping A2DP volume update: 73
bluealsa: [1384] D: ../../src/dbus.c:66: Called: org.bluealsa.PCM1.Open() on /org/bluealsa/hci0/dev_F4_4E_FD_AC_5B_C9/a2dpsrc/sink
bluealsa: [1384] D: ../../src/ba-transport.c:651: New A2DP transport: 18
bluealsa: [1384] D: ../../src/ba-transport.c:652: A2DP socket MTU: 18: R:672 W:672
bluealsa: [1373] D: ../../src/bluez.c:1343: Signal: org.freedesktop.DBus.Properties.PropertiesChanged(): org.bluez.MediaTransport1: State
bluealsa: [1373] D: ../../src/ba-transport.c:1271: Starting transport: A2DP Source (SBC)
bluealsa: [1373] D: ../../src/ba-transport.c:360: Created BT socket duplicate: [18]: 19
bluealsa: [1373] D: ../../src/ba-transport.c:1622: Created new IO thread [ba-a2dp-sbc]: A2DP Source (SBC)
bluealsa: [1385] D: ../../src/codec-sbc.c:234: SBC setup: 48000 Hz JointStereo allocation=SNR blocks=16 sub-bands=8 bit-pool=35 => 249000 bps
bluealsa: [1385] D: ../../src/a2dp-sbc.c:223: IO loop: START: a2dp_sbc_enc_thread: A2DP Source (SBC)

bluealsa: [1373] D: ../../src/ba-transport.c:1512: PCM resume: 14
bluealsa: [1373] D: ../../src/bluealsa-dbus.c:846: Setting volume: 3 [-54.03 dB] <> 13 [-32.88 dB]
bluealsa: [1373] D: ../../src/bluealsa-dbus.c:846: Setting volume: 3 [-54.03 dB] <> 3 [-54.03 dB]
bluealsa: [1373] D: ../../src/ba-transport.c:1553: PCM drop: 14
bluealsa: [1373] D: ../../src/ba-transport.c:1512: PCM resume: 14

**when i change the volume using the shairport (on my iPhone)**

bluealsa: [1373] D: ../../src/bluealsa-dbus.c:846: Setting volume: 4 [-49.88 dB] <> 3 [-54.03 dB]
bluealsa: [1373] D: ../../src/bluealsa-dbus.c:846: Setting volume: 4 [-49.88 dB] <> 4 [-49.88 dB]
bluealsa: [1373] D: ../../src/upower.c:36: Signal: org.freedesktop.DBus.Properties.PropertiesChanged()
bluealsa: [1373] D: ../../src/bluealsa-dbus.c:846: Setting volume: 6 [-44.03 dB] <> 4 [-49.88 dB]
bluealsa: [1373] D: ../../src/bluealsa-dbus.c:846: Setting volume: 6 [-44.03 dB] <> 6 [-44.03 dB]
bluealsa: [1373] D: ../../src/bluealsa-dbus.c:846: Setting volume: 8 [-39.88 dB] <> 6 [-44.03 dB]
bluealsa: [1373] D: ../../src/bluealsa-dbus.c:846: Setting volume: 8 [-39.88 dB] <> 8 [-39.88 dB]
bluealsa: [1373] D: ../../src/bluealsa-dbus.c:846: Setting volume: 13 [-32.88 dB] <> 8 [-39.88 dB]
[... cut ...]
bluealsa: [1373] D: ../../src/bluez.c:1361: Skipping A2DP volume update: 57
bluealsa: [1373] D: ../../src/bluez.c:1343: Signal: org.freedesktop.DBus.Properties.PropertiesChanged(): org.bluez.MediaTransport1: Volume
bluealsa: [1373] D: ../../src/bluez.c:1361: Skipping A2DP volume update: 49
bluealsa: [1373] D: ../../src/bluez.c:1343: Signal: org.freedesktop.DBus.Properties.PropertiesChanged(): org.bluez.MediaTransport1: Volume
bluealsa: [1373] D: ../../src/bluez.c:1361: Skipping A2DP volume update: 40

arkq, 2023-01-30:
> bluealsa: [1373] D: ../../src/bluez.c:1361: Skipping A2DP volume update: 65

That is the reason, the PCM is in the "software" mode. Please use:
```
bluealsa-cli soft-volume /org/bluealsa/hci0/dev_F4_4E_FD_AC_5B_C9/a2dpsrc/sink no
```

It seems that the manual page (and/or bluealsa help message) will have to be updated to describe this behavior in bold :D Right now the `--a2dp-volume` works only for the "first" connection. If the device was connected before this commadn line flag and later one will start `bluealsa --a2dp-volume`, the soft-volume will still be `true`, because it was restored from the previous (first) connection... and in such a case the flag `--a2dp-volume` will not apply. As I think of this now it is very "issue" prone :/

Jakes9, 2023-01-30:
Maybe it is stupid question but where i should use this command ?

Homebridge@Dom:~$ bluealsa-cli soft-volume /org/bluealsa/hci0/dev_F4_4E_FD_AC_5B_C9/a2dpsrc/sink no
bluealsa-cli: [4204] E: ../../../utils/cli/cmd-softvol.c:64: CMD "soft-volume": Invalid BlueALSA PCM path: /org/bluealsa/hci0/dev_F4_4E_FD_AC_5B_C9/a2dpsrc/sink

Jakes9, 2023-01-30:
OK It works Thank You Borine and Arek

closed 2023-01-30 by Jakes9

arkq, 2023-01-30:
OK, that's great that now it works for you.

@borine what do you think about dropping/deprecating `--a2dp-volume` option? For previous versions of bluez-alsa it was needed for switching PCM to native volume control for A2DP. Now, when there is a persistent storage and `...,EXT=on`, this option is not required, and what's worse it's behavior is misleading...

borine, 2023-01-30:
> what do you think about dropping/deprecating --a2dp-volume option?

Hmm, not sure. Definitely for a2dp-source systems this option is now redundant and can be misleading; for a2dp-sinks it still has a role for use with bluealsa-aplay. Perhaps an option should be added to bluealsa-aplay to say "use pass-through volume control for all PCMs" ?