arkq/bluez-alsa #614: Couldn't open PCM: Device or resource busy.

fizista, 2023-02-13:
I'm trying to record audio from my cell phone on my computer. Unfortunately I get something like this:

```
> arecord -D bluealsa:34:1C:F0:F6:76:07 -f s16_le -c 2 -r 48000 --dump-hw-params out.wav
D: bluealsa-pcm.c:1309: Getting BlueALSA PCM: CAPTURE 34:1C:F0:F6:76:07 a2dp
D: bluealsa-pcm.c:1062: /org/bluealsa/hci0/dev_34_1C_F0_F6_76_07/a2dpsnk/source: Setting constraints
Recording WAVE 'out.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
HW Params of device "bluealsa:34:1C:F0:F6:76:07":
--------------------
ACCESS:  MMAP_INTERLEAVED MMAP_NONINTERLEAVED MMAP_COMPLEX RW_INTERLEAVED RW_NONINTERLEAVED
FORMAT:  S8 U8 S16_LE S16_BE U16_LE U16_BE S24_LE S24_BE U24_LE U24_BE S32_LE S32_BE U32_LE U32_BE FLOAT_LE FLOAT_BE FLOAT64_LE FLOAT64_BE MU_LAW A_LAW IMA_ADPCM S20_LE S20_BE U20_LE U20_BE S24_3LE S24_3BE U24_3LE U24_3BE S20_3LE S20_3BE U20_3LE U20_3BE S18_3LE S18_3BE U18_3LE U18_3BE
SUBFORMAT:  STD
SAMPLE_BITS: [4 64]
FRAME_BITS: [4 640000]
CHANNELS: [1 10000]
RATE: [4000 4294967295)
PERIOD_TIME: [10000 5944309)
PERIOD_SIZE: [40 4294967295)
PERIOD_BYTES: [20 4294967295)
PERIODS: (0 107374183)
BUFFER_TIME: [1 4294967295]
BUFFER_SIZE: [80 4294967294]
BUFFER_BYTES: [40 4294967295]
TICK_TIME: ALL
--------------------
D: bluealsa-pcm.c:532: /org/bluealsa/hci0/dev_34_1C_F0_F6_76_07/a2dpsnk/source: Initializing HW
**D: bluealsa-pcm.c:551: /org/bluealsa/hci0/dev_34_1C_F0_F6_76_07/a2dpsnk/source: Couldn't open PCM: Device or resource busy**
arecord: set_params:1416: Unable to install hw params:
ACCESS:  RW_INTERLEAVED
FORMAT:  S16_LE
SUBFORMAT:  STD
SAMPLE_BITS: 16
FRAME_BITS: 32
CHANNELS: 2
RATE: 48000
PERIOD_TIME: 125000
PERIOD_SIZE: 6000
PERIOD_BYTES: 24000
PERIODS: 4
BUFFER_TIME: 500000
BUFFER_SIZE: NONE
BUFFER_BYTES: 96000
TICK_TIME: 0
D: bluealsa-pcm.c:443: /org/bluealsa/hci0/dev_34_1C_F0_F6_76_07/a2dpsnk/source: Closing
```

```
> bluetoothctl
[bluetooth]# connect 34:1C:F0:F6:76:07
Attempting to connect to 34:1C:F0:F6:76:07
[CHG] Device 34:1C:F0:F6:76:07 Connected: yes
[NEW] Endpoint /org/bluez/hci0/dev_34_1C_F0_F6_76_07/sep1 
[NEW] Transport /org/bluez/hci0/dev_34_1C_F0_F6_76_07/sep1/fd6 
Connection successful
[NEW] Player /org/bluez/hci0/dev_34_1C_F0_F6_76_07/player0 [default]
[CHG] Device 34:1C:F0:F6:76:07 ServicesResolved: yes
[CHG] Player /org/bluez/hci0/dev_34_1C_F0_F6_76_07/player0 Type: Audio
[CHG] Player /org/bluez/hci0/dev_34_1C_F0_F6_76_07/player0 Subtype: None
[CHG] Player /org/bluez/hci0/dev_34_1C_F0_F6_76_07/player0 Status: paused
[CHG] Player /org/bluez/hci0/dev_34_1C_F0_F6_76_07/player0 Name: YouTube
[CHG] Player /org/bluez/hci0/dev_34_1C_F0_F6_76_07/player0 Title: Dr. Alban - Its My Life (DJ Savin, Alex Pushkarev Remix) #Shuffle #Dance
[CHG] Player /org/bluez/hci0/dev_34_1C_F0_F6_76_07/player0 TrackNumber: 0x00000000 (0)
[CHG] Player /org/bluez/hci0/dev_34_1C_F0_F6_76_07/player0 NumberOfTracks: 0x00000000 (0)
[CHG] Player /org/bluez/hci0/dev_34_1C_F0_F6_76_07/player0 Duration: 0x000395f8 (235000)
[CHG] Player /org/bluez/hci0/dev_34_1C_F0_F6_76_07/player0 Artist: Fommy Music
[CHG] Player /org/bluez/hci0/dev_34_1C_F0_F6_76_07/player0 Position: 0x0002cb04 (183044)
[CHG] Player /org/bluez/hci0/dev_34_1C_F0_F6_76_07/player0 Status: playing
[CHG] Player /org/bluez/hci0/dev_34_1C_F0_F6_76_07/player0 Position: 0x0002cb23 (183075)
[CHG] Transport /org/bluez/hci0/dev_34_1C_F0_F6_76_07/sep1/fd6 State: pending
[CHG] Transport /org/bluez/hci0/dev_34_1C_F0_F6_76_07/sep1/fd6 State: active
[... 72 lines cut ...]
[    9.762295] Bluetooth: SCO socket layer initialized
[    9.877414] Bluetooth: hci0: CSR: Setting up dongle with HCI ver=3 rev=0c5c; LMP ver=3 subver=0c5c; manufacturer=10
[   14.548566] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   14.548585] Bluetooth: BNEP filters: protocol multicast
[   14.548601] Bluetooth: BNEP socket layer initialized
[   14.559864] Bluetooth: MGMT ver 1.22
[  949.173685] Bluetooth: ISO socket layer initialized
[ 1184.355349] Bluetooth: RFCOMM TTY layer initialized
[ 1184.355378] Bluetooth: RFCOMM socket layer initialized
[ 1184.355405] Bluetooth: RFCOMM ver 1.11
```

Operating system: Debian (sid|testing)

"Bluealsa" compiled from source packages for Debian unstable. For the debian sid version of "blualsa" 3.x did not want to connect at all.

```
dpkg -l | grep blu
ii  bluetooth                        5.66-1                         all          Bluetooth support (metapackage)
ii  bluez                            5.66-1                         i386         Bluetooth tools and daemons
**ii  bluez-alsa-utils                 4.0.0-1                        i386         Bluetooth Audio ALSA Backend (utils)
ii  libasound2-plugin-bluez:i386     4.0.0-1                        i386         Bluetooth Audio ALSA Backend (plugins)**
ii  libbluetooth-dev:i386            5.66-1                         i386         Development files for using the BlueZ Linux Bluetooth library
ii  libbluetooth3:i386               5.66-1                         i386         Library to use the BlueZ Linux Bluetooth stack
```

arkq, 2023-02-14:
OK, I'll check that - maybe something was broken in recent commits...

arkq, 2023-02-14:
> bluealsa: D: ba-transport.c:1494: Created new IO thread [ba-a2dp-sbc]: A2DP Sink (SBC)

According to quick check, you've got code from around bluez-alsa v4.0.0 :
```sh
git clone https://github.com/arkq/bluez-alsa.git && cd bluez-alsa
while [ $(git grep 'Created new IO thread' |awk -F: '{print $2}') != 1494 ]; do git checkout HEAD^; done
```

I've just checked that version and I was able to record audio using the same command as you: `arecord -D ...`. With bluez-alsa (from this repo) there is one constrain - you can open the PCM device only once. Trying to open it more than once will result in "Device or resource busy" error. However, it seems that it's not the case here - there is only one `org.bluealsa.PCM1.Open()` in bluealsa logs.

Anyway, please try the most recent version of bluez-alsa. Maybe you've hit some non-common bug which potentially might be fixed already. If master branch will not work for you as well, we will have to do some "printf" debugging :)

borine, 2023-02-15:
I notice that the Debian bluez-alsa-utils package includes bluealsa-aplay with a systemd service file. Just to be absolutely sure that this is not being started silently in the background, what is the output of 
```sh
systemctl status bluealsa-aplay.service
```

fizista, 2023-02-15:
Everything is already working. I can't repeat this error.

Thank you all for your help.

closed 2023-02-15 by fizista