arkq/bluez-alsa #615: bluealsa.service: Start request repeated too quickly.

diadntjr, 2023-02-22:
Hi everyone.
I met a problem to active bluez-alsa service in jetson xavier Ubuntu 18.04.

The problem is appear when I after reading a wiki about 'Systemd integration' and setting up on my own system, so I typing `sudo journalctl -xeu bluealsa.servie` in my shell to watch log. 

And I can see that major error log,

```
bluealsa.service: Main process exited, code=exited, status=127/n/a
bluealsa.service: Failed with result 'exit-code'
```

And when I typing `bluealsa-aplay`,I just only see just Warning log.
```
bluealsa-aplay: W: Couldn't get BlueALSA PCM list: The name org.bluealsa was not provided by any .service files
```

What should I do? Please give me some advice.

And sorry to everyone about my terrible English writting.
Hwang

arkq, 2023-02-22:
Can you provide logs from bluealsa.service? If in the journal there is nothing else than "Main process exited", please compile bluez-alsa with debug enabled (`../configure --enable-debug`) and post debug messages from the journal.

borine, 2023-02-22:
I think that wiki page needs updating. The bluealsa systemd service file now sends its messages to syslog by default, but I think the journalctl log only contains STDERR output. So you may have to look in '/var/log/syslog' for the bluealsa error messages.

I wonder whether defaulting to syslog is correct when running under systemd? For example `bluetoothd` logs to the systemd journal, not to syslog, when run by systemd on Ubuntu.

arkq, 2023-02-22:
>  The bluealsa systemd service file now sends its messages to syslog by default, but I think the journalctl log only contains STDERR output.

On my RPi the journal contains everything... and what's more, logs are colorized based on severity, debug is gray, error is red. To be honest, that's the reason why I've changed the default from stderr to syslog, so syslog messages are colorized by systemd. To be specific, I'm running Debian 11.6 on my RPi.

borine, 2023-02-22:
> On my RPi the journal contains everything

Ah, OK, my mistake, on Ubuntu also it seems systemd itself logs to syslog,  so journalctl is actually reading from syslog anyway. That's my lesson learned for today ! :smile:

diadntjr, 2023-02-23:
Hi, and thanks for your reply.

Well, I did remove bluez-alsa service and files,(`cd /path/to/bluez-alsa/build/ ; sudo make uninstall ; rm -rf *`) then I tried to configure with `--enable-debug` in my clear build folder.
But it makes another error log. I installed all dependencies file in the installation wiki, and also  I can't find the error log in trouble shooting.

This is full error log.

```
Making all in asound
make[3]: Entering directory '/home/soda/bluez-alsa/build/src/asound'
make[3]: Nothing to be done for 'all'.
make[3]: Leaving directory '/home/soda/bluez-alsa/build/src/asound'
make[3]: Entering directory '/home/soda/bluez-alsa/build/src'
/bin/bash ../libtool  --tag=CC   --mode=link gcc -pthread -I/usr/include/gio-unix-2.0/ -I/usr/include/glib-2.0 -I/usr/lib/aarch64-linux-gnu/glib-2.0/include -I/usr/include/glib-2.0 -I/usr/lib/aarch64-linux-gnu/glib-2.0/include  -g -O2   -o bluealsa shared/a2dp-codecs.o shared/ffb.o shared/log.o shared/rt.o shared/nv.o a2dp.o a2dp-sbc.o at.o audio.o ba-adapter.o ba-device.o ba-rfcomm.o ba-transport.o bluealsa-config.o bluealsa-dbus.o bluealsa-iface.o bluealsa-skeleton.o bluez.o bluez-iface.o bluez-skeleton.o codec-sbc.o dbus.o hci.o hfp.o io.o rtp.o sco.o storage.o utils.o main.o        a2dp-mpeg.o codec-msbc.o   -lbluetooth -lgio-2.0 -lgobject-2.0 -lglib-2.0 -lglib-2.0 -lmp3lame -lsbc -lspandsp -lpthread -lm -lSegFault
libtool: link: gcc -pthread -I/usr/include/gio-unix-2.0/ -I/usr/include/glib-2.0 -I/usr/lib/aarch64-linux-gnu/glib-2.0/include -I/usr/include/glib-2.0 -I/usr/lib/aarch64-linux-gnu/glib-2.0/include -g -O2 -o bluealsa shared/a2dp-codecs.o shared/ffb.o shared/log.o shared/rt.o shared/nv.o a2dp.o a2dp-sbc.o at.o audio.o ba-adapter.o ba-device.o ba-rfcomm.o ba-transport.o bluealsa-config.o bluealsa-dbus.o bluealsa-iface.o bluealsa-skeleton.o bluez.o bluez-iface.o bluez-skeleton.o codec-sbc.o dbus.o hci.o hfp.o io.o rtp.o sco.o storage.o utils.o main.o a2dp-mpeg.o codec-msbc.o  -lbluetooth -lgio-2.0 -lgobject-2.0 -lglib-2.0 -lmp3lame -lsbc -lspandsp -lpthread -lm -lSegFault -pthread
shared/log.o: In function `vlog':
/home/soda/bluez-alsa/build/src/asound/../../../src/asound/../shared/log.c:109: undefined reference to `gettid'
collect2: error: ld returned 1 exit status
Makefile:653: recipe for target 'bluealsa' failed
make[3]: *** [bluealsa] Error 1
make[3]: Leaving directory '/home/soda/bluez-alsa/build/src'
Makefile:765: recipe for target 'all-recursive' failed
make[2]: *** [all-recursive] Error 1
make[2]: Leaving directory '/home/soda/bluez-alsa/build/src'
Makefile:460: recipe for target 'all-recursive' failed
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory '/home/soda/bluez-alsa/build'
Makefile:392: recipe for target 'all' failed
make: *** [all] Error 2
```

What can I do..?

Diaa

arkq, 2023-02-23:
> undefined reference to `gettid'

It seems that Ubuntu 18 is not recent enough to be shipped with `gettid()`... I will add a fix for that, but in the meantime you can remove these lines:

https://github.com/arkq/bluez-alsa/blob/master/src/shared/log.c#L108-L110

diadntjr, 2023-02-23:
Thanks for fix, and I did pull and autoconf install. And I also added config option `--enable-debug`, and enabled daemon system.

This is my full error logs.

```
-- Unit bluealsa.service has finished shutting down.
 2월 23 16:38:19 alto systemd[1]: Starting BlueALSA service...
-- Subject: Unit bluealsa.service has begun start-up
-- Defined-By: systemd
-- Support: http://www.ubuntu.com/support
--
-- Unit bluealsa.service has begun starting up.
 2월 23 16:38:20 alto systemd[1]: bluealsa.service: Main process exited, code=exited, status=127/n/a
 2월 23 16:38:20 alto systemd[1]: bluealsa.service: Failed with result 'exit-code'.
 2월 23 16:38:20 alto systemd[1]: Failed to start BlueALSA service.
-- Subject: Unit bluealsa.service has failed
-- Defined-By: systemd
-- Support: http://www.ubuntu.com/support
--
-- Unit bluealsa.service has failed.
--
-- The result is RESULT.
 2월 23 16:38:20 alto systemd[1]: bluealsa.service: Service hold-off time over, scheduling restart.
 2월 23 16:38:20 alto systemd[1]: bluealsa.service: Scheduled restart job, restart counter is at 5.
-- Subject: Automatic restarting of a unit has been scheduled
-- Defined-By: systemd
-- Support: http://www.ubuntu.com/support
--
-- Automatic restarting of the unit bluealsa.service has been scheduled, as the result for
-- the configured Restart= setting for the unit.
 2월 23 16:38:20 alto systemd[1]: Stopped BlueALSA service.
-- Subject: Unit bluealsa.service has finished shutting down
-- Defined-By: systemd
-- Support: http://www.ubuntu.com/support
--
-- Unit bluealsa.service has finished shutting down.
 2월 23 16:38:20 alto systemd[1]: bluealsa.service: Start request repeated too quickly.
 2월 23 16:38:20 alto systemd[1]: bluealsa.service: Failed with result 'exit-code'.
 2월 23 16:38:20 alto systemd[1]: Failed to start BlueALSA service.
-- Subject: Unit bluealsa.service has failed
-- Defined-By: systemd
-- Support: http://www.ubuntu.com/support
--
-- Unit bluealsa.service has failed.
--
-- The result is RESULT.
```

And this is my systemd config file about bluealsa

```
[Unit]
Description=BlueALSA service
Documentation=man:bluealsa(8)
Requisite=dbus.service
After=bluetooth.service

# In order to customize BlueALSA D-Bus service one should create an override
# for this systemd unit file. Please note, that in the override file one will
# have to explicitly clear the ExecStart before setting it again. See the
# bluez-alsa wiki for more options.
#
# $ sudo systemctl edit bluealsa
# [Service]
# ExecStart=
# ExecStart=/usr/bin/bluealsa -S --keep-alive=5 -p a2dp-sink

[Service]
Type=dbus
BusName=org.bluealsa
User=root
ExecStart=/usr/bin/bluealsa -S -p a2dp-source -p a2dp-sink
Restart=on-failure

# Sandboxing
AmbientCapabilities=CAP_NET_RAW
CapabilityBoundingSet=CAP_NET_RAW
IPAddressDeny=any
LockPersonality=true
MemoryDenyWriteExecute=true
NoNewPrivileges=true
PrivateDevices=true
PrivateTmp=true
PrivateUsers=false
ProtectClock=true
ProtectControlGroups=true
ProtectHome=true
ProtectHostname=true
ProtectKernelLogs=true
ProtectKernelModules=true
ProtectKernelTunables=true
ProtectProc=invisible
ProtectSystem=strict
RemoveIPC=true
RestrictAddressFamilies=AF_UNIX AF_BLUETOOTH
RestrictNamespaces=true
RestrictRealtime=true
RestrictSUIDSGID=true
SystemCallArchitectures=native
SystemCallErrorNumber=EPERM
SystemCallFilter=@system-service
SystemCallFilter=~@resources @privileged
UMask=0077

# Setup state directory for persistent storage
ReadWritePaths=/var/lib/bluealsa
StateDirectory=bluealsa

[Install]
WantedBy=bluetooth.target
```

borine, 2023-02-23:
I recommend to debug first without using systemd or syslog, just to see if the problem lies within the bluealsa daemon:
```
sudo systemctl stop bluealsa
sudo /usr/bin/bluealsa -p a2dp-source -p a2dp-sink
```

diadntjr, 2023-02-24:
I did use your shell command... but I don't know is it working.

The log:

```
bluealsa: D: ../../src/main.c:590: Starting main dispatching loop
bluealsa: D: ../../src/main.c:114: Acquired D-Bus service name: org.bluealsa
bluealsa: D: ../../src/bluealsa-dbus.c:380: Registering D-Bus manager: /org/bluealsa
bluealsa: D: ../../src/bluez.c:776: Registering battery provider: /org/bluez/hci0/battery
bluealsa: D: ../../src/bluez.c:796: BlueZ battery provider support not available
bluealsa: D: ../../src/bluez.c:590: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/source/1
bluealsa: D: ../../src/bluez.c:501: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/source/1
bluealsa: D: ../../src/bluez.c:590: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/source/2
bluealsa: D: ../../src/bluez.c:501: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/source/2
bluealsa: D: ../../src/bluez.c:590: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/sink/1
bluealsa: D: ../../src/bluez.c:501: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/sink/1
bluealsa: D: ../../src/bluez.c:590: Creating media endpoint object: /org/bluez/hci0/A2DP/SBC/sink/2
bluealsa: D: ../../src/bluez.c:501: Registering media endpoint: /org/bluez/hci0/A2DP/SBC/sink/2
```

And nothing to show me.

arkq, 2023-02-24:
According to logs you've provided, bluealsa service has started correctly. Now you should try to connect Bluetooth device. However, since you are using Ubuntu, I expect that you've got PulseAudio installed. And such setup might behave in such a way, that even though Bluetooth device has connected, you will not see any more logs in bluealsa output. Please, see troubleshooting section in the README file.

diadntjr, 2023-02-24:
Okay, but that is my second question. If I disable pulseaudio-module-bluetooth, my hci0 device didn't provide a2dp and hsp. So I can't connect my own bluetooth headset or anything bluetooth audio device. 

Am I missing some part of the wiki? Or what do I don't know?

Sorry for making you frustrated, but I really confused about it.

borine, 2023-02-24:
> Sorry for making you frustrated, but I really confused about it

That's OK, the only way to learn is to ask questions.

Using BlueALSA at the same time as PulseAudio is a very complicated, and confusing, thing to do. Are you sure you really want to do this? If you can explain why PulseAudio on its own is not sufficient for you, then we may be able to help. Normally on a desktop system you would use PulseAudio for bluetooth audio (or Pipewire on more recent OS), BlueALSA is really meant for  smaller dedicated audio or audio/visual systems.

diadntjr, 2023-02-24:
> If you can explain why PulseAudio on its own is not sufficient for you, then we may be able to help

Actually, I want to use Google Assistant SDK on my Jetson Xavier board thought bluetooth mic. But that SDK only use ALSA or PortAudio, and I counldn't find any connect with PortAudio or ALSA(Except bluez-alsa), Google Docs too. I did setting up successfully in PulseAudio, but that SDK didn't find my connected bluetoooth mic. So I thought I needed config ALSA configuration myself, then I tried to setting up bluez-alsa on my board.

borine, 2023-02-24:
As far as I know, google assistant does work with PulseAudio (but I have never used it myself). I suspect that the lack of microphone my be related to the chosen bluetooth profile. You would need to read up on using Pulseaudio tools such as `pavucontrol` or `pacmd`; that is out of scope for this project.

I believe it also works with bluealsa (there are several guides on the web for using them together on a raspberry pi). 

But you must choose one or the other; unless you already have a deep understanding of both PulseAudio and ALSA then trying to have both at the same time will prove very difficult.

diadntjr, 2023-02-24:
Okay, those audio systems are more huge than I thought. I will study more, then I try to do something.

I think the reason for the error is operating at the same time pulseaudio-module-bluetooth and bluez-alsa too.

closed 2023-02-28 by diadntjr